linux4wilc/driver #2: Incorrect MAC address is read from WILC1000 module

ChristianHackAMS, 2017-11-09:
Firstly the release version of the driver (same as the one in this repo) doesn't seem to work properly on our system, so I've taken the version from linux4wilc/linux-at91 but it doesn't have an issues section. So I'm reporting the issue here since I expect this version will be the same.

Using that driver against 4.9.30 of the kernel (commit hash of a487a26a1f19), the MAC address does not report correctly. The first byte F8 on the WILC1000 modules is always changed to FA.

i.e. F8:F0:05:xx:yy:zz becomes FA:F0:05:xx:yy:zz

example log and ifconfig:

```root@sama5d4-ams-sm01:~# grep wilc /var/log/syslog
Oct 20 00:00:42 sama5d4-ams-sm01 kernel: wilc1000_sdio mmc1:0001:1: Driver Initializing success
Oct 20 00:00:42 sama5d4-ams-sm01 kernel: wilc1000_sdio mmc1:0001:1: chipid (001003a0)
Oct 20 00:00:42 sama5d4-ams-sm01 kernel: wilc1000_sdio mmc1:0001:1: has_thrpt_enh3 = 1...
Oct 20 00:00:42 sama5d4-ams-sm01 kernel: wilc1000_sdio mmc1:0001:1 wlan0: Detect chip WILC1000
Oct 20 00:00:42 sama5d4-ams-sm01 kernel: wilc1000_sdio mmc1:0001:1 wlan0: loading firmware wilc1000_wifi_firmware.bin
Oct 20 00:00:42 sama5d4-ams-sm01 kernel: wilc1000_sdio mmc1:0001:1 wlan0: WILC Firmware Ver = WILC_WIFI_FW_REL_15_00_RC4 Build: 8837
Oct 20 00:00:42 sama5d4-ams-sm01 kernel: wilc1000_sdio mmc1:0001:1 wlan0:  his_state = HOST_IF_CONNECTED
Oct 20 00:00:42 sama5d4-ams-sm01 kernel: wilc1000_sdio mmc1:0001:1 wlan0:  EAPOL transmitted
Oct 20 00:00:42 sama5d4-ams-sm01 kernel: wilc1000_sdio mmc1:0001:1 wlan0:  EAPOL transmitted
root@sama5d4-ams-sm01:~# ifconfig wlan0
wlan0     Link encap:Ethernet  HWaddr fa:f0:05:xx:yy:zz
          inet addr:10.1.2.127  Bcast:10.1.2.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:41 errors:0 dropped:0 overruns:0 frame:0
          TX packets:13 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:4676 (4.5 KiB)  TX bytes:2138 (2.0 KiB)
```

With the old driver in the main linux-at91 repo, the MAC address is read correctly from the device.

AdhamAbozaeid, 2017-11-09:
Hi Chirstian
WILC has 2 interfaces, wlan, and p2p0, each with a different mac address.
p2p0 will have the mac address on WILC's efuse, and wlan will have the same address with bit[1] of the first byte flipped.

ChristianHackAMS, 2017-11-09:
Well that's not how the old driver worked and you are using an OUI that is not allocated to you (or anyone for that matter). You shouldn't make up your own OUI. How do you justify creating your own OUI?

ChristianHackAMS, 2017-11-09:
Also your statement about the bit being flipped is not working. **Both** interfaces are using invalid MAC addresses:

```
root@sama5d4-ams-sm01:~# ifconfig p2p0
p2p0      Link encap:Ethernet  HWaddr fa:f0:05:xx:yy:zz
          inet6 addr: fe80::f8f0:5ff:feff:cfa0/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:8 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:648 (648.0 B)

root@sama5d4-ams-sm01:~# ifconfig wlan0
wlan0     Link encap:Ethernet  HWaddr fa:f0:05:xx:yy:zz
          inet addr:10.1.2.127  Bcast:10.1.2.255  Mask:255.255.255.0
          inet6 addr: fd94:b3bc:6078:0:f8f0:5ff:feff:cfa0/64 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:6543 errors:0 dropped:131 overruns:0 frame:0
          TX packets:874 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:602888 (588.7 KiB)  TX bytes:93995 (91.7 KiB)

AdhamAbozaeid, 2017-11-09:
The old driver has a problem with running the 2 interfaces concurently.
For you OUI's comment, I'm not making up my own OUI! The bit changed to identify the interface is the unviersal/local bit of the mac address format.
This is how they are identified on air, I'll trace why they are appearing as the same address on ifconfig

ChristianHackAMS, 2017-11-10:
Ah OK, sorry. I find it wierd that the unversally administered address is not assigned to the wlan0 interface then as that would be the more common use.

deanoati, 2017-11-13:
I agree with ChristianHackAMS - the universally administered IEEE should be assigned to the primary wlan0 interface.

FWIW, I also had reported this issue to MicroChip and they said it was a known issue and they were working on it.

AdhamAbozaeid, 2017-11-14:
Thanks ChristianHackAMS and Deanoati for the feedback.
Actually my statement wasn't very accurate; the first interface up-ed will get the FA address, and the second will get the F8 address.
We'll definitely take your input into consideration.

ChristianHackAMS, 2017-11-30:
Not sure how much this helps but I noticed this:

If I bring up the wlan0 interface first I get the same MAC address (and seem to have issues with the connecting):

```
root@sama5d4-ams-sm01:~# ifconfig wlan0 up
wilc1000_sdio mmc1:0001:1: chipid (001003a0)
wilc1000_sdio mmc1:0001:1: has_thrpt_enh3 = 1...
wilc1000_sdio mmc1:0001:1 wlan0: Detect chip WILC1000
wilc1000_sdio mmc1:0001:1 wlan0: loading firmware wilc1000_wifi_firmware.bin
wilc1000_sdio mmc1:0001:1 wlan0: WILC Firmware Ver = WILC_WIFI_FW_REL_15_00_RC4 Build: 8837
root@sama5d4-ams-sm01:~# ifconfig
wlan0     Link encap:Ethernet  HWaddr fa:f0:05:xx:yy:zz
          inet6 addr: fe80::f8f0:5ff:feff:d478/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:508 (508.0 B)

root@sama5d4-ams-sm01:~# ifconfig p2p0 up
root@sama5d4-ams-sm01:~# ifconfig
p2p0      Link encap:Ethernet  HWaddr fa:f0:05:xx:yy:zz
          inet6 addr: fe80::f8f0:5ff:feff:d478/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:508 (508.0 B)

wlan0     Link encap:Ethernet  HWaddr fa:f0:05:xx:yy:zz
          inet6 addr: fe80::f8f0:5ff:feff:d478/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:8 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:648 (648.0 B)
````

If I bring up the p2p0 first and then wlan0 it's ok (software rebooting the system first):

```
root@sama5d4-ams-sm01:~# ifconfig p2p0 up
wilc1000_sdio mmc1:0001:1: chipid (001003a0)
wilc1000_sdio mmc1:0001:1: has_thrpt_enh3 = 1...
wilc1000_sdio mmc1:0001:1 p2p0: Detect chip WILC1000
wilc1000_sdio mmc1:0001:1 p2p0: loading firmware wilc1000_wifi_firmware.bin
wilc1000_sdio mmc1:0001:1 p2p0: WILC Firmware Ver = WILC_WIFI_FW_REL_15_00_RC4 Build: 8837
root@sama5d4-ams-sm01:~# ifconfig
p2p0      Link encap:Ethernet  HWaddr fa:f0:05:xx:yy:zz
          inet6 addr: fe80::f8f0:5ff:feff:d478/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:508 (508.0 B)

root@sama5d4-ams-sm01:~# ifconfig wlan0 up
root@sama5d4-ams-sm01:~# ifconfig
p2p0      Link encap:Ethernet  HWaddr fa:f0:05:xx:yy:zz
          inet6 addr: fe80::f8f0:5ff:feff:d478/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:8 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:648 (648.0 B)

wlan0     Link encap:Ethernet  HWaddr f8:f0:05:xx:yy:zz
          inet6 addr: fe80::faf0:5ff:feff:d478/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:508 (508.0 B)
```

kvams1, 2018-06-20:
this issue is fixed in  driver version 15.1

closed 2018-06-20 by AdhamAbozaeid

AdhamAbozaeid, 2018-06-24:
@deanoati, this is fixed in [dc3d3cb](https://github.com/linux4wilc/linux-at91/commit/dc3d3cb877e17f12b473e9b2ed9de48fa44a69fb)

ShonkyCH, 2018-06-24:
@AdhamAbozaeid 
So the "release" version is called 15_01_RC3?
And the next version with the MAC address fix is also called 15_01_RC3?
That's great.

AdhamAbozaeid, 2018-06-24:
@ShonkyCH, fixes on the development branch are not released yet.