linux4wilc/driver #4: Can I use this driver for ATWINC1500?

semofa, 2018-01-17:
I have ATWINC1500 connected to imx6. I can not found any linux driver fo ATWINC1500. I it possible use this driver for ATWINC1500?

AdhamAbozaeid, 2018-01-17:
@semofa , you can use ATWINC1500 HW for testing, but this will be an unnecessary cost when going into production.
ATWINC1500 and ATWILC1000 are basically the same chip, with ATWINC1500 having a stacked flash to store the FW at. This flash isn't used for Linux; since the FW is stored on the host.

semofa, 2018-01-19:
Thanks, I thought so. I searched more and realize that ATWILC1000 store TCP/IP stack on chip and ATWINC1500 does not. so ATWINC1500 is better choice for Linux. Beside that ATWINC1500 and ATWILC1000 have same price.

AdhamAbozaeid, 2018-01-19:
@semofa , the opposite is correct.
ATWILC1000 DOESN'T store the TCP/IP on chip, which makes it a better choice for Linux.

semofa, 2018-01-20:
sorry, I edited. thanks for your help.

semofa, 2018-01-20:
I searched and I realized that `ATWINC1500  `has an Ethernet mode that can enabled with set the `rHAVE_ETHERNET_MODE_BIT`(0x80) bit in `rNMI_GP_REG_1`(0x14A0) register.
cat I set `ATWINC1500  `to ethernet mode and use your driver? or maybe update `ATWINC1500  `firmware to `ATWILC1000 `firmware and use your driver?

AdhamAbozaeid, 2018-01-20:
You can use the ATWINC1500 HW with this driver and firmware with zero changes, but as I mentioned, this will come at an extra cost.
Please explain why you want to use ATWINC1500, not ATWILC1000

semofa, 2018-01-21:
I have a board which has ATWINC1500 and for now I have to setting it up. I know it is mush easier if I use ATWILC1000, but for know I stuck with ATWINC1500.
thanks

semofa, 2018-01-21:
I managed to setup `ATWINC1500` with `ATWILC1000` firmware and driver.
know I can connect to my cellphone AP and get IP. but when i ping `www.google.com` I received a lot of `Failed cmd response, cmd (c4), resp (e2)` error and can not ping. but I can ping my cellphon.

semofa, 2018-01-22:
I tested "dev" branch of driver with "dev" branch of firmware. but I cannot connect to hotspot. I received "wlan0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1" error.

semofa, 2018-01-23:
one more things: I cannot find a GPIO setup for `wakeup` in driver codes. How should I manage `wakeup` ?

AdhamAbozaeid, 2018-01-23:
For SPI the IRQ line is used both for regular interrupts, and for waking up the host.
For SDIO, the in-band SDIO interrupt is using for regular interrupts, and the IRQ line is used for waking up the host. However, the dev branch has a known issue with waking the host on linux 4.9 that we are working on. It works for other kernel versions like 3.18 for example

semofa, 2018-01-23:
Is it mean the wake up line than go out of module and connect to host is useless?

AdhamAbozaeid, 2018-01-23:
The wake up line is not used for now

closed 2018-01-24 by AdhamAbozaeid