linux4wilc/driver #5: Compilation error using kernel 4.9

bpussacq, 2018-01-23:
Hi,

I'm trying to use the atwilc1000 driver on PetaLinux distro for Xilinx (kernel 4.9).
During compilation, I've got the following error:
```
wilc_wfi_cfgoperations.c:2284:25: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
|   .change_virtual_intf = change_virtual_intf,
|                          ^~~~~~~~~~~~~~~~~~~
| build/tmp/work/plnx_aarch64-xilinx-linux/wilc/1.0-r0/wilc_wfi_cfgoperations.c:2284:25: note: (near initialization for 'wilc_cfg80211_ops.change_virtual_intf')
```

This issue can be resolve by modifying the change_virtual_intf definition at line 1869 as follow:
```
static int change_virtual_intf(struct wiphy *wiphy, struct net_device *dev,
			       enum nl80211_iftype type, u32 *flags, struct vif_params *params)
```
u32 *flags shall be placed after nl80211_iftype and not before.

I hope it can help.

Kind regards,
Benjamin

AdhamAbozaeid, 2018-01-23:
Thanks Benjamin for reporting this.
This error was fixed on the dev branch on commit 68f4d26676b43c8608ef30831640ce5c3fe877fa

closed 2018-01-23 by AdhamAbozaeid

bpussacq, 2018-01-23:
My bad... I used the master branch, not the dev branch... 
Sorry for that.
Thanks for your reply

AdhamAbozaeid, 2018-01-23:
No worries.
It's actually recommended to use the master branch. The dev branch isn't tested fully until officially released.