linux4wilc/driver #6: Linux 3.10 compatibility broken

bmoraine, 2018-02-05:
Hello,

It seems the driver is no more compatible with linux 3.10 whereas "ATWILC Devices Linux Porting Guide" pretends that driver is compatible.

 Please find build output :
 
[wilc_build_error.log](https://github.com/linux4wilc/driver/files/1696339/wilc_build_error.log)

Regards

AdhamAbozaeid, 2018-02-05:
Hi @bmoraine 

The dependency  on the linux kernel version comes from the dependency of wilc_wfi_cfgoperations.c on the kernel's cfg operations, that changes slightly from one version to another, making it difficult to keep the driver compliant with all versions.

For example, the error you are getting is because cfg80211_inform_bss changes from 
cfg80211_inform_bss(struct wiphy *wiphy,
                   struct ieee80211_channel *channel,
                   const u8 *bssid, u64 tsf, u16 capability,
                   u16 beacon_interval, const u8 *ie, size_t ielen,
                   s32 signal, gfp_t gfp);
in kernel 3.10 to 
static inline struct cfg80211_bss * __must_check
cfg80211_inform_bss(struct wiphy *wiphy,
                   struct ieee80211_channel *rx_channel,
                   _enum cfg80211_bss_frame_type ftype,_
                   const u8 *bssid, u64 tsf, u16 capability,
                   u16 beacon_interval, const u8 *ie, size_t ielen,
                   s32 signal, gfp_t gfp)
in kernel 4.9

While we will do our best to keep the driver compliant with as much kernel versions as possible, it's hard to cover them all, specially that the linux community opposes the usage of LINUX_VERSION_CODE macros in the code.
For now, it's compliant with 3.18 and 4.x kernel's, and we will add as many version as possible. Meanwhile, please do this required changes on your side. The changes should be straightforward and easy, since it's mostly related to new parameters add in the newer Linux versions.

Please point me to where the Linux porting guide gave the impression that Linux 3.10 is supported to correct that. The Linux porting guide is a general porting guide to any new Linux system and shouldn't be dependent on any specific Linux version

bmoraine, 2018-02-06:
Hello,

To answer you last question about 'Linux porting guide', please have a look on document DS70005329A page 12 chapter 3.1 TARGET : Linux 3.10 is quoted  as a compatible kernel version compatible. I assume this chapter is no more up to date.

Today, I succesfully build the diver with linux 3.10 using following patch
[old_kernel_support_patch.txt](https://github.com/linux4wilc/driver/files/1701158/old_kernel_support_patch.txt)

When I test it, after driver downloads successfully the firmware, the driver fails in linux_wlan_start_firmware() as wait_for_completion_timeout() returns an error. I was wondering if it could be related to an SDIO IRQ management issue? So I had a look on WILC1000_HW_OOB_INTR option and notices a Kconfig issue. Indeed WILC1000_HW_OOB_INTR depends on WILC1000_SDIO instead of WILC_SDI0 so it could not be selected in menuconfig.

Do you think using a Out of band interrupt could fix my issue in  linux_wlan_start_firmware() or I have to look for other(s) cause(s)?

 Regards

AdhamAbozaeid, 2018-02-06:
Thanks for the patch. Will work on integrating that on the repo.
Section 3.1 was in the document as an example of how to build the driver with different kernels and different toolchains. Sorry for the confusion. It was also targeting building the driver separately without building the whole kernel.
We are also working on updating the porting guide and user guide, and will post them once they are ready.
When WILC1000_HW_OOB_INTR isn't selected, the in-band interrupt mechanism will be used, which is recommended.
However, your observation is correct; it looks like your problem is related to the interrupt being received.
Some things you can check are:
* See the FW logs (debug UART on WILC3000 shield board or WILC1000 SD board) to make sure it started
* Double check the IRQ registration.
* See if the IRQ was received, but for some reason didn't unlock the waiting completion at linux_wlan_start_firmware()

bmoraine, 2018-02-08:
Hello,

My issue was not related to IRQ management but to a clock issue. 

Indeed, when I use a 26 MHz TCXO instead of providing the clock by my microprocessor (that can not provide a 26 MHz but only 12, 16, 32 or 33 MHz) driver seems to load correctly :
# modprobe  wilc-sdio
[ 1253.310000] wilc1000_sdio mmc0:0001:1: Initializing netdev
[ 1253.430000] wilc1000_sdio mmc0:0001:1: Driver Initializing success
# wpa_supplicant -D nl80211 -i wlan0 -c /etc/wpa.conf
Successfully initialized wpa_supplicant
rfkill: Cannot open RFKILL control device
[ 1258.400000] wilc1000_sdio mmc0:0001:1 wlan0: MAC OPEN[c2d22800]
[ 1258.410000] wilc1000_sdio mmc0:0001:1: chipid (001003a0)
[ 1258.410000] wilc1000_sdio mmc0:0001:1: has_thrpt_enh3 = 1...
[ 1258.420000] wilc1000_sdio mmc0:0001:1 wlan0: Detect chip WILC1000
[ 1258.430000] wilc1000_sdio mmc0:0001:1 wlan0: loading firmware wilc1000_wifi_firmware.bin
[ 1258.440000] wilc1000_sdio mmc0:0001:1 wlan0: Download Start
[ 1258.470000] wilc1000_sdio mmc0:0001:1 wlan0: Download Succeeded
[ 1258.470000] wilc1000_sdio mmc0:0001:1 wlan0: wlan_start...
[ 1258.550000] wilc1000_sdio mmc0:0001:1 wlan0: WILC Firmware Ver = WILC_WIFI_FW_REL_15_00_RC4 Build: 8837
[ 1258.560000] wilc1000_sdio mmc0:0001:1 wlan0: Start configuring Firmware
[ 1258.560000] wilc1000_sdio mmc0:0001:1 wlan0: Host = c35bf000
[ 1258.660000] wilc1000_sdio mmc0:0001:1 wlan0: Mac address: 00:50:c2:5e:10:8e
[ 1258.670000] wilc1000_sdio mmc0:0001:1 wlan0: Entry[0]: 33:33:0:0:0:1
[ 1258.670000] wilc1000_sdio mmc0:0001:1 wlan0: Entry[0]: 33:33:0:0:0:1
[ 1258.680000] wilc1000_sdio mmc0:0001:1 wlan0: Entry[0]: 33:33:0:0:0:1
[ 1258.690000] wilc1000_sdio mmc0:0001:1 wlan0: Entry[1]: 1:0:5e:0:0:1
[ 1258.690000] wilc1000_sdio mmc0:0001:1 wlan0: Entry[0]: 33:33:0:0:0:1
[ 1258.700000] wilc1000_sdio mmc0:0001:1 wlan0: Entry[1]: 1:0:5e:0:0:1
[ 1258.710000] wilc1000_sdio mmc0:0001:1 wlan0: Entry[2]: 33:33:ff:5e:10:8e
nl80211: Could not re-add multicast membership for vendor events: -2 (No such file or directory)
wlan0: Trying to associate with e4:95:6e:40:51:2a (SSID='Kapelse-Wifi' freq=2462 MHz)
[ 1259.920000] wilc1000_sdio mmc0:0001:1 wlan0:  his_state = HOST_IF_CONNECTED
[ 1259.930000] wilc1000_sdio mmc0:0001:1 wlan0: netif_rx ret value is: 0
wlan0: Associated with e4:95:6e:40:51:2a
[ 1259.940000] wilc1000_sdio mmc0:0001:1 wlan0:  EAPOL transmitted
[ 1260.900000] wilc1000_sdio mmc0:0001:1 wlan0: netif_rx ret value is: 0
[ 1260.900000] wilc1000_sdio mmc0:0001:1 wlan0:  EAPOL transmitted
[ 1261.900000] wilc1000_sdio mmc0:0001:1 wlan0: netif_rx ret value is: 0
[ 1261.910000] wilc1000_sdio mmc0:0001:1 wlan0:  EAPOL transmitted
[ 1262.900000] wilc1000_sdio mmc0:0001:1 wlan0: netif_rx ret value is: 0
[ 1262.900000] wilc1000_sdio mmc0:0001:1 wlan0:  EAPOL transmitted
[ 1263.920000] cfg80211: Calling CRDA to update world regulatory domain
wlan0: CTRL-EVENT-DISCONNECTED bssid=e4:95:6e:40:51:2a reason=0
wlan0: WPA: 4-Way Handshake failed - pre-shared key may be incorrect
wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=3 ssid="Kapelse-Wifi" auth_failures=1 duration=10 reason=WRONG_KEY
`
It seems I have trouble with wpa to connect to my hotspot. It's weird as this set up works with another wifi module from TI.

Any idea to undertand why I face such behaviour.

Regards

AdhamAbozaeid, 2018-02-08:
The 26Mhz crystal is essential for correct operation.
From the logs, I see the problem is more related to a wrong password. Can you please double check the password you are using, and the wpa_cli commands you are using to set the password?
You can also try changing the password to a simpler one, maybe there's a problem with complex passwords or unicode characters.
Does it also work if you tried connecting to an open security AP?

bmoraine, 2018-02-12:
Hello Adham,

For information, I build my rootfs using  buildroot and I have the following version of wifi tools
- wpa_supplicant : 2.5
- iw : 4.3
- libnl : 3.2.27

When using a hotspot without security, connection is successful and I can ping the gateway if I used a fix IP as I can not  get an IP address using udhcpc.
`# modprobe wilc-sdio
[   26.050000] wilc1000_sdio mmc0:0001:1: Driver Initializing success
# ifconfig wlan0 up
[   33.290000] wilc1000_sdio mmc0:0001:1: chipid (001003a0)
[   33.300000] wilc1000_sdio mmc0:0001:1: has_thrpt_enh3 = 1...
[   33.310000] wilc1000_sdio mmc0:0001:1 wlan0: Detect chip WILC1000
[   33.310000] wilc1000_sdio mmc0:0001:1 wlan0: loading firmware wilc1000_wifi_firmware.bin
[   33.450000] wilc1000_sdio mmc0:0001:1 wlan0: WILC Firmware Ver = WILC_WIFI_FW_REL_15_00_RC4 Build: 8837
# iw dev wlan0 link
Not connected.
# iw dev wlan0 scan | grep -i airport
        SSID: KapelseAirport
# iw dev wlan0 connect KapelseAirport
# [   80.250000] wilc1000_sdio mmc0:0001:1 wlan0:  his_state = HOST_IF_CONNECTED 
# iw dev wlan0 link
Connected to 74:1b:b2:d5:4a:fc (on wlan0)
        SSID: KapelseAirport
        freq: 2462
        tx bitrate: 13.0 MBit/s
# ifconfig wlan0 10.0.1.10
# ping -I wlan0 10.0.1.1
PING 10.0.1.1 (10.0.1.1): 56 data bytes
64 bytes from 10.0.1.1: seq=0 ttl=255 time=1004.577 ms
64 bytes from 10.0.1.1: seq=1 ttl=255 time=3.739 ms
64 bytes from 10.0.1.1: seq=2 ttl=255 time=6.788 ms
64 bytes from 10.0.1.1: seq=3 ttl=255 time=12.068 ms
64 bytes from 10.0.1.1: seq=4 ttl=255 time=9.263 ms
^C
--- 10.0.1.1 ping statistics ---
5 packets transmitted, 5 packets received, 0% packet loss
round-trip min/avg/max = 3.739/207.287/1004.577 ms

# udhcpc -i wlan0
udhcpc (v1.24.1) started
Sending discover...
Sending discover...
Sending discover...
^C
ith 
# iw dev wlan0 link
Connected to 74:1b:b2:d5:4a:fc (on wlan0)
        SSID: KapelseAirport
        freq: 2462
        tx bitrate: 39.0 MBit/s`

When I can establisf a connection to an open hot spot using  wpa_suppplicant.

I do not understand why I have troubles using  udhcpc and wpa_supplicant with wpa/wpa2 enabled hot-spot.

I can provide trace with more vebosity using dmesg -n 8 to help your understand what happens.

Regards

AdhamAbozaeid, 2018-02-12:
Hi @bmoraine 
Regarding the wpa2 connection problem, please note that iw tool doesn't support wpa/wpa2 connections.
https://wireless.wiki.kernel.org/en/users/documentation/iw#getting_link_status

To connect to wpa2 AP, you need to the wpa_cli commands. Please refer to the WILC linux user guide for more information about how to use it.

bmoraine, 2018-02-12:
Hi Adham,

Sorry to not have been clear enough.

I am able  : 
- to connect to an open hotspot using either iw or wpa_supplicant
- to set manually the IP address and ping the gateway 

I am not able :
- to connect to WPA/WPA2 secured hot spot using wpa_supplicant
- to get an IP over DHCP using udhcpc

with a rootfs and wpa.conf file that is fully functional with a TI WL1271L based module.

I will provide detailed traces tomorrow.

Regards

bmoraine, 2018-02-15:
Hi Adam, 

Please find console and WILC1000 Uart Debug logs attached of an unsuccessful attempt to connect to a WPA/WPA2 secured Hotspot.

[DBG_wilc1000.log](https://github.com/linux4wilc/driver/files/1728191/DBG_wilc1000.log)
[wpa_wilc1000.log](https://github.com/linux4wilc/driver/files/1728192/wpa_wilc1000.log)

Please let me know if it helps you to understand my issue.

Regards

Baptiste Moraine

AdhamAbozaeid, 2018-02-15:
Hi bmoraine
I tested this issue on my setup but couldn't reproduce it. Can you try with another AP, or a hotspot? Can also try with the sama5d4 reference platform with the prebuilt linux images?

bmoraine, 2018-02-15:
Hi Adham,

For you information,
- The SOC I used is a SAMA5D31 
- I have tried 3 differents hotspot without success that I used for several years for interoperability tests for my previous WiFi based product using the same SOC, the same kernel and the same rootffs.

I am deeply convinced this is a driver issue either introduced by my patch or due to an incompability issue with Linux 3.10.

Did you test my patch on a Linux 3.10 on a SAMA5D4XPLAINED board?

If you use a more recent kernel with the current wilc1000 driver did you notice any differences on UART debug  or WPA output?

Could you add -d parameter to wpa_supllicant and share wpa_supplicant and UART debug output  when successfully connecting to an WPA/WPA2 enables Hotsptot. I will do the same on my side then we could see differences.

By the way, If Microchip has for the moment no bandwidth to add/check Linux 3.10 compatibility, is there any previous couple of driver and firmware that are known compatible with Linux 3.10? I could use them while Microchip find the time to fix Linux 3.10 compatibility issue?

Regards

Baptiste Moraine

AdhamAbozaeid, 2018-02-22:
Hi Baptiste

Unfortunately I don't have a setup with Linux 3.10 ready at the moment , but I'm attaching the logs from kernel 4.9 with -d as you requested.
[SAMA5D4_Linux_2018-02-21-172729.log](https://github.com/linux4wilc/driver/files/1746214/SAMA5D4_Linux_2018-02-21-172729.log)

Microchip can check linux 3.10 compatibility, but I'll ask you to open a ticket on Microchip's support. They should be able to help more with porting issues. http://microchipsupport.force.com/ 

Thanks,
Adham

AdhamAbozaeid, 2018-02-22:
Also, please send me more details on the security type you are using, WPA/WPA2/mixed? TKIP/AES?

bmoraine, 2018-02-22:
Hi,

For information a ticket (00278322)  is already open on Microchip's  support.

My resquest is handled by both HW and SW support teams so I close this issue.

Thanks for your help

Regards

Baptiste Moraine

closed 2018-02-22 by bmoraine