linux4wilc/driver #9: error while doing ifdown and ifup

prattom, 2018-03-26:
I get following errors while doing ifup and ifdown wlan0
Trying to free already-free IRQ 64
Modules linked in: wilc_spi(O) wilc(O)
CPU: 0 PID: 324 Comm: wpa_supplicant Tainted: G           O    4.1.15-1.2.0+g77f6154 #2
Hardware name: Freescale i.MX6 Ultralite (Device Tree)
[<80015d84>] (unwind_backtrace) from [<80012728>] (show_stack+0x10/0x14)
[<80012728>] (show_stack) from [<807da36c>] (dump_stack+0x84/0xc4)
[<807da36c>] (dump_stack) from [<80034a40>] (warn_slowpath_common+0x80/0xb0)
[<80034a40>] (warn_slowpath_common) from [<80034aa0>] (warn_slowpath_fmt+0x30/0x40)
[<80034aa0>] (warn_slowpath_fmt) from [<8006d7b0>] (__free_irq+0xa4/0x1ec)
[<8006d7b0>] (__free_irq) from [<8006d98c>] (free_irq+0x4c/0xa4)
[<8006d98c>] (free_irq) from [<7f003748>] (deinit_irq+0x1c/0x28 [wilc])
[<7f003748>] (deinit_irq [wilc]) from [<7f004ed0>] (wilc_wlan_deinitialize+0x88/0xd8 [wilc])
[<7f004ed0>] (wilc_wlan_deinitialize [wilc]) from [<7f004ff4>] (wilc_mac_close+0xd4/0x108 [wilc])
[<7f004ff4>] (wilc_mac_close [wilc]) from [<80643950>] (__dev_close_many+0x88/0xd0)
[<80643950>] (__dev_close_many) from [<80643aa8>] (__dev_close+0x24/0x38)
[<80643aa8>] (__dev_close) from [<8064af8c>] (__dev_change_flags+0x94/0x144)
[<8064af8c>] (__dev_change_flags) from [<8064b054>] (dev_change_flags+0x18/0x48)
[<8064b054>] (dev_change_flags) from [<806ab0e4>] (devinet_ioctl+0x664/0x738)
[<806ab0e4>] (devinet_ioctl) from [<80631d84>] (sock_ioctl+0x1bc/0x290)
[<80631d84>] (sock_ioctl) from [<800f9e6c>] (do_vfs_ioctl+0x3e8/0x608)
[<800f9e6c>] (do_vfs_ioctl) from [<800fa0c0>] (SyS_ioctl+0x34/0x5c)
[<800fa0c0>] (SyS_ioctl) from [<8000f480>] (ret_fast_syscall+0x0/0x3c)
---[ end trace 344f134da92d4389 ]---
WILC_SPI spi0.0: Failed cmd response, cmd (c9), resp (c0)
WILC_SPI spi0.0: Failed cmd, write reg (00001400)...
wlan0: wilc deinitialization Done
genirq: Flags mismatch irq 64. 00002008 (WILC_IRQ) vs. 00002008 (WILC_IRQ)
wlan0: Failed to request IRQ GPIO: 34
wlan0: WLAN Iinitialization FAILED
wlan0: Failed to set driver handler
wlan0: Host cmd 24 failed
wlan0: [Host Interface] undefined
I am using Atwilc3000 wifi module with imx6ul. The old drivers used to work fine but this driver is not working. The wifi is connected to router but I am not able to send received packets from router.

AdhamAbozaeid, 2018-03-26:
This looks like an issue that was fixed on the dev branch on commit 1a5d06de2b22e3d1d366c82afd7cd672b308281a
I retested WILC3000 SPI on the dev branch and could reinit successfully. Please give it a try and let me know if you are still having issues

prattom, 2018-03-27:
Thanks for your reply. I wan using the maste branch instead of dev branch. Now I am using latest code from dev branch and I am getting following error
WILC_SPI spi0.0 wlan0: INFO [linux_wlan_start_firmware]Firmware start timed out
WILC_SPI spi0.0 wlan0: ERR [wilc_wlan_initialize: 1223] Failed to start firmware
WILC_SPI spi0.0 wlan0: INFO [wlan_deinitialize_threads]Deinitializing Threads
WILC_SPI spi0.0 wlan0: INFO [wlan_deinitialize_threads]Deinitializing debug Thread
WILC_SPI spi0.0 wlan0: INFO [debug_thread]Exit debug thread
WILC_SPI spi0.0 wlan0: INFO [wlan_deinitialize_threads]Deinitializing Threads
WILC_SPI spi0.0 wlan0: ERR [wilc_wlan_cleanup: 1717] Writing ABORT reg
WILC_SPI spi0.0 wlan0: INFO [wlan_deinit_locks]De-Initializing Locks
WILC_SPI spi0.0 wlan0: ERR [wilc_wlan_initialize: 1264] WLAN initialization FAILED
WILC_SPI spi0.0 wlan0: ERR [wilc_mac_open: 1314] Failed to initialize wilc
WILC_SPI spi0.0 wlan0: ERR [wilc_send_config_pkt: 1880] [Sendconfigpkt]Set Timed out
WILC_SPI spi0.0 wlan0: ERR [handle_set_wfi_drv_handler: 446] Failed to set driver handler
WILC_SPI spi0.0 wlan0: Host cmd 24 failed
WILC_SPI spi0.0 wlan0: ERR [host_if_work: 3012] [Host Interface] undefined
Could not set interface wlan0 flaWILC_SPI spi0.0 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 0
gs (UP): Input/output error
nl80WILC_SPI spi0.0 wlan0: ERR [wilc_frame_register: 4091] wilc mq send fail
211: Could not set interface 'wlaWILC_SPI spi0.0 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 0
n0' UP
nl80211: deinit ifname=wlWILC_SPI spi0.0 wlan0: ERR [wilc_frame_register: 4091] wilc mq send fail
an0 disabled_11b_rates=0
WILC_SPI spi0.0 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 0
WILC_SPI spi0.0 wlan0: ERR [wilc_frame_register: 4091] wilc mq send fail
WILC_SPI spi0.0 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 0
WILC_SPI spi0.0 wlan0: ERR [wilc_frame_register: 4091] wilc mq send fail
WILC_SPI spi0.0 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 0
WILC_SPI spi0.0 wlan0: ERR [wilc_frame_register: 4091] wilc mq send fail
WILC_SPI spi0.0 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 0
WILC_SPI spi0.0 wlan0: ERR [wilc_frame_register: 4091] wilc mq send fail
wlan0: Failed to initialize driver interface

As we can see from above error the firmware fails to start. I am also using latest firmware from dev branch. I am using SPI mode.

prattom, 2018-03-27:
can you please tell me  which commit of driver and firmware is most stable currently for wilc3000 in spi mode?

AdhamAbozaeid, 2018-03-27:
For my tests, I'm also using the dev branch on the /linux4wilc/firmware. Once testing is finalized, all dev branches will be merged back to the master
Please note that one of the changes in the driver's dev branch would look for the firmware under /lib/firmware/mchp, rather than /lib/firmware

prattom, 2018-03-27:
Can you please tell me the reason for error I am getting "firmware not able to start" since I am also using latest code from dev branch.

AdhamAbozaeid, 2018-03-27:
From the logs, it looks like the FW didn't start, or the driver didn't receive the first interrupt from the firmware that indicates that the FW is alive, hence you are getting the 
WILC_SPI spi0.0 wlan0: INFO [linux_wlan_start_firmware]Firmware start timed out
error.
you can double check the IRQ initialization, and make sure that the GPIO number is defined correctly and matching the GPIO that is connected to WILC's module

prattom, 2018-03-28:
The IRQ GPIO is initialized correctly. If I use latest master branch code then firmware loads fine and I am also able to scan for wifi access points (other things don't work ifup, ifdown etc.) but dev branch firmware doesn't start. So I think it's definitely some issue with drivers and firmware.

AdhamAbozaeid, 2018-03-28:
Hi @prattom 
I tested the dev branch on a SAMA5D4 platform using linux from /linux4wilc/linux-at91 branch linux-4.9-at91_wilc_dev and it works, so my guess is that the problem is related to one of the changes in the branch that conflicts with something on your platform.
At the stage where the driver fails, it's waiting for an interrupt from the chip. Can you probe the interrupt line and see if it gets asserted? you can also read the FW logs from the debug UART on WILC to make sure the FW was downloaded and started correctly. 
If the interrupt line is asserted, you can check the ISR and see why the IRQ isn't handled correctly

prattom, 2018-06-08:
Hi @AdhamAbozaeid  I tried the latest main branch drivers now. But I an facing the same problem Firmware start timed out. I tried checking with oscilloscope but interrupt doesn't get asserted at all. Also when I load drivers wilc_spi.ko I get following output
WILC_SPI spi0.0: spiModalias: WILC_SPI, spiMax-Speed: 6000000
(unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
(unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
(unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Successful Registering
(unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
(unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
(unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Successful Registering
WILC_SPI spi0.0: WILC setting default Reset GPIO to 35.  Got -2
WILC_SPI spi0.0: WILC setting default Chip Enable GPIO to 33.  Got -2
WILC_SPI spi0.0: WILC setting default IRQ GPIO to 34.  Got -2
wifi_pm : 0 
wifi_pm : 1 
WILC_SPI spi0.0: WILC SPI probe success

I got following output on wifi module debug UART when I ran following command
wpa_supplicant -B -Dnl80211 -iwlan0 -c/etc/wpa_supplicant.conf
00 60 04 2C 62 82 E2 DA   72 92 FB 32 7C 1B 1A 82  .`.,b...r..2|...
62 04 8C 1B E3 4E 72 FF   12 6F DC 03 64 04 04 24  b....Nr..o..d..$
20 03 20 0C 64 22 83 1C   92 72 9B FB 13 7C 1B 12   . .d"...r...|..
82 63 0C 6C 6C 80 0F 02   24 83 83 23 02 04 23 F2  .c.ll...$..#..#.
8E 9B 83 DB 64 64 23 24   0C 8C 9C 84 84 9C 8C 0C  ....dd#$........
6C 8F 3E 0F 02 84 8C 8C   84 84 8E E0 72 6C 20 0C  l.>.........rl .
64 23 83 1C 93 7B DA FA   12 7C 12 1B 83 22 04 84  d#...{...|..."..
90 7C 13 1B 02 E4 24 9C   DC 0C 22 F3 8F 92 82 9B  .|....$...".....
2C 64 23 24 04 EC 84 64   84 24 2C 60 02 20 04 24  ,d#$...d.$,`. .$
23 82 1C 9B 7A DA F2 12   7C 13 1B 82 22 04 04 B2  #...z...|..."...
03 E4 9C E0 62 E2 2F 0C   62 F2 7A DA 82 92 0C 2C  ....b./.b.z....,
22 64 04 04 2C 60 02 28   30 29 3E 3E 20 53 6C 65  "d..,`.(0)>> Sle
65 70 20 63 6C 6B 20 73   72 63 20 3C 3D 20 52 54  ep clk src <= RT
43 0D 0A

After this when I do ifup wlan0 I get following on debug UART
(0)Efuse_read: bank_idx = 0
(0)Efuse_read: MAC addr (used=1) = f8:f0:05:c0:22:6a
(0)Efuse_read: Freq Off (used=1) = 32121
(0)Efuse_read: PA gain (used=0) = 0
(0)>> Sleep clk src <= RTC

But the IRQ line doesn't get asserted at all.

AdhamAbozaeid, 2018-06-09:
are you using the buildroot and linux kernel from linux4wilc?

prattom, 2018-06-11:
We are using customised board based on NXP imx6UL processor. Thus we are using yocto BSP by NXP with linux kernel 4.1.15.

AdhamAbozaeid, 2018-06-11:
Please open a support ticket on https://microchipsupport.force.com/

prattom, 2018-06-12:
I have already done so. Case no. is 00303795

closed 2018-06-26 by AdhamAbozaeid