linux4wilc/driver #10: Issue compiling on raspbian

rocky134, 2018-06-17:
I'm trying to compile the drivers on branch master but it seems that there is not a __init nor a module_init() so it never gets compiled.
My process for compilation is add the driver sources  to drivers/staging/wilc1000 
make menuconfig -> Device Drivers -> staging drivers -> Microchip devices -> WILC SDIO 
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf-
Everything compiles without error and a zImage along with all the module *.ko and *.dtb* files are available for transfer except the wilc driver has no *.ko files

ShonkyCH, 2018-06-18:
Could you be having issues with the fact that they have renamed the driver from wilc1000 to just wilc?

Are your Kconfig and Makefiles correct one level up in drivers/staging/ ? In particular the line in Makefile that looks similar to this (I've updated fully to the new name)

```
obj-$(CONFIG_WILC)		+= wilc/
```

rocky134, 2018-06-18:
Yes, I've copied all the source files into the wilc1000 directory as well as tried adding the wilc/ directory and renaming the entries in the Makefile from 
```c
obj-$(CONFIG_WILC1000)		+= wilc1000/
```
to 
```c
obj-$(CONFIG_WILC)		+= wilc/
```
and the Kconfig
```c
source "drivers/staging/wilc1000/Kconfig"
```
to 
```c
source "drivers/staging/wilc/Kconfig"
```
From what I understand, an init function needs to be defined similarly to this
```c
static int __init wilc_init(void)
{
...
}
module_init(wilc_init);
```

ShonkyCH, 2018-06-18:
I'm not a kernel expert either but it compiles modules fine for me in with kernel 4.9.52 (linux4sam 5.7). I'm using the 15.01 release which is not far behind master.

module_init is not a requirement and I expect the driver is simply not being compiled at all. Put some bad code in one of the driver source files and see if it bombs out.

rocky134, 2018-06-18:
in linux4sam 5.7 the following code snippet is found in [wilc_debugfs.c](https://github.com/linux4sam/linux-at91/blob/973820d8c61f3b858c0cf46e69dcf7c5fdb2a661/drivers/staging/wilc1000/wilc_debugfs.c)

```c
static int __init wilc_debugfs_init(void)
{
	int i;
	struct wilc_debugfs_info_t *info;

	wilc_dir = debugfs_create_dir("wilc_wifi", NULL);
	for (i = 0; i < ARRAY_SIZE(debugfs_info); i++) {
		info = &debugfs_info[i];
		debugfs_create_file(info->name,
				    info->perm,
				    wilc_dir,
				    &info->data,
				    &info->fops);
	}
	return 0;
}
module_init(wilc_debugfs_init);
```
which leads me to the the origin of this question, are the __init and module_init placed on the correct function? I am able to compile the code as it is above however it does nothing except create a file in /sys/kernel/debug/wilc_wifi/ that says "debug level: 8" with nothing else happening. 

There are no debug or error messages in either dmesg or /var/log/messages related to this module from any of the printk() or dev_err() functions which leads me to believe that the core driver is never loaded.

ShonkyCH, 2018-06-18:
@rocky134 
I'm not using the wilc driver in linux4sam-5.7 - that is just my base. I have replaced it with 15.01 as I said. 15.01 does not have this entry. @AdhamAbozaeid has stated how the driver initialises.

If you are compiling as a module and not getting a wilc-sdio.ko or similar file then I'd still say it's not compiling it at all.

Have you confirmed you are actually compiling the module?

rocky134, 2018-06-19:
@AdhamAbozaeid @ShonkyCH  I was able to get the module compiled and loaded successfully, however the module still does not show up on ifconfig as a device.
The dev board I'm using is [this](https://www.microchip.com/developmenttools/ProductDetails/ATWILC3000-SHLD) with a wilc3000 module.

Upon booting pin 3 which is connected to EN is IN and HIGH. pin 32 which is RST is IN and LOW.
I manually set both of these pins to OUT and HIGH before loading the driver and the only thing I can see is this message when dmesg is called
```
wilc_sdio: module is from the staging directory, the quality is unknown, you have been warned.
```

The SDIO interface is configured on GPIO22-27 as ALT3 SD1_x which is referenced [here](https://elinux.org/RPi_BCM2835_GPIOs)

/boot/config.text has the following parameters added
```
dtoverlay=sdio
poll_once=off
```
On the board R311 and R218-R221 are placed while R214-R217 are open to connect the SDIO interface to the breakout connector, additionally R227-R232 are connected to access the SDIO from the Raspberry Pi header.

AdhamAbozaeid, 2018-06-19:
Hi @rocky134 
Please install resistor R312 with 120K Ohm on the shield board as described in section 1 of the user guide 
http://ww1.microchip.com/downloads/en/DeviceDoc/70005328A.pdf

rocky134, 2018-06-19:
@AdhamAbozaeid R312 was already on the board when I purchased it, it's value was measured at 100k ohm. On the Gerber files it looks like this resistor is used as a pull-up to 3.3V connected to CHIP_EN. The CHIP_EN pin is also connected to pin 3 which is initially LOW on boot. 

The [datasheet](http://ww1.microchip.com/downloads/en/DeviceDoc/70005327A.pdf) on page 9 at the bottom says to "Connect to a host output that defaults low at power-up." so I believe this is correct. This is also why I manually set the GPIO pin to OUT and HIGH prior to loading the driver.

when I look at the code I see this line
```c
gpio_chip_en = of_get_named_gpio_flags(cnp, "gpio_chip_en", 0, NULL);
dev_warn(wilc->dev,
			 "WILC setting default Chip Enable GPIO to %d. Got %d\r\n",
gpio_chip_en, ret);
```
I don't believe there is any GPIO label on the pi called "gpio_chip_en" so it would default to GPIO_NUM_CHIP_EN which I can easily change and recompile.

The issue is right afterwards dev_warn() should print something but the kernel log no messages. I even added a printk() with KERN_ALERT at the beginning of the function and set the system logging level to 8 and still dmesg was empty.

This makes me believe for whatever reason linux_sdio_probe() is never called.

ShonkyCH, 2018-06-19:
Sound a bit like your SD/MMC interface may not be working. Do you have anything like the 5th line below reported on your hardware?

My dmesg looks like:

```
Jun 15 17:21:46 dev kernel: atmel_mci fc000000.mmc: version: 0x600
Jun 15 17:21:46 dev kernel: atmel_mci fc000000.mmc: using dma0chan9 for DMA transfers
Jun 15 17:21:46 dev kernel: mmc_host mmc1: Setting MMC_CAP_4_BIT_DATA
Jun 15 17:21:46 dev kernel: atmel_mci fc000000.mmc: Atmel MCI controller at 0xfc000000 irq 32, 1 slots
Jun 15 17:21:47 dev kernel: mmc1: new high speed SDIO card at address 0001
Jun 15 17:21:47 dev kernel: wilc_sdio mmc1:0001:1: WILC setting default Reset GPIO to 128. Got -2^M
Jun 15 17:21:47 dev kernel: wilc_sdio mmc1:0001:1: WILC setting default Chip Enable GPIO to 156. Got -2^M
Jun 15 17:21:47 dev kernel: wilc_sdio mmc1:0001:1: WILC setting default IRQ GPIO to 155. Got -2^M
Jun 15 17:21:47 dev kernel: wilc_sdio mmc1:0001:1: Error requesting GPIOs for CHIP_EN and RESET
Jun 15 17:21:47 dev kernel: wilc_sdio mmc1:0001:1: Error requesting GPIOs for CHIP_EN and RESET
Jun 15 17:21:47 dev kernel: wilc_sdio mmc1:0001:1: Driver Initializing success
Jun 15 17:21:47 dev kernel: wilc_sdio mmc1:0001:1: SDIO speed: 50000000
Jun 15 17:21:47 dev kernel: wilc_sdio mmc1:0001:1: chipid 001003a0
```

The GPIOs will differ as thats for my custom board. I have them hardcoded in the driver although it does support supplying them via the device tree. You will definitely need this at a minimum or the device won't show up.

Not sure about those errors in mine actually. It does seem to work correctly. The "Got -2" is just from not having the GPIOs configured in the device tree.

Are your CHIP_EN and RESET being enabled in the right order? I have manually controlled them at the moment at startup.

AdhamAbozaeid, 2018-06-19:
Thanks @ShonkyCH 
@rocky134 , as Shonky described the linux kernel should find the SDIO controller even before WILC's driver to initialize. If you don't get the following log message 
mmc1: new high speed SDIO card at address 0001
then the linux kernel will not call sdio_probe function, which is the driver's init point.

This is the reason a pull up resistor is required on the chip_en line; because the WILC driver can't control the chip_en line until sdio_probe is called, and sdio_probe called can't be executed unless the WILC is powered on.

In your case, the error is as you expected; the device tree doesn't have the correct configuration for chip_en and reset_n GPIOs, and hence uses the hardcoded values from the driver.

rocky134, 2018-06-19:
@ShonkyCH @AdhamAbozaeid tha'ts it, there is no mmc1 in dmesg, strange since I enabled 
```
dtoverlay=sdio
poll_once=off
```
in /boot/config.txt

Thanks for the help!

closed 2018-06-19 by rocky134

ShonkyCH, 2018-06-20:
To be clear the "mmc1" is simply because I'm using MCI1 on my particular CPU. I might be mmc0 on yours.

rocky134, 2018-06-20:
mmc0 is for the eMMC flash chip. I should specify im using a CM3. On a normal raspberry pi3 model b with the same broadcom processor the following is in the device dts
```
/* SDHCI is used to control the SDIO for wireless */
&sdhci {
	pinctrl-names = "default";
	pinctrl-0 = <&emmc_gpio34>;
	status = "okay";
	bus-width = <4>;
	non-removable;
};
```
and dmesg confirms that mmc1 is used for wireless. Since the CM3 does not have this wireless component, the mmc1 interface should be free to add my own wifi module which I am trying to do.

But it seems that this line of code is missing from the CM3 dts so the interface is never loaded on boot so I have to add it in.

atorresdi, 2018-07-02:
Hi @rocky134,

Were you able to make the WILC driver work in Raspbian? I am using a Raspberry Pi 2 (running Raspbian Stretch Lite) and the WILC1000/3000 evaluation boards.

I have been able to compile and load the driver, the WILC1000/3000 boards are being recognized as SDIO devices (mmc1) but the driver fails to bring up the 'wlan0' interface. 

Regards.

Alvaro

rocky134, 2018-07-02:
@atorresdi I see your thread [here](https://github.com/linux4wilc/driver/issues/12)
I was able to get it to load but it was extremely unreliable, one out of every 10 hard power cycles made the device load as wlan0, the errors i was getting are the same as yours. Clock still off and -110. However, it would only load 169.254.xxx.xxx ip addresses. I was only able to get it to work once.

rocky134, 2018-07-02:
Moreover the UART debug interface was throwing maloc out of memory errors so it might be a firmware issue with the wilc3000. I've ordered the 1000 model to test and will be able to get back with more logs at a later date, most likely this weekend.

AdhamAbozaeid, 2018-07-02:
@rocky134 , can you attach the memory errors you are seeing?
if they happen during scan, that shouldn't be a problem. It happens because the the FW will be sending too many probe requests to the AP, and some of the requests will not find enough memory to be allocated.
the FW scans each channel a few times, so it shouldn't be a problem, and at the extreme, the only effect would be that the AP you are looking for might not appear on the scan results.
The clock still  off is also a warning, so you can ignore. The only serious issue will be with the -110 error, which indicates a bus communication error

AdhamAbozaeid, 2018-07-02:
As I advised rocky, the clock still off and bus errors might be a problem with powersave, which looks like Raspberry Pi has it enabled by default.
On sama5d4 + linux 4.9 + bulidroot, it's disabled by default, and only enables when you use:

rocky134, 2018-07-02:
@AdhamAbozaeid 
[cutecom.log](https://github.com/linux4wilc/driver/files/2156903/cutecom.log)