linux4wilc/driver #12: Initialization error

atorresdi, 2018-06-22:
Hi, I am trying to evaluate the ATWILC1000SD using my laptop. I have successfully compiled the driver against the source tree of the kernel version I am running (3.19.0-32-generic). The .config options for building the driver are the following:

```
CONFIG_WILC=m
CONFIG_WILC_SDIO=m
# CONFIG_WILC_SPI is not set
# CONFIG_WILC_HW_OOB_INTR is not set
```

I have placed all the firmware files in `/lib/firmware/mchp` and the module file `wilc.ko` in my home directory. I can load the driver without problem using `insmod`, but when I insert the ATWILC1000SD module in the SD card slot of my laptop I get an initialization error and the `wlan` intergace is not created. The following is some of the output of `dmesg` right after inserting the ATWILC1000SD module:

```
Jun 21 15:00:29 alvaro-K55VD kernel: [ 2731.550465] wilc_sdio: module is from the staging directory, the quality is unknown, you have been warned.
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.549034] mmc0: queuing unknown CIS tuple 0x91 (2 bytes)
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.549082] mmc0: new high speed SDIO card at address 0001
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.549312] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.549315] (unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.549464] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Successful Registering
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.549964] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.549969] (unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.550206] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Successful Registering
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.550700] wilc_sdio mmc0:0001:1: WILC setting default Reset GPIO to 60. Got -38
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.550704] wilc_sdio mmc0:0001:1: WILC setting default Chip Enable GPIO to 94. Got -38
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.550707] wilc_sdio mmc0:0001:1: WILC setting default IRQ GPIO to 91. Got -38
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.550710] wifi_pm : 0 
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.550713] wilc_sdio mmc0:0001:1: Error requesting GPIOs for CHIP_EN and RESET
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.550716] wifi_pm : 1 
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.550719] wilc_sdio mmc0:0001:1: Error requesting GPIOs for CHIP_EN and RESET
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.550829] wilc_sdio mmc0:0001:1: Driver Initializing success
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.569247] Driver 'mmcblk' needs updating - please use bus_type methods
Jun 21 15:01:19 alvaro-K55VD NetworkManager[1241]: <info> rfkill4: found WiFi radio killswitch (at /sys/devices/pci0000:00/0000:00:1c.3/0000:04:00.0/rtsx_pci_sdmmc.0/mmc_host/mmc0/mmc0:0001/mmc0:0001:1/ieee80211/phy2/rfkill4) (driver wilc_sdio)
Jun 21 15:01:19 alvaro-K55VD NetworkManager[1241]: <info> rfkill3: found WiFi radio killswitch (at /sys/devices/pci0000:00/0000:00:1c.3/0000:04:00.0/rtsx_pci_sdmmc.0/mmc_host/mmc0/mmc0:0001/mmc0:0001:1/ieee80211/phy1/rfkill3) (driver wilc_sdio)
Jun 21 15:01:19 alvaro-K55VD NetworkManager[1241]:    SCPlugin-Ifupdown: devices added (path: /sys/devices/pci0000:00/0000:00:1c.3/0000:04:00.0/rtsx_pci_sdmmc.0/mmc_host/mmc0/mmc0:0001/mmc0:0001:1/net/wlan1, iface: wlan1)
Jun 21 15:01:19 alvaro-K55VD NetworkManager[1241]:    SCPlugin-Ifupdown: device added (path: /sys/devices/pci0000:00/0000:00:1c.3/0000:04:00.0/rtsx_pci_sdmmc.0/mmc_host/mmc0/mmc0:0001/mmc0:0001:1/net/wlan1, iface: wlan1): no ifupdown configuration found.
Jun 21 15:01:19 alvaro-K55VD NetworkManager[1241]: <info> (wlan1): using nl80211 for WiFi device control
Jun 21 15:01:19 alvaro-K55VD NetworkManager[1241]: <info> (wlan1): driver supports Access Point (AP) mode
Jun 21 15:01:19 alvaro-K55VD NetworkManager[1241]: <info> (wlan1): new 802.11 WiFi device (driver: 'wilc_sdio' ifindex: 4)
Jun 21 15:01:19 alvaro-K55VD NetworkManager[1241]: <info> (wlan1): exported as /org/freedesktop/NetworkManager/Devices/2
Jun 21 15:01:19 alvaro-K55VD NetworkManager[1241]: <info> (wlan1): device state change: unmanaged -> unavailable (reason 'managed') [10 20 2]
Jun 21 15:01:19 alvaro-K55VD NetworkManager[1241]: <info> (wlan1): bringing up device.
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.579231] wilc_sdio mmc0:0001:1 wlan1: INFO [wilc_mac_open]MAC OPEN[ffff88008e416000] wlan1
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.579238] WILC POWER UP
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.579245] wilc_sdio mmc0:0001:1 wlan1: INFO [wilc_init_host_int]Host[ffff88008e416000][ffff880221147800]
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.582306] wilc_sdio mmc0:0001:1 wlan1: INFO [wilc_mac_open]*** re-init ***
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.582313] wilc_sdio mmc0:0001:1 wlan1: INFO [wlan_init_locks]Initializing Locks ...
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.582318] wilc_sdio mmc0:0001:1 wlan1: INFO [wilc_wlan_init]Initializing WILC_Wlan
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.582323] wilc_sdio mmc0:0001:1: SDIO speed: 50000000
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.584737] wilc_sdio mmc0:0001:1: chipid 001003a0
Jun 21 15:01:19 alvaro-K55VD NetworkManager[1241]: <info> (wlan1): deactivating device (reason 'managed') [2]
Jun 21 15:01:19 alvaro-K55VD NetworkManager[1241]:    SCPlugin-Ifupdown: devices added (path: /sys/devices/pci0000:00/0000:00:1c.3/0000:04:00.0/rtsx_pci_sdmmc.0/mmc_host/mmc0/mmc0:0001/mmc0:0001:1/net/p2p0, iface: p2p0)
Jun 21 15:01:19 alvaro-K55VD NetworkManager[1241]:    SCPlugin-Ifupdown: device added (path: /sys/devices/pci0000:00/0000:00:1c.3/0000:04:00.0/rtsx_pci_sdmmc.0/mmc_host/mmc0/mmc0:0001/mmc0:0001:1/net/p2p0, iface: p2p0): no ifupdown configuration found.
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.587247] wilc_sdio mmc0:0001:1 wlan1: INFO [wilc_wlan_initialize]WILC Initialization done
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.587254] wilc_sdio mmc0:0001:1 wlan1: ERR [init_irq: 434] could not obtain gpio for WILC_INTR
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.587258] wilc_sdio mmc0:0001:1 wlan1: INFO [wlan_deinit_locks]De-Initializing Locks
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.587261] wilc_sdio mmc0:0001:1 wlan1: ERR [wilc_wlan_initialize: 1244] WLAN initialization FAILED
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.587264] wilc_sdio mmc0:0001:1 wlan1: ERR [wilc_mac_open: 1294] Failed to initialize wilc
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.587282] wilc_sdio mmc0:0001:1 wlan1: ERR [wilc_send_config_pkt: 1929] Set Timed out
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.587289] wilc_sdio mmc0:0001:1 wlan1: ERR [handle_set_wfi_drv_handler: 445] Failed to set driver handler
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.587294] wilc_sdio mmc0:0001:1 wlan1: Host cmd 24 failed
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.587301] wilc_sdio mmc0:0001:1 wlan1: ERR [host_if_work: 3000] [Host Interface] undefined
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.588587] wilc_sdio mmc0:0001:1 wlan1: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.588593] wilc_sdio mmc0:0001:1 wlan1: INFO [wilc_mgmt_frame_register]Return since mac is closed
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.588607] wilc_sdio mmc0:0001:1 wlan1: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.588610] wilc_sdio mmc0:0001:1 wlan1: INFO [wilc_mgmt_frame_register]Return since mac is closed
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.588622] wilc_sdio mmc0:0001:1 wlan1: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.588625] wilc_sdio mmc0:0001:1 wlan1: INFO [wilc_mgmt_frame_register]Return since mac is closed
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.588636] wilc_sdio mmc0:0001:1 wlan1: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.588639] wilc_sdio mmc0:0001:1 wlan1: INFO [wilc_mgmt_frame_register]Return since mac is closed
[... cut ...]
Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.595551] wilc_sdio mmc0:0001:1 wlan1: ERR [host_if_work: 3000] [Host Interface] undefined
Jun 21 15:01:19 alvaro-K55VD wpa_supplicant[1290]: Could not set interface wlan1 flags (UP): Input/output error
Jun 21 15:01:19 alvaro-K55VD wpa_supplicant[1290]: nl80211: Could not set interface 'wlan1' UP 
```
From this output I can see that the module is failing to respond to cmd24, later in the ouput (not included here) I can see also errors with cmd53 and cmd52. 

Do you know what could be causing this issue? Am I missing something to make the driver work properly?

Regards.

Alvaro

AdhamAbozaeid, 2018-06-24:
Hi @atorresdi 
I see from the logs that your setup has a problem registering the IRQ line
`Jun 21 15:01:19 alvaro-K55VD kernel: [ 2781.587254] wilc_sdio mmc0:0001:1 wlan1: ERR [init_irq: 434] could not obtain gpio for WILC_INTR`
Please make sure you have the correct GPIO number defined either as a hardcoded value in the driver or in the dts file. 

WILC has 2 configurations to deliver packets to the host, either to use in-band interrupts, or Out Of Band interrupt.
However, even in the in-band interrupt configuration, an OOB interrupt is required to wakeup the host if it went into sleep mode.

Thanks,
Adham

atorresdi, 2018-06-26:
Hi @AdhamAbozaeid,

Thanks for your reply. 

Now I am connecting the ATWILC1000SD to a Raspberry Pi 2 and I have hardcoded the GPIO numbers in the driver. The initialization seems to work now but after loading the firmware `cmd52` fails multiple times and then the module crashes. I am using the master branch.

Here is the output of `dmesg`:

```
[  136.840338] wilc_sdio: loading out-of-tree module taints kernel.
[  136.840355] wilc_sdio: module is from the staging directory, the quality is unknown, you have been warned.
[  139.702642] mmc1: new high speed SDIO card at address 0001
[  139.703103] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
[  139.703111] (unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
[  139.703479] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Successful Registering
[  139.704337] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
[  139.704349] (unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
[  139.706449] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Successful Registering
[  139.707459] wilc_sdio mmc1:0001:1: WILC setting default Reset GPIO to 16. Got -2
[  139.707475] wilc_sdio mmc1:0001:1: WILC setting default Chip Enable GPIO to 6. Got -2
[  139.707485] wilc_sdio mmc1:0001:1: WILC setting default IRQ GPIO to 5. Got -2
[  139.707492] wifi_pm : 0 
[  139.707546] wifi_pm : 1 
[  139.714425] wilc_sdio mmc1:0001:1: Driver Initializing success
[  139.931036] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  139.931047] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  139.931091] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  139.931097] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  139.931139] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  139.931146] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  139.931188] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  139.931194] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  139.931236] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  139.931242] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  139.931283] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  139.931290] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  139.931331] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  139.931338] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  139.931379] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  139.931385] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  139.931474] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  139.931491] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  139.931549] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  139.931559] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  139.931604] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  139.931611] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  139.931652] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  139.931659] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  139.931701] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  139.931708] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  139.931749] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  139.931755] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  139.931798] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  139.931808] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  139.931852] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  139.931859] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  139.931903] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  139.931912] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  139.932028] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[b76dd000] wlan0
[  139.932033] WILC POWER UP
[... cut ...]
[  230.231207] wilc_sdio mmc1:0001:1 wlan0: INFO [debug_thread]*** Debug Thread Running ***
[  236.631193] wilc_sdio mmc1:0001:1 wlan0: INFO [debug_thread]*** Debug Thread Running ***
[  243.031218] wilc_sdio mmc1:0001:1 wlan0: INFO [debug_thread]*** Debug Thread Running ***
[  249.431198] wilc_sdio mmc1:0001:1 wlan0: INFO [debug_thread]*** Debug Thread Running ***
[  255.831172] wilc_sdio mmc1:0001:1 wlan0: INFO [debug_thread]*** Debug Thread Running ***
```

Do you know what could be the problem?

Thanks.
Alvaro

AdhamAbozaeid, 2018-06-26:
Hi @atorresdi 
This is a bus error that might be related to HW. Can you make sure the SD board is plugged in all the way into the mmc slot, and isn't loose?

AdhamAbozaeid, 2018-06-28:
Hi @atorresdi 
For the shield board, this looks like a problem with powersave.
Are you enabling powersave in your sequence? Can you try disabling it and check if the errors go away?

AdhamAbozaeid, 2018-06-29:
Hi @atorresdi 
Are you still getting this log after disabling powersave?
`wilc_sdio mmc1:0001:1 wlan0: INFO [set_power_mgmt] Power save Enabled= 1 , TimeOut = -1`
If yes, you can hack the driver to disable powersave for testing purposes, by  overriding power_mode to NO_POWERSAVE in handle_power_management just before wilc_send_config_pkt() is called

atorresdi, 2018-07-02:
Hi @AdhamAbozaeid,

I was still seeing `Power save Enabled= 1` in the log. I have overridden the `power_mode` variable to `NO_POWERSAVE` in the `handle_power_management` function:

```
static void handle_power_management(struct wilc_vif *vif,
				    struct power_mgmt_param *pm_param)
{
	s32 result = 0;
	struct wid wid;
	s8 power_mode;

	wid.id = (u16)WID_POWER_MANAGEMENT;

//	if (pm_param->enabled)
//		power_mode = MIN_FAST_PS;
//	else
		power_mode = NO_POWERSAVE;
	PRINT_INFO(vif->ndev, HOSTINF_DBG, "Handling power mgmt to %d\n", power_mode);
	wid.val = &power_mode;
	wid.size = sizeof(char);

	PRINT_INFO(vif->ndev, HOSTINF_DBG, "Handling Power Management\n");
	result = wilc_send_config_pkt(vif, SET_CFG, &wid, 1,
				      wilc_get_vif_idx(vif));
	if (result) {
		PRINT_ER(vif->ndev, "Failed to send power management\n");
		return;
	}
	store_power_save_current_state(vif, power_mode);
}
```
After recompiling the driver and loading it I am getting the this error:

```
[  772.225674] mmc1: new high speed SDIO card at address 0001
[  810.220813] wilc_sdio: module is from the staging directory, the quality is unknown, you have been warned.
[  810.225306] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
[  810.225319] (unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
[  810.225785] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Successful Registering
[  810.234226] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
[  810.234242] (unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
[  810.235317] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Successful Registering
[  810.236598] wilc_sdio mmc1:0001:1: WILC setting default Reset GPIO to 5. Got -2
[  810.236617] wilc_sdio mmc1:0001:1: WILC setting default Chip Enable GPIO to 6. Got -2
[  810.236627] wilc_sdio mmc1:0001:1: WILC setting default IRQ GPIO to 4. Got -2
[  810.236633] wifi_pm : 0 
[  810.236686] wifi_pm : 1 
[  810.243254] wilc_sdio mmc1:0001:1: Driver Initializing success
[  810.333248] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  810.333481] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[b909c000] wlan0
[  810.333485] WILC POWER UP
[  810.333494] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_init_host_int]Host[b909c000][b64bb000]
[  810.334258] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
[  810.334438] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_init_locks]Initializing Locks ...
[  810.334453] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
[  810.334465] wilc_sdio mmc1:0001:1: SDIO speed: 50000000
[  810.335640] wilc_sdio mmc1:0001:1: chipid 003000d0
[  810.336243] wilc_sdio mmc1:0001:1 wlan0: INFO [init_chip]Bootrom sts = d
[  810.336403] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_initialize]WILC Initialization done
[  810.336559] wilc_sdio mmc1:0001:1 wlan0: INFO [init_irq]IRQ request succeeded IRQ-NUM= 169 on GPIO: 4
[  810.336567] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Initializing Threads ...
[  810.336574] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Creating kthread for transmission
[  810.338252] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Creating kthread for Debugging
[  810.343500] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]Detect chip WILC3000
[  810.343512] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]loading firmware mchp/wilc3000_wifi_firmware.bin
[  810.343521] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]WLAN firmware: mchp/wilc3000_wifi_firmware.bin
[  810.354151] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_firmware_download]Downloading Firmware ...
[  810.354165] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_firmware_download]Downloading firmware size = 136076
[  810.382316] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_firmware_download]Offset = 119296
[... 246 lines cut ...]
[  813.054146] bf60: b9cd3000 8013778c b9c3f29c b9cefe78 b9d0bfac b9d0bf80 8013d860 80137798
[  813.054157] bf80: ffffffff b9c3f1c0 8013d724 00000000 00000000 00000000 00000000 00000000
[  813.054167] bfa0: 00000000 b9d0bfb0 8010810c 8013d730 00000000 00000000 00000000 00000000
[  813.054176] bfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  813.054186] bfe0: 00000000 00000000 00000000 00000000 00000013 00000000 55555555 55555555
[  813.054403] [<7f28a2e4>] (handle_rcvd_ntwrk_info [wilc_sdio]) from [<7f28bad4>] (host_if_work+0x7f4/0x1c14 [wilc_sdio])
[  813.054514] [<7f28bad4>] (host_if_work [wilc_sdio]) from [<80137490>] (process_one_work+0x158/0x454)
[  813.054534] [<80137490>] (process_one_work) from [<801377f0>] (worker_thread+0x64/0x5b8)
[  813.054549] [<801377f0>] (worker_thread) from [<8013d860>] (kthread+0x13c/0x16c)
[  813.054567] [<8013d860>] (kthread) from [<8010810c>] (ret_from_fork+0x14/0x28)
[  813.054579] Code: 00000002 00000000 00000000 00000000 (b72fe000) 
[  813.054587] ---[ end trace f5b66a5b62a07e02 ]---
[  817.424463] Voltage normalised (0x00000000)
[  819.024411] wilc_sdio mmc1:0001:1 wlan0: INFO [debug_thread]*** Debug Thread Running ***
[  825.424473] wilc_sdio mmc1:0001:1 wlan0: INFO [debug_thread]*** Debug Thread Running ***
[  831.824561] wilc_sdio mmc1:0001:1 wlan0: INFO [debug_thread]*** Debug Thread Running ***
```

I have tried powering the WILC3000-SHIELD from the Raspberry Pi header and from an external source (Imax = 1 A) with the same result.

Do you know what else could I try to make it work?

Regards.

Alvaro

AdhamAbozaeid, 2018-07-02:
Hi Alvaro
It's ok if you are still seeing Power save Enabled= 1  in the logs, since this is printed in an earlier stage when the supplicant is enabling powersave, but you are already overwritting the powersave value before sending the configuration packet to the chip.

From the logs, I see that the interface is being closed 
INFO [wilc_mac_close]Mac close
can you run the supplicant with debug flag and see if it provides more information on why the interface is being closed? do you have any scripts running that might be closing the interface?
Also I see this log message:
ERR [host_if_work: 3000] [Host Interface] undefined
which is alarming. Basically it means that a message was sent to host_if_work with an invalid message id.
Can you log the message ID and try to trace back the source of this message?

rocky134, 2018-07-03:
I am able to get the driver to at least load and create a wlan0 interface, however the IP address it assigns itself is 169.254.xxx.xxx not a lan IP. I have done the same modifications to the code as mentioned above and ran
```
sudo iwconfig wlan0 power off
```
to disable powersave. 
The device can see my network via iwlist wlan0 scanning but for some reason cannot connect through wla_supplicant and I am unable to force the connection with 
```
wpa_supplicant -Dwext -iwlan0 -c/etc/wpa_supplicant/wpa_supplicant.conf
```
Here is my kernel log as well as the debug serial output from the devices
[kern.log](https://github.com/linux4wilc/driver/files/2157264/kern.log)
[cutecom1.log](https://github.com/linux4wilc/driver/files/2157266/cutecom1.log)

ShonkyCH, 2018-07-03:
I did have issues with a number of access points with the 15.00 release version of the firmware/driver. Was able to see access points no problem but couldn't connect and/or DHCP an IP. With wpa_supplicant verbose debugging it wasn't clear what was happening other than it would end up "blacklisting" the AP I had configured.

I never figured out what sort of AP it had issues with (brands etc) but had the problem with at least 3 access points (1 Netgear, one Dlink and one Linksys) out of about 8-10 in total that I've tried. TP-link, my Galaxy S8+ phone and an old WRT54G were ok). I don't seem to have the problem any more with the 15.01 driver and the "RC3" release firmware. I see you're already using that but thought I'd mention it.

Sooo tried another AP?

AdhamAbozaeid, 2018-07-03:
@rocky134, as Shonky suggested, try other APs, and try other security options as well. Start with open security and see if you are able to obtain IP.
From WILC's logs, I see that connection was successful, and DHCP packets were exchanged.

rocky134, 2018-07-04:
@atorresdi @AdhamAbozaeid 
I was able to get it to successfully connect and work on every reboot. The issue was the package ```dhcpcd5``` which seems to not play nice with this module. The solution was to use ```wicd-curses``` as the network manager.

Also it seems the external function ```gpio_request``` returns 0 even if it fails to physically claim the GPIO pin and thus all subsequent functions to physically modify the GPIO pin state do not work, mentioned [here](https://patchwork.kernel.org/patch/72526/). The solution was to change all instances of the function to ```gpio_request_one``` and initialize the state as well as claim the pin for use.
Example being the following code in function ```wilc_wlan_power``` in [linux_wlan.c](https://github.com/linux4wilc/driver/blob/master/wilc/linux_wlan.c#L1802)
```c
if (gpio_request(wilc->gpio_chip_en, "CHIP_EN") == 0 &&
	    gpio_request(wilc->gpio_reset, "RESET") == 0) {
		gpio_direction_output(wilc->gpio_chip_en, 0);
		gpio_direction_output(wilc->gpio_reset, 0);
......
```
becomes
```c
if (gpio_request_one(wilc->gpio_chip_en, GPIOF_INIT_LOW, "CHIP_EN") == 0 &&
	    gpio_request_one(wilc->gpio_reset, GPIOF_INIT_LOW, "RESET") == 0) {
......
```
Thus the physical state is able to transition from high->low and vice versa to reset the device. Allowing to to operate normally.

rocky134, 2018-07-05:
@atorresdi 
This is the driver I am using with a raspberry pi model 3 B. 
https://github.com/rocky134/driver/commit/8e55bb875d7b4b1f2502289cb64f44a637f547bb
From what I understand the BCM2835 and BCM2837 have the same SDIO overlays so the following code should be compatable with the model 2.
Just add the following to ```sdio-overlay.dts``` under the sdio_ovl. Be sure to replace the pin numbers with your own.
```
gpio_reset = <&gpio 6 0>;
gpio_chip_en = <&gpio 5 0>;
gpio_irq = <&gpio 4 0>;
```
so it would look something like this
```
sdio_ovl: sdio@7e300000 {
	compatible = "brcm,bcm2835-mmc",
				 "brcm,bcm2835-sdhci";
	reg = <0x7e300000 0x100>;
	interrupts = <2 30>;
	clocks = <&clocks 28/*BCM2835_CLOCK_EMMC*/>;
	dmas = <&dma 11>;
	dma-names = "rx-tx";
	brcm,overclock-50 = <0>;
	status = "okay";
	pinctrl-names = "default";
	pinctrl-0 = <&sdio_ovl_pins>;
	gpio_reset = <&gpio 6 0>;
	gpio_chip_en = <&gpio 5 0>;
	gpio_irq = <&gpio 4 0>;
	non-removable;
	bus-width = <4>;
};
```

atorresdi, 2018-07-05:
@rocky134

Thanks for sharing your solution. I have disabled `dhcpcd5` using `systemctl disable dhcpcd5.service` and now I am able to connect to an AP using `wicd-curses`. The GPIO initialization is also working fine, I no longer need to set the RESET and CHIP_ENA pins to a fixed value.

closed 2018-07-05 by atorresdi