linux4wilc/driver #13: Laggy response over wlan0 (wilc1000)

ShonkyCH, 2018-06-25:
This is seen on our custom hardware using the SAMA5D42 + wilc1000 module, but can be reproduced with the SAMA5D4Xplained dev board with the ATWILC1000SD dev board plugged into the the full size SD card socket and using the release 15.01 demo image from: 
https://github.com/linux4wilc/wilc_demo

I simply create a /etc/wpa_supplicant.conf file like so:
```
ctrl_interface=/run/wpa_supplicant
ctrl_interface_group=0

network={
        ssid="AccessPoint"
        psk="PSK"
}
```
then run:
```
/usr/sbin/wpa_supplicant -Dnl80211 -iwlan0 -c /etc/wpa_supplicant.conf -B
dhclient wlan0
```

Pinging the unit from other machines on the LAN show up to 1 second of delay. The ramping up indicates the unit waits up to 1 second before processing a packet depending on the point in time that it's sent:
```
64 bytes from 10.1.2.234: icmp_seq=296 ttl=64 time=935 ms
64 bytes from 10.1.2.234: icmp_seq=297 ttl=64 time=976 ms
64 bytes from 10.1.2.234: icmp_seq=298 ttl=64 time=7.01 ms
64 bytes from 10.1.2.234: icmp_seq=299 ttl=64 time=16.4 ms
64 bytes from 10.1.2.234: icmp_seq=300 ttl=64 time=55.9 ms
64 bytes from 10.1.2.234: icmp_seq=301 ttl=64 time=94.9 ms
64 bytes from 10.1.2.234: icmp_seq=302 ttl=64 time=134 ms
64 bytes from 10.1.2.234: icmp_seq=303 ttl=64 time=174 ms
64 bytes from 10.1.2.234: icmp_seq=304 ttl=64 time=213 ms
64 bytes from 10.1.2.234: icmp_seq=305 ttl=64 time=253 ms
64 bytes from 10.1.2.234: icmp_seq=306 ttl=64 time=293 ms
64 bytes from 10.1.2.234: icmp_seq=307 ttl=64 time=333 ms
64 bytes from 10.1.2.234: icmp_seq=308 ttl=64 time=383 ms
64 bytes from 10.1.2.234: icmp_seq=309 ttl=64 time=412 ms
64 bytes from 10.1.2.234: icmp_seq=310 ttl=64 time=452 ms
64 bytes from 10.1.2.234: icmp_seq=311 ttl=64 time=496 ms
64 bytes from 10.1.2.234: icmp_seq=312 ttl=64 time=531 ms
64 bytes from 10.1.2.234: icmp_seq=313 ttl=64 time=421 ms
64 bytes from 10.1.2.234: icmp_seq=314 ttl=64 time=611 ms
64 bytes from 10.1.2.234: icmp_seq=315 ttl=64 time=651 ms
64 bytes from 10.1.2.234: icmp_seq=316 ttl=64 time=696 ms
64 bytes from 10.1.2.234: icmp_seq=317 ttl=64 time=732 ms
64 bytes from 10.1.2.234: icmp_seq=318 ttl=64 time=460 ms
64 bytes from 10.1.2.234: icmp_seq=319 ttl=64 time=813 ms
64 bytes from 10.1.2.234: icmp_seq=320 ttl=64 time=852 ms
64 bytes from 10.1.2.234: icmp_seq=321 ttl=64 time=892 ms
64 bytes from 10.1.2.234: icmp_seq=322 ttl=64 time=932 ms
64 bytes from 10.1.2.234: icmp_seq=323 ttl=64 time=501 ms
64 bytes from 10.1.2.234: icmp_seq=324 ttl=64 time=6.77 ms
64 bytes from 10.1.2.234: icmp_seq=325 ttl=64 time=11.9 ms
64 bytes from 10.1.2.234: icmp_seq=326 ttl=64 time=51.8 ms
64 bytes from 10.1.2.234: icmp_seq=327 ttl=64 time=90.9 ms
64 bytes from 10.1.2.234: icmp_seq=328 ttl=64 time=131 ms
64 bytes from 10.1.2.234: icmp_seq=329 ttl=64 time=171 ms
64 bytes from 10.1.2.234: icmp_seq=330 ttl=64 time=210 ms
64 bytes from 10.1.2.234: icmp_seq=331 ttl=64 time=250 ms
64 bytes from 10.1.2.234: icmp_seq=332 ttl=64 time=138 ms
64 bytes from 10.1.2.234: icmp_seq=333 ttl=64 time=330 ms
64 bytes from 10.1.2.234: icmp_seq=334 ttl=64 time=372 ms
64 bytes from 10.1.2.234: icmp_seq=335 ttl=64 time=409 ms
64 bytes from 10.1.2.234: icmp_seq=336 ttl=64 time=449 ms
64 bytes from 10.1.2.234: icmp_seq=337 ttl=64 time=489 ms
64 bytes from 10.1.2.234: icmp_seq=338 ttl=64 time=534 ms
64 bytes from 10.1.2.234: icmp_seq=339 ttl=64 time=569 ms
64 bytes from 10.1.2.234: icmp_seq=340 ttl=64 time=611 ms
64 bytes from 10.1.2.234: icmp_seq=341 ttl=64 time=650 ms
64 bytes from 10.1.2.234: icmp_seq=342 ttl=64 time=690 ms
64 bytes from 10.1.2.234: icmp_seq=343 ttl=64 time=658 ms
64 bytes from 10.1.2.234: icmp_seq=344 ttl=64 time=770 ms
64 bytes from 10.1.2.234: icmp_seq=345 ttl=64 time=810 ms
64 bytes from 10.1.2.234: icmp_seq=346 ttl=64 time=850 ms
64 bytes from 10.1.2.234: icmp_seq=347 ttl=64 time=890 ms
64 bytes from 10.1.2.234: icmp_seq=348 ttl=64 time=699 ms
64 bytes from 10.1.2.234: icmp_seq=349 ttl=64 time=971 ms
64 bytes from 10.1.2.234: icmp_seq=350 ttl=64 time=7.21 ms
64 bytes from 10.1.2.234: icmp_seq=351 ttl=64 time=11.3 ms
64 bytes from 10.1.2.234: icmp_seq=352 ttl=64 time=60.7 ms
```

In the reverse direction (pinging from the unit) is fine:
```
64 bytes from 10.1.2.1: icmp_seq=2 ttl=64 time=0.418 ms
64 bytes from 10.1.2.1: icmp_seq=3 ttl=64 time=0.360 ms
64 bytes from 10.1.2.1: icmp_seq=4 ttl=64 time=0.360 ms
64 bytes from 10.1.2.1: icmp_seq=5 ttl=64 time=0.412 ms
64 bytes from 10.1.2.1: icmp_seq=6 ttl=64 time=0.475 ms
64 bytes from 10.1.2.1: icmp_seq=7 ttl=64 time=0.403 ms
64 bytes from 10.1.2.1: icmp_seq=8 ttl=64 time=0.485 ms
64 bytes from 10.1.2.1: icmp_seq=9 ttl=64 time=0.428 ms
64 bytes from 10.1.2.1: icmp_seq=10 ttl=64 time=0.502 ms
64 bytes from 10.1.2.1: icmp_seq=11 ttl=64 time=0.444 ms
```

Interactive performance via a shell shows a lagging response that makes manual interaction over a shell awkward.

General throughput is OK but not great at around 10Mbit/s (~1.1MB/sec) sending data (using scp) to the device.
Receiving data from the device via scp is much lower without throughput around 400kbit/sec (~350kB/sec).

I believe this issue started around the 15.xx driver. The 14.xx (?) driver in the mainline linux-at91 kernal had much better throughputaround 30-40Mbit/sec (and I don't think lagged)  but would hard lock the module requiring a power cycle of the WILC1000 to recover it. 

I note though that the whole driver was recently deleted from linux-at91 
https://github.com/linux4sam/linux-at91/commit/a6411e708c4488ec6fd7c4f9e6cc12659ff36292
... whilst it remains in Linus' kernel with some recent modifications that also appear here.

AdhamAbozaeid, 2018-06-26:
I'm able to reproduce the lag in pinging to SAMA5D4. Please open a support ticket for further debugging on http://microchipsupport.force.com.
For the throughput, release 15.1 on sama5d4 has a UDP downnlink throughput of 46Mbps and uplink of 21Mbps in a shielded environment

ShonkyCH, 2018-06-26:
Case number 00308992

I also note that two threads:
```
root 2195 0.0 0.0 0 0 ? D Jun26 0:11 [K_TXQ_TASK]
root 2196 0.0 0.0 0 0 ? D Jun26 0:00 [WILC_DEBUG]
```
... run in uninterruptible sleep (resulting in a permanent system load average of 2.00 I think). Not sure if this is related.

ShonkyCH, 2018-09-11:
This has basically gone nowhere in Salesforce. How do we get this fixed?

AdhamAbozaeid, 2018-09-11:
Hi Shonky

Further investigation showed that there's a latency on the specific combination of linux 4.9 + SAMA5d4 boards, specifically while handling interrupts.
Such latency wasn't observed with other linux versions on sama5d4 + same WILC driver/FW, nor on linux 4.9 + other hosts + same WILC driver/FW.
Accordingly, this was assigned a lower priority for now since most probably it's not related to WILC's implementation.

ShonkyCH, 2018-09-11:
Which Linux kernel versions do not have the issue then please.

AdhamAbozaeid, 2018-09-20:
> So Microchip are not prepared to give any priority to an issue with a Microchip CPU running a Microchip Wifi module with the Microchip driver+firmware on the current major version of the Linux kernel? Great support.

Sorry that your priorities doesn't align with ours at the moment. You can escalate further on the salesforce ticket, and again the issue will be prioritized appropriately.

> Or are you saying this is a hardware issue with the Microchip SAMA5D4? If you're going to say it's "probably not the WILC implementation", then where is the likely issue?
It's not clear yet where the problem is, but we don't think it's a HW issue, since as I said before, the problem isn't observed on SAMA5D4 + linux 3.18 for example.

> Does this happen with A5D3 or A5D2 CPUs?
The same issue wasn't ovserved on A5D2 + kernel 4.19

> Which interrupts are suffering from latency (which I presume is up to 1 second given the issue we are seeing)?
The WILC has a single interrupt to the host, that is an in band SDIO interrupt

ShonkyCH, 2018-09-21:
Well rolling back to kernel 3.anything is not acceptable.

Does Microchip have a supported Wifi device on the A5D4 or do I need to look at changing the design to another vendor?

The Salesforce ticket has been basically ignored for 3 months and there is no way to escalate.

AdhamAbozaeid, 2018-09-21:
Sorry to hear that @ShonkyCH .
I'll follow up with the salesforce support team

ssuraci, 2018-12-28:
I can confirm that on a SAMA5D3 derived board (kernel 3.10-at91)  the laggy response problem is resolved with the latest patch which fixes high latency on interrupt handling.

However the problem with the two kernel threads stuck in "D" state that result in permanent load average over 2.00 is still here ... 

> Case number 00308992
> 
> I also note that two threads:
> 
> ```
> root 2195 0.0 0.0 0 0 ? D Jun26 0:11 [K_TXQ_TASK]
> root 2196 0.0 0.0 0 0 ? D Jun26 0:00 [WILC_DEBUG]
> ```
> ... run in uninterruptible sleep (resulting in a permanent system load average of 2.00 I think). Not sure if this is related.

commit 130cdeb4 (2018-12-20): staging: wilc1000: High latency on handling FW interrupts

With pm_runtime enabled, the host shutsdown its SD clock which causes
high latency on handling interrupts on SAMA5D4 and new kernels.
The impact of that was high ping latency when pinging from other STAs
to WILC.

Fixes: #13
commit dcece8ad (2018-12-20): staging: wilc1000: High latency on handling FW interrupts

With pm_runtime enabled, the host shutsdown its SD clock which causes
high latency on handling interrupts on SAMA5D4 and new kernels.
The impact of that was high ping latency when pinging from other STAs
to WILC.

Fixes #13
commit 06108eeb (2018-12-20): staging: wilc1000: High latency on handling FW interrupts

With pm_runtime enabled, the host shutsdown its SD clock which causes
high latency on handling interrupts on SAMA5D4 and new kernels.
The impact of that was high ping latency when pinging from other STAs
to WILC.

Closes #13
commit efe02b80 (2019-01-11): staging: wilc1000: High latency on handling FW interrupts

With pm_runtime enabled, the host shutsdown its SD clock which causes
high latency on handling interrupts on SAMA5D4 and new kernels.
The impact of that was high ping latency when pinging from other STAs
to WILC.

Closes #13
closed 2019-02-13 by AdhamAbozaeid