linux4wilc/driver #14: Issue compiling in SAMA5D2

mcoromac, 2018-07-05:
Hello, I'm trying to use ATWILC3000 on a custom board with a SAMA5D2 processor.
We are using an sdio interface. The connections are the following:
Processor                    ATWILC3000
SDMMC_CK      ->     SD_CLK (Pin 22)
SDMMC_DAT0   ->    SD_DAT0 (Pin 24)
SDMMC_DAT1  ->     SD_DAT1 (Pin 25)
SDMMC_DAT2  ->      SD_DAT2 (Pin 26)
SDMMC_DAT3  ->     SD_DAT3 (Pin 27)
SDMMC_RSTN  ->    RESETEN (Pin 7)
SDMMC_CMD   ->     SD_CMD (Pin 23)
SDMMC_CD   ->        CHIP_EN (Pin 19)
PB24               ->        IRQN (Pin 33)  

Device Tree configuration:
&sdmmc1 {
	compatible= "atmel,sama5d2-sdhci";
	bus-width = <4>;
	pinctrl-names = "default";
	pinctrl-0 = <&pinctrl_sdmmc1_default>;
	max-frequency=<2000000>;
	non-removable;
	cd-inverted;
	status = "okay";
};

pinctrl_sdmmc1_default: sdmmc1_default {
		cmd_data {
			pinmux = <PIN_PA28__SDMMC1_CMD>,
				 <PIN_PA18__SDMMC1_DAT0>,
				 <PIN_PA19__SDMMC1_DAT1>,
				 <PIN_PA20__SDMMC1_DAT2>,
				 <PIN_PA21__SDMMC1_DAT3>;
			bias-pull-up;
		};

		ck_cd_vddsel {
			pinmux = <PIN_PA22__SDMMC1_CK>,
				<PIN_PA27__SDMMC1_RSTN>,
				 <PIN_PA30__SDMMC1_CD>; 
			bias-disable;
		};
	};

I have also already added the driver and the firmware in my kernel. 
However this is my last log: 
[   21.130000] [wilc start]: fail read reg 0x3b0000
[   21.130000] wilc_sdio mmc1:0001:1: wilc_sdio_cmd53..failed, err(-110)
[   21.140000] wilc_sdio mmc1:0001:1: Failed cmd53, read reg (00001000)...
[   21.150000] wilc_sdio mmc1:0001:1: wilc_sdio_cmd53..failed, err(-110)
[   21.150000] wilc_sdio mmc1:0001:1: Failed cmd53, read reg (00001118)...
[   21.160000] wilc_sdio mmc1:0001:1 p2p0: ERR [init_chip: 1954] fail read reg 0x1118
[   21.170000] wilc_sdio mmc1:0001:1 p2p0: ERR [wilc_wlan_initialize: 1157] Initializing WILC_Wlan FAILED
[   21.180000] wilc_sdio mmc1:0001:1 p2p0: INFO [wlan_deinit_locks]De-Initializing Locks
[   21.180000] wilc_sdio mmc1:0001:1 p2p0: ERR [wilc_wlan_initialize: 1244] WLAN initialization FAILED
[   21.190000] wilc_sdio mmc1:0001:1 p2p0: ERR [wilc_mac_open: 1294] Failed to initialize wilc
[   21.320000] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_mac_open]MAC OPEN[c41f9000] p2p0
[   21.320000] power up request for already powered up source Wifi
[   21.330000] Device already up. request source is Wifi
[   21.330000] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_mac_open]*** re-init ***
[   21.340000] wilc_sdio mmc1:0001:1 p2p0: INFO [wlan_init_locks]Initializing Locks ...
[   21.350000] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_wlan_init]Initializing WILC_Wlan
[   21.360000] wilc_sdio mmc1:0001:1: clocks still OFF. Retrying
[   21.370000] wilc_sdio mmc1:0001:1: clocks still OFF. Retrying
[   21.370000] wilc_sdio mmc1:0001:1: clocks still OFF. Retrying
[   21.380000] wilc_sdio mmc1:0001:1: clocks still OFF. Retrying
[   21.390000] wilc_sdio mmc1:0001:1: wilc_sdio_cmd53..failed, err(-110)
[   21.390000] wilc_sdio mmc1:0001:1: Failed cmd53, read reg (003b0000)...

I have attached the full log. 
[iwlan_log.txt](https://github.com/linux4wilc/driver/files/2168377/iwlan_log.txt)
The first strange behavior I notice are these lines:
[    2.770000] wilc_sdio mmc1:0001:1: WILC setting default Reset GPIO to 60. Got -2
[    2.780000] wilc_sdio mmc1:0001:1: WILC setting default Chip Enable GPIO to 94. Got -2
[    2.790000] wilc_sdio mmc1:0001:1: WILC setting default IRQ GPIO to 91. Got -2

Am I missing something in my device tree?

mcoromac, 2018-07-06:
My linux kernel version is 5.8 and the driver I added came from the master branch. I didn't notice any differences between master and dev branches.

AdhamAbozaeid, 2018-07-06:
HI @mcoromac 
How are the chip_en and reset_n lines controlled on your setup?

The lines you highlighted from the logs means that the GPIOs that controls power and IRQ weren't found in the device tree, and harcoded values for the GPIO numbers were used in the driver. The hardcoded values correspond to:
* reset_n: GPIO 60
* chip_en: GPIO 94
* irq: GPIO 91 (used for host wake up only if you are using in-band interrupt)

If chip_en and reset_n are connected to your host, then you need to either change the hardcoded values in the driver to match your setup, or change the device tree as follows:
`mmc1: mmc@fc000000 {
	pinctrl-names = "default";
	pinctrl-0 = <&pinctrl_mmc1_clk_cmd_dat0 &pinctrl_mmc1_dat1_3>;
	non-removable;
	vmmc-supply = <&vcc_mmc1_reg>;
	vqmmc-supply = <&vcc_3v3_reg>;
	status = "okay";
	wilc_sdio@0{
		compatible = "atmel,wilc_sdio";
		gpio_reset = <&pioB 28 0>;
		gpio_chip_en = <&pioC 30 0>;
		gpio_irq = <&pioC 27 0>;
		status = "okay";
	};
	slot@0 {
	reg = <0>;
	bus-width = <4>;
	};
};`

mcoromac, 2018-07-09:
Hi @AdhamAbozaeid 
Ok, I corrected them and it displays this:
[    2.790000] wilc_sdio mmc1:0001:1: WILC got 27 for gpio_reset
[    2.800000] wilc_sdio mmc1:0001:1: WILC got 30 for gpio_chip_en
[    2.800000] wilc_sdio mmc1:0001:1: WILC got 56 for gpio_irq
Atleast now is not on an infinite loop displaying a bunch of errors. Ignore the pin numbers they are handled different on sama5d2.  

This is my latest log: 
[    6.370000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[c7a08000] wlan0
[    6.370000] WILC POWER UP
[    6.380000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_init_host_int]Host[c7a08000][c7a0b800]
forked to background, child pid 181
[    6.400000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
[    6.400000] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_init_locks]Initializing Locks ...
[    6.410000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
[    6.420000] wilc_sdio mmc1:0001:1: SDIO speed: 2000000
[    6.440000] wilc_sdio mmc1:0001:1: wilc_sdio_cmd52..failed, err(-110)
[    6.440000] wilc_sdio mmc1:0001:1: Fail cmd 52, enable csa...
[    6.450000] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_initialize: 1157] Initializing WILC_Wlan FAILED
[    6.460000] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinit_locks]De-Initializing Locks
[    6.460000] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_initialize: 1244] WLAN initialization FAILED
[    6.470000] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_mac_open: 1294] Failed to initialize wilc
[    6.500000] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_send_config_pkt: 1929] Set Timed out
[    6.500000] wilc_sdio mmc1:0001:1 wlan0: ERR [handle_set_wfi_drv_handler: 445] Failed to set driver handler
[    6.510000] wilc_sdio mmc1:0001:1 wlan0: Host cmd 24 failed
[    6.520000] wilc_sdio mmc1:0001:1 wlan0: ERR [host_if_work: 3000] [Host Interface] undefined
[    6.540000] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_mac_open]MAC OPEN[c7a6c800] p2p0
[    6.550000] power up request for already powered up source Wifi
[    6.550000] Device already up. request source is Wifi
[    6.560000] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_init_host_int]Host[c7a6c800][c7a6b400]
[    6.570000] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_mac_open]*** re-init ***
[    6.580000] wilc_sdio mmc1:0001:1 p2p0: INFO [wlan_init_locks]Initializing Locks ...
[    6.580000] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_wlan_init]Initializing WILC_Wlan
[    6.590000] wilc_sdio mmc1:0001:1: SDIO speed: 2000000
[    6.610000] wilc_sdio mmc1:0001:1: wilc_sdio_cmd52..failed, err(-110)
[    6.620000] wilc_sdio mmc1:0001:1: Fail cmd 52, enable csa...
[    6.620000] wilc_sdio mmc1:0001:1 p2p0: ERR [wilc_wlan_initialize: 1157] Initializing WILC_Wlan FAILED
[    6.630000] wilc_sdio mmc1:0001:1 p2p0: INFO [wlan_deinit_locks]De-Initializing Locks
[    6.640000] wilc_sdio mmc1:0001:1 p2p0: ERR [wilc_wlan_initialize: 1244] WLAN initialization FAILED
[    6.650000] wilc_sdio mmc1:0001:1 p2p0: ERR [wilc_mac_open: 1294] Failed to initialize wilc
[    6.660000] wilc_sdio mmc1:0001:1 p2p0: ERR [wilc_send_config_pkt: 1929] Set Timed out
[    6.660000] wilc_sdio mmc1:0001:1 p2p0: ERR [handle_set_wfi_drv_handler: 445] Failed to set driver handler
[    6.670000] wilc_sdio mmc1:0001:1 p2p0: Host cmd 24 failed
[    6.690000] wilc_sdio mmc1:0001:1 p2p0: ERR [host_if_work: 3000] [Host Interface] undefined

Could you help me out with this?

AdhamAbozaeid, 2018-07-09:
Hi @mcoromac 
It looks like there's SDIO failures on this setup. Possibilities are:
* The SD controller configuration on the host, or the SD data lines aren't connected properly.
* The chip isn't powered on correctly, which I doubt since the sdio probe function is executed, which won't happen if WILC was powered down.

Please open a support ticket on http://microchipsupport.force.com for further more support

mcoromac, 2018-07-10:
Hi, @AdhamAbozaeid 
How can I be sure if I'm making use of the interrupt pin? 
I thought of adding some printks to the driver.

AdhamAbozaeid, 2018-07-10:
The sdio doesn't use the IRQ line; it rather uses the muxed interrupt, which muxes SD Dat1 line to deliver interrupts, as well as data.
I don't expect that you would receive any interrupts in your case, since the FW wasn't downloaded.
From the logs I see it's failing with the first cmd52 sent in sdio_init(), which makes me suspect the sd lines.

AdhamAbozaeid, 2018-07-10:
Last log I see here was the one sent yesterday here https://github.com/linux4wilc/driver/issues/14#issuecomment-403635698

In this one, I'm seeing this log:
[ 6.370000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[c7a08000] wlan0
[ 6.370000] WILC POWER UP
[ 6.380000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_init_host_int]Host[c7a08000][c7a0b800]
forked to background, child pid 181
[ 6.400000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
[ 6.400000] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_init_locks]Initializing Locks ...
[ 6.410000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
[ 6.420000] wilc_sdio mmc1:0001:1: SDIO speed: 2000000
[ 6.440000] wilc_sdio mmc1:0001:1: **wilc_sdio_cmd52..failed, err(-110)**
[ 6.440000] wilc_sdio mmc1:0001:1: Fail cmd 52, enable csa...
[ 6.450000] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_initialize: 1157] Initializing WILC_Wlan FAILED
[ 6.460000] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinit_locks]De-Initializing Locks
[ 6.460000] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_initialize: 1244] WLAN initialization FAILED

In this one i'm not seeing the FW download logs. Am I missing something?
If the FW downloads successfully, you should see wilc_sdio_interrupt() executed, which calls wilc_handle_isr()

mcoromac, 2018-07-10:
Oh no, I meant the .txt attached at the beggining of this thread.
This one, this happens when I leave the default gpio pins. 
[    6.350000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[c41f7800] wlan0
[    6.350000] WILC POWER UP
[    6.360000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_init_host_int]Host[c41f7800][c41ee800]
forked to background, child pid 180
[    6.380000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
[    6.380000] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_init_locks]Initializing Locks ...
[    6.390000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
[    6.400000] wilc_sdio mmc1:0001:1: SDIO speed: 2000000
[    6.430000] wilc_sdio mmc1:0001:1: chipid 003000d0
[    6.440000] wilc_sdio mmc1:0001:1 wlan0: INFO [init_chip]Bootrom sts = d
[    6.440000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_initialize]WILC Initialization done
[    6.450000] wilc_sdio mmc1:0001:1 wlan0: INFO [init_irq]IRQ request succeeded IRQ-NUM= 128 on GPIO: 91
[    6.460000] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Initializing Threads ...
[    6.470000] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Creating kthread for transmission
[    6.480000] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Creating kthread for Debugging
[    6.500000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]Detect chip WILC3000
[    6.500000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]loading firmware mchp/wilc3000_wifi_firmware.bin
[    6.520000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]WLAN firmware: mchp/wilc3000_wifi_firmware.bin
[    6.530000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_firmware_download]Downloading Firmware ...
[    6.530000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_firmware_download]Downloading firmware size = 136076
Starting sshd: [    6.730000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_firmware_download]Offset = 119296
[    6.760000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_firmware_download]Offset = 136076
[    6.760000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_firmware_download]Download Succeeded
[    6.770000] wilc_sdio mmc1:0001:1 wlan0: INFO [linux_wlan_start_firmware]Starting Firmware ...
[    6.780000] wilc_sdio mmc1:0001:1 wlan0: INFO [linux_wlan_start_firmware]Waiting for FW to get ready ...
[    6.870000] wilc_sdio mmc1:0001:1 wlan0: INFO [linux_wlan_start_firmware]Firmware successfully started
[    6.890000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_initialize]WILC Firmware Ver = WILC_WIFI_FW_REL_15_01_RC3 Build: 9792
[    6.900000] wilc_sdio mmc1:0001:1 wlan0: INFO [linux_wlan_init_test_config]Start configuring Firmware
[    7.010000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]Mac address: fa:f0:05:c0:2e:ba
[    7.020000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 1. 
[    7.030000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[0]: 33:33:0:0:0:1
[    7.040000] wilc_sdio mmc1:0001:1 wlan0: INFO [set_power_mgmt] Power save Enabled= 1 , TimeOut = -1
[    7.050000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 2. 
[    7.060000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[0]: 33:33:0:0:0:1
[    7.060000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[1]: 1:0:5e:0:0:1
[    7.090000] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_mac_open]MAC OPEN[c41f9000] p2p0
[    7.100000] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_init_host_int]Host[c41f9000][c41fbc00]
[    7.100000] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_mac_open]*** re-init ***
[    7.110000] wilc_sdio mmc1:0001:1 p2p0: WRN [wilc_wlan_initialize: 1246]wilc already initialized
[    7.150000] wilc_sdio mmc1:0001:1: clocks still OFF. Retrying
[    7.160000] wilc_sdio mmc1:0001:1: clocks still OFF. Retrying
[    7.190000] wilc_sdio mmc1:0001:1: clocks still OFF. Retrying
[    7.190000] wilc_sdio mmc1:0001:1: clocks still OFF. Retrying
[    7.210000] wilc_sdio mmc1:0001:1: clocks still OFF. Retrying
[    7.230000] wilc_sdio mmc1:0001:1: clocks still OFF. Retrying
[    7.230000] wilc_sdio mmc1:0001:1: clocks still OFF. Retrying
[    7.250000] wilc_sdio mmc1:0001:1: clocks still OFF. Retrying
[    7.250000] wilc_sdio mmc1:0001:1: clocks still OFF. Retrying
OK
[    7.310000] wilc_sdio mmc1:0001:1: clocks still OFF. Retrying
[    7.310000] wilc_sdio mmc1:0001:1: clocks still OFF. Retrying
[    7.320000] wilc_sdio mmc1:0001:1: clocks still OFF. Retrying
[    7.340000] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_mac_open]Mac address: f8:f0:05:c0:2e:ba
[    7.340000] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_set_multicast_list]Setting mcast List with count = 1. 
[    7.350000] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_set_multicast_list]Entry[0]: 33:33:0:0:0:1
[    7.360000] wilc_sdio mmc1:0001:1 p2p0: INFO [set_power_mgmt] Power save Enabled= 1 , TimeOut = -1
[    7.370000] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_set_multicast_list]Setting mcast List with count = 2. 
[    7.380000] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_set_multicast_list]Entry[0]: 33:33:0:0:0:1
[    7.390000] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_set_multicast_list]Entry[1]: 1:0:5e:0:0:1
[    7.400000] wilc_sdio mmc1:0001:1: clocks still OFF. Retrying
[    7.400000] wilc_sdio mmc1:0001:1: clocks still OFF. Retrying
[    7.410000] wilc_sdio mmc1:0001:1: clocks still OFF. Retrying
[... cut ...]
[   14.620000] wilc_sdio mmc1:0001:1: Failed cmd53, read reg (0000106c)...
[   14.630000] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_handle_txq: 1073] fail read reg vmm_tbl_entry..
[   14.970000] wilc_sdio mmc1:0001:1: clocks still OFF. Retrying
[   14.980000] wilc_sdio mmc1:0001:1: clocks still OFF. Retrying
[   14.980000] wilc_sdio mmc1:0001:1: clocks still OFF. Retrying
[   14.990000] wilc_sdio mmc1:0001:1: clocks still OFF. Retrying
[   15.000000] wilc_sdio mmc1:0001:1: wilc_sdio_cmd53..failed, err(-110)

AdhamAbozaeid, 2018-07-10:
Oh I see. I'm not sure how the chip_en and reset_n are controlled in this case, but looks like these GPOIs are pulled high or configured high by default, which makes the chip powered up.

In this log, it looks like there's a problem with powersave. I see that the supplicant is enabling it by default while initializing, which isn't the case on my setup 
Try disabling it by overriding power_mode to be NO_POWERSAVE in handle_power_management()

AdhamAbozaeid, 2018-07-10:
It means DHCP operation is ongoing, but the "clocks still off" and intermittent sdio errors means the driver is failing to communicate with the chip

AdhamAbozaeid, 2018-07-10:
"The debug thread running" is a periodic debug message from the driver, but looks like changing the loglevel is causing a problem on your platform. Maybe you can disable this message in the driver, or set it to PRINT_D instead of PRINT_INFO

For the second problem you are having, this is failure to communicate with the chip, I guess maybe because of the determinant status of chip_en

mcoromac, 2018-07-10:
Hmmm... how can I set it to PRINT_D?

Yeah since with the current configuration the pins are the driver's default. 
I think it could be a problem with this:
gpio_reset = <&pioA 27 0>;
gpio_chip_en = <&pioA 30 0>;
gpio_irq = <&pioA 56 0>;

I tried to export the pins as gpio (the ones above and the drivers default) and it was succesful meaning that the driver is not using those pins.

AdhamAbozaeid, 2018-07-10:
In debug_thread(), change
`PRINT_INFO(dev, GENERIC_DBG,
					   "*** Debug Thread Running ***\n");`
to
`PRINT_D(dev, GENERIC_DBG,
					   "*** Debug Thread Running ***\n");`
or comment it out all together

mcoromac, 2018-07-13:
@AdhamAbozaeid Hi Adham, I disable powersave by overiding power_mode in line 2702. 
Here is my latest log: 
[    2.740000] mmc1: new high speed SDIO card at address 0001
[    2.740000] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
[    2.750000] (unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
[    2.760000] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Successful Registering
[    2.770000] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
[    2.780000] (unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
[    2.790000] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Successful Registering
[    2.800000] wilc_sdio mmc1:0001:1: WILC setting default Reset GPIO to 60. Got -2
[    2.810000] wilc_sdio mmc1:0001:1: WILC setting default Chip Enable GPIO to 94. Got -2
[    2.820000] wilc_sdio mmc1:0001:1: WILC setting default IRQ GPIO to 91. Got -2
[    2.830000] wifi_pm : 0 
[    2.830000] wifi_pm : 1 
[    2.840000] Bluetooth: RFCOMM socket layer initialized
[    2.840000] Bluetooth: RFCOMM ver 1.11
[    2.850000] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    2.850000] Bluetooth: BNEP socket layer initialized
[    2.860000] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[    2.860000] Bluetooth: HIDP socket layer initialized
[    2.880000] wilc_sdio mmc1:0001:1: Driver Initializing success
[    2.880000] Bluetooth: Starting self testing
[    2.890000] Bluetooth: Finished self testing
[    2.900000] EXT4-fs (mmcblk0p2): couldn't mount as ext3 due to feature incompatibilities
[    2.970000] EXT4-fs (mmcblk0p2): recovery complete
[    2.970000] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
[    2.980000] VFS: Mounted root (ext4 filesystem) on device 179:2.
[    2.990000] devtmpfs: mounted
[    2.990000] Freeing unused kernel memory: 1024K
[    3.090000] EXT4-fs (mmcblk0p2): re-mounted. Opts: data=ordered
Starting logging: OK
Populating /dev using udev: /etc/init.d/S10udev: line 47: can't create /proc/sys/kernel/hotplug: nonexistent directory
[    3.370000] udevd[123]: starting version 3.2.5
[    3.510000] udevd[124]: starting eudev-3.2.5
done
Initializing random number generator... done.
Starting system message bus: done
Starting network: [    5.890000] (NULL net_device): INFO [dev_state_ev_handler]Interface is neither WLAN0 nor P2P0
OK
Starting dhcpcd...
dev: loaded udev
no interfaces have a carrier
[    6.380000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[c7a09000] wlan0
[    6.380000] WILC POWER UP
[    6.390000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_init_host_int]Host[c7a09000][c7a0bc00]
forked to background, child pid 180
[    6.410000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
[    6.410000] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_init_locks]Initializing Locks ...
[    6.420000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
[    6.430000] wilc_sdio mmc1:0001:1: SDIO speed: 4000000
Starting gpsd: [    6.460000] wilc_sdio mmc1:0001:1: chipid 003000d0
[    6.460000] wilc_sdio mmc1:0001:1 wlan0: INFO [init_chip]Bootrom sts = d
[    6.480000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_initialize]WILC Initialization done
[    6.480000] wilc_sdio mmc1:0001:1 wlan0: INFO [init_irq]IRQ request succeeded IRQ-NUM= 128 on GPIO: 91
[    6.490000] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Initializing Threads ...
[    6.500000] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Creating kthread for transmission
[    6.510000] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Creating kthread for Debugging
[    6.520000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]Detect chip WILC3000
[    6.530000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]loading firmware mchp/wilc3000_wifi_firmware.bin
[    6.540000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]WLAN firmware: mchp/wilc3000_wifi_firmware.bin
[    6.550000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_firmware_download]Downloading Firmware ...
[    6.560000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_firmware_download]Downloading firmware size = 136076
OK
[    6.750000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_firmware_download]Offset = 119296
Starting sshd: [    6.780000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_firmware_download]Offset = 136076
[    6.780000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_firmware_download]Download Succeeded
[    6.790000] wilc_sdio mmc1:0001:1 wlan0: INFO [linux_wlan_start_firmware]Starting Firmware ...
[    6.820000] wilc_sdio mmc1:0001:1 wlan0: INFO [linux_wlan_start_firmware]Waiting for FW to get ready ...
[    6.910000] wilc_sdio mmc1:0001:1 wlan0: INFO [linux_wlan_start_firmware]Firmware successfully started
[    6.920000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_initialize]WILC Firmware Ver = WILC_WIFI_FW_REL_15_01_RC3 Build: 9792
[... 44 lines cut ...]
[   18.590000] wilc_sdio mmc1:0001:1 p2p0: INFO [dev_state_ev_handler]dev_state_ev_handler +++
[   18.600000] wilc_sdio mmc1:0001:1 p2p0: INFO [dev_state_ev_handler]dev_state_ev_handler event=NETDEV_UP c7a13800
[   18.610000] wilc_sdio mmc1:0001:1 p2p0: INFO [handle_pwrsave_during_obtainingIP]IP obtained , Enable (Scan-Set PowerSave)
[... cut ...]
[   68.140000] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_mac_xmit]DHCP Message transmitted, type:35 1 1
[   69.610000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_xmit]DHCP Message transmitted, type:35 1 1
[   71.520000] wilc_sdio mmc1:0001:1 wlan0: INFO [debug_thread]*** Debug Thread Running ***

Does this seems normal to you? Now it seems more reliable to be honest. 

Just one question, since apparently the device tree labels didn't work, where do I need to hardcode them at the code? My guess is that in linux_sdio_probe().

AdhamAbozaeid, 2018-07-13:
Hi @mcoromac 
Yeah, that looks ok to me. you should have an IP by now.
If you will not be using the device tree to get the GPIO numbers, you can hardcode them in the driver in
#define GPIO_NUM
#define GPIO_NUM_CHIP_EN
#define GPIO_NUM_RESET
in wilc_wlan.h

When the driver fails to get them from the device tree, it will go ahead and use these definitions.

mcoromac, 2018-07-17:
@AdhamAbozaeid I think that's all. Thanks for your help.

closed 2018-07-17 by mcoromac

AdhamAbozaeid, 2018-07-17:
@mcoromac anytime :)