linux4wilc/driver #15: iMX6 - master branch - WILC_SPI spi1.0: clocks still OFF. Retrying

prabillard, 2018-08-08:
I'm trying to use the ATWILC3000 evaluation board with an Imx6Rex board (same base as SabreSD) and I get (as an infinite loop) error : " WILC_SPI spi1.0: clocks still OFF. Retrying"  (can't stop it with ctrl+c ).

- SPI communication is OK
- Firmware is correctly loaded and started
- the issue appears when the driver attempts to wake-up the chip.

imx6-rexultra login: root
root@imx6-rexultra:~# ifconfig wlan0 up
WILC_SPI spi1.0 wlan0: INFO [wilc_mac_open]MAC OPEN[d62a4800] wlan0
WILC POWER UP
WILC_SPI spi1.0 wlan0: INFO [wilc_init_host_int]Host[d62a4800][d625f800]
WILC_SPI spi1.0 wlan0: INFO [wilc_mac_open]*** re-init ***
WILC_SPI spi1.0 wlan0: INFO [wlan_init_locks]Initializing Locks ...
WILC_SPI spi1.0 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
WILC_SPI spi1.0 wlan0: INFO [init_chip]Bootrom sts = c
WILC_SPI spi1.0 wlan0: INFO [wilc_wlan_initialize]WILC Initialization done
WILC_SPI spi1.0 wlan0: INFO [init_irq]IRQ request succeeded IRQ-NUM= 220 on GPIO: 176
WILC_SPI spi1.0 wlan0: INFO [wlan_initialize_threads]Initializing Threads ...
WILC_SPI spi1.0 wlan0: INFO [wlan_initialize_threads]Creating kthread for transmission
WILC_SPI spi1.0 wlan0: INFO [wlan_initialize_threads]Creating kthread for Debugging
WILC_SPI spi1.0 wlan0: INFO [wilc_wlan_get_firmware]Detect chip WILC3000
WILC_SPI spi1.0 wlan0: INFO [wilc_wlan_get_firmware]loading firmware mchp/wilc3000_wifi_firmware.bin
WILC_SPI spi1.0 wlan0: INFO [wilc_wlan_get_firmware]WLAN firmware: mchp/wilc3000_wifi_firmware.bin
WILC_SPI spi1.0 wlan0: INFO [wilc_firmware_download]Downloading Firmware ...
WILC_SPI spi1.0 wlan0: INFO [wilc_wlan_firmware_download]Downloading firmware size = 136076
WILC_SPI spi1.0 wlan0: INFO [wilc_wlan_firmware_download]Offset = 119296
WILC_SPI spi1.0 wlan0: INFO [wilc_wlan_firmware_download]Offset = 136076
WILC_SPI spi1.0 wlan0: INFO [wilc_firmware_download]Download Succeeded
WILC_SPI spi1.0 wlan0: INFO [linux_wlan_start_firmware]Starting Firmware ...
WILC_SPI spi1.0 wlan0: INFO [linux_wlan_start_firmware]Waiting for FW to get ready ...
random: nonblocking pool is initialized
WILC_SPI spi1.0 wlan0: INFO [linux_wlan_start_firmware]Firmware successfully started
WILC_SPI spi1.0 wlan0: INFO [wilc_wlan_initialize]WILC Firmware Ver = WILC_WIFI_FW_REL_15_01_RC3 Build: 9792
WILC_SPI spi1.0 wlan0: INFO [linux_wlan_init_test_config]Start configuring Firmware
WILC_SPI spi1.0 wlan0: INFO [wilc_mac_open]Mac address: fa:f0:05:c0:26:c8
WILC_SPI spi1.0 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 1.
WILC_SPI spi1.0 wlan0: INFO [wilc_set_multicast_list]Entry[0]: 33:33:0:0:0:1
WILC_SPI spi1.0 wlan0: INFO [set_power_mgmt] Power save Enabled= 1 , TimeOut = -1
WILC_SPI spi1.0 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 2.
WILC_SPI spi1.0 wlan0: INFO [wilc_set_multicast_list]Entry[0]: 33:33:0:0:0:1
WILC_SPI spi1.0 wlan0: INFO [wilc_set_multicast_list]Entry[1]: 1:0:5e:0:0:1
WILC_SPI spi1.0 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 3.
WILC_SPI spi1.0 wlan0: INFO [wilc_set_multicast_list]Entry[0]: 33:33:0:0:0:1
WILC_SPI spi1.0 wlan0: INFO [wilc_set_multicast_list]Entry[1]: 1:0:5e:0:0:1
WILC_SPI spi1.0 wlan0: INFO [wilc_set_multicast_list]Entry[2]: 33:33:ff:c0:26:c8
root@imx6-rexultra:~# WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
[... 196 lines cut ...]
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying
WILC_SPI spi1.0: clocks still OFF. Retrying

AdhamAbozaeid, 2018-08-08:
Hi @prabillard 
There's an issue that we are aware of when powersave is enabled while the driver is initializing, that doesn't happen if powersave was enabled after initializatin.
For now, you can either disable wpa_cli from enable powersave while initializing, and enable it afterwards, or disable powersave all together till we work on a fix for that.

closed 2018-09-29 by AdhamAbozaeid

howey, 2019-03-20:
In my case I had to go into my Linux kernel config and disable CONFIG_CFG80211_DEFAULT_PS. This was default enabled. When I disabled this I could load the module fine.

AdhamAbozaeid, 2019-03-20:
I tested this on SAMA5D4 + Kernel 4.9 using the master branch, and couldn't see failures while loading the module. I could see the "clocks still OFF" message, but this is warning only message