linux4wilc/driver #16: Unable to Compile driver in linux 3.10.108

GokulTechno, 2018-08-13:
I'm compiling driver for ATWILC3000 module interfaced through SDIO interface. While try to compile the linux driver. I'm receiving following error message. Help me resolve this issue. And how to define SDIO of the platform in the driver ?

Linux kernel version: 3.10.108

```
  CC      drivers/net/wireless/wilc/wilc_wfi_cfgoperations.o
drivers/net/wireless/wilc/wilc_wfi_cfgoperations.c: In function 'wilc_create_wiphy':
drivers/net/wireless/wilc/wilc_wfi_cfgoperations.c:2921:13: error: 'struct wiphy' has no member named 'wowlan'
scripts/Makefile.build:308: recipe for target 'drivers/net/wireless/wilc/wilc_wfi_cfgoperations.o' failed
make[4]: *** [drivers/net/wireless/wilc/wilc_wfi_cfgoperations.o] Error 1
scripts/Makefile.build:455: recipe for target 'drivers/net/wireless/wilc' failed
make[3]: *** [drivers/net/wireless/wilc] Error 2
scripts/Makefile.build:455: recipe for target 'drivers/net/wireless' failed
make[2]: *** [drivers/net/wireless] Error 2
scripts/Makefile.build:455: recipe for target 'drivers/net' failed
make[1]: *** [drivers/net] Error 2
Makefile:800: recipe for target 'drivers' failed
make: *** [drivers] Error 2
```

AdhamAbozaeid, 2018-08-13:
Hi @GokulTechno 
It looks like the linux kernel version you are using doesn't have a wowlan member in struct wiphy, or CONFIG_PM isn't defined on your linux kernel. (check https://elixir.bootlin.com/linux/v4.9/source/include/net/cfg80211.h#L3446)

Gokulgithhub, 2018-08-17:
Hi @AdhamAbozaeid 
Thanks for the support. I'm using Linux 3.10.108. I'm able to compile the driver after enabling **CONFIG_PM** in the linux kernel.
How do I **define the SDIO** interface in the driver ? (I'm having two mmc/sdio interface mmc0, mmc1)

GokulTechno, 2018-08-17:
Hi @AdhamAbozaeid 
Thanks for the support. I'm using Linux 3.10.108. I'm able to compile the driver after enabling **CONFIG_PM** in the linux kernel.
How do I define the **SDIO** interface in the driver ? (I'm having two mmc/sdio interface mmc0, mmc1)

AdhamAbozaeid, 2018-08-18:
Hi @GokulTechno 
Linux will know which mmc# to use using the vendor and device IDs defined in wilc_sdio.c while registering the sdio driver wilc_sdio_driver
Linux will read the corresponding IDs from WILC's HW, and map the correct mmc# to the driver

AdhamAbozaeid, 2018-08-23:
Please reopen if you are still having problems

closed 2018-08-23 by AdhamAbozaeid