linux4wilc/driver #17: wilc1000 on linux 3.10

wrdec, 2018-09-20:
Hello, I'm currently using wilc1000 (master version) on my linux 3.10.

During kernel boot, wilc driver initialization is leading to a kernel NULL pointer dereference.

```
[    2.020000] (unregistered net_device): INFO [wilc_create_wiphy]Registering wifi device
[    2.030000] (unregistered net_device): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
[    5.730000] macb f802c000.ethernet eth0: link up (100/Full)
[    5.730000] (unregistered net_device): INFO [wilc_create_wiphy]Successful Registering
[    5.740000] (unregistered net_device): INFO [wilc_create_wiphy]Registering wifi device
[    5.750000] (unregistered net_device): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
[    5.760000] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[    5.760000] (NULL net_device): INFO [dev_state_ev_handler]Interface is neither WLAN0 nor P2P0
[    5.770000] (NULL net_device): INFO [dev_state_ev_handler]Interface is neither WLAN0 nor P2P0
[    5.780000] Unable to handle kernel NULL pointer dereference at virtual address 00000272
[    5.790000] pgd = c0004000
[    5.790000] [00000272] *pgd=00000000
[    5.800000] Internal error: Oops: 5 [#1] ARM
[    5.800000] Modules linked in:
[    5.800000] CPU: 0 PID: 1 Comm: swapper Not tainted 3.10.0+ #27
[    5.800000] task: c344dc00 ti: c344e000 task.ti: c344e000
[    5.800000] PC is at __netdev_printk+0x14/0xc0
[    5.800000] LR is at netdev_info+0x30/0x44
[    5.800000] pc : [<c0272450>]    lr : [<c027252c>]    psr: 20000013
[    5.800000] sp : c344fd90  ip : c0436482  fp : 00000008
[    5.800000] r10: c344fecc  r9 : 00000000  r8 : 00000001
[    5.800000] r7 : c37dcd40  r6 : c37dcd6b  r5 : c344fdb8  r4 : 0000005a
[    5.800000] r3 : c344fdb4  r2 : c344fdb8  r1 : 0000005a  r0 : c0436482
[    5.800000] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment kernel
[    5.800000] Control: 10c53c7d  Table: 20004059  DAC: 00000015
[    5.800000] Process swapper (pid: 1, stack limit = 0xc344e230)
[    5.800000] Stack: (0xc344fd90 to 0xc3450000)
[    5.800000] fd80:                                     c0436482 0000005a c344fdb8 c344fdb4
[    5.800000] fda0: c36e3800 c37dcd40 c37dcd6b c027252c 0000005a c344fdc8 c04621a7 c344fdb4
[    5.800000] fdc0: c01fd834 c04621a7 c039da35 00000613 c053ca70 00000001 c37dcd40 c01fd7b0
[    5.800000] fde0: fffffffe 00000000 00000001 c37dcd40 00000000 00000000 c344fecc c0035c5c
[    5.800000] fe00: c05438ac c05439bc ffffffff c37dcd40 00000001 c0035eac 00000000 00000001
[    5.800000] fe20: c37dcd40 62015b0a 00000000 00000000 c36f708c c0035ed4 00000000 00000001
[    5.800000] fe40: c37dcd40 c02ba284 c37dcd40 c36f708c c36f7080 ffffff0a c36e3800 c02bb9fc
[    5.800000] fe60: 30687465 00000000 00000000 00000000 00000002 00ffffff 00000000 00000000
[    5.800000] fe80: c37cb580 c344e000 00000000 c05187a0 00000002 ffff8f7f 00001002 c05415e0
[    5.800000] fea0: 00000000 c04e8ae8 00ffffff c0544554 00000000 c04e9000 00000000 c344ff28
[    5.800000] fec0: 00000010 c37c8800 c052fc24 30687465 00000000 00000000 00000000 00000002
[    5.800000] fee0: 00ffffff 00000000 00000000 c013407c c0133fb0 c37cb9c0 00000010 c344ff28
[    5.800000] ff00: c04e774c c37cb9c4 00000000 c02b1a98 00000000 00000007 c04ee740 c0519764
[    5.800000] ff20: 00000007 c04ee740 c0519768 c0547640 c04e8bcc c04ce164 00000000 c0008668
[    5.800000] ff40: c04e8bcc c344ff38 c051976c 00000008 00000007 c04ee740 c0519768 c0547640
[    5.800000] ff60: 00000068 c04ce164 c04ee748 c04ce868 00000007 00000007 c04ce164 c344ff90
[    5.800000] ff80: c0038624 c0037f58 00000000 c03793d4 00000000 00000000 00000000 00000000
[    5.800000] ffa0: 00000000 c03793dc 00000000 c000db18 00000000 00000000 00000000 00000000
[    5.800000] ffc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[    5.800000] ffe0: 00000000 00000000 00000000 00000000 00000013 00000000 6264d758 4c380e5c
[    5.800000] [<c0272450>] (__netdev_printk+0x14/0xc0) from [<c027252c>] (netdev_info+0x30/0x44)
[    5.800000] [<c027252c>] (netdev_info+0x30/0x44) from [<c01fd834>] (dev_state_ev_handler+0x84/0x2c
4)
[    5.800000] [<c01fd834>] (dev_state_ev_handler+0x84/0x2c4) from [<c0035c5c>] (notifier_call_chain+
0x38/0x68)
[    5.800000] [<c0035c5c>] (notifier_call_chain+0x38/0x68) from [<c0035eac>] (__blocking_notifier_ca
ll_chain+0x40/0x54)
[    5.800000] [<c0035eac>] (__blocking_notifier_call_chain+0x40/0x54) from [<c0035ed4>] (blocking_no
tifier_call_chain+0x14/0x18)
[    5.800000] [<c0035ed4>] (blocking_notifier_call_chain+0x14/0x18) from [<c02ba284>] (__inet_insert
_ifa+0x190/0x1b4)
[    5.800000] [<c02ba284>] (__inet_insert_ifa+0x190/0x1b4) from [<c02bb9fc>] (devinet_ioctl+0x4d8/0x
620)
[    5.800000] [<c02bb9fc>] (devinet_ioctl+0x4d8/0x620) from [<c04e8ae8>] (ic_devinet_ioctl+0x30/0x3c
)
[    5.800000] [<c04e8ae8>] (ic_devinet_ioctl+0x30/0x3c) from [<c04e9000>] (ip_auto_config+0x434/0x7d
c)
[    5.800000] [<c04e9000>] (ip_auto_config+0x434/0x7dc) from [<c0008668>] (do_one_initcall+0x90/0x14
0)
[    5.800000] [<c0008668>] (do_one_initcall+0x90/0x140) from [<c04ce868>] (kernel_init_freeable+0xe4
/0x1a8)
[    5.800000] [<c04ce868>] (kernel_init_freeable+0xe4/0x1a8) from [<c03793dc>] (kernel_init+0x8/0xe0
)
[    5.800000] [<c03793dc>] (kernel_init+0x8/0xe0) from [<c000db18>] (ret_from_fork+0x14/0x3c)
[    5.800000] Code: e2514000 e1a0c000 e1a05002 0a00001f (e5940218) 
[    6.170000] ---[ end trace 738d9990e44c1d84 ]---
[    6.180000] Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
[    6.180000] 
```

I can avoid this issue by register dev_state_ev_handler callback at the end of wilc_netdev_init(), which seems safer.

```
diff --git a/drivers/net/wireless/atmel/wilc1000/linux_wlan.c b/drivers/net/wireless/atmel/wilc1000/linux_wlan.c
index 78254cb..f660f41 100644
--- a/drivers/net/wireless/atmel/wilc1000/linux_wlan.c
+++ b/drivers/net/wireless/atmel/wilc1000/linux_wlan.c
@@ -1737,10 +1737,6 @@ int wilc_netdev_init(struct wilc **wilc, struct device *dev, int io_type,
        wl->io_type = io_type;
        wl->hif_func = ops;
 
-#ifdef DISABLE_PWRSAVE_AND_SCAN_DURING_IP
-       register_inetaddr_notifier(&g_dev_notifier);
-#endif
-
        for (i = 0; i < NUM_CONCURRENT_IFC; i++) {
                ndev = alloc_etherdev(sizeof(struct wilc_vif));
                if (!ndev)
@@ -1793,6 +1789,10 @@ int wilc_netdev_init(struct wilc **wilc, struct device *dev, int io_type,
        }
        wilc_sysfs_init(wl->vif[0], wl->vif[1]);
 
+#ifdef DISABLE_PWRSAVE_AND_SCAN_DURING_IP
+       register_inetaddr_notifier(&g_dev_notifier);
+#endif
+
        return 0;
 }
```

But then, when I try to launch udhcpc on my target, I get a new kernel NULL pointer due to dev_state_ev_handler.

Any idea?

Thanks by advance,

AdhamAbozaeid, 2018-09-20:
Hi @wrdec 

From the backtrace, it looks like a problem with the netdevice passed to the PRINT_INFO is null in the following line:
`PRINT_INFO(vif->ndev, GENERIC_DBG,"Interface is neither WLAN0 nor P2P0\n");`
dev_state_ev_handler will be called when any interface -including eth0- gets an IP, which happened in this case, but the function should nothing unless it was a WILC interface that triggered the call.

The bug here is that the PRINT_INFO uses vif->ndev which is invalid in this case. This was fixed in the dev branch by changing the log to:
`pr_info("Interface is neither WLAN0 nor P2P0\n");`

wrdec, 2018-09-26:
Hi AdhamAbozaeid,

Thanks for your advice, indeed there is no more crashes at startup.
By the way, I still encounter some issues. 

First issue is that I do not succeed to connect to a secured network when wifi chip MAC address is not already fused. Where I can easily join a network when the MAC address is fused.

I set mac address with the following command:
`ifconfig wlan0 hw ether XX:XX:XX:XX:XX:XX`
When MAC address is set, my chip tries to connect but is disconnected after few seconds.

The second issue is when I try to ifconfig down / up. Crashes do not happen every time, but often.
```
# ifconfig wlan0 down
[  409.160000] wilc_sdio mmc0:0001:1 wlan0: INFO [remove_network_from_shadow]Num
ber of cached networks: 6
# ifconfig wlan0 up
[  416.070000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[c2cbd800] wlan0
[  416.080000] power up request for already powered up source Wifi
[  416.080000] Device already up. request source is Wifi
[  416.090000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_init_host_int]Host[c2cbd800][c2ceb600]
[  416.100000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
[  416.100000] wilc_sdio mmc0:0001:1 wlan0: INFO [wlan_init_locks]Initializing Locks ...
[  416.110000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
[  416.130000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_initialize]WILC Initialization done
[  416.140000] wilc_sdio mmc0:0001:1 wlan0: INFO [init_irq]IRQ request succeeded IRQ-NUM= 44 on GPIO:
 91
[  416.150000] wilc_sdio mmc0:0001:1 wlan0: INFO [wlan_initialize_threads]Initializing Threads ...
[  416.160000] wilc_sdio mmc0:0001:1 wlan0: INFO [wlan_initialize_threads]Creating kthread for transm
ission
[  416.170000] wilc_sdio mmc0:0001:1 wlan0: INFO [wlan_initialize_threads]Creating kthread for Debugg
ing
[  416.180000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_get_firmware]Detect chip WILC1000
[  416.190000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_get_firmware]loading firmware mchp/wilc10
00_wifi_firmware.bin
[  416.200000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_get_firmware]WLAN firmware: mchp/wilc1000
_wifi_firmware.bin
[  416.220000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_firmware_download]Downloading Firmware ...
[  416.230000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_firmware_download]Downloading firmware si
ze = 134964
[  416.260000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_firmware_download]Offset = 119660
[  416.280000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_firmware_download]Offset = 134964
[  416.290000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_firmware_download]Download Succeeded
[  416.300000] wilc_sdio mmc0:0001:1 wlan0: INFO [linux_wlan_start_firmware]Starting Firmware ...
[  416.310000] wilc_sdio mmc0:0001:1 wlan0: INFO [linux_wlan_start_firmware]Waiting for FW to get rea
dy ...
[  416.360000] wilc_sdio mmc0:0001:1 wlan0: INFO [linux_wlan_start_firmware]Firmware successfully sta
rted
[  416.390000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_initialize]WILC Firmware Ver = WILC_WIFI_
FW_REL_15_01_RC3 Build: 9792
[  416.400000] wilc_sdio mmc0:0001:1 wlan0: INFO [linux_wlan_init_test_config]Start configuring Firmw
are
[  416.520000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mac_open]Mac address: 00:50:c2:5e:10:8e
[  416.530000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Ty
pe: d0: Boolean: 1
[  416.540000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with cou
nt = 0. 
[  416.550000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_set_multicast_list]Enable mcast filter retrive
 directed pkts only
[  416.560000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with cou
nt = 0. 
[  416.570000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_set_multicast_list]Enable mcast filter retrive
 directed pkts only
[  416.580000] wilc_sdio mmc0:0001:1 wlan0: INFO [set_power_mgmt] Power save Enabled= 1 , TimeOut = -
1
[  416.590000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with cou
nt = 1. 
[  416.600000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[0]: 1:0:5e:0:0:1
[  416.610000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with cou
nt = 2. 
[... 82 lines cut ...]
[  417.760000] bf40: c053eb00 00000001 c05287d8 410fc051 c0528834 00000000 00000008 c051bf70
[  417.760000] bf60: c003e49c c023f1a8 60000013 ffffffff 442cb4da 00000061 00000000 c053eb00
[  417.760000] bf80: ffffffff c053eb00 00000001 c05287d8 c057b94c 00000000 00000000 c023f308
[  417.760000] bfa0: c051a000 ffffffff c0522078 c060fc00 20004059 c000e43c 00000000 c003da5c
[  417.760000] bfc0: 00000004 c04cd730 ffffffff ffffffff c04cd1d0 00000000 00000000 c04ed958
[  417.760000] bfe0: 00000000 10c53c7d c0522028 c04ed954 c05256dc 20008070 00000000 00000000
[  417.760000] [<c0023c54>] (cascade+0x64/0x78) from [<c0023eb8>] (run_timer_softirq+0x4c/0x174)
[  417.760000] [<c0023eb8>] (run_timer_softirq+0x4c/0x174) from [<c001f2e4>] (__do_softirq+0xb0/0x194
)
[  417.760000] [<c001f2e4>] (__do_softirq+0xb0/0x194) from [<c001f68c>] (irq_exit+0x70/0x84)
[  417.760000] [<c001f68c>] (irq_exit+0x70/0x84) from [<c000e340>] (handle_IRQ+0x64/0x84)
[  417.760000] [<c000e340>] (handle_IRQ+0x64/0x84) from [<c000d740>] (__irq_svc+0x40/0x50)
[  417.760000] [<c000d740>] (__irq_svc+0x40/0x50) from [<c023f1a8>] (cpuidle_enter_state+0x54/0xe4)
[... cut ...]
[  417.760000] [<c023f308>] (cpuidle_idle_call+0xd0/0x138) from [<c000e43c>] (arch_cpu_idle+0x8/0x38)
[  417.760000] [<c000e43c>] (arch_cpu_idle+0x8/0x38) from [<c003da5c>] (cpu_startup_entry+0x68/0xc4)
[  417.760000] [<c003da5c>] (cpu_startup_entry+0x68/0xc4) from [<c04cd730>] (start_kernel+0x27c/0x2d0
)
[  417.760000] Code: e7f001f2 e1a00006 ebffff01 e1a01005 (e5955000) 
[  417.760000] ---[ end trace 1a513e73567f4acf ]---
[  417.760000] Kernel panic - not syncing: Fatal exception in interrupt
```

Thank you.

AdhamAbozaeid, 2018-09-26:
Hi @wrdec 
Can you try with:
`ip link set dev wlan0 address xx:xx:xx:xx:xx:xx`

For the second problem, I'm trying to reproduce it while using the development branch, but can't. I tried for around 10 times the following sequence
```
modprobe wilc-sdio
ifconfig wlan0 up
ifconfig wlan0 down
ifconfig wlan0 up
ifconfig wlan0 down
......
```

Can you try to reproduce it on the dev branch and let me know if you can? The development branch has many changes implemented while working on mainlining the driver.

bmoraine, 2018-09-28:
Hello Adham,

As @wrdec is on hollidays, I take the lead on this subject.

For information, during my test I use a wilc1000 chip with a fused MAC address

Using dev branch with 15.01 firmware seems to solve the " second problem" issue.
I launch following command for a long time without facing any problem 
"watch -n 10 "ifconfig wlan0 up; sleep 5; ifconfig wlan0 down;"

Still using dev branch with 15.01 firmware, I did not find the right sequence to modify MAC Address from User Space

**Example 1**
# modprobe wilc-sdio
# ip link set dev wlan0 address 12:34:56:78:00:12 
< SYSTEM HANG SOME TIMES>

**Example 2**
# modprobe wilc-sdio
# ifconfig wlan0 up
# ip link set dev wlan0 address 12:34:56:78:00:12
# ifconfig
eth0      Link encap:Ethernet  HWaddr C6:18:1C:54:BC:2C  
          inet addr:192.168.1.133  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::c418:1cff:fe54:bc2c/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:3989 errors:1 dropped:0 overruns:1 frame:1
          TX packets:3798 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:2322528 (2.2 MiB)  TX bytes:626986 (612.2 KiB)
          Interrupt:25 Base address:0xc000 

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

wlan0     Link encap:Ethernet  HWaddr 12:34:56:78:00:12  
          inet6 addr: fe80::f8f0:5ff:fee1:43f1/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:4 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:308 (308.0 B)

# wpa_supplicant -D nl80211 -i wlan0 -c /etc/wpa.conf  -B
Successfully initialized wpa_supplicant
rfkill: Cannot open RFKILL control device
nl80211: Could not re-add multicast membership for vendor events: -2 (No such file or directory)
# iw dev wlan0 link
Connected to e4:95:6e:40:51:2a (on wlan0)
	SSID: Kapelse-Wifi
	freq: 2437
	signal: -64 dBm
	tx bitrate: 72.0 MBit/s
# udhcpc
udhcpc (v1.24.1) started
<COMMAND NEVER RETURN> 

Could you please provide the right commands sequence?

Regards

Baptiste

AdhamAbozaeid, 2018-09-29:
Hi @bmoraine 

Are you using a sama5d4 board? can you try the prebuilt image from the demo repo?
I just flashed this image and it worked fine:
```
# ip link set dev wlan0 address 12:34:56:78:00:12
#
# ifconfig
eth0      Link encap:Ethernet  HWaddr FC:C2:3D:0C:E1:19
          inet addr:10.16.32.29  Bcast:10.16.32.255  Mask:255.255.255.0
          inet6 addr: fe80::fec2:3dff:fe0c:e119/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:89 errors:0 dropped:0 overruns:0 frame:0
          TX packets:11 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:15925 (15.5 KiB)  TX bytes:1642 (1.6 KiB)
          Interrupt:29

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

wlan0     Link encap:Ethernet  HWaddr 12:34:56:78:00:12
          inet6 addr: fe80::f8f0:5ff:fef6:5388/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:9 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:958 (958.0 B)

```

AdhamAbozaeid, 2018-09-29:
Hi @bmoraine 
For example 1, you need to bring the interface up before being able to set the MAC address.

For example 2, I see from the logs that the address was changed successfully.
I usually use _dhcpcd wlan0_ while testing, but now when I tried with udhcpc, I could see the same problem. It goes in a loop printing that discovers are being sent, but never returns.

Can you use dhcpcd instead for now?

Thanks,
Adham

bmoraine, 2018-10-02:
Hi Adham,

I now used also the latest firmware from dev branch andI face no more issue with udhcpc with example 2 use case.

I have to test now with a wilc1000 chip without a fused MAC address to confirm  that is no more mandatory to fuse MAC address during production to successfully use wilc1000  in our product.

Regards

Baptiste Moraine

AdhamAbozaeid, 2018-10-03:
Thanks @bmoraine for confirming that.
Please keep me updated with your test results

bmoraine, 2018-10-03:
Hi Adham,

I use today a PCB with a wilc1000 without fused mac address using driver & firmware dev branch and it works fine (wpa, udhcpc & ping) using ip link set dev wlan0 address xx:xx:xx:xx:xx:xx or not.

I will process deeper tests in the next days to confirm everything is ok.

I did not have time to review commits on the dev branch, maybe you could help me : 
- To understand how the Mac address is generated when no mac address is fused. I noticed it is always the same address after reset or power on/off cycle
- How to diminish Info level driver verbosity 

Regards

AdhamAbozaeid, 2018-10-03:
Hi bmoraine

- If no efuse mac address is available, the user will need to set the mac address everytime the interface is re-initialized
Driver's log messages follows the generic kernel logging mechanism, so you can use dmesg -n x, where x is the maximum log level you want to see

bmoraine, 2018-10-03:
Hi Adham,

It’s weird as I do not need to set the MAC address to have wilc1000 working fine

Maybe the  wilc1000 I used for my test has a prefused MAC address. 

Is there an easy way (without using Totalphase USB to I2C tool) to check is a MAC address is fused.

Regards

AdhamAbozaeid, 2018-10-03:
Hi bmoraine
WILC will use a hardcoded mac address if no efussed mac address was available. This will make it work if the user didn't set a new mac address, but would run into conflict mac address problems.
You can know if the mac address is effused or not by looking into the WILC's logs coming out on the uart port, or by reading the mac address from ifconfig wlan0. If the address is in the form 00:50:C2:5E:10:8b then this is the hardcoded address, and it means reading the mac address from the efuse failed

AdhamAbozaeid, 2018-10-03:
RE the issue where udhcpc never returns, you need to specify the interface name:
`udhcpc -i wlan0`

bmoraine, 2018-10-03:
Hi Adham,

Thank you for these clarifications regarding Mac address used when not fused.

It confirms my tests today. I had 00:50:C2:5E:10:8b mac address by default and my tests (WPA / udhcpc  -i wlan0 / ping) were successful then I set a new mac address and my tests were still successful.

Regards

closed 2018-10-03 by AdhamAbozaeid