linux4wilc/driver #18: Firmware changing incoming MAC address on broadcast packets (ARP in particular)

ShonkyCH, 2018-09-28:
We are running a custom board with the WILC1000 connected in SDIO mode on kernel  4.9.52

We currently have concurrent mode running on some test units

/etc/network/interfaces looks like
```
# Wireless interfaces
iface wlan0 inet dhcp
        wireless_mode managed
        wireless_essid any
        wpa-driver wext
        wpa-conf /etc/wpa_supplicant.conf

auto p2p0
iface p2p0 inet static
        address 192.168.100.1
        netmask 255.255.255.0
        network 192.168.100.0
        broadcast 192.168.100.255
````

Appropriate parts of ifconfig
```
p2p0      Link encap:Ethernet  HWaddr f8:f0:05:d0:9e:be
          inet addr:192.168.100.1  Bcast:192.168.100.255  Mask:255.255.255.0
          inet6 addr: fe80::faf0:5ff:fed0:9ebe/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:15 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:1198 (1.1 KiB)

wlan0     Link encap:Ethernet  HWaddr fa:f0:05:d0:9e:be
          inet addr:10.1.2.171  Bcast:10.1.2.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1168 errors:0 dropped:0 overruns:0 frame:0
          TX packets:74 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:236535 (230.9 KiB)  TX bytes:6137 (5.9 KiB)
```
ie p2p0 gets the F8:F0:05 MAC address,  wlan0 gets the FA:F0:05 MAC address which I still feel this is the wrong way around. 

Watching the incoming traffic on the unit with tcpdump (note the -n is important as it prevents the unit doing a DNS lookup of the IP which creates outgoing traffic)

tcpdump -iwlan0 -nvvvXSe arp

```
<lots of ignored ARP requests to f8:f0:05:d0:9e:be >
23:08:32.696726 1c:c1:de:34:01:97 > f8:f0:05:d0:9e:be, ethertype ARP (0x0806), length 60: Ethernet (len 6), IPv4 (len 4), Request who-has 10.1.2.171 tell 10.1.2.8, length 46
        0x0000:  0001 0800 0604 0001 1cc1 de34 0197 0a01  ...........4....
        0x0010:  0208 0000 0000 0000 0a01 02ab 0000 0000  ................
        0x0020:  0000 0000 0000 0000 0000 0000 0000       ..............
23:08:33.736692 1c:c1:de:34:01:97 > f8:f0:05:d0:9e:be, ethertype ARP (0x0806), length 60: Ethernet (len 6), IPv4 (len 4), Request who-has 10.1.2.171 tell 10.1.2.8, length 46
        0x0000:  0001 0800 0604 0001 1cc1 de34 0197 0a01  ...........4....
        0x0010:  0208 0000 0000 0000 0a01 02ab 0000 0000  ................
        0x0020:  0000 0000 0000 0000 0000 0000 0000       ..............
23:08:34.697548 1c:c1:de:34:01:97 > f8:f0:05:d0:9e:be, ethertype ARP (0x0806), length 60: Ethernet (len 6), IPv4 (len 4), Request who-has 10.1.2.171 tell 10.1.2.8, length 46
        0x0000:  0001 0800 0604 0001 1cc1 de34 0197 0a01  ...........4....
        0x0010:  0208 0000 0000 0000 0a01 02ab 0000 0000  ................
        0x0020:  0000 0000 0000 0000 0000 0000 0000       ..............
23:08:35.816735 1c:c1:de:34:01:97 > f8:f0:05:d0:9e:be, ethertype ARP (0x0806), length 60: Ethernet (len 6), IPv4 (len 4), Request who-has 10.1.2.171 tell 10.1.2.8, length 46
        0x0000:  0001 0800 0604 0001 1cc1 de34 0197 0a01  ...........4....
        0x0010:  0208 0000 0000 0000 0a01 02ab 0000 0000  ................
        0x0020:  0000 0000 0000 0000 0000 0000 0000       ..............
23:08:36.856708 1c:c1:de:34:01:97 > f8:f0:05:d0:9e:be, ethertype ARP (0x0806), length 60: Ethernet (len 6), IPv4 (len 4), Request who-has 10.1.2.171 tell 10.1.2.8, length 46
        0x0000:  0001 0800 0604 0001 1cc1 de34 0197 0a01  ...........4....
        0x0010:  0208 0000 0000 0000 0a01 02ab 0000 0000  ................
        0x0020:  0000 0000 0000 0000 0000 0000 0000       ..............
23:08:37.896733 1c:c1:de:34:01:97 > fa:f0:05:d0:9e:be, ethertype ARP (0x0806), length 60: Ethernet (len 6), IPv4 (len 4), Request who-has 10.1.2.171 tell 10.1.2.8, length 46
        0x0000:  0001 0800 0604 0001 1cc1 de34 0197 0a01  ...........4....
        0x0010:  0208 0000 0000 0000 0a01 02ab 0000 0000  ................
        0x0020:  0000 0000 0000 0000 0000 0000 0000       ..............
23:08:37.897448 fa:f0:05:d0:9e:be > 1c:c1:de:34:01:97, ethertype ARP (0x0806), length 42: Ethernet (len 6), IPv4 (len 4), Reply 10.1.2.171 is-at fa:f0:05:d0:9e:be, length 28
        0x0000:  0001 0800 0604 0002 faf0 05d0 9ebe 0a01  ................
        0x0010:  02ab 1cc1 de34 0197 0a01 0208            .....4......
23:08:42.935777 fa:f0:05:d0:9e:be > 1c:c1:de:34:01:97, ethertype ARP (0x0806), length 42: Ethernet (len 6), IPv4 (len 4), Request who-has 10.1.2.8 tell 10.1.2.171, length 28
        0x0000:  0001 0800 0604 0001 faf0 05d0 9ebe 0a01  ................
        0x0010:  02ab 0000 0000 0000 0a01 0208            ............
23:08:42.949462 1c:c1:de:34:01:97 > fa:f0:05:d0:9e:be, ethertype ARP (0x0806), length 60: Ethernet (len 6), IPv4 (len 4), Reply 10.1.2.8 is-at 1c:c1:de:34:01:97, length 46
        0x0000:  0001 0800 0604 0002 1cc1 de34 0197 0a01  ...........4....
        0x0010:  0208 faf0 05d0 9ebe 0a01 02ab 0000 0000  ................
        0x0020:  0000 0000 0000 0000 0000 0000 0000       ..............
```

Note that the MAC address changes to the correct address for wlan0 and as soon as it does, it correctly responds to the ARP request. 

It would seem the WILC1000 firmware is (incorrectly) assigning the device’s base MAC address to incoming packets on wlan0. As the MAC address doesn’t match it’s ignored. Why is it even doing this? It shouldn’t need to change the MAC address as it should be coming in as a broadcast destination address.

If the unit creates *any* outgoing traffic on the wlan0 interface, it appears to fix itself and starts correctly responding to ARP requests. It eventually reoccurs but if there is enough outgoing traffic it seems to remember the right MAC address. Sometimes it eventually corrects itself, but that could simply be something else creating an outgoing packet being enough to bring it in line. It will also revert after a period of time back to the wrong MAC address and so starts ignoring ARP requests again and presumably any other broadcast messages too.

Thanks

AdhamAbozaeid, 2018-09-29:
Hi @ShonkyCH 

This was recently fixed on the dev branch. Specifically, this is commit 5264c2e on the firmware repo

ChristianHackAMS, 2018-09-29:
Thanks @adhamabozaeid. Sorry meant to put this in the firmware project.

Does that fix just transfer the problem to the p2p0 interface though? The commit really just says the two MAC addresses were swapped around or was it sometimes assigning the wrong address?

Even then why does it change the destination address?

AdhamAbozaeid, 2018-09-29:
Hi @ShonkyCH 
This fix fixes the issue where the wlan0 is assigned the fa:xx:xx address instead of the f8:xx:xx.
Sorry I overlooked the second part of your question about overriding the destination address of the broadcast packets.
I'll need to check that and get back to you.

ChristianHackAMS, 2018-09-30:
Are the recent firmware and driver versions interchangeable without issue or does the -dev firmware need the -dev driver?

ShonkyCH, 2018-09-30:
Does -dev firmware need the -dev driver or are the recent versions fairly interchangeable without issue?

AdhamAbozaeid, 2018-10-01:
They are fairly interchangeable, but some fixes/new features might require -dev pairs if the fix spans both.

ShonkyCH, 2018-10-03:
Further to this we have determined that the new firmware in the referenced patch does indeed switch the MAC addresses around.

*However*, it still rewrites the wrong MAC addresses. Incoming broadcast packets on wlan0 are changed to FA:F0:05 when wlan0 is on F8:F0:05. So still completely broken.

Even worse, we have found it changes *any* incoming packet's MAC address not just broadcast ones.  If we manually enter the MAC address on another machine into the arp table and send a ping packet (or any other kind of packet) to the unit's F8:F0:05 MAC address on wlan0, the destination address is changed and ignored too.

This is fundamentally flawed and makes this device somewhat unusable if it's ignoring packets addressed to it. Why is the firmware changing incoming packet destination addresses at all? Basically unless the unit creates outgoing packets first it is impossible to communicate with the device.

AdhamAbozaeid, 2018-10-03:
Hi Shonky

I agree that the firmware shouldn't change the mac address on the packet. Please open a salesforce ticket for that with steps to reproduce.

Thanks,
Adham

ShonkyCH, 2018-10-04:
Salesforce case : 00338069

AdhamAbozaeid, 2018-10-16:
Fixed on linux4wilc/firmware commit b0537c273e821e1655e1f906d55eb0a9de1b7274

closed 2018-10-16 by AdhamAbozaeid