linux4wilc/driver #19: crash with 4.14.73-mainline

hsdenx, 2018-10-01:
Hello all,

I just tried to use this driver (spi version) on imx6 based hardware and see the following crash dump:

# ifconfig wlan0 up
[   98.564711] WILC_SPI spi0.0 wlan0: INFO [wilc_mac_open]MAC OPEN[c3312000] wlan0
[   98.573226] WILC POWER UP
[   98.576097] WILC_SPI spi0.0 wlan0: INFO [wilc_init_host_int]Host[c3312000][c3420800]
[   98.590026] WILC_SPI spi0.0 wlan0: INFO [wilc_mac_open]*** re-init ***
[   98.596633] WILC_SPI spi0.0 wlan0: INFO [wlan_init_locks]Initializing Locks ...
[   98.604177] WILC_SPI spi0.0 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
[   98.619614] WILC_SPI spi0.0 wlan0: INFO [wilc_wlan_initialize]WILC Initialization done
[   98.632042] WILC_SPI spi0.0 wlan0: INFO [init_irq]IRQ request succeeded IRQ-NUM= 166 on GPIO: 116
[   98.641150] WILC_SPI spi0.0 wlan0: INFO [wlan_initialize_threads]Initializing Threads ...
[   98.649524] WILC_SPI spi0.0 wlan0: INFO [wlan_initialize_threads]Creating kthread for transmission
[   98.661109] WILC_SPI spi0.0 wlan0: INFO [wlan_initialize_threads]Creating kthread for Debugging
[   98.671602] WILC_SPI spi0.0 wlan0: INFO [wilc_wlan_get_firmware]Detect chip WILC1000
[   98.679556] WILC_SPI spi0.0 wlan0: INFO [wilc_wlan_get_firmware]loading firmware mchp/wilc1000_wifi_firmware.bin
[   98.689846] WILC_SPI spi0.0 wlan0: INFO [wilc_wlan_get_firmware]WLAN firmware: mchp/wilc1000_wifi_firmware.bin
[   98.961858] WILC_SPI spi0.0 wlan0: INFO [wilc_firmware_download]Downloading Firmware ...
[   98.970155] WILC_SPI spi0.0 wlan0: INFO [wilc_wlan_firmware_download]Downloading firmware size = 134964
[   99.340134] WILC_SPI spi0.0 wlan0: INFO [wilc_wlan_firmware_download]Offset = 119660
[   99.394077] WILC_SPI spi0.0 wlan0: INFO [wilc_wlan_firmware_download]Offset = 134964
[   99.402010] WILC_SPI spi0.0 wlan0: INFO [wilc_firmware_download]Download Succeeded
[   99.413702] WILC_SPI spi0.0 wlan0: INFO [linux_wlan_start_firmware]Starting Firmware ...
[   99.426286] WILC_SPI spi0.0 wlan0: INFO [linux_wlan_start_firmware]Waiting for FW to get ready ...
[   99.499185] WILC_SPI spi0.0 wlan0: INFO [linux_wlan_start_firmware]Firmware successfully started
[   99.533490] WILC_SPI spi0.0 wlan0: INFO [wilc_wlan_initialize]WILC Firmware Ver = WILC_WIFI_FW_REL_15_01_RC3 Build: 9792
[   99.544553] WILC_SPI spi0.0 wlan0: INFO [linux_wlan_init_test_config]Start configuring Firmware
[   99.714468] WILC_SPI spi0.0 wlan0: INFO [wilc_mac_open]Mac address: 00:50:c2:5e:10:8e
[   99.722614] WILC_SPI spi0.0 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 1. 
[   99.732115] WILC_SPI spi0.0 wlan0: INFO [wilc_set_multicast_list]Entry[0]: 33:33:0:0:0:1
[   99.754104] WILC_SPI spi0.0 wlan0: INFO [set_power_mgmt] Power save Enabled= 1 , TimeOut = -1
[   99.770011] WILC_SPI spi0.0 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 2. 
[   99.779302] WILC_SPI spi0.0 wlan0: INFO [wilc_set_multicast_list]Entry[0]: 33:33:0:0:0:1
[   99.787444] WILC_SPI spi0.0 wlan0: INFO [wilc_set_multicast_list]Entry[1]: 1:0:5e:0:0:1
[   99.797107] WILC_SPI spi0.0 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 3. 
[   99.806556] WILC_SPI spi0.0 wlan0: INFO [wilc_set_multicast_list]Entry[0]: 33:33:0:0:0:1
[   99.814759] WILC_SPI spi0.0 wlan0: INFO [wilc_set_multicast_list]Entry[1]: 1:0:5e:0:0:1
[   99.822869] WILC_SPI spi0.0 wlan0: INFO [wilc_set_multicast_list]Entry[2]: 33:33:ff:5e:10:8e
# [   99.858703] 
[   99.860280] ================================
[   99.864596] WARNING: inconsistent lock state
[   99.868922] 4.14.73-mainline #1 Tainted: G           O   
[   99.874364] --------------------------------
[   99.878681] inconsistent {SOFTIRQ-ON-W} -> {IN-SOFTIRQ-W} usage.
[   99.884747] K_TXQ_TASK/514 [HC0[0]:SC1[5]:HE1:SE0] takes:
[   99.890188]  (fs_reclaim){+.?.}, at: [<c01accb0>] fs_reclaim_acquire.part.2+0x0/0x48
[   99.898072] {SOFTIRQ-ON-W} state was registered at:
[   99.903017]   lock_acquire+0x74/0x90
[   99.906656]   fs_reclaim_acquire.part.2+0x38/0x48
[   99.911417]   fs_reclaim_acquire+0x1c/0x20
[   99.915571]   kmem_cache_alloc+0x30/0x118
[   99.919643]   alloc_worker.constprop.11+0x24/0x58
[   99.924406]   create_worker+0x3c/0x178
[   99.928213]   workqueue_init+0x8c/0x120
[   99.932113]   kernel_init_freeable+0x60/0x1d8
[   99.936526]   kernel_init+0x10/0x114
[   99.940160]   ret_from_fork+0x14/0x20
[   99.943867] irq event stamp: 15904
[   99.947336] hardirqs last  enabled at (15904): [<c01671e8>] __call_rcu+0x74/0x7c
[   99.954796] hardirqs last disabled at (15903): [<c01671a0>] __call_rcu+0x2c/0x7c
[   99.962252] softirqs last  enabled at (13242): [<c010170c>] __do_softirq+0x24c/0x2e0
[   99.970057] softirqs last disabled at (15879): [<c012183c>] irq_exit+0xe4/0x138
[   99.977407] 
[   99.977407] other info that might help us debug this:
[   99.983986]  Possible unsafe locking scenario:
[   99.983986] 
[   99.989951]        CPU0
[   99.992434]        ----
[... 411 lines cut ...]
[  249.157835] WILC_SPI spi0.0 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 0
[  249.169351] WILC_SPI spi0.0 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 0
[  249.180181] WILC_SPI spi0.0 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 0
[  249.195674] WILC_SPI spi0.0 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 0
[  249.207393] WILC_SPI spi0.0 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  249.219181] WILC_SPI spi0.0 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  249.234308] WILC_SPI spi0.0 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  249.246875] WILC_SPI spi0.0 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  249.263471] WILC_SPI spi0.0 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  249.275408] WILC_SPI spi0.0 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  249.286292] WILC_SPI spi0.0 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  249.301984] WILC_SPI spi0.0 wlan0: INFO [connect]Connecting to SSID [Emblux] on netdev [c3312000] host if [c39c9000]
[  249.312649] WILC_SPI spi0.0 wlan0: INFO [connect]Required bss is in scan results
[  249.320148] WILC_SPI spi0.0 wlan0: INFO [connect]Adding key with cipher group fac04
[  249.327832] WILC_SPI spi0.0 wlan0: INFO [connect]Authentication Type = 0
[  249.334592] WILC_SPI spi0.0 wlan0: INFO [connect]In OPEN SYSTEM
[  249.340574] WILC_SPI spi0.0 wlan0: INFO [wilc_wlan_set_bssid]set bssid on[c3312000]
[  249.348296] WILC_SPI spi0.0 wlan0: INFO [wilc_wlan_set_bssid]set bssid [50][c7][bf]
[  249.548530] WILC_SPI spi0.0 wlan0: INFO [handle_connect]send HOST_IF_WAITING_CONN_RESP
[  249.633931] WILC_SPI spi0.0 wlan0: INFO [handle_connect]set HOST_IF_WAITING_CONN_RESP

Any hints / ideas?

Thanks in advance,
Heiko

AdhamAbozaeid, 2018-10-01:
Hi Heiko

For the first problem, this isn't a crash, it's rather a warning of possible deadlock in kmem_cache_alloc()
I wonder if this and https://patchwork.kernel.org/patch/10207013/ 

For the second problem, it's hard to tell from the log what was the reason for connection failure. Can you try with open security AP and see if that works? If this woks, you can can double check the password you using on the secured network.
If that still doesn't work, please open a ticket on http://microchipsupport.force.com/ and provide the driver, FW and air captures

hsdenx, 2018-10-03:
Hello Adham,

Am 02.10.2018 um 00:19 schrieb Adham Abozaeid:
> Hi Heiko
> 
> For the first problem, this isn't a crash, it's rather a warning of possible deadlock in 
> kmem_cache_alloc()
> I wonder if this and https://patchwork.kernel.org/patch/10207013/

Yes, but it pollutes dmesg and a warning is a warning ...

Hmm.. above mentioned patch is already in 4.14.73 LTS:

https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/mm/page_alloc.c?h=linux-4.14.y&id=ef006d43bc617d5dde8c3076a9b49bbce7dc3f82

or do you mean, if this patch is "buggy" ?

Just reverted this patch ... same warning ...

> For the second problem, it's hard to tell from the log what was the reason for connection failure.
> Can you try with open security AP and see if that works? If this woks, you can can double check [...]

Checked with 3 different APs, all without a password, no connection

Interesting ... tried HostAP mode, and I can connect for example with
my smartphone. So Host AP mode works, also HW problems can be excluded!

bye,
Heiko
-- 
DENX Software Engineering GmbH,      Managing Director: Wolfgang Denk
HRB 165235 Munich, Office: Kirchenstr.5, D-82194 Groebenzell, Germany
Phone: +49-8142-66989-52   Fax: +49-8142-66989-80   Email: hs@denx.de

AdhamAbozaeid, 2018-10-04:
Hi Heiko

Can you send me the commands you are using to configure the network using wpa_cli?

hsdenx, 2018-10-05:
Hello Adham,

Am 04.10.2018 um 18:48 schrieb Adham Abozaeid:
> Hi Heiko
> 
> Can you send me the commands you are using to configure the network using wpa_cli?

Of course ... so, I enable AP on my smartphone SSID "schocher-mobil"
no password. Tested access to it from another smartphone, works.

First try with already made /etc/wpa_supplicant.conf

# cat /etc/wpa_supplicant.conf
ctrl_interface=/var/run/wpa_supplicant
ctrl_interface_group=0
update_config=1

network={
         ssid="schocher-mobil"
         key_mgmt=NONE
}

root@ckey00:00:00:00:00:00:~# ifconfig wlan0 up
[  310.423233] WILC_SPI spi0.0 wlan0: INFO [wilc_mac_open]MAC OPEN[c32ca000] wlan0
[  310.432350] power up request for already powered up source Wifi
[  310.438340] Device already up. request source is Wifi
[  310.443503] WILC_SPI spi0.0 wlan0: INFO [wilc_init_host_int]Host[c32ca000][c3656000]
[  310.452568] WILC_SPI spi0.0 wlan0: INFO [wilc_mac_open]*** re-init ***
[  310.459143] WILC_SPI spi0.0 wlan0: INFO [wlan_init_locks]Initializing Locks ...
[  310.466616] WILC_SPI spi0.0 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
[  310.476312] WILC_SPI spi0.0 wlan0: INFO [wilc_wlan_initialize]WILC Initialization done
[  310.485478] WILC_SPI spi0.0 wlan0: INFO [init_irq]IRQ request succeeded IRQ-NUM= 166 on GPIO: 116
[  310.494466] WILC_SPI spi0.0 wlan0: INFO [wlan_initialize_threads]Initializing Threads ...
[  310.502736] WILC_SPI spi0.0 wlan0: INFO [wlan_initialize_threads]Creating kthread for transmission
[  310.512762] WILC_SPI spi0.0 wlan0: INFO [wlan_initialize_threads]Creating kthread for Debugging
[  310.521797] WILC_SPI spi0.0 wlan0: INFO [wilc_wlan_get_firmware]Detect chip WILC1000
[  310.529576] WILC_SPI spi0.0 wlan0: INFO [wilc_wlan_get_firmware]loading firmware 
mchp/wilc1000_wifi_firmware.bin
[  310.539892] WILC_SPI spi0.0 wlan0: INFO [wilc_wlan_get_firmware]WLAN firmware: 
mchp/wilc1000_wifi_firmware.bin
[  310.551966] WILC_SPI spi0.0 wlan0: INFO [wilc_firmware_download]Downloading Firmware ...
[  310.560326] WILC_SPI spi0.0 wlan0: INFO [wilc_wlan_firmware_download]Downloading firmware size = 
134964
[  310.665519] WILC_SPI spi0.0 wlan0: INFO [wilc_wlan_firmware_download]Offset = 119660
[  310.695392] WILC_SPI spi0.0 wlan0: INFO [wilc_wlan_firmware_download]Offset = 134964
[  310.703288] WILC_SPI spi0.0 wlan0: INFO [wilc_firmware_download]Download Succeeded
[  310.713637] WILC_SPI spi0.0 wlan0: INFO [linux_wlan_start_firmware]Starting Firmware ...
[  310.723864] WILC_SPI spi0.0 wlan0: INFO [linux_wlan_start_firmware]Waiting for FW to get ready ...
[  310.784232] WILC_SPI spi0.0 wlan0: INFO [linux_wlan_start_firmware]Firmware successfully started
[  310.809833] WILC_SPI spi0.0 wlan0: INFO [wilc_wlan_initialize]WILC Firmware Ver = 
WILC_WIFI_FW_REL_15_01_RC3 Build: 9792
[  310.820921] WILC_SPI spi0.0 wlan0: INFO [linux_wlan_init_test_config]Start configuring Firmware
[  310.943824] WILC_SPI spi0.0 wlan0: INFO [wilc_mac_open]Mac address: 00:50:c2:5e:10:8e
[  310.951776] WILC_SPI spi0.0 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 0.
[  310.961016] WILC_SPI spi0.0 wlan0: INFO [wilc_set_multicast_list]Enable mcast filter retrive 
directed pkts only
[  310.975695] WILC_SPI spi0.0 wlan0: INFO [set_power_mgmt] Power save Enabled= 1 , TimeOut = -1
[  310.991572] WILC_SPI spi0.0 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 1.
[  311.000815] WILC_SPI spi0.0 wlan0: INFO [wilc_set_multicast_list]Entry[0]: 1:0:5e:0:0:1
[  311.008937] WILC_SPI spi0.0 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 2.
[  311.018166] WILC_SPI spi0.0 wlan0: INFO [wilc_set_multicast_list]Entry[0]: 1:0:5e:0:0:1
[  311.026233] WILC_SPI spi0.0 wlan0: INFO [wilc_set_multicast_list]Entry[1]: 33:33:0:0:0:1
[  311.034836] WILC_SPI spi0.0 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 3.
[  311.044087] WILC_SPI spi0.0 wlan0: INFO [wilc_set_multicast_list]Entry[0]: 1:0:5e:0:0:1
[  311.052156] WILC_SPI spi0.0 wlan0: INFO [wilc_set_multicast_list]Entry[1]: 33:33:0:0:0:1
[  311.060302] WILC_SPI spi0.0 wlan0: INFO [wilc_set_multicast_list]Entry[2]: 33:33:ff:5e:10:8e
root@ckey00:00:00:00:00:00:~#

root@ckey00:00:00:00:00:00:~# wpa_supplicant -B -D nl80211 -i wlan0 -c /etc/wpa_supplicant.conf
Successfully initialized wpa_supplicant
[... 328 lines cut ...]
[  550.335088] WILC_SPI spi0.0 wlan0: INFO [connect]Adding key with cipher group 0
[  550.342483] WILC_SPI spi0.0 wlan0: INFO [connect]Authentication Type = 0
[  550.349229] WILC_SPI spi0.0 wlan0: INFO [connect]In OPEN SYSTEM
[  550.355212] WILC_SPI spi0.0 wlan0: INFO [wilc_wlan_set_bssid]set bssid on[c32ca000]
[  550.362938] WILC_SPI spi0.0 wlan0: INFO [wilc_wlan_set_bssid]set bssid [62][f1][89]
[  550.538748] WILC_SPI spi0.0 wlan0: INFO [handle_connect]send HOST_IF_WAITING_CONN_RESP
[  550.624712] WILC_SPI spi0.0 wlan0: INFO [handle_connect]set HOST_IF_WAITING_CONN_RESP

 > > quit
 > root@ckey00:00:00:00:00:00:~#

Dummy question:

I see:

[  341.050306] WILC_SPI spi0.0 wlan0: INFO [get_tx_power]Got tx power 0

tx power 0 ? May this is a problem ?

bye,
Heiko
-- 
DENX Software Engineering GmbH,      Managing Director: Wolfgang Denk
HRB 165235 Munich, Office: Kirchenstr.5, D-82194 Groebenzell, Germany
Phone: +49-8142-66989-52   Fax: +49-8142-66989-80   Email: hs@denx.de