linux4wilc/driver #20: interface reinitialisation failed, wilc_firmware_download over sdio

rom-dim, 2018-10-08:
Hello,

i tested actual dev branch, last pull from today. 
after approx. 2-3 sequences ifconfig up/down , without wpa_supplicant. I get error on initialization wilc_sdio_cmd52..failed, err(-84). First initialization is always ok.

wilc1000 is connected to imx6sx oder sdio interface

- test: ifconfig wlan0 up && sleep 1 && ifconfig wlan0 down
- debug thread is deactivated
- last down -> up sequence 
ifconfig wlan0 down
[   70.391598] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Deinitializing wilc  ...
[   70.400242] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Disabling IRQ
[   70.407922] wilc_sdio mmc1:0001:1: wilc_sdio_disable_interrupt
[   70.416841] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Deinitializing Threads
[   70.433772] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinitializing Threads
[   70.451223] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinit debug Thread
[   70.468257] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinitializing Threads
[   70.485768] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Deinitializing IRQ
[   70.504380] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Deinitializing WILC Wlan
[   70.513312] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Deinitializing Locks
[   70.521618] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinit_locks]De-Initializing Locks
[   70.529560] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_deinitialize]wilc deinitialization Done
[   72.597416] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[883ac000] wlan0
ifconfig  wlan0 up
[   72.605337] power up request for already powered up source Wifi
[   72.611336] Device already up. request source is Wifi
[   72.616437] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_init_host_int]Host[883ac000][886123c0]
[   72.624732] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
[   72.631803] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_init_locks]Initializing Locks ...
[   72.639654] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
[   72.657797] wilc_sdio mmc1:0001:1: failed to get IRQ GPIO, load default
[   72.664531] wilc_sdio mmc1:0001:1: got gpio_irq default successfully
[   72.671026] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Initializing Threads ...
[   72.679754] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Creating kthread for transmission
[   72.691780] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]Detect chip WILC1000
[   72.700087] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]loading firmware mchp/wilc1000_wifi_firmware.bin
[   72.710843] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]WLAN firmware: mchp/wilc1000_wifi_firmware.bin
[   72.721702] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_firmware_download]Downloading Firmware ...
[   72.730439] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_firmware_download]Downloading firmware size = 135284
[   72.751956] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_firmware_download]Offset = 119976
[   72.763809] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_firmware_download]Offset = 135284
[   72.772105] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_firmware_download]Download Succeeded
[   72.780287] wilc_sdio mmc1:0001:1 wlan0: INFO [linux_wlan_start_firmware]Starting Firmware ...
**[   72.793215] wilc_sdio mmc1:0001:1: wilc_sdio_cmd52..failed, err(-84)**
[   72.799594] wilc_sdio mmc1:0001:1: Failed cmd 52, read reg (000000fc) ...
[   72.806403] wilc_sdio mmc1:0001:1 wlan0: INFO [linux_wlan_start_firmware]Waiting for FW to get ready ...
[   72.875055] wilc_sdio mmc1:0001:1 wlan0: INFO [linux_wlan_start_firmware]Firmware successfully started
[   72.896580] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_initialize]WILC Firmware Ver = WILC_WIFI_FW_REL_15_1_RC3 Build: 10501
[   72.907999] wilc_sdio mmc1:0001:1 wlan0: INFO [linux_wlan_init_test_config]Start configuring Firmware
[   73.034786] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]Mac address: f8:f0:05:ff:f3:22
[   73.043177] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 0.
[   73.052764] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Enable mcast filter retrive directed pkts only
[   73.068445] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 1.
[   73.078078] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry0: 1:0:5e:0:0:1
[   73.093829] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 2.
[   73.103441] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry0: 1:0:5e:0:0:1
[   73.111813] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry1: 33:33:0:0:0:1
[   73.127762] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 3.
[   73.137387] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry0: 1:0:5e:0:0:1
[   73.145764] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry1: 33:33:0:0:0:1
[   73.154223] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry2: 33:33:ff:ff:f3:22
[   74.906382] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 4.
[   74.916014] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry0: 1:0:5e:0:0:1
[   74.924392] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry1: 33:33:0:0:0:1
[   74.932858] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry2: 33:33:ff:ff:f3:22
[   74.941663] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry3: 33:33:0:0:0:fb

rom-dim, 2018-10-08:
I executed other test where I rmove wilc-sdio driver and insert if before ip/down.. I didn't found any problems. It's a problem of reset and chip_en ? _init_wilc_sdio do reset and chip en.. ifconfig up/down not, but will load firmware to wilc.

Test: insmod /rw/wilc-sdio.ko && ifconfig wlan0 up && sleep 5 &&  ifconfig wlan0 down && rmmod /rw/wilc-sdio.ko

AdhamAbozaeid, 2019-03-18:
Not reproducible on release 15.2

closed 2019-03-18 by AdhamAbozaeid

rom-dim, 2019-05-13:
> Not reproducible on release 15.2 

Hi @AdhamAbozaeidm, today i tried new version (15_2 driver + firmware). I still have the same problem with same script :(.

Here my dmesg output (debug output is disbaled)

[   18.622573] wifi_pm : 0
[   18.629209] wifi_pm : 1
[   18.713089] wilc_sdio mmc1:0001:1: Driver Initializing success
[   18.718722] WILC POWER UP
[   18.718770] wilc_sdio mmc1:0001:1: SDIO speed: 50000000
[   18.719425] wilc_sdio mmc1:0001:1: chipid 001003a0
[   18.728873] wilc_sdio mmc1:0001:1: got gpio_irq successfully
[   24.009021] wilc_sdio mmc1:0001:1: wilc_sdio_disable_interrupt
[   29.938400] power up request for already powered up source Wifi
### [   29.944440] Device already up. request source is Wifi
[   29.954982] wilc_sdio mmc1:0001:1: got gpio_irq successfully
**[   29.987110] wilc_sdio mmc1:0001:1: wilc_sdio_cmd52..failed, err(-84)
[   29.993505] wilc_sdio mmc1:0001:1: Failed cmd 52, read reg (000000fc) ...**
[   35.271738] wilc_sdio mmc1:0001:1: wilc_sdio_disable_interrupt
[   36.815393] power up request for already powered up source Wifi
[   36.821423] Device already up. request source is Wifi
[   36.831945] wilc_sdio mmc1:0001:1: got gpio_irq successfully
[   42.159161] wilc_sdio mmc1:0001:1: wilc_sdio_disable_interrupt
[   43.182935] power up request for already powered up source Wifi
[   43.189171] Device already up. request source is Wifi
[   43.206683] wilc_sdio mmc1:0001:1: got gpio_irq successfully
[   48.527015] wilc_sdio mmc1:0001:1: wilc_sdio_disable_interrupt

AdhamAbozaeid, 2019-05-13:
Hi @rom-dim 
Can you try with CONFIG_CFG80211_DEFAULT_PS disabled in your kernel .config file?

AdhamAbozaeid, 2019-05-13:
You will also need a sleep between the modprobe and ifconfig up. Below is the script i'm using for testing
for i in \`seq 1 50\`
do
        echo "*****************************************"
        echo "***************** Loop $i ***************"
        echo ""
        modprobe wilc-sdio && sleep 1 && ifconfig wlan0 up && sleep 3 && ifconfig wlan0 down && rmmod wilc-sdio
done

The attached test log is from a run of the linux-4.9-at91_wilc_dev branch, with CONFIG_CFG80211_DEFAULT_PS enabled, and it ran successfully for 50 iterations.
You can try to disable CONFIG_CFG80211_DEFAULT_PS just in case your problem was related to powersave.

[init_deinit_log.txt](https://github.com/linux4wilc/driver/files/3174929/init_deinit_log.txt)