linux4wilc/driver #24: "Invalid handle" followed by "wilc_netdev in wilc is NULL"

ShonkyCH, 2018-10-29:
'''
Oct 29 05:08:18 SM01-9999000002 kernel: wilc_sdio mmc1:0001:1 wlan0: WRN [get_if_handler: 294]Invalid handle
Oct 29 05:08:18 SM01-9999000002 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [get_if_handler:296] BSSID : 00:50:C2:5E:10:82
Oct 29 05:08:18 SM01-9999000002 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [get_if_handler:298] BSSID1 : FF:FF:FF:FF:FF:FF
Oct 29 05:08:18 SM01-9999000002 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_handle_rx_buff:1345] wilc_netdev in wilc is NULL
'''

Dug into this whilst looking at other issues. The middle two lines above are my debug messages in get_if_handler() in wilc_wlan.c i.e. 

'''
	if (!mon_netdev) {
		PRINT_WRN(wilc->vif[0]->ndev, GENERIC_DBG, "Invalid handle\n");
		PRINT_ER(wilc->vif[0]->ndev, "BSSID : %02X:%02X:%02X:%02X:%02X:%02X\n",
			bssid[0], bssid[1], bssid[2], bssid[3], bssid[4], bssid[5]);
		PRINT_ER(wilc->vif[0]->ndev, "BSSID1 : %02X:%02X:%02X:%02X:%02X:%02X\n",
			bssid1[0], bssid1[1], bssid1[2], bssid1[3], bssid1[4], bssid1[5]);
	}
'''

I see this MAC address embedded in the firmware. 00:50:C2:5E:10:8E is also in there.

https://hwaddress.com/mac-address-range/00-50-C2-5E-10-00/00-50-C2-5E-1F-FF

Who/what are Ittiam systems and what is this packet that appears to be sent by the firmware? This is being reported as an error on our units in the log but what is going on here?

This packet or whatever it is then appears to be simply dropped / ignored by wilc_wlan_handle_rx_buff()

Running latest dev of firmware and driver but fairly certain this was always happening.

AdhamAbozaeid, 2018-10-30:
Hi @ShonkyCH 
It looks like your WILC device doesn't have a mac address in its non-volatile memory, so it's defaulting to a hardcoded mac address.
In cases when the non-volatile memory doesn't have a mac address, the host should store a unique mac address for the device in its flash, and program it in WILC everytime it initializes the wifi interfaces, by using command
`ifconfig wlan0 hw ether fa:f0:05:f6:53:88`
or
`ip link set wlan0 address fa:f0:05:f6:53:88`
Same commands can be used for interface p2p0

ShonkyCH, 2018-10-30:
These are production units that have MAC addresses printed on the labels and this error is not always occurring. The MAC address is definitely programmed in and working.

It seems to happen more so when there is significant amount of traffic running through the Wifi module e.g. Doing an scp copy to/from the unit or a wget download. This error just randomly appears every now and then and it seems like the firmware is providing packets with the wrong MAC somehow. I haven't determined if these are packets intended for the specifically for the unit or just broadcast ones but we are also seeing packets going missing causing lots of TCP Dup Acks and poor performance as TCP retries. Doing say an iperf -u also drops large numbers of packets frequently too.

ShonkyCH, 2018-10-31:
I think 
https://github.com/linux4wilc/driver/blob/dev/wilc/wilc_wlan.c#L1340

Should read

```
wilc_netdev = get_if_handler(wilc, buff_ptr);
```

AdhamAbozaeid, 2018-10-31:
Well spotted!
That's correct. This should use buff_ptr.

AdhamAbozaeid, 2018-10-31:
Does this fix the main problem you were having?

ShonkyCH, 2018-11-01:
I believe this will fix the problem in this particular issue. It was a fairly specific case where some kind of special config packet came first *and* there were multiple packets queued to be processed. So not something I could easily reproduce.

What are these packets with this strange MAC? They seem to be called config packets

It doesn't fix the reason I was digging into the driver, however. I'm still debugging significant performance issues (lost packets, TCP retries, Dup Acks, spurious TCP retransmissions etc) though.

ZonKnoZ, 2018-11-01:
> It doesn't fix the reason I was digging into the driver, however. I'm still debugging significant performance issues (lost packets, TCP retries, Dup Acks, spurious TCP retransmissions etc) though.

Same here.
I have just tested todays' commit and we still seem to have a lot of issues with lost packets.

I have a case open ( 00328657) at salesforce: https://microchipsupport.force.com/5001N00000dan2T where I provided a couple of logs & details.

Two issues still remain rather problematic:

1. 
We have a high system load of 1.X (single core) on an idle system after loading thw wilc driver. This actually dropped from 2.X since the last two dev-driver-commits(!!!), but still makes performance analysis quite difficult.

2.
The other effect I'm trying to capture (via wireshark & the WILC300-debug-uart) is:
After booting 6 to 10 boards with a WILC3000 all wifi connectivity seems to come to a halt.

AdhamAbozaeid, 2018-11-02:
Hi @ZonKnoZ 
What are the module loading/unloading problems you are seeing?

ShonkyCH, 2018-11-03:
ifconfig has never frozen I don't think. Certainly unloading modules was quite unstable until fairly recently. The current official release on the demo boards falls over pretty badly taking down the whole system as an example.

My debugging involves running as a module and removing and then insmodding a freshly compiled version of the module (4.9.52 kernel) and it's much more stable now than previously to the point where I can run all day doing this without a reboot.

ShonkyCH, 2018-11-05:
My dropping/delayed packet issue is documented in #25 

Appreciate your thoughts @ZonKnoZ if you are seeing some similar problems.

ZonKnoZ, 2018-11-05:
Thanks for the hint at issue #25 ! I'll try to find similarities as well as post some Wireshark captures of our congestion issue.

@AdhamAbozaeid this is what I see in dmesg. The system cannot send any Data although a 3G data connection is still open as well. As soon as I type ifconfig, the whole system freezes:

[83468.721838] WILC_SPI spi0.2 wlan0: INFO [change_station]HT Extended params= 0 [83468.729291] WILC_SPI spi0.2 wlan0: INFO [change_station]Tx Beamforming Cap= 0 [83468.736748] WILC_SPI spi0.2 wlan0: INFO [change_station]Antenna selection info = 0 [83468.744653] WILC_SPI spi0.2 wlan0: INFO [change_station]Flag Mask = 2 [83468.751373] WILC_SPI spi0.2 wlan0: INFO [change_station]Flag Set = 2 [83470.345366] WILC_SPI spi0.2 wlan0: INFO [get_station]Getting station parameters [83470.353119] WILC_SPI spi0.2 wlan0: INFO [handle_get_inactive_time]SETING STA inactive time [83470.387152] WILC_SPI spi0.2 wlan0: INFO [handle_get_inactive_time]Getting inactive time : 0 [83470.396057] WILC_SPI spi0.2 wlan0: INFO [get_station]Inactive time 0 [83547.206316] WILC_SPI spi0.2 wlan0: INFO [change_station]BSSID = d4e6b79ef53 [83547.213644] WILC_SPI spi0.2 wlan0: INFO [change_station]ASSOC ID = 0 [83547.220331] WILC_SPI spi0.2 wlan0: INFO [change_station]Number of supported rates = 0 [83547.228539] WILC_SPI spi0.2 wlan0: INFO [change_station]IS HT supported = 0 [83547.235828] WILC_SPI spi0.2 wlan0: INFO [change_station]Capability Info = 0 [83547.243126] WILC_SPI spi0.2 wlan0: INFO [change_station]AMPDU Params = 0 [83547.250141] WILC_SPI spi0.2 wlan0: INFO [change_station]HT Extended params= 0 [83547.257615] WILC_SPI spi0.2 wlan0: INFO [change_station]Tx Beamforming Cap= 0 [83547.265068] WILC_SPI spi0.2 wlan0: INFO [change_station]Antenna selection info = 0 [83547.272996] WILC_SPI spi0.2 wlan0: INFO [change_station]Flag Mask = 2 [83547.279739] WILC_SPI spi0.2 wlan0: INFO [change_station]Flag Set = 0 [83547.300985] WILC_SPI spi0.2 wlan0: INFO [del_station]Deleting station [83547.307792] WILC_SPI spi0.2 wlan0: INFO [del_station]With mac
address: d4e6b79ef53
[83547.317106] WILC_SPI spi0.2 wlan0: INFO [change_beacon]Setting beacon [83547.324205] WILC_SPI spi0.2 wlan0: INFO [change_bss]Changing Bss parametrs [83547.332298] WILC_SPI spi0.2 wlan0: INFO [change_bss]Changing Bss parametrs [83633.846762] WILC_SPI spi0.2 wlan0: INFO [del_station]Deleting station [83633.853547] WILC_SPI spi0.2 wlan0: INFO [del_station]All associated stations [83633.860963] WILC_SPI spi0.2 wlan0: INFO [wilc_del_allstation]BSSID = d4e6b79ef53 [83634.633490] WILC_SPI spi0.2 wlan0: INFO [stop_ap]Deleting beacon [83634.639928] WILC_SPI spi0.2 wlan0: INFO [wilc_wlan_set_bssid]set bssid on[dd780000] [83634.647983] WILC_SPI spi0.2 wlan0: INFO [wilc_wlan_set_bssid]set bssid [0][0][0] [83634.658625] WILC_SPI spi0.2 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 1.
[83634.668173] WILC_SPI spi0.2 wlan0: INFO
[wilc_set_multicast_list]Entry0: 1:0:5e:0:0:1 [83634.676461] WILC_SPI spi0.2 wlan0: INFO [wilc_mac_close]Mac close

commit e9d0eccc (2018-10-31): staging: wilc1000: Incorrect pointer passed while getting if handler

In wilc_wlan.c, buffer is being passed to get_if_handler to obtain the
wilc_netdev, while it should be buff_ptr that points to the current
position in buffer that is being processed

Fixes #24
commit 52ee1cd3 (2018-11-08): staging: wilc1000: Incorrect pointer passed while getting if handler

In wilc_wlan.c, buffer is being passed to get_if_handler to obtain the
wilc_netdev, while it should be buff_ptr that points to the current
position in buffer that is being processed

Fixes #24
closed 2018-11-27 by AdhamAbozaeid