linux4wilc/driver #27: Initialization problem on Kernel version 4.14.73

drewwestrick, 2018-11-21:
### Introduction
I'm trying to get a WILC3000 module working on **linux4sam_6.0** which is kernel **4.14.73** on a **SAMA5D27-D1G-CU** (SIP). I'm using the **ubuntu-core 18.04.01** as my root file system. I have the WILC3000 initializing during the kernel boot sequence but when I get to the end of the ubuntu loading sequence I'm getting some errors and then a driver panic. If I run ifconfig the command just hangs.

### Relevant DTS file information
```
sdmmc1: sdio-host@b0000000 {
	pinctrl-names = "default";
	pinctrl-0 = <&pinctrl_sdmmc1_default>;
	non-removable;
	status = "okay";
	#address-cells = <1>;
	#size-cells = <0>;
	wilc_sdio@0{
		compatible = "atmel,wilc_sdio";
		gpio_reset = <&pioA PIN_PB25 GPIO_ACTIVE_HIGH>;
		gpio_chip_en = <&pioA PIN_PB24 GPIO_ACTIVE_HIGH>;
		gpio_irq = <&pioA PIN_PB3 GPIO_ACTIVE_HIGH>;
		pinctrl-names = "default";
		pinctrl-0 = <&pinctrl_wilc_gpio>;
		status = "okay";
	};
	slot@0 {
		reg = <0>;
		bus-width = <4>;
	};
};

flx0: flexcom@f8034000 {
	atmel,flexcom-mode = <ATMEL_FLEXCOM_MODE_USART>;
	status = "okay";

	uart7: serial@200 {
		compatible = "atmel,at91sam9260-usart";
		reg = <0x200 0x200>;
		interrupts = <19 IRQ_TYPE_LEVEL_HIGH 7>;
		clocks = <&flx0_clk>;
		clock-names = "usart";
		atmel,use-dma-rx;
		atmel,use-dma-tx;
		dmas = <&dma0
			(AT91_XDMAC_DT_MEM_IF(0) | AT91_XDMAC_DT_PER_IF(1) |
			 AT91_XDMAC_DT_PERID(11))>,
		       <&dma0
			(AT91_XDMAC_DT_MEM_IF(0) | AT91_XDMAC_DT_PER_IF(1) |
			 AT91_XDMAC_DT_PERID(12))>;
		dma-names = "tx", "rx";
		pinctrl-names = "default";
		pinctrl-0 = <&pinctrl_flx0_default>;
		atmel,fifo-size = <32>;
		status = "okay";
	};
};

pinctrl_sdmmc1_default: sdmmc1_default {
	cmd_data {
		pinmux = <PIN_PA28__SDMMC1_CMD>,
			 <PIN_PA18__SDMMC1_DAT0>,
			 <PIN_PA19__SDMMC1_DAT1>,
			 <PIN_PA20__SDMMC1_DAT2>,
			 <PIN_PA21__SDMMC1_DAT3>;
		bias-disable;
	};

	conf-ck_cd {
		pinmux = <PIN_PA22__SDMMC1_CK>;
		bias-disable;
	};
};

pinctrl_wilc_gpio: wilc_gpio {
[... 427 lines cut ...]
1f40: c68bdb00 c68bdb00 c0107764 c0562084 c7e91510 c0ffa088 00000000 00000007
1f60: c0b33b40 00080000 00000000 00000013 c68bdb00 c0563cc0 c4b61f8c 00000000
1f80: 00000000 00000000 00000014 c68bdb00 bef3b0b2 00000000 bef3af78 0000000c
1fa0: 00000013 c0107580 bef3af78 0000000c 00000013 bef3af94 00000028 00000000
1fc0: bef3af78 0000000c 00000013 00000122 00000003 00000014 00000002 bef3af94
1fe0: 00000122 bef3af18 b6cec7cd b6c6a206 20010030 00000013 00000000 00000000
Function entered at [<c057c5e0>] from [<c057ca84>]
Function entered at [<c057ca84>] from [<c0469a34>]
Function entered at [<c0469a34>] from [<c01306fc>]
Function entered at [<c01306fc>] from [<c0130ec8>]
Function entered at [<c0130ec8>] from [<c05e3d0c>]
Function entered at [<c05e3d0c>] from [<c05e4804>]
Function entered at [<c05e4804>] from [<c0592660>]
Function entered at [<c0592660>] from [<c05a779c>]
Function entered at [<c05a779c>] from [<c05a6f84>]
Function entered at [<c05a6f84>] from [<c05a7410>]
Function entered at [<c05a7410>] from [<c056335c>]
Function entered at [<c056335c>] from [<c0564438>]
Function entered at [<c0564438>] from [<c0107580>]
Code: e92d47f0 e2514000 e24dd010 0a000071 (e5946270)
---[ end trace bf0517439132d2a7 ]---
```

### Conclusion
I'm not sure if this NULL pointer issue is something related to the driver, or just my implementation and porting of it. I would appreciate any help or feedback.

drewwestrick, 2018-11-22:
@AdhamAbozaeid 

Sure thing. I'm out of the office until Monday, but I will enable debug symbols and post a log first thing on Monday.

One thing I've been wondering is how does the BLE portion of the driver know which serial port is to be used for communication? I didn't see anything in the porting guide to indicate where that is set or inferred. We are using the flexcom 0 for the UART (ttyS3). I have RTS and CTS hardware pins available but the device tree is currently configured for just RX/TX operation. I also enabled the DMA per the porting guide which I have confirmed is working properly.

AdhamAbozaeid, 2018-11-26:
@drewwestrick 
The BLE driver doesn't need to know the UART port since the serial communication goes directly from blueZ to the device, without passing through the driver.
You specify this while attaching the BLE device to BlueZ
`hciattach ttyS1 any 115200 noflow`

You can find more information about using BlueZ on Linux in the ATWILC Linux user guide.

drewwestrick, 2018-11-26:
@AdhamAbozaeid 

Now that you mention it, I do recall something in the user guide indicating this. I figured there must have been a method to pass this information. I'm working on getting you a kernel log with debug symbols enabled right now.

drewwestrick, 2018-11-26:
### New oops with symbols enabled:

```
Unable to handle kernel NULL pointer dereference at virtual address 00000472
pgd = c50a0000
[00000472] *pgd=00000000
Internal error: Oops: 5 [#1] ARM
Modules linked in: ov2640 atmel_usba_udc
CPU: 0 PID: 299 Comm: connmand Not tainted 4.14.73-linux4sam_6.0-dirty #12
Hardware name: Atmel SAMA5
task: c6e8d300 task.stack: c4d9c000
PC is at __netdev_printk+0x10/0x290
LR is at netdev_info+0x38/0x48
pc : [<c057e0c0>]    lr : [<c057e564>]    psr: 20010013
sp : c4d9dca8  ip : c4d9dcf0  fp : 00000000
r10: ffffffff  r9 : c7acc90c  r8 : 00000001
r7 : c6602f30  r6 : c6602f00  r5 : c6602f00  r4 : 00000202
r3 : c4d9dcdc  r2 : c4d9dce0  r1 : 00000202  r0 : c0954460
Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
Control: 10c53c7d  Table: 250a0059  DAC: 00000051
Process connmand (pid: 299, stack limit = 0xc4d9c208)
Stack: (0xc4d9dca8 to 0xc4d9e000)
dca0:                   00000000 00000c00 00000000 c0c32a80 c7dc0000 c6602f00
dcc0: c6602f00 c6602f30 00000001 c7acc90c ffffffff c057e564 0d0110ac c4d9dcf0
dce0: c0995dec c4d9dcdc c046b51c c0995dec c0849b28 00000613 c6602f00 c7dc0000
dd00: 00000000 c7acc90c fffffffe 00000000 c6602f00 00000001 00000000 c7acc90c
dd20: ffffffff c01307bc 00000001 c0c34748 c6602f00 de69057d c7acc90c c0130f88
dd40: 00000000 c7acc90c c6602f00 00000000 c127ee00 c05e57ec 0d0110ac c7acc900
dd60: c0c68b68 00000000 c6602f00 c127ee00 c0b73d80 c0c32a80 ffffffff c05e62e4
dd80: c086b0f8 00000000 00000200 00000000 c127ee18 c127ee18 00000000 c127ee20
dda0: 00000000 00000000 00000000 00000000 c4c809c0 c0c68b68 c127ee00 c0b73d80
ddc0: 00000030 c4d9de20 00000004 c0c32a80 00000000 c0594140 c4ed6688 c6132e38
dde0: c01d4464 00000000 00000148 c4d9dec0 00000148 00000148 c4d9df08 c0b73d80
de00: c127ee00 c0593f2c c0b73d80 c4d9de54 00000000 00000000 00000000 c05a927c
de20: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
de40: c7e47c00 c114f400 00000028 c05a8a64 00000000 7fffffff 00000028 c4d9dee8
de60: 00000028 c114f400 00000008 c0b73d80 014000c0 c05a8ef0 00000148 c4d9dcf4
de80: c01d4464 00000100 00000200 c7b63940 00000000 0000012b 00000000 00000000
dea0: 00000051 00000000 00000000 c10f1040 00000122 c0107804 c4d9c000 00000000
dec0: be9a1f94 c0564e3c 00000000 c0565f18 c4d9def0 00000000 00000000 00000000
dee0: be9a1f94 00000028 c4d9df18 0000000c 00000001 00000000 00000000 c4d9dee8
df00: 00000000 c509df00 00000000 00000000 00000000 c01c37c0 00000010 00000000
df20: 00000000 c0107804 c4d9c000 00000000 be9a1f94 c01c38a0 c0c03694 00080000
df40: c10f1040 c10f1040 c0107804 c0563b64 c7e91510 c4385ee0 00000000 00000007
df60: c0c33b40 00080000 00000000 00000013 c10f1040 c05657a0 c4d9df8c 00000000
df80: 00000000 00000000 00000014 c10f1040 be9a20b2 00000000 be9a1f78 0000000c
dfa0: 00000013 c0107620 be9a1f78 0000000c 00000013 be9a1f94 00000028 00000000
dfc0: be9a1f78 0000000c 00000013 00000122 00000003 00000014 00000002 be9a1f94
dfe0: 00000122 be9a1f18 b6c4d7cd b6bcb206 20010030 00000013 00000000 00000000
[<c057e0c0>] (__netdev_printk) from [<c057e564>] (netdev_info+0x38/0x48)
[<c057e564>] (netdev_info) from [<c046b51c>] (dev_state_ev_handler+0x254/0x318)
[<c046b51c>] (dev_state_ev_handler) from [<c01307bc>] (notifier_call_chain+0x48/0x84)
[<c01307bc>] (notifier_call_chain) from [<c0130f88>] (blocking_notifier_call_chain+0x44/0x5c)
[<c0130f88>] (blocking_notifier_call_chain) from [<c05e57ec>] (__inet_insert_ifa+0x1b4/0x290)
[<c05e57ec>] (__inet_insert_ifa) from [<c05e62e4>] (inet_rtm_newaddr+0x300/0x380)
[<c05e62e4>] (inet_rtm_newaddr) from [<c0594140>] (rtnetlink_rcv_msg+0x214/0x290)
[<c0594140>] (rtnetlink_rcv_msg) from [<c05a927c>] (netlink_rcv_skb+0x9c/0xe8)
[<c05a927c>] (netlink_rcv_skb) from [<c05a8a64>] (netlink_unicast+0x17c/0x20c)
[<c05a8a64>] (netlink_unicast) from [<c05a8ef0>] (netlink_sendmsg+0x330/0x370)
[<c05a8ef0>] (netlink_sendmsg) from [<c0564e3c>] (sock_sendmsg+0x14/0x24)
[<c0564e3c>] (sock_sendmsg) from [<c0565f18>] (SyS_sendto+0x88/0xe4)
[<c0565f18>] (SyS_sendto) from [<c0107620>] (ret_fast_syscall+0x0/0x54)
Code: e92d47f0 e2514000 e24dd010 0a000071 (e5946270)
---[ end trace 0c0d488901df069e ]---
```

AdhamAbozaeid, 2018-11-27:
Hi @drewwestrick 
Looks like your issue is with dev_state_ev_handler() in linux_wlan.c
This function gets called when any network device gets an IP, and code below checks if it's a wireless interface
```
	if (memcmp(dev_iface->ifa_label, IFC_0, 5) &&
	    memcmp(dev_iface->ifa_label, IFC_1, 4)) {
		PRINT_INFO(vif->ndev, GENERIC_DBG,
			   "Interface is neither WLAN0 nor P2P0\n");
		return NOTIFY_DONE;
	}
```
However, it prints the info using PRINT_INFO(vif->ndev, ) while ndev isn't valid.
This was changed on the dev branch and will be released  in version 15.2
As a quick fix, you can change remove the PRINT_INFO, or use 
`pr_info("Interface is neither WLAN0 nor P2P0\n");`
instead

closed 2018-11-27 by AdhamAbozaeid

drewwestrick, 2018-11-27:
@AdhamAbozaeid 

Thanks for the help so far. As predicted that solves the kernel panic problem. Below is now what I get when I build the wilc-sdio as a module and do a **modprobe wilc-sdio**.

```
[ +29.150000] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
[  +0.000000] (unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
[  +0.000000] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Successful Registering
[  +0.000000] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
[  +0.000000] (unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
[  +0.000000] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Successful Registering
[  +0.000000] wilc_sdio mmc1:0001:1: WILC got 57 for gpio_reset
[  +0.000000] wilc_sdio mmc1:0001:1: WILC got 56 for gpio_chip_en
[  +0.000000] wilc_sdio mmc1:0001:1: WILC got 35 for gpio_irq
[  +0.000000] wifi_pm : 0
[  +0.000000] wifi_pm : 1
[  +0.080000] wilc_sdio mmc1:0001:1: Driver Initializing success
```

When I then do a **ifconfig wlan0 up** I get the following:

```
[Nov27 21:45] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[c742c800] wlan0
[  +0.000000] WILC POWER UP
[  +0.000000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_init_host_int]Host[c742c800][c0dba500]
[  +0.000000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
[  +0.000000] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_init_locks]Initializing Locks ...
[  +0.000000] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
[  +0.000000] wilc_sdio mmc1:0001:1: SDIO speed: 50000000
[  +0.000000] wilc_sdio mmc1:0001:1: wilc_sdio_cmd52..failed, err(-110)
[  +0.010000] wilc_sdio mmc1:0001:1: Fail cmd 52, enable csa...
[  +0.000000] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_initialize: 1158] Initializing WILC_Wlan FAILED
[  +0.010000] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinit_locks]De-Initializing Locks
[  +0.000000] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_initialize: 1245] WLAN initialization FAILED
[  +0.010000] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_mac_open: 1295] Failed to initialize wilc
[  +0.020000] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_send_config_pkt: 1929] Set Timed out
[  +0.000000] wilc_sdio mmc1:0001:1 wlan0: ERR [handle_set_wfi_drv_handler: 445] Failed to set driver handler
[  +0.010000] wilc_sdio mmc1:0001:1 wlan0: Host cmd 24 failed
[  +0.040000] wilc_sdio mmc1:0001:1 wlan0: ERR [host_if_work: 3000] [Host Interface] undefined
```

Looking at an o-scope I see the Reset and Enable pins working correctly. I can also see the SDIO clock running during the ifconfig wlan0 up but then it stops as sits a zero volts.

AdhamAbozaeid, 2018-11-28:
Hi @drewwestrick 
This looks like an SDIO bus failure. At this stage
```
[  +0.000000] wilc_sdio mmc1:0001:1: SDIO speed: 50000000
[  +0.000000] wilc_sdio mmc1:0001:1: wilc_sdio_cmd52..failed, err(-110)
[  +0.010000] wilc_sdio mmc1:0001:1: Fail cmd 52, enable csa...
```
The driver should be using the bus to identify the chip, then start downloading the FW.
It looks like at least the host can identify the WILC SDIO controller since it probes successfully.
I think that might point to a problem with some of the DAT lines, because the SDIO probe doesn't use the 4 DAT lines, but reading the chip id needs them.
How is the WILC connected to your host? Is it a WILC3000 shield board plugged in the host's arduino shield connector, a WILC3000 SD board that's plugged in the host's mmc slot, or your own design with WILC3000 module?
I would check the DAT lines and make sure they are connected correctly, and isn't using fly wires.
If that still didn't work, please open a ticket on [Microchip Salesforce](http://microchipsupport.force.com/) and the team will help you more with checking your design and other porting issues.

The way the documentation is separated is that the porting guide explains how to integrate the driver into a new kernel and host, and do the required config and dts changes, while the linux user guide mainly explains how to use the demo reference platform which is the SAMA5D4 + at-wilc kernel 4.9. The later is more or less a user guide to linux wireless and BLE  more than it's a guide to WILC.

Thanks,
Adham

drewwestrick, 2018-11-28:
@AdhamAbozaeid 

This seems to line up with what I'm seeing. On the oscilloscope, I can see the clock and command lines working properly during the **modprobe** and also during the **ifconfig**. I do not see any activity at all on the data lines during the **ifconfig** nor the **modprobe** (it sounds like there shouldn't be during the modprobe).

I am connecting the module via MMC1. I'm using the CLK, CMD, and DAT0-3 lines. I'm not currently using the chip detect or the reset line from the MMC module. My understanding is that the **non-removable;** flag in the dtc should tell the driver not to look for a chip detect line. Are there any example dts files you are aware of for the SAMA5D2 family that show how the module should be setup from a device tree perspective?

I'm using the module on a custom board with relatively short traces so signal integrity shouldn't be an issue. I've traced the pins on the layout and they are connected to the proper pins on the SAMA5 SIP part as well as the ATWILC3000 part. 

At this point, I'm thinking this may be a device tree issue or some other problem with data communication on MMC1. I will take your advice and open a support case on salesforce. I can also reach out to my applications engineer and see if he can help support this issue too. Let me know if you have any further ideas or guidance.

AdhamAbozaeid, 2018-11-28:
@drewwestrick 

You can check the dts file for sama5d4.
And you are correct, **non-removable** will ignore the line detect, which is already ok in your case since the probe is successful.
Please go ahead and open a salesforce ticket and the respective team will help you review the design. My guess is that it might be a noise problem on the SDIO lines.
Meanwhile, maybe you can try lowering the SDIO clock. if that works, then it's definitely a noise problem

drewwestrick, 2018-11-28:
@AdhamAbozaeid 

I'm putting a ticket together right now. I have already tried lowering the clock to 20 Mhz as well as 2 Mhz. I'm not sure it is noise issue since I don't see any signals on the data pins at all. I will post an update for others to see if I'm able to come up with a solution.

drewwestrick, 2018-11-29:
@AdhamAbozaeid 

I'll update my support ticket as well with this information, but I believe there may be an issue with the GPIO reset and enable pin timing. When I take the gpio_reset and gpio_chip_en definitions out of the dts file I get to the point where the firmware is downloaded into the device. I then get a timeout on the firmware starting. Any ideas why removing the gpio control allows the device to at least accept a firmware download? It appears that the gpio drives the reset line low while the SDIO is attempting to communicate with the ATWILC3000.

```
# modprobe wilc-sdio
(unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
(unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
(unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Successful Registering
(unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
(unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
(unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Successful Registering
wilc_sdio mmc1:0001:1: WILC setting default Reset GPIO to 60. Got -2
wilc_sdio mmc1:0001:1: WILC setting default Chip Enable GPIO to 94. Got -2
wilc_sdio mmc1:0001:1: WILC got 35 for gpio_irq
wifi_pm : 0
wifi_pm : 1
wilc_sdio mmc1:0001:1: Driver Initializing success
# wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[c7941800] wlan0
WILC POWER UP
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_init_host_int]Host[c7941800][c7833000]
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_init_locks]Initializing Locks ...
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
wilc_sdio mmc1:0001:1: SDIO speed: 25000000
wilc_sdio mmc1:0001:1: chipid 003000d0
wilc_sdio mmc1:0001:1 wlan0: INFO [init_chip]Bootrom sts = d
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_initialize]WILC Initialization done
wilc_sdio mmc1:0001:1 wlan0: INFO [init_irq]IRQ request succeeded IRQ-NUM= 86 on GPIO: 35
wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Initializing Threads ...
wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Creating kthread for transmission
wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Creating kthread for Debugging
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]Detect chip WILC3000
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]loading firmware mchp/wilc3000_wifi_firmware.bin
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]WLAN firmware: mchp/wilc3000_wifi_firmware.bin
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_firmware_download]Downloading Firmware ...
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_firmware_download]Downloading firmware size = 136076
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_firmware_download]Offset = 119296
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_firmware_download]Offset = 136076
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_firmware_download]Download Succeeded
wilc_sdio mmc1:0001:1 wlan0: INFO [linux_wlan_start_firmware]Starting Firmware ...
wilc_sdio mmc1:0001:1 wlan0: INFO [linux_wlan_start_firmware]Waiting for FW to get ready ...
wilc_sdio mmc1:0001:1 wlan0: INFO [linux_wlan_start_firmware]Firmware start timed out
wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_initialize: 1197] Failed to start firmware
wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinitializing Threads
wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinit debug Thread
wilc_sdio mmc1:0001:1 wlan0: INFO [debug_thread]Exit debug thread
wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinitializing Threads
wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinit_locks]De-Initializing Locks
wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_initialize: 1245] WLAN initialization FAILED
wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_mac_open: 1295] Failed to initialize wilc
wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_send_config_pkt: 1929] Set Timed out
wilc_sdio mmc1:0001:1 wlan0: ERR [handle_set_wfi_drv_handler: 445] Failed to set driver handler
wilc_sdio mmc1:0001:1 wlan0: Host cmd 24 failed
wilc_sdio mmc1:0001:1 wlan0: ERR [host_if_work: 3002] [Host Interface] undefined
wilc_sdio mmc1:0001:1 p2p0: ERR [wilc_get_tx_power: 4586] Failed to get TX PWR
```

AdhamAbozaeid, 2018-11-30:
@drewwestrick 
Not really.. but maybe there's a problem with how the driver is controlling the chip_en and reset_n GPIOs, and when you take them out of the dts, the driver uses the hardcoded GPIO numbers, which aren't connected to WILC chip_en and reset_n on your setup, and thus the chip_en and reset_n would be pulled high all the time.
It'll be useful if you can probe these two signals