linux4wilc/driver #28: Problem with driver init sequence

bht-patrick-schneider, 2018-12-02:
Hi,
I am having a bit of a strange behaviour when trying to bring bluetooth and wifi up. As this would be too long and confusing to be posted here I attached a zip file with different scenarios and their outcome.
Basically the problem is that I have to trigger the driver initialization two times and then bring bluetooth up before wifi - otherwise something crashes. I have to follow this exact procedure to prevent any errors.
This doesn't seem right to me - should it?
Every scenario has the corresponding script which reproduces the behaviour, the log from our product and a readme for explanation.
Any insights you guys might have would be very helpful in locating the problem!
Thanks in advance,
Patrick

[atwilc3000_driver_init_logs.zip](https://github.com/linux4wilc/driver/files/2637078/atwilc3000_driver_init_logs.zip)

AdhamAbozaeid, 2018-12-03:
Hi
Sure not. Having to initialize the driver twice before bringing bluetooth up isn't a normal behavior.

1. single_bt_bad: the correct commands should be echoing the "BT_POWER_UP" and other commands to /dev/wilc_bt, which I don't see happening in 1_atwilc_startup_single_bt_bad.log
single_wifi_bad: Looks like a bus error is happening with the first 4-line SDIO communication happening. 2. Same as #27 , I suspect a problem with the DAT[3:0] lines. However, I can't explain why it works if you did the probe twice.
5. The eth0 crach, this is a known issue that was fixed in the dev branch, and will be released soon. You can find more information about it [here](https://github.com/linux4wilc/driver/issues/27#issuecomment-442210519)
 
For issues 2,3 & 4, please open a request on MCHP [salesforce](http://microchipsupport.force.com/), and the team will be able to dig deeper with you.

Thanks,
Adham

bht-patrick-schneider, 2018-12-04:
Hi @AdhamAbozaeid,
thanks for the quick reply!
1. 1_atwilc_startup_single_bt_bad.log if you look in the script I am echoing them to /dev/wilc_bt but NOTHING happens, thats the strange/erratic behaviour here.
atwilc_single_wifi_bad: yes there is only communication on the CMD line while the error persists, if probed twice I can measure the Data lines as well. That's probably the "root" error here which I have to find.
2. Ah okay, sorry for not seeing that in past issues. At this point - can you recommend using the dev-branch, is it stable, should I switch? If so, should I also use the firmware from the dev branch?

I already have an ongoing ticket at salesforce for several weeks now, this is very new information though so I thought maybe someone here could also help!

Another question, I suspect it could all be related to something going on with the SLEEP modes. How can I "completely" disable that in linux and/or the driver?

Thanks again,
Patrick

AdhamAbozaeid, 2018-12-04:
Hi @bht-patrick-schneider 
I wonder why the echo commands doesn't look  correct in the log file. It looks like the commands aren't reaching the driver because you should be seeing some logs if they did. Can you try executing them manually and check if there's a difference in behavior?

The dev branch will be merged back to the master as release 15.2 after enough being QAed. The planned timeline for that is by end of December. I don't think any of the fixes there would fix your problems -other than the print info crash-, but it's still worth giving it a try.

Bus errors always point to either bad HW connections, or problem with sleep modes. However, I don't see sleep mode being enabled in the logs before the bus errors you are seeing. It's still worth trying.
The sleep modes is by default disabled in WILC, and is controlled by 2 ways:
* Kernel config: When the supplicant starts, it automatically enables sleep mode if CONFIG_CFG80211_DEFAULT_PS is defined in the kernel's config file. If not, it will disable sleep mode once it starts
* Command line: by using command
`iw dev wlan0 set power_save on`
or
`iw dev wlan0 set power_save off`
If you want, you can also hack the driver in handle_power_management, by hardcoding power_mode to NO_POWERSAVE

closed 2019-01-14 by AdhamAbozaeid