linux4wilc/driver #29: interface mode combinations supported or not?

ZonKnoZ, 2018-12-17:
Hi,

I'm trying to bring up an access point while bein connected to another Hotspot.
Now I realized that the interface itself reports:
```
        Supported interface modes:
                 * managed
                 * AP
                 * monitor
                 * P2P-client
                 * P2P-GO
        software interface modes (can always be added):
        interface combinations are not supported
```
While this driver page is stating:
Concurrent modes
STA-STA
STA-AP
STA-P2 Client
STA-P2P GO
AP-P2P Client

Could someone explain to me if I'm misunderstanding the "STA-AP" part?
I was assuming that this means being an access point and a client simultaneously is possible with the WILC3000

ShonkyCH, 2018-12-17:
I get the same response on a WILC1000, but have successfully had both STA mode (wlan0) and AP mode (p2p0) working concurrently.

AdhamAbozaeid, 2019-01-14:
Hi @ZonKnoZ 
Yes, concurrent STA-AP mode is supported. We will look into the reported interface capabilities, maybe they aren't accurate enough

closed 2019-01-14 by AdhamAbozaeid

ShonkyCH, 2019-01-15:
Surely this should still be open given both devices are reporting incorrect capabilities?