linux4wilc/driver #30: Mainline kernel wilc1000 driver

ShonkyCH, 2019-01-14:
What is happening in the mainline kernel with the wilc1000 driver?
https://github.com/torvalds/linux/tree/master/drivers/staging/wilc1000

There is significant dev work there by a Microchip employee (well someone with a microchip.com email address) based around the older firmware versions.

So there are two drivers that have now diverged significantly particularly due to lots of code formatting/refactoring etc.

If linux4wilc is the official version why are improvements being made to the mainline version? Is anyone working on the linux4wilc codebase following the changes there?

Whilst there have been some improvements, the current linux4wilc driver is still not ready for any level of production use. Throughput is poor compared to any other network device (particularly in the presence of multiple wilc1000 devices). It's also fairly easy to get the driver to block in the kernel which ultimately causes the whole system to become unstable.

AdhamAbozaeid, 2019-01-14:
The wilc1000 driver is currently in staging on the linux kernel, and we are working on mainlining it, hence all the changes on the linux kernel you are referring to.
Changes implemented to the staging driver are merged one by one to linux4wilc, but this process stopped in November 2018 since linux4wilc is on freeze preparing for release 15.2.
The rest of the changes on staging will be merged to the dev branch after 15.2 is released, and will be officially released in 15.3

The ultimate goal is to unify the two drivers, and lots of efforts was done already in that direction if you are following up the history of commits on both repos. However such goal will take some time to achieve, and till then, linux4wilc will remain the official WILC diver.

Throughput measured on the reference platform in standalone mode is 45Mbps for UDP downlink.
For the issues you are seeing in case of multiple WILC devices, and the driver blocking, please open tickets for these issues on [salesforce](http://microchipsupport.force.com/) if you haven't done already.

closed 2019-01-15 by AdhamAbozaeid

AntoineBore, 2019-07-09:
Do you have some news about the mainlining process ? Can we expect the driver to be added in the Linux kernel soon or this process will take a lot of time ?

AdhamAbozaeid, 2019-07-12:
The driver has been submitted for review in [September 2018](https://patchwork.kernel.org/cover/10615585/), and feedback was received from the community. The driver was then modified, and resubmitted in [February 2019](https://patchwork.kernel.org/cover/10804197/), but no feedback was received from the community since then.
The driver was then resubmitted this week, and hopefully will get more attention from the community.

It's a very long process and we don't have much control over it timeline, so I can't comment on when to expect the driver to be in the mainline kernel. You can follow the wireless subsystem kernel forum if interested.