linux4wilc/driver #31: Kernel module fatal exception / stack trace

ShonkyCH, 2019-01-15:
Hardware: SAMA5D4Xplained dev board + ATWILC1000-SD (rev 4) dev board

Software: wilc_demo 15.2RC1 from:
https://github.com/linux4wilc/wilc_demo/tree/009b36048d18ffb64b2140eeecd0194cc21cbfd4

```
# cat /etc/wpa_supplicant.conf
ctrl_interface=/run/wpa_supplicant
ctrl_interface_group=0

network={
ssid="AP"
psk="01234567"
}
```

There is no "AP" access point in the vicinity i.e. the unit has nothing to connect to. There are quite a few other APs around though as you can see in the log.

Steps to reproduce:
```
modprobe wilc-sdio; \
ifconfig wlan0 up; \
/usr/sbin/wpa_supplicant -Dnl80211 -iwlan0 -c /etc/wpa_supplicant.conf -B; \
rmmod wilc-sdio
```

Repeatable every time. Yes this is a somewhat harsh test, but the driver shouldn't be taking down the whole system.

Full log from boot:
```
AT91Bootstrap 3.8.8 (Thu May 31 12:57:18 MST 2018)

EEPROM: Loading AT24xx information ...
EEPROM: BoardName | [Revid] | VendorName
HW Info: The total size: 0xff isn't correct

EEPROM: Using default information

EEPROM: Board sn: 0x1012420 revision: 0x680820

HDMI SiI9022: Enter D3 Cold mode
NAND: ONFI flash detected
NAND: Manufacturer ID: 0x2c Chip ID: 0x34
NAND: Page Bytes: 4096, Spare Bytes: 224
NAND: ECC Correctability Bits: 8, ECC Sector Bytes: 512
NAND: Disable On-Die ECC
NAND: Initialize PMECC params, cap: 8, sector: 512
NAND: Image: Copy 0xa0000 bytes from 0x40000 to 0x26f00000
NAND: Done to load image
<debug_uart>

U-Boot 2017.03-linux4sam_5.6 (May 31 2018 - 12:57:41 -0700)

CPU: SAMA5D44
Crystal frequency:       12 MHz
CPU clock        :      600 MHz
Master clock     :      200 MHz
DRAM:  512 MiB
NAND:  512 MiB
MMC:   Atmel mci: 0
In:    serial
Out:   serial
Err:   serial
Net:   eth0: ethernet@f8020000
Hit any key to stop autoboot:  0

NAND read: device 0 offset 0x180000, size 0x817c
 33148 bytes read: OK

NAND read: device 0 offset 0x200000, size 0x3df7c8
[... 478 lines cut ...]
1e80: c0a0c3c0 c0a0c3d4 c0a0c414 c01497d8 00000000 c0a0cc40 00000000 d40c1600
1ea0: 00000000 c0a01ef4 0000001c c04af66c d411b440 c01410b0 00000000 00000000
1ec0: 00000001 c0a00000 c0a31a44 c0a31a40 40000001 00000100 c0a31a40 c0119234
1ee0: d4002200 c0a00000 00000000 c0a31a00 0000000a ffffac49 c0a0cc40 00200100
1f00: d40c1600 c0a1329c 0000001c 00000000 00000000 d4002200 c0a00000 00000000
1f20: 00000000 c0140a58 c0487450 800f0013 ffffffff c0a01f74 f2547580 c010b2cc
1f40: 00000000 00000013 f2547580 c0a098f0 f2547580 00000013 c0a24230 00000001
1f60: f2547580 00000013 00000000 00000000 98968000 c0a01f90 c048744c c0487450
1f80: 800f0013 ffffffff 00000051 00000000 c0a24230 c0a24230 c0a00000 c0a0286c
1fa0: c0a24228 c0a02874 c0a0de07 c0a2462c 00000000 c01392fc ffffffff c0900c38
1fc0: ffffffff ffffffff 00000000 c0900684 00000000 c092ca28 c0a31414 c0a02818
1fe0: c092ca24 c0a065f4 20004059 410fc051 00000000 20008078 00000000 00000000
[<c0149614>] (call_timer_fn.constprop.6) from [<c01497d8>] (run_timer_softirq+0x154/0x170)
[<c01497d8>] (run_timer_softirq) from [<c0119234>] (__do_softirq+0xac/0x21c)
[<c0119234>] (__do_softirq) from [<c0140a58>] (__handle_domain_irq+0x58/0xa8)
[<c0140a58>] (__handle_domain_irq) from [<c010b2cc>] (__irq_svc+0x6c/0x90)
[<c010b2cc>] (__irq_svc) from [<c0487450>] (cpuidle_enter_state+0x17c/0x29c)
[<c0487450>] (cpuidle_enter_state) from [<c01392fc>] (cpu_startup_entry+0xb8/0x144)
[<c01392fc>] (cpu_startup_entry) from [<c0900c38>] (start_kernel+0x35c/0x368)
Code: bad PC value
---[ end trace cbc3a1b43c6edad8 ]---
Kernel panic - not syncing: Fatal exception in interrupt
---[ end Kernel panic - not syncing: Fatal exception in interrupt
```
Second stack trace comes a few seconds after the initial one.

AdhamAbozaeid, 2019-01-15:
This look like the same issue fixed in 6b33f29.
I tried the following sequence on the head of dev branch, and it succeeded 10/10 times.
`modprobe wilc_spi
wpa_supplicant -Dnl80211 -iwlan0 -c/etc/wilc_wpa_supplicant.conf &

killall wpa_supplicant
rmmod wilc_spi`

AdhamAbozaeid, 2019-01-17:
The latest demo maps to release 15.01 release as per the commit message. This release is tagged on the driver, firmware and linux-at91 repos, and it doesn't have the fix i'm pointing to

ShonkyCH, 2019-01-21:
Using -dev "updated demo binaries for release 15.2 RC1" is the commit message so no it doesn’t give the driver version. The master driver is much too old/broken to use. 

Have upgraded the firmware binaries to 15.2 RC3 (build 10924) firmware.

Seems to crash every time on the second loop when modprobing module:

'''
# cat wifitest.sh
#!/bin/sh

while true; do
        modprobe wilc-sdio
        sleep 3
        ifconfig wlan0 up
        sleep 1
        /usr/sbin/wpa_supplicant -Dnl80211 -iwlan0 -c /etc/wpa_supplicant.conf -B
        sleep 20
        rmmod wilc-sdio
        sleep 1
        killall wpa_supplicant
        sleep 5
done
'''

Full log
'''
AT91Bootstrap 3.8.8 (Thu May 31 12:57:18 MST 2018)

EEPROM: Loading AT24xx information ...
EEPROM: BoardName | [Revid] | VendorName
HW Info: The total size: 0xff isn't correct

EEPROM: Using default information

EEPROM: Board sn: 0x1012420 revision: 0x680820

HDMI SiI9022: Enter D3 Cold mode
NAND: ONFI flash detected
NAND: Manufacturer ID: 0x2c Chip ID: 0x34
NAND: Page Bytes: 4096, Spare Bytes: 224
NAND: ECC Correctability Bits: 8, ECC Sector Bytes: 512
NAND: Disable On-Die ECC
NAND: Initialize PMECC params, cap: 8, sector: 512
NAND: Image: Copy 0xa0000 bytes from 0x40000 to 0x26f00000
NAND: Done to load image
<debug_uart>

U-Boot 2017.03-linux4sam_5.6 (May 31 2018 - 12:57:41 -0700)

CPU: SAMA5D44
Crystal frequency:       12 MHz
CPU clock        :      600 MHz
Master clock     :      200 MHz
DRAM:  512 MiB
NAND:  512 MiB
MMC:   Atmel mci: 0
In:    serial
Out:   serial
Err:   serial
Net:   eth0: ethernet@f8020000
Hit any key to stop autoboot:  0

NAND read: device 0 offset 0x180000, size 0x817c
 33148 bytes read: OK

NAND read: device 0 offset 0x200000, size 0x3df7c8
 4061128 bytes read: OK
## Flattened Device Tree blob at 21000000
   Booting using the fdt blob at 0x21000000
   Loading Device Tree to 3f951000, end 3f95c17b ... OK
[... 543 lines cut ...]
1e60:                                                       c0a01ef4 0000001c
1e80: c0a0c3c0 c0a0c3d4 c0a0c414 c01497d8 00000000 c0a0cc40 00000000 d40c1600
1ea0: 00000000 c0a01ef4 0000001c c04af66c d411b440 c01410b0 c0a01f40 00000000
1ec0: 00000001 c0a00000 c0a31a44 c0a31a40 40000001 00000100 c0a31a40 c0119234
1ee0: d4002200 c0a00000 00000000 c0a31a00 0000000a ffffadc1 c0a0cc40 00200100
1f00: d40c1600 c0a1329c 0000001c 00000000 00000000 d4002200 c0a00000 00000000
1f20: 00000000 c0140a58 c0487450 80000013 ffffffff c0a01f74 d2718180 c010b2cc
1f40: 00000000 00000014 d2718180 c0a098f0 d2718180 00000014 c0a24230 00000001
1f60: d2718180 00000014 00000000 00000000 98968000 c0a01f90 c048744c c0487450
1f80: 80000013 ffffffff 00000051 00000000 c0a24230 c0a24230 c0a00000 c0a0286c
1fa0: c0a24228 c0a02874 c0a0de07 c0a2462c 00000000 c01392fc ffffffff c0900c38
1fc0: ffffffff ffffffff 00000000 c0900684 00000000 c092ca28 c0a31414 c0a02818
1fe0: c092ca24 c0a065f4 20004059 410fc051 00000000 20008078 00000000 00000000
[<c0149614>] (call_timer_fn.constprop.6) from [<c01497d8>] (run_timer_softirq+0x154/0x170)
[<c01497d8>] (run_timer_softirq) from [<c0119234>] (__do_softirq+0xac/0x21c)
[<c0119234>] (__do_softirq) from [<c0140a58>] (__handle_domain_irq+0x58/0xa8)
[<c0140a58>] (__handle_domain_irq) from [<c010b2cc>] (__irq_svc+0x6c/0x90)
[<c010b2cc>] (__irq_svc) from [<c0487450>] (cpuidle_enter_state+0x17c/0x29c)
[<c0487450>] (cpuidle_enter_state) from [<c01392fc>] (cpu_startup_entry+0xb8/0x144)
[<c01392fc>] (cpu_startup_entry) from [<c0900c38>] (start_kernel+0x35c/0x368)
Code: bad PC value
---[ end trace 76663530cd67116c ]---
Kernel panic - not syncing: Fatal exception in interrupt
---[ end Kernel panic - not syncing: Fatal exception in interrupt
'''

AdhamAbozaeid, 2019-03-18:
@ShonkyCH , are you still seeing this problem?

tsifb, 2019-03-18:
I have also been testing for this issue yesterday and again today. Yes, there is still an issue with this sequence.
More in a few hours....
Summary:
- occasional crash in the driver, especially when wild is under heavy rx load ( rx of udp iperf traffic )
- occasionally stops the entire networking system. Seems to be locked up dead. All interfaces.

ShonkyCH, 2019-03-19:
Yes we are still seeing this. It seems more of a problem when the module is receiving traffic e.g. flood ping it or do a constant UDP iperf from another machine to the module so it's always receiving packets.

ksathara, 2019-05-21:
I had the similar experience with trying to integrate ATWILC1000 with a Raspberry Pi CM3+. I could see the kernel oops saying null kernel reference. When I dig deeper I could see this oops was happening when CMD52 read error occurs. I had wired the two dev kits with wires and I thought this could be a signal integrity issue as my sdio clock was running at 50MHz, So I reduced the sdio clock to 25MHz just to test and it solved the problem. I have tested the driver with iperf, flood pings, wget large files and it seems to hold.

this is the bus error
[ 345.520541] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_xmit]DHCP Message transmitted, type:35 1 1
[ 345.870812] wilc_sdio mmc1:0001:1: wilc_sdio_cmd52..failed, err(-84)
[ 345.870826] wilc_sdio mmc1:0001:1: Failed cmd 52, read reg (000000f1) ...
[ 345.870835] Bus error (5).0 0
[ 346.390762] wilc_sdio mmc1:0001:1: wilc_sdio_cmd52..failed, err(-84)
[ 346.390775] wilc_sdio mmc1:0001:1: Failed cmd 52, read reg (000000f1) ...
[ 346.390784] Bus error (5).0 0
[ 346.393224] Unable to handle kernel NULL pointer dereference at virtual address 0000001a
[ 346.405472] pgd = c5a7bbf3
[ 346.410225] [0000001a] *pgd=00000000

the kernel oops is 
[  89.163586] Unable to handle kernel NULL pointer dereference at virtual address 00000004
[  89.176083] pgd = 53ee692f
[  89.180981] [00000004] *pgd=00000000
[  89.186715] Internal error: Oops: 17 [#1] SMP ARM
[  89.193532] Modules linked in: wilc_sdio sha256_generic cfg80211 rfkill snd_bcm2835(C) bcm2835_codec(C) bcm2835_v4l2(C) v4l2_mem2mem v4l2_common bcm2835_mmal_vchiq(C) snd_pcm videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 snd_timer videobuf2_common snd raspberrypi_hwmon hwmon videodev media vc_sm_cma(C) fixed uio_pdrv_genirq uio ip_tables x_tables ipv6
[  89.238347] CPU: 2 PID: 623 Comm: K_TXQ_TASK Tainted: G     C    4.19.30-v7+ #6
[  89.251273] Hardware name: BCM2835
[  89.257259] PC is at wilc_wlan_txq_remove_from_head+0x60/0xa8 [wilc_sdio]

I think the driver is not gracefully handling a CMD52 read failure. This was failing (in my case) in kernel thread "K_TXQ_TASK " and executing wilc_wlan_txq_remove_from_head () function.

Cheers,
Kaushalyas

Ezhik92, 2019-12-25:
> 
> 
> Yes we are still seeing this. It seems more of a problem when the module is receiving traffic e.g. flood ping it or do a constant UDP iperf from another machine to the module so it's always receiving packets.

@ShonkyCH I have the same problem. Did you solve it?

ShonkyCH, 2019-12-25:
> > Yes we are still seeing this. It seems more of a problem when the module is receiving traffic e.g. flood ping it or do a constant UDP iperf from another machine to the module so it's always receiving packets.
> 
> @ShonkyCH I have the same problem. Did you solve it?

We never had a 100% reliable way to reproduce it - just it seemed that pulling the module whilst receiving packets caused it. I would say it's better in newer versions but can't 100% guarantee it's fixed.  Definitely try the latest release if you haven't already.

We have spent way too much time chasing issues with driver and firmware on this thing, we're just not in a position to burn more time on it whilst it's largely working.