linux4wilc/driver #32: wilc_wlan_txq_add_net_pkt triggers "sleeping function called from invalid context at"

benhegarty, 2019-01-15:
I'm developing with a WILC3000 and am utilising an SPI connection with the wilc_spi driver. I am able to modprobe wilc_spi, but as soon as I initiate a connection or send data I see lots of stack traces in dmesg. If I ping my router, for example, I see the stack trace once per second as the packets are sent out.

Note that the packets _are_ successfully sent and a response is received, so the communication itself is working. It is a bit slower than expected though. There are also some `Failed cmd response` messages in the logs. I'm not sure if the stack traces are causing the failed command responses, or vice versa.

Any help is greatly appreciated.

See kernel output below.

```
[    0.000000] Linux version 3.18.44 (ben@ben-pc) (gcc version 6.2.0 (GCC) ) #20 PREEMPT Tue Jan 15 08:37:47 AEDT 2019
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c53c7d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
```

`$ modprobe wilc_spi`

```
[  307.485632] WILC_SPI spi1.0: spiModalias: wilc_spi, spiMax-Speed: 48000000
[  307.485761] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
[  307.485777] (unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
[  307.486804] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Successful Registering
[  307.490891] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
[  307.490916] (unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
[  307.502328] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Successful Registering
[  307.512967] of_get_named_gpiod_flags: parsed 'gpio_reset' property of node '/soc/spi@78b6000/wilc_spi@0[0]' - status (0)
[  307.513004] WILC_SPI spi1.0: WILC got 58 for gpio_reset
[  307.513044] of_get_named_gpiod_flags: parsed 'gpio_chip_en' property of node '/soc/spi@78b6000/wilc_spi@0[0]' - status (0)
[  307.513059] WILC_SPI spi1.0: WILC got 76 for gpio_chip_en
[  307.513092] of_get_named_gpiod_flags: parsed 'gpio_irq' property of node '/soc/spi@78b6000/wilc_spi@0[0]' - status (0)
[  307.513107] WILC_SPI spi1.0: WILC got 16 for gpio_irq
[  307.513118] wifi_pm : 0
[  307.513186] wifi_pm : 1
[  307.578544] WILC_SPI spi1.0: WILC SPI probe success
```

`$ ifup wlan0`
```
[  335.891003] WILC_SPI spi1.0 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  335.891024] WILC_SPI spi1.0 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  335.891094] WILC_SPI spi1.0 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  335.891108] WILC_SPI spi1.0 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  335.891165] WILC_SPI spi1.0 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  335.891178] WILC_SPI spi1.0 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  335.891231] WILC_SPI spi1.0 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  335.891245] WILC_SPI spi1.0 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  335.891299] WILC_SPI spi1.0 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  335.891312] WILC_SPI spi1.0 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  335.891363] WILC_SPI spi1.0 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  335.891376] WILC_SPI spi1.0 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  335.891430] WILC_SPI spi1.0 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  335.891444] WILC_SPI spi1.0 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  335.892456] WILC_SPI spi1.0 wlan0: INFO [wilc_mac_open]MAC OPEN[c8b89000] wlan0
[  335.892473] WILC POWER UP
[  335.892488] WILC_SPI spi1.0 wlan0: INFO [wilc_init_host_int]Host[c8b89000][c818d400]
[  335.892727] WILC_SPI spi1.0 wlan0: INFO [wilc_mac_open]*** re-init ***
[  335.892744] WILC_SPI spi1.0 wlan0: INFO [wlan_init_locks]Initializing Locks ...
[  335.892757] WILC_SPI spi1.0 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
[  335.894529] WILC_SPI spi1.0 wlan0: INFO [init_chip]Bootrom sts = c
[  335.894899] WILC_SPI spi1.0 wlan0: INFO [wilc_wlan_initialize]WILC Initialization done
[  335.895186] WILC_SPI spi1.0 wlan0: INFO [init_irq]IRQ request succeeded IRQ-NUM= 68 on GPIO: 16
[  335.895205] WILC_SPI spi1.0 wlan0: INFO [wlan_initialize_threads]Initializing Threads ...
[  335.895217] WILC_SPI spi1.0 wlan0: INFO [wlan_initialize_threads]Creating kthread for transmission
[  335.895366] WILC_SPI spi1.0 wlan0: INFO [wlan_initialize_threads]Creating kthread for Debugging
[  335.895510] WILC_SPI spi1.0 wlan0: INFO [wilc_wlan_get_firmware]Detect chip WILC3000
[  335.895525] WILC_SPI spi1.0 wlan0: INFO [wilc_wlan_get_firmware]loading firmware mchp/wilc3000_wifi_firmware.bin
[  335.895538] WILC_SPI spi1.0 wlan0: INFO [wilc_wlan_get_firmware]WLAN firmware: mchp/wilc3000_wifi_firmware.bin
[  335.927406] WILC_SPI spi1.0 wlan0: INFO [wilc_firmware_download]Downloading Firmware ...
[  335.927435] WILC_SPI spi1.0 wlan0: INFO [wilc_wlan_firmware_download]Downloading firmware size = 136076
[  336.085623] WILC_SPI spi1.0 wlan0: INFO [wilc_wlan_firmware_download]Offset = 119296
[  336.109816] WILC_SPI spi1.0 wlan0: INFO [wilc_wlan_firmware_download]Offset = 136076
[... 1026 lines cut ...]
[  371.810454] WILC_SPI spi1.0 wlan0: INFO [dev_state_ev_handler]IP add=192:168:69:183
[  372.481046] WILC_SPI spi1.0 wlan0: INFO [remove_network_from_shadow]Number of cached networks: 7
[  372.930894] WILC_SPI spi1.0 wlan0: INFO [debug_thread]*** Debug Thread Running ***
[  378.930274] WILC_SPI spi1.0 wlan0: INFO [debug_thread]*** Debug Thread Running ***
[  381.501302] WILC_SPI spi1.0 wlan0: INFO [remove_network_from_shadow]Number of cached networks: 7
[  384.931352] WILC_SPI spi1.0 wlan0: INFO [debug_thread]*** Debug Thread Running ***
[  390.521192] WILC_SPI spi1.0 wlan0: INFO [remove_network_from_shadow]Network expired in ScanShadow: OPTUS_D00739
[  390.530291] WILC_SPI spi1.0 wlan0: INFO [remove_network_from_shadow]Network expired in ScanShadow: DODO-1F52
[  390.540086] WILC_SPI spi1.0 wlan0: INFO [remove_network_from_shadow]Network expired in ScanShadow: Mohit Wireless Canb 2.4Ghz
[  390.551371] WILC_SPI spi1.0 wlan0: INFO [remove_network_from_shadow]Network expired in ScanShadow: Ms Whovian
[  390.561263] WILC_SPI spi1.0 wlan0: INFO [remove_network_from_shadow]Number of cached networks: 3
[  390.931601] WILC_SPI spi1.0 wlan0: INFO [debug_thread]*** Debug Thread Running ***
[  396.931574] WILC_SPI spi1.0 wlan0: INFO [debug_thread]*** Debug Thread Running ***
[  399.541243] WILC_SPI spi1.0 wlan0: INFO [remove_network_from_shadow]Network expired in ScanShadow: TPG 5273
[  399.550012] WILC_SPI spi1.0 wlan0: INFO [remove_network_from_shadow]Network expired in ScanShadow: Prestige Worldwide
[  399.560657] WILC_SPI spi1.0 wlan0: INFO [remove_network_from_shadow]Number of cached networks: 1
[  402.931607] WILC_SPI spi1.0 wlan0: INFO [debug_thread]*** Debug Thread Running ***
[  408.560146] WILC_SPI spi1.0 wlan0: INFO [remove_network_from_shadow]Network expired in ScanShadow: DangerZone
[  408.569040] WILC_SPI spi1.0 wlan0: INFO [remove_network_from_shadow]Number of cached networks: 0
[  408.577855] WILC_SPI spi1.0 wlan0: INFO [remove_network_from_shadow]No need to restart Aging timer
[  408.930407] WILC_SPI spi1.0 wlan0: INFO [debug_thread]*** Debug Thread Running ***
[  414.931167] WILC_SPI spi1.0 wlan0: INFO [debug_thread]*** Debug Thread Running ***
[  420.930988] WILC_SPI spi1.0 wlan0: INFO [debug_thread]*** Debug Thread Running ***
[  426.931320] WILC_SPI spi1.0 wlan0: INFO [debug_thread]*** Debug Thread Running ***
```

AdhamAbozaeid, 2019-01-15:
Hi @benhegarty 

1- For the **debug messages**, you can either change the kernel's verbosity using dmesg -n 1, or you can compile your kernel with less verbosity level

2- **Sleeping function called from invalid context** Looks like this happens because while transmitting a packet, wilc_mac_xmit adds the packet to the tx Q using wilc_wlan_txq_add_net_pkt, which kmalloc a txq_entry_t  object to be posted to the Q.
I wonder why I'm not seeing similar behavior on my setup!

3- **SPI Fails** Probably this might be related to powersave. I've seen this problem before on platforms that have powersave enabled by default at init, but not observed on platforms that enables powersave manually using 
`iw dev wlan0 set power_save on`
To confirm this is your problem, you can hack the driver to disable powersave in  wilc_set_power_mgmt(), or compile your linux with CONFIG_CFG80211_DEFAULT_PS undefined from your config

benhegarty, 2019-01-19:
Hi @AdhamAbozaeid 

1: **Debug messages** - Thanks for the tip :smile: 

2: **Sleeping function error** - I was able to resolve this by changing the kmalloc type in `wilc_wlan_txq_add_net_pkt()`  from `GFP_KERNEL` to `GFP_ATOMIC`. I am getting much better performance now and the stack-traces have gone. Are there any negative side-affects which this change could cause? It seems stable so far...

3: **SPI Fails** - I hacked the driver and now the SPI errors are gone! If I'd like to utilise sleep mode in the future how can I resolve this? Could this problem be caused by an incorrectly wired RTC? 

Again, thanks for the help!

AdhamAbozaeid, 2019-01-21:
Thanks for confirming.

2. The only problem with GFP_ATOMIC is that it might fail to allocate memory if your host is struggling with dynamic memory, since it doesn't block. I believe this should be the correct behavior if ndo_start_xmit() doesn't allow blocking.

3. Instead of hacking the driver, you can undefine CONFIG_CFG80211_DEFAULT_PS in your kernel .config.  In this case, the supplicant will not enable PS by default, and then you will be able to enable it manually after starting the supplicant using 
`iw dev wlan0 set power_save on`
This case doesn't have problem as per our testing.
We still need to fix the root cause of this problem, so keep tuned for a proper fix that doesn't require you to undefine CONFIG_CFG80211_DEFAULT_PS

benhegarty, 2019-02-05:
Thank you for your help @AdhamAbozaeid 

1. Is there another option you'd suggest I explore (aside from using GFP_ATOMIC) before closing this issue?

2. Crystal clear :) Thanks!

AdhamAbozaeid, 2019-02-15:
@benhegarty, I think GFP_ATOMIC should be the way to go. I should also change it in on our code since ndo_start_xmit doesn't allow blocking

benhegarty, 2019-02-16:
Great! Thanks for your help!

ShonkyCH, 2019-04-10:
Seeing this just now. Is this issue to be fixed still? I still see in -dev the line

`
	tqe = kmalloc(sizeof(*tqe), GFP_KERNEL);
`

in wilc_wlan.c/txq_add_net_pkt(). Is this still to be fixed?

AdhamAbozaeid, 2019-04-10:
Yes, this will be pushed soon on the dev branch

ShonkyCH, 2019-04-11:
Why not push it now? This was two months ago.

AdhamAbozaeid, 2019-04-11:
I understand this can be a critical fix for you, and though should be for us as well.
However, because of already set plans that we are trying to meet, we weren't able to implement and test the change yet. Also the change has to be approved by the Linux staging community, which is still pending.
Since the change is simple, I advise that you do the change on your local copy for now, till we push it to the development branch.

ShonkyCH, 2019-04-12:
That's what the dev branch is for, right?. You have previously stated that this repo is the official wilc1000 driver. Why would you withhold known fixes from the official repo's dev branch?

How many other fixes are you holding on to? Do we have to go one by one through the mainline fixes where it has very significantly diverged from this (official) driver. Why keep updating the mainline driver? What purpose does it even serve?

Yes we can patch the driver. We already usually run the dev version due to the numerous issues we are finding. Yes, it's not ideal but what choice do we have?

Can the "set plans" you mention be communicated to us somehow? It seems there's something going on with some kind of timeline but we are in the dark. Saying 3 months for testing the firmware for example. Given the issues we have found I find it hard to believe that level of testing has ever been done.

AdhamAbozaeid, 2019-04-12:
Patches submitted to dev branch goes through basic testing before they are made public. Also if the change applies to the WILC driver on linux kernel repo, it's also submitted there to get the community's feedback and ensure the quality of the fix, and also to converge the difference between the driver on github and the diverged version on staging.
You don't have to go through the mainline fixes since we take care of patching this driver with all patches submitted to the staging.
Updating the mainline driver is important because we need the WILC driver to be available with the mainline kernel that has the same stability and functionality of this driver.
I understand your frustration about the long testing cycles, but the testing cycles involves too many combinations for each release:
* WILC 1000 SDIO
* WILC 3000 SDIO
* WILC 1000 SPI
* WILC 3000 SPI
and different features:
* STA
* AP
* P2P
* Monitor
* Concurrency
* BLE
* Wifi/BLE coexistence

I won't convince you whether this level of testing have ever been done, but that's the truth :)
Most of the issues found so far was either related to different hosts/kernel version being used (like this one), or requires special setup to be observed, like the CF-End packet problem where multiple WILCs had to be operating in the same environment and the sniffer logs had to be examined to see the issue clearly.

The "set plans" are the release plans we are working on and that we submit changes for on the dev branch.

tsifb, 2019-04-12:
+1 for pushing this to the dev branch as soon as possible... its a safe, easy, no-brainer fix that will impact every user.
and after all.... that is what a development branch is for.

Also would like to understand the set plans mentioned, at whatever level can be shared here.

tsifb, 2019-04-12:
@AdhamAbozaeid , maybe another dev-staging branch to update more frequently?
As you know, there is help available here to test.

tsifb, 2019-04-12:
@benhegarty @AdhamAbozaeid this issue should probably remain open until a fix is pushed. (or a new one opened.)

AdhamAbozaeid, 2019-04-12:
Testing before pushing to the dev branch is basic and not a bottleneck, what delays a specific fix is for it get enough priority for the team to work on

AdhamAbozaeid, 2019-04-12:
Agree.
I'll reopen this one.

ShonkyCH, 2019-05-21:
Shouldn't

https://github.com/linux4wilc/driver/blob/dev/wilc/wilc_wlan.c#L1434

...also have the same GFP_ATOMIC malloc?

AdhamAbozaeid, 2019-05-21:
I didn't see the kernel throwing warning about that as it did with the kmalloc in the calls from cfg80211, so i assume this isn't an atomic context.
Tracing that back to where it's called from sdio_irq.c, it looks like it's being called from sdio_irq_thread process context

commit bff2ae70 (2019-04-16): staging: wilc1000: use atomic kmalloc in atomic context

kmalloc_array in wilc_set_multicast_list() should be atomic since it is
called via ops->ndo_set_rx_mode()from __dev_set_rx_mode() from
dev_set_rx_mode() ops->is with a spinlock held

same for txq_add_net_pkt and txq_add_mgmt_pkt

Fixes #32
closed 2019-09-18 by AdhamAbozaeid