linux4wilc/driver #36: WILC3000 Shield and Raspberry Pi compatibility

omenlabs, 2019-02-01:
I have so far been unable to get the driver in this repository to work with the Raspberry Pi.  There is a forked driver by @rocky134 that does work.  Are there plans to integrate his changes back into this driver (or is it actually known to work and I'm messing something subtle up)?

Buried in issue #12 from @rocky134 is what appears to be the underlying problem:

https://github.com/linux4wilc/driver/issues/12#issuecomment-402562755

The comment below points to the changes and was extremely helpful in getting the WILC3000 up and running:

https://github.com/linux4wilc/driver/issues/12#issuecomment-402576264

rocky134, 2019-02-01:
Hi,
the fix you mention while does work was more of a quick dirty patch I did for a senior design project.
I wouldn't recommend it in its current state for any real use case as there were issues such as connection dropping for seemingly no reason and when used as a hotspot dropped users frequently.
If you were to use this, it would probably need some rework.

Also I learned on an unrelated project that the `gpio_request_one` calls I used are for [legacy backwards compatibility](https://www.kernel.org/doc/html/v4.18/driver-api/gpio/legacy.html) and `gpio_request` should have worked on modern kernels, why it does not on the pi is unknown to me.

AdhamAbozaeid, 2019-02-01:
@omenlabs, the comment from [#12 (comment)](https://github.com/linux4wilc/driver/issues/12#issuecomment-402576264) points to a fix done by defining the GPIOs (chip_en, reset.. etc) to the host's dts file, which is an essential step for porting to your platform as described in the porting guide.
This modification shouldn't be integrated here since it's platform dependent.

AdhamAbozaeid, 2019-02-01:
@ZonKnoZ , I think this is a separate issue, since @rocky134 isn't using multiple WILC STA. I don't see connection drops when I use one WILC in STA or softAP modes

omenlabs, 2019-02-01:
@AdhamAbozaeid Buried in issue #12 from @rocky134 is what appears to be the underlying problem:

https://github.com/linux4wilc/driver/issues/12#issuecomment-402562755

I have the gpios set correctly afaik in my clone of the official driver, but the chip doesn't detect.  I am also using the same device tree overlay between the two drivers.  It works with @rocky134 changes (and it is properly getting the gpios from the device tree based on the printks in the driver), but drops connections as he describes.

My point about the second comment is that although the shield comes with a PI header, it isn't well documented for people that don't regularly work with device tree overlays.

AdhamAbozaeid, 2019-02-01:
It's not very clear to me on why gpio_request_one() works, while gpio_request() doesn't. If gpio_request() fails to retrieve the gpio, how would supplying an initial configuration make it work?
Do you have more information on why gpio_request() fails on Pi?

The shield board is also compatible with other boards like the SAMA5, so changes to the device tree are inevitable.
The porting guide describes the required modifications to be done to the device tree. If it's not clear enough, I would appreciate if you suggest modifications to make it clearer.

omenlabs, 2019-02-01:
I'm just starting to scratch the surface on this.  I'll try to look into why `gpio_request_one()` works while `gpio_request()` doesn't.

I am happy to do a small write up and make a pull request with a device tree overlay for the Raspberry Pi.  How would you like it structured?  Should I make a sub directory for it?

AdhamAbozaeid, 2019-02-01:
That would be great!
If all your changes are portable then it doesn't have to be structured. You can do them directly on the driver files.
For the Pi device tree, you can add a new dts for PI

tsifb, 2019-02-04:
> @ZonKnoZ , I think this is a separate issue, since @rocky134 isn't using multiple WILC STA. I don't see connection drops when I use one WILC in STA or softAP modes

@AdhamAbozaeid even with a single WILC1000 STA, you will see invalid packets being transmitted, as you could also see in the packet capture file I posted in #33. While this may not perturb the network enough to disconnect in your case or mine, it does appear to cause some packet loss (somehow?) which will play havoc with a high throughput TCP stream. 
To me, it is entirely reasonable that these issues may be related as @ZonKnoZ suggested, especially given that we do not understand the exact trigger that causes the firmware to transmit erroneous traffic, and we are unsure which firmware versions cause this issue.

AdhamAbozaeid, 2019-02-04:
The only way to make sure is to examine the air sniffer logs and see what the reason of these disconnections are.
The point here is that to me @rocky134 's problem is different, so he shouldn't expect 100% that it will go away when we fix the CF-end packets issue.

AdhamAbozaeid, 2019-03-18:
@omenlabs , @rocky134 , are you still having problems related to using WILC with PI?

closed 2019-04-22 by AdhamAbozaeid