linux4wilc/driver #37: Starting hostapd -> ERR [wilc_gnrl_async_info_received:4013] there is no current Connect Request

ShonkyCH, 2019-03-05:
When starting hostapd this error is logged by the driver. 

Driver version:
https://github.com/linux4wilc/driver/commit/720f7d4fe2cc8bb17ca9e06775d09768572dbbb6

15.2 RC3 "release" firmware:
https://github.com/linux4wilc/firmware/commit/3e6c54d0449390cc3f0280d219f4aebcacab2ad0

wlan0 is up and attached to an access point

Starting hostapd results in error log

```
2019-03-05T16:17:12.172+10:00 SM01-1839000092 kernel: wilc_sdio mmc1:0001:1 p2p0: ERR [wilc_gnrl_async_info_received:4013] there is no current Connect Request
```

Happens every time

deanoati, 2019-03-05:
FWIW, I also receive this exact same notice. I am using the officially tagged/released 15.2 driver and firmware.

```
root@xxx:~# /etc/init.d/apmode_manager forceup
apmode_manager Running as PID 9405
signal sent
root@xxx:~# WILC POWER UP
wilc_sdio mmc0:0001:1: SDIO speed: 50000000
wilc_sdio mmc0:0001:1: chipid 001003a0
wilc_sdio mmc0:0001:1: got gpio_irq successfully
wilc_sdio mmc0:0001:1 p2p0: ERR [wilc_gnrl_async_info_received:4013] there is no current Connect Request

root@xxx:~#
```

AdhamAbozaeid, 2019-03-05:
Does the hostapd fail to start?

deanoati, 2019-03-05:
For me, hostapd works properly, the error seems spurious.

deanoati, 2019-03-05:
> Does the hostapd fail to start?

For me, hostapd works properly, the error seems spurious.

ShonkyCH, 2019-03-05:
hostapd appears to work here too.

However, the driver is logging an ERROR level message. So that's an ERROR, right? If this is defensive coding, it the root cause should still not be happening in my opinion. Is something else wiping a pointer or incorrectly overwriting a data structure etc?

The specific line is 
```
	if (!hif_drv->usr_conn_req.conn_result) {
		PRINT_ER(vif->ndev, "there is no current Connect Request\n");
		mutex_unlock(&hif_deinit_lock);
		return;
	}
```
How is that value becoming NULL?

AdhamAbozaeid, 2019-03-05:
My question was only to assess the priority of the issue