linux4wilc/driver #38: WILC3000 Initialisation failure

BDP42, 2019-04-04:
I'm using the WILC3000 SD extension board to evaluate the WILC3000 WiFi Device. It's connected to our evaluation platform via the SD interface. The evaluation platform is an iMX6-based platform running Android7/Linux 4.9.
The WILC driver was pulled from the master branch - release 15.2
Firmware - release 15.2
The WILC3000 SD module is set for SD mode.

I'm having a number of problems bringing up the device. I've tried 3 different scenarios and getting different behaviour but it fails to initialise in all cases.

1- I modified the kernel using the Linux Porting Guide document as a guide. I've added the device tree for this platform and I've connected the IRQ, EN and RESET pins to GPIOs on the evaluation platform and I've verified they are working. The device is detected by the kernel on startup. When I try to bring up the wlan0 interface I get SDIO timeouts and communication between the kernel and the device stops. I've attached the full console output here
[console_output_sdio_timeout.txt](https://github.com/linux4wilc/driver/files/3043152/console_output_sdio_timeout.txt)

One thing I see is that the driver resets the device via the RESET, WL_EN controls and after this communication stops with cmd52 timeouts. I can see the cmd 52 transmitted on the CMD line but there is no response from the device. 
I would expect that an SDIO rescan is required after the device is reset to get the kernel SDIO driver and device back in sync but I don't see this anywhere.

2 - Based on the above I removed the resetting of the device in the WILC3000 driver via the RESET,WL_EN lines (in function wilc_wlan_power() in file linux_wlan.c). It gets further through the startup process but is timing out waiting for the firmware. I verified that the ISR is being triggered but the status register the driver reads from the device contains 0. I've attached the console log for this scenario. 
[console_output_firmware_timeout.txt](https://github.com/linux4wilc/driver/files/3043077/console_output_firmware_timeout.txt)

I do get a small amount of output from the WILC3000 debug. This is also attached.
[wilc3000_debug_output.txt](https://github.com/linux4wilc/driver/files/3043029/wilc3000_debug_output.txt)

3 - I modified U-Boot to reset the WILC3000 device at the very start of platform power up. The reset in the driver itself is still removed. I'm using the WL_EN and RESET lines as per the data sheet. In this case the firmware appears to get loaded but then the driver also times out. I do see a small (< 0.17us) pulse on the IRQ line but no ISR is triggered in the driver.

It could be there is some issue with the way the device is being reset/initialised as it appears to get to a certain stage and then stops. I'm confused about the resetting of the device in the driver. It's obviously there for a reason but it seems to make things worse and I would expect to see an SDIO rescan after this reset. Your help would be appreciated here as I think I am following the guidelines but I could be missing something.

Thanks,
Bruno

AdhamAbozaeid, 2019-04-05:
Please open a ticket on [Microchip Salesforce ](https://microchipsupport.force.com) and the team will support you more on that

BDP42, 2019-04-17:
For the sake of closure I'll document what I found with this issue.

It seems there are a number of problems at play here.

I had the driver configured to use the out-of-band interrupt (i.e. select CONFIG_WILC_HW_OOB_INTR) and it looks like this does not work properly. It was suggested to use the in-band interrupt and this fixed the problem with timing out on firmware start. I don't fully understand why out-of-band does not work as it is a valid configuration option.

I then hit another problem and this turned out to be a known issue with enabling powersave while the driver is initialising. See https://github.com/linux4wilc/driver/issues/15
The problem went away when this was changed. 
The workaround is to enable powersave after initialisation is completed.

As to the issue of the device being reset in the driver when starting up, the problem here is that an SDIO rebind is required after a device reset to rebind the driver and the device. I had to add this to the driver to get this to work.
This appears to be the same issue here -  https://github.com/linux4wilc/driver/issues/39
I don't understand why this is not a problem for everyone.

Bruno

AdhamAbozaeid, 2019-04-17:
For the last point, I observed that the behavior of the SD controller on the host is different. On SAMA5D4 for example, I see that the rebinding happens automatically once the device is reset.
A fix for that can be to avoid powering down/up the chip (wilc_wlan_power_on_sequence() and wilc_wlan_power_off_sequence(), and use external pullups on the chip_en and reset_n lines

BDP42, 2019-04-18:
It looks like the rebind is platform dependent. 
For our evaluation platform I've added an MMC re-scan after the device reset and this works.
I'm going to close off on this issue.

closed 2019-04-18 by BDP42