linux4wilc/driver #39: Race condition in linux_sdio_probe()

BillPlunkett, 2019-04-05:
I'm seeing an issue with linux4wilc 15.2, similar to the linux_sdio_probe() issue described in one of the closed issues:  [https://github.com/linux4wilc/driver/issues/21](url)

Before I learned of the following commit, [https://github.com/linux4wilc/driver/commit/ba2e53afb765aefc1cc5a5f890df88ed85521682](url), I attempted to work around the problem by calling a modified version of wilc_wlan_power_on_sequence() prior to wilc_netdev_init().

Has the commit been verified to correct the issue?

AdhamAbozaeid, 2019-04-05:
I'm not able to reproduce the problem on my setup, but the reporter confirmed that the proposed change fixed his issue. If you are seeing similar behavior when you use this fix, please share your logs and analysis and I'll try to help.
Note that fix isn't on the master, so you will need to use the dev branch to use this fix

BillPlunkett, 2019-04-07:
Below is the console output from two sessions using the dev branch driver.  I'm using a sama5d2-xplained with WILC100SD and jumper wires connecting the gpios between the two.  First session was with the reset and chip-en gpios connected; second session had them disconnected.  I've verified with an oscilloscope that the reset and chip-en signals behave as expected.

Poky (Yocto Project Reference Distro) 2.5.1 sama5d2-xplained-sd ttyS0

sama5d2-xplained-sd login: root
root@sama5d2-xplained-sd:~# uname -a
Linux sama5d2-xplained-sd 4.14.73-linux4sam_6.0-dirty #1 Sat Apr 6 14:52:48 UTC 2019 armv7l GNU/Linux

---
--- with reset and chip-en gpios connected
---
root@sama5d2-xplained-sd:~# modprobe wilc-sdio
wilc_sdio: module is from the staging directory, the quality is unknown, you have been warned.
Initializing Locks ...
(unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
(unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
(unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
(unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
wifi_pm : 0
wilc_sdio mmc1:0001:1: failed to get Reset GPIO, try default
wilc_sdio mmc1:0001:1: failed to get chip_en GPIO, try default
wifi_pm : 1
wilc_sdio mmc1:0001:1: failed to get Reset GPIO, try default
wilc_sdio mmc1:0001:1: failed to get chip_en GPIO, try default
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[dacd4000] wlan0
WILC POWER UP
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_init_host_int]Host[dacd4000][ddf23400]
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
wilc_sdio mmc1:0001:1: SDIO speed: 50000000
wilc_sdio mmc1:0001:1: wilc_sdio_cmd52..failed, err(-110)
wilc_sdio mmc1:0001:1: Fail cmd 52, enable csa...
wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_initialize:1065] Initializing WILC_Wlan FAILED
wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_initialize:1145] WLAN initialization FAILED
wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_mac_open:1189] Failed to initialize wilc
wilc_sdio mmc1:0001:1: Driver Initializing success

---
--- with reset and chip-en gpios disconnected
---
root@sama5d2-xplained-sd:~# modprobe wilc-sdio
wilc_sdio: module is from the staging directory, the quality is unknown, you have been warned.
Initializing Locks ...
(unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
(unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
(unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
(unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
wifi_pm : 0
wilc_sdio mmc1:0001:1: failed to get Reset GPIO, try default
wilc_sdio mmc1:0001:1: failed to get chip_en GPIO, try default
wifi_pm : 1
wilc_sdio mmc1:0001:1: failed to get Reset GPIO, try default
wilc_sdio mmc1:0001:1: failed to get chip_en GPIO, try default
wilc_sdio mmc1:0001:1: Driver Initializing success
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[dacc8000] wlan0
WILC POWER UP
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_init_host_int]Host[dacc8000][ddebd400]
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
wilc_sdio mmc1:0001:1: SDIO speed: 50000000
root@sama5d2-xplained-sd:~# wilc_sdio mmc1:0001:1: chipid 001003a0
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_initialize]WILC Initialization done
wilc_sdio mmc1:0001:1: failed to get IRQ GPIO, load default
wilc_sdio mmc1:0001:1 wlan0: INFO [init_irq]IRQ request succeeded IRQ-NUM= 144
wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Initializing Threads ...
wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Creating kthread for transmission
wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Creating kthread for Debugging
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]Detect chip WILC1000
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]loading firmware mchp/wilc1000_wifi_firmware.bin
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]WLAN firmware: mchp/wilc1000_wifi_firmware.bin
[... 23 lines cut ...]
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[1]: 01:00:5e:00:00:01
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[2]: 33:33:ff:e5:47:1c
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[3]: 01:80:c2:00:00:00
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 5.
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[0]: 33:33:00:00:00:01
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[1]: 01:00:5e:00:00:01
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[2]: 33:33:ff:e5:47:1c
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[3]: 01:80:c2:00:00:00
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[4]: 01:80:c2:00:00:03
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 6.
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[0]: 33:33:00:00:00:01
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[1]: 01:00:5e:00:00:01
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[2]: 33:33:ff:e5:47:1c
[... cut ...]
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[2]: 33:33:ff:e5:47:1c
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[3]: 01:80:c2:00:00:00
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[4]: 01:80:c2:00:00:03
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[5]: 01:80:c2:00:00:0e
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[6]: 33:33:00:01:00:03
wilc_sdio mmc1:0001:1 wlan0: INFO [debug_thread]*** Debug Thread Running ***

AdhamAbozaeid, 2019-04-08:
It looks from the log that the SDIO is failing, and since this happens only when the chip_en and reset_n GPIOs controlled by the host, then most probably the driver is trying to access the bus while the chip isn't enabled correctly.

BillPlunkett, 2019-04-09:
I agree that the SDIO is failing as a result of the driver manipulation of the chip_en and reset_n GPIOS in the wilc_wlan_power() function.  After reading some of the other issues, I did some experimenting with the driver gpio manipulation disabled.  In summary, what I found is that resetting the WILC requires an SDIO driver unbind/bind, otherwise the driver fails with SDIO errors.

# for the following tests, the April 5, 2019 dev branch of the driver was modified as follows:
void wilc_wlan_power_on_sequence(struct wilc *wilc)
{
	#if 0
	wilc_wlan_power(wilc, 0);
	wilc_wlan_power(wilc, 1);
	#endif
}

void wilc_wlan_power_off_sequence(struct wilc *wilc)
{
	#if 0
	wilc_wlan_power(wilc, 0);
	#endif
}

# SAMA5D2 PB28 = reset_n
# SAMA5D2 PB29 = chip_en

# reset_n and chip_en were monitored with an oscilloscope during the test

root@sama5d2-xplained-sd:~# uname -a
Linux sama5d2-xplained-sd 4.14.73-linux4sam_6.0-dirty #1 Sun Apr 7 15:51:23 UTC 2019 armv7l GNU/Linux

# resetting the WILC in conjunction with SDIO driver unbind/bind works:

root@sama5d2-xplained-sd:~# cd /sys/class/gpio
root@sama5d2-xplained-sd:/sys/class/gpio# echo 60 > export
root@sama5d2-xplained-sd:/sys/class/gpio# echo 61 > export
root@sama5d2-xplained-sd:/sys/class/gpio# ls
PB28       PB29       export     gpiochip0  unexport
root@sama5d2-xplained-sd:/sys/class/gpio# echo out > PB28/direction
root@sama5d2-xplained-sd:/sys/class/gpio# echo out > PB29/direction
root@sama5d2-xplained-sd:/sys/class/gpio# echo "reset and chip_en both low now"
reset and chip_en both low now
root@sama5d2-xplained-sd:/sys/class/gpio# pushd /sys/devices/platform/ahb/b0000000.sdio-host/driver/
/sys/devices/platform/ahb/b0000000.sdio-host/driver /sys/class/gpio
root@sama5d2-xplained-sd:/sys/devices/platform/ahb/b0000000.sdio-host/driver# echo -n "b0000000.sdio-host" > unbind
mmc1: card 0001 removed
root@sama5d2-xplained-sd:/sys/devices/platform/ahb/b0000000.sdio-host/driver# echo 1 > /sys/class/gpio/PB29/value
root@sama5d2-xplained-sd:/sys/devices/platform/ahb/b0000000.sdio-host/driver# echo "chip_en high now"
chip_en high now
root@sama5d2-xplained-sd:/sys/devices/platform/ahb/b0000000.sdio-host/driver# echo 1 > /sys/class/gpio/PB28/value
root@sama5d2-xplained-sd:/sys/devices/platform/ahb/b0000000.sdio-host/driver# echo "both reset and chip_en high now"
both reset and chip_en high now
root@sama5d2-xplained-sd:/sys/devices/platform/ahb/b0000000.sdio-host/driver# modprobe wilc-sdio
wilc_sdio: module is from the staging directory, the quality is unknown, you have been warned.
root@sama5d2-xplained-sd:/sys/devices/platform/ahb/b0000000.sdio-host/driver# echo -n "b0000000.sdio-host" > bind
mmc1: SDHCI controller on b0000000.sdio-host [b0000000.sdio-host] using ADMA
root@sama5d2-xplained-sd:/sys/devices/platform/ahb/b0000000.sdio-host/driver# mmc1: new high speed SDIO card at address 0001
Initializing Locks ...
(unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
(unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
(unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
(unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
wilc_sdio mmc1:0001:1: Driver Initializing success
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[ddcb2000] wlan0
WILC POWER UP
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_init_host_int]Host[ddcb2000][ddf9a800]
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
wilc_sdio mmc1:0001:1: SDIO speed: 50000000
wilc_sdio mmc1:0001:1: chipid 001003a0
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_initialize]WILC Initialization done
wilc_sdio mmc1:0001:1: failed to get IRQ GPIO, load default
wilc_sdio mmc1:0001:1 wlan0: INFO [init_irq]IRQ request succeeded IRQ-NUM= 144
wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Initializing Threads ...
wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Creating kthread for transmission
[... 34 lines cut ...]
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Deinitializing IRQ
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_stop]Read RESET Reg fffffbff : Retry10
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_stop]Bit 10 reset after : Retry 10
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_stop]Read RESET Reg fffffbff : Retry10
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Deinitializing WILC Wlan
wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_deinitialize]wilc deinitialization Done
wilc_sdio mmc1:0001:1 wlan0 (unregistered): INFO [wilc_netdev_cleanup]Freeing Wiphy...
wilc_sdio mmc1:0001:1 wlan0 (unregistered): INFO [wilc_free_wiphy]Unregistering wiphy
wilc_sdio mmc1:0001:1 wlan0 (unregistered): INFO [wilc_free_wiphy]Freeing wiphy
wilc_sdio mmc1:0001:1 wlan0 (unregistered): INFO [wilc_netdev_cleanup]Freeing netdev...
De-Initializing Locks
Module_exit Done.
at_pwr_dev: deinit
at_pwr_dev: unregistered

# resetting the WILC without SDIO driver unbind/bind does not work:

root@sama5d2-xplained-sd:/sys/devices/platform/ahb/b0000000.sdio-host/driver# echo 0 > /sys/class/gpio/PB28/value
root@sama5d2-xplained-sd:/sys/devices/platform/ahb/b0000000.sdio-host/driver# echo 0 > /sys/class/gpio/PB29/value
root@sama5d2-xplained-sd:/sys/devices/platform/ahb/b0000000.sdio-host/driver# echo 1 > /sys/class/gpio/PB29/value
root@sama5d2-xplained-sd:/sys/devices/platform/ahb/b0000000.sdio-host/driver# echo 1 > /sys/class/gpio/PB28/value
root@sama5d2-xplained-sd:/sys/devices/platform/ahb/b0000000.sdio-host/driver# modprobe wilc-sdio
wilc_sdio: module is from the staging directory, the quality is unknown, you have been warned.
wilc_sdio: probe of mmc1:0001:1 failed with error -110
root@sama5d2-xplained-sd:/sys/devices/platform/ahb/b0000000.sdio-host/driver#

AdhamAbozaeid, 2019-04-22:
SDIO bind/unbind depends on the host environment used. Some hosts (like sama5d4 with kernel 4.9) do the bind/unbind automatically when the SDIO device is registered by the host driver, and some doesn't

closed 2019-04-22 by AdhamAbozaeid

BillPlunkett, 2019-04-22:
I understand that there is a difference between 4.9 and 4.14, but how can this issue be considered closed?  Seems a bit premature, since it will become an issue in the very near future.

AdhamAbozaeid, 2019-04-22:
My understanding is that using manual bind/unbind works for you, so for cases when the host doesn't bind automatically, you will need to manually bind after the device reset.
Another option is to avoid resetting the device, and keeping the chip_en pulled high.

BillPlunkett, 2019-04-22:
You understand correctly.  However, all of the existing driver "power sequencing" code will be broken as soon as you move to 4.14.

AdhamAbozaeid, 2019-04-22:
Does the current power sequencing code works if you rescan the sdio devices after wilc resets?

BillPlunkett, 2019-04-22:
I don't think I tested the following case, if that's what you're asking:

unbind sdio driver
load wilc driver (which resets wilc via gpio)
(re)bind sdio driver

I was just surprised to see this issue closed, since I was expecting it to remain open until such workarounds are no longer necessary.  No big deal.  Is there another issue I should be monitoring instead for linux 4.14 support?

AdhamAbozaeid, 2019-04-22:
As per #38, this sequence should work.
There's no other issues that I'm aware of that relates to 4.14

Mateusz-Gwara, 2019-04-23:
@BillPlunkett could you please explain what 4.14 kernel specifics you mean that could cause problems compared to which version?

BillPlunkett, 2019-04-23:
@Mateusz-Gwara, the linux4wilc driver resets the WILC via gpio pins on startup.  That poses a problem for the SDIO driver in linux 4.14 - the workaround is to manually unbind the SDIO driver, load the wilc driver, then (re)bind the SDIO driver.  The workaround is not required with linux 4.9, in which apparently the kernel automatically rebinds the SDIO driver.

Mateusz-Gwara, 2019-04-24:
Thanks @BillPlunkett . I just wanted to make sure if there was something else I might be missing as the current kernel I use is 4.14.

BillPlunkett, 2019-06-20:
FYI, this issue was investigated by Microchip support, who found an effective solution - delete the "non-removable" flag from the device tree entry.

For reference, here is the device tree entry I tested with on the SAMA5D2-Xplained using linux kernel 4.14.73-linux4sam_6.0:

    sdmmc1: sdio-host@b0000000 {

      #address-cells = <1>;

      #size-cells = <0>;

      bus-width = <4>;

      pinctrl-names = "default";

      pinctrl-0 = <&pinctrl_sdmmc1_default>;

      no-1-8-v;

      //non-removable;

      status = "okay"; /* conflict with qspi0 */

      vqmmc-supply = <&vdd_3v3_reg>;

      vmmc-supply = <&vdd_3v3_reg>;

      wilc_sdio@0 {

              compatible = "microchip,wilc1000";

              gpio_reset = <&pioA PIN_PB28 GPIO_ACTIVE_HIGH>;

              gpio_chip_en = <&pioA PIN_PB29 GPIO_ACTIVE_HIGH>;

              //gpio_irq = <&pioA PIN_P??? GPIO_ACTIVELOWS>;

              //interrupt-parent = <&pioA>;

              //interrupts = <PIN_P??? IRQ_TYPE_EDGE_RISING>;

              status = "okay";

              reg = <0>;

              bus-width = <4>;

      };

    };

AdhamAbozaeid, 2019-10-30:
Hi @BillPlunkett 
The pwrseq isn't supported yet in the WILC driver