linux4wilc/driver #40: Unable to start AP mode if wpa_supplicant does not exit cleanly

ShonkyCH, 2019-04-15:
When using P2P mode on a wilc1000 connected via SDIO :-
```
wpa_supplicant -i p2p0 -B -Dnl80211 -c /opt/wpa_p2p.conf
wpa_cli -ip2p0 p2p_group_add freq=2437
wpa_cli -ip2p0 wps_pin any 12345678
```
P2P GO mode works OK, however, if wpa_supplicant is terminated using kill -9, subsequent attempts to create the P2P group fail with "Failed to start AP functionality":-

```
SM01-1839000017:/opt# ps axf | grep wpa_p2p | grep -v grep | awk '{ print $1 }' | xargs -i kill -9 {}
SM01-1839000017:/opt# wpa_supplicant -i p2p0 -B -Dnl80211 -c /opt/wpa_p2p.conf
Successfully initialized wpa_supplicant
SM01-1839000017:/opt# wpa_cli -i p2p0
wpa_cli v2.6
Copyright (c) 2004-2016, Jouni Malinen <j@w1.fi> and contributors

This software may be distributed under the terms of the BSD license.
See README for more details.

Interactive mode

> p2p_group_add freq=2437
OK
<3>Failed to start AP functionality
```

The wilc_sdio kernel module has to be removed and reinserted to fix the problem. The driver should not get into a stuck state just because an application went away unexpectedly.

AdhamAbozaeid, 2019-04-15:
Thanks Shonky for the bug report. Please report it on Salesforce as well

ShonkyCH, 2019-04-15:
Salesforce ID 00405975

tsifb, 2019-08-15:
Retested with latest driver / fw and does not appear to happen.

root@SM01-9999000003:~# iw phy1 interface add p2p0 type managed
root@SM01-9999000003:~# wpa_supplicant -i p2p0 -B -Dnl80211 -c /opt/wpa_p2p.conf
Successfully initialized wpa_supplicant
root@SM01-9999000003:~# wpa_cli -ip2p0 p2p_group_add freq=2437
OK
root@SM01-9999000003:~# wpa_cli -ip2p0 wps_pin any 12345678
root@SM01-9999000003:~# ps axf | grep wpa_p2p | grep -v grep | awk '{ print $1 }' | xargs -i kill -9 {}
root@SM01-9999000003:~# iw phy1 interface add p2p0 type managed
root@SM01-9999000003:~# wpa_supplicant -i p2p0 -B -Dnl80211 -c /opt/wpa_p2p.conf
Successfully initialized wpa_supplicant
root@SM01-9999000003:~# wpa_cli -i p2p0
wpa_cli v2.6
Copyright (c) 2004-2016, Jouni Malinen <j@w1.fi> and contributors

This software may be distributed under the terms of the BSD license.
See README for more details.

Interactive mode

> p2p_group_add freq=2437
OK
<3>AP-ENABLED
<3>CTRL-EVENT-CONNECTED - Connection to fa:f0:05:e5:4d:9a completed [id=0 id_str=]
<3>P2P-GROUP-STARTED p2p0 GO ssid="DIRECT-AG" freq=2437 passphrase="M6I0amcG" go_dev_addr=fa:f0:05:e5:4d:9a

Note: additional commands " iw phy1 interface add.... " needed because p2p0 interface is no longer statically created.

AdhamAbozaeid, 2019-08-15:
Thanks @tsifb for the updates

closed 2019-08-15 by AdhamAbozaeid