linux4wilc/driver #41: wilc1000 on Raspberry Compute Module 3+

AntoineBore, 2019-04-17:
Hi,

I bought the atwilc1000-SD card and I would like to connect it with my Raspberry pi compute model 3+. I already have booted my raspberry with Raspbian 9.4.

As a newbie, I struggle with your tutorial and I am a bit lost. Can I just install the driver on my boot or do I need to build a new kernel ?

Thank you.

AdhamAbozaeid, 2019-04-17:
Hi @AntoineBore 
You can build the driver module (wilc_sdio.ko) externally, using the same toolchain you used to build the kernel, and against the same kernel headers, then push it to your file system.
However, you will still need to rebuild the kernel to do the device tree modifications described in the porting guide, so rebuilding the kernel might be the easier way to go.

AntoineBore, 2019-04-18:
Hi @AdhamAbozaeid 
Thank you. I will try to rebuild the kernel.

AntoineBore, 2019-04-22:
Hi again,

I'm sorry to bother you but I'm really struggling with the driver. I think I succeed in rebuilding the kernel (I've followed this tutorial : []( https://www.raspberrypi.org/documentation/linux/kernel/building.md)) and selected wilc1000 which is already available with [](https://github.com/raspberrypi/linux).

Now, I tried to configure an AP mode with your wiki [](https://github.com/linux4wilc/driver/wiki/wifi#ap-mode). The problem is that there is no file such as wilc_hostapd_open.conf, wilc_hostapd_wep.conf and wilc_hostapd_wpa.conf in "/etc/".

Do you know what I did wrong ?
Thank you.

AdhamAbozaeid, 2019-04-22:
No worries. You'll need to create these files yourself.
I added them to the reference buildroot at buildroot4wilc under system/skeleton/etc so that they would be available automatically on the target filesystem when you use buildroot4wilc

AntoineBore, 2019-04-23:
So I just have to copy those files, after the cross-compilation, into the Raspberry ?
If yes, can I just use a tool like FileZila and a SFTP protocol ? Do I need to configure a special file permissions ?

AntoineBore, 2019-04-23:
Hi Adham.
Thank you. I think I have made some progress thanks to you !

Although, I have a new issue. I'm following the [tutorial for the AP mode](https://github.com/linux4wilc/driver/wiki/wifi#ap-mode).

When I try to execute the commands of the step "Run hostapd as user's configuration", I have the following error : 
`
pi@raspberrypi:/lib/modules/4.19.34-v7+/kernel/drivers/staging/wilc1000 $ sudo insmod /lib/modules/4.19.34-v7+/kernel/drivers/staging/wilc1000/wilc1000.ko
insmod: ERROR: could not insert module /lib/modules/4.19.34-v7+/kernel/drivers/staging/wilc1000/wilc1000.ko: Invalid module format
`

I don't understand because the version of my kernel seems to be the same as the version of the driver.

`pi@raspberrypi:/lib/modules/4.19.34-v7+/kernel/drivers/staging/wilc1000 $ modinfo wilc1000.ko`
`
filename:       /lib/modules/4.19.34-v7+/kernel/drivers/staging/wilc1000/wilc1000.ko
license:        GPL
srcversion:     103A341599BB336B8000271
depends:        cfg80211
staging:        Y
intree:         Y
name:           wilc1000
vermagic:       4.19.34-v7+ SMP mod_unload modversions ARMv7 p2v8`

`pi@raspberrypi:/lib/modules/4.19.34-v7+/kernel/drivers/staging/wilc1000 $ uname -a`
`Linux raspberrypi 4.19.34-v7+ #1211 SMP Mon Apr 8 22:56:37 BST 2019 armv7l GNU/Linux
`

AdhamAbozaeid, 2019-04-23:
Was that module built while building the linux source, or was it build externally, and transferred to the file system later?

AntoineBore, 2019-04-23:
I have uploaded raspbian into my raspberry pi, downloaded https://github.com/raspberrypi/linux and I then I have built it directly on the raspberry after including Wilc1000 modules (both spi and sdio) with menuconfig.

AntoineBore, 2019-04-24:
A hard reboot solve the problem. Thank you for your help, the installation is a success.

AntoineBore, 2019-04-24:
I reopen the issue again for another problem.  This problem is surely because of me but I can't find a solution on the internet.

As I said, I was able to execute both of the "insmod" instructions. I connected the wilc1000-sd card to my compute module 3+ with SDIO: I connected the WiFi module to the GPIO22-27 of the Compute Mobule IO Board.

On my raspberry pi, I add dtoverlay=sdio,poll_once=off to /boot/config.txt. I think that the configuration is OK because, when I print the GPIO with "gpio readall", the GPIO22-27 are in the ALT3 mode.

My problem is that when i try to execute "ifconfig wlan0 up", i get the error message: "wlan0: ERROR while getting interface flags: No such device".

If I check the clock signal with an oscilloscope, I have [this result ](http://zupimages.net/viewer.php?id=19/17/di26.jpg) which does not seem good.

Any idea of what I could have done wrong ?

AdhamAbozaeid, 2019-04-24:
from the logs you shared, it looks like the host wasn't able to detect the SD-board. I wasn't able to view the clock signal at this link (I assume you are referring to the SD CLK signal), but from your comment I assume it was dead.
Unfortunately I can't help much with this problem since the SD clk originates from the host side. Did you check the device tree file, specifically the mmc node?
Also I see other people were able to use WILC with Raspberry pi and got past the SDIO probe stage, maybe they can help you more on that. You can check the closed issues for more information.

AntoineBore, 2019-04-25:
Hi,
It turns out that the signal problem was due to a bad wiring. Indeed, on the CM IO Board you can't just plug the SD pin, you have to solder wire and connect them to the Raspberry. 

The log "mmc-bcm2835 3f300000.sdio: could not get clk, deferring probe" isn't an error : "This isn't an error - it's just an artifact of the way Linux manages dependencies between devices. Between the first message and the other two the kernel has loaded the clock driver, allowing the second probe attempt to succeed."

What do you mean with:

> check the device tree file, specifically the mmc node?

AdhamAbozaeid, 2019-04-25:
does it work for you now?
I meant to check the mmc node in your dts file, just in case the mmc (sdio) device isn't configured correctly somehow

AntoineBore, 2019-04-26:
Yes, it work now. Thank you.

Now, when I try to execute "ifconfig wlan0 up", i get the error: "SIOCSIFFLAGS: Input/output error".

Here is the output of dmesg : 

`[   86.072133] mmc1: card 0001 removed
[   91.235805] mmc1: new high speed SDIO card at address 0001
[  104.874646] wilc1000: module is from the staging directory, the quality is unknown, you have been warned.
[  108.935793] wilc1000_sdio: module is from the staging directory, the quality is unknown, you have been warned.
[  108.939960] wilc1000_sdio mmc1:0001:1: Driver Initializing success
[  109.123846] wilc1000_sdio mmc1:0001:1: chipid (001003a0)
[  109.123855] wilc1000_sdio mmc1:0001:1: has_thrpt_enh3 = 1...
[  109.124419] wilc1000_sdio mmc1:0001:1 wlan0: loading firmware atmel/wilc1003_firmware.bin
[  114.232507] wilc1000_sdio mmc1:0001:1 wlan0: WLAN initialization FAILED
[  114.232550] wilc1000_sdio mmc1:0001:1 wlan0: Failed to set driver handler
[  114.234049] wilc1000_sdio mmc1:0001:1: chipid (001003a0)
[  114.234058] wilc1000_sdio mmc1:0001:1: has_thrpt_enh3 = 1...
[  114.234617] wilc1000_sdio mmc1:0001:1 wlan0: loading firmware atmel/wilc1003_firmware.bin
[  119.272902] wilc1000_sdio mmc1:0001:1 wlan0: WLAN initialization FAILED
[  119.272951] wilc1000_sdio mmc1:0001:1 wlan0: Failed to set driver handler
[  119.353050] wilc1000_sdio mmc1:0001:1: chipid (001003a0)
[  119.353065] wilc1000_sdio mmc1:0001:1: has_thrpt_enh3 = 1...
[  119.354040] wilc1000_sdio mmc1:0001:1 wlan0: loading firmware atmel/wilc1003_firmware.bin
[  124.393012] wilc1000_sdio mmc1:0001:1 wlan0: WLAN initialization FAILED
[  124.393063] wilc1000_sdio mmc1:0001:1 wlan0: Failed to set driver handler
[  124.537890] wilc1000_sdio mmc1:0001:1: chipid (001003a0)
[  124.537899] wilc1000_sdio mmc1:0001:1: has_thrpt_enh3 = 1...
[  124.538473] wilc1000_sdio mmc1:0001:1 p2p0: loading firmware atmel/wilc1003_firmware.bin
[  129.592966] wilc1000_sdio mmc1:0001:1 p2p0: WLAN initialization FAILED
[  129.593013] wilc1000_sdio mmc1:0001:1 p2p0: Failed to set driver handler
[  129.596044] wilc1000_sdio mmc1:0001:1: chipid (001003a0)
[  129.596057] wilc1000_sdio mmc1:0001:1: has_thrpt_enh3 = 1...
[  129.597010] wilc1000_sdio mmc1:0001:1 p2p0: loading firmware atmel/wilc1003_firmware.bin
[  134.712921] wilc1000_sdio mmc1:0001:1 p2p0: WLAN initialization FAILED
[  134.712965] wilc1000_sdio mmc1:0001:1 p2p0: Failed to set driver handler
[  134.786330] wilc1000_sdio mmc1:0001:1: chipid (001003a0)
[  134.786343] wilc1000_sdio mmc1:0001:1: has_thrpt_enh3 = 1...
[  134.787336] wilc1000_sdio mmc1:0001:1 p2p0: loading firmware atmel/wilc1003_firmware.bin
[  139.833007] wilc1000_sdio mmc1:0001:1 p2p0: WLAN initialization FAILED
[  139.833048] wilc1000_sdio mmc1:0001:1 p2p0: Failed to set driver handler
[  159.199360] wilc1000_sdio mmc1:0001:1: chipid (001003a0)
[  159.199375] wilc1000_sdio mmc1:0001:1: has_thrpt_enh3 = 1...
[  159.200338] wilc1000_sdio mmc1:0001:1 wlan0: loading firmware atmel/wilc1003_firmware.bin
[  164.312961] wilc1000_sdio mmc1:0001:1 wlan0: WLAN initialization FAILED
[  164.313004] wilc1000_sdio mmc1:0001:1 wlan0: Failed to set driver handler
[  249.181083] wilc1000_sdio mmc1:0001:1: chipid (001003a0)
[  249.181100] wilc1000_sdio mmc1:0001:1: has_thrpt_enh3 = 1...
[  249.182409] wilc1000_sdio mmc1:0001:1 wlan0: loading firmware atmel/wilc1003_firmware.bin
[  254.232962] wilc1000_sdio mmc1:0001:1 wlan0: WLAN initialization FAILED
[  254.233003] wilc1000_sdio mmc1:0001:1 wlan0: Failed to set driver handler
`

AdhamAbozaeid, 2019-04-26:
Try to trace the error returned from wilc_wlan_init() under wilc_wlan_initialize()

AntoineBore, 2019-04-29:
How can I do that ?

I have realized that the version of https://github.com/linux4wilc/driver and https://github.com/raspberrypi/linux/tree/rpi-4.19.y/drivers/staging/wilc1000 are different. I will try to build the kernel with the version of the linux4wilc git.

AntoineBore, 2019-04-29:
I have tried to build the kernel with https://github.com/linux4wilc/driver and it fail with this error :
"error: ‘struct wilc_vif’ has no member named ‘obtaining_ip’"

If I try to define DISABLE_PWRSAVE_AND_SCAN_DURING_IP so obtaining_ip is defined for the structure wilc_vif, I get this error: 
"drivers/staging/wilc1000/linux_wlan.c:259:30: error: dereferencing pointer to incomplete type ‘struct in_ifaddr’
  if (!dev_iface || !dev_iface->ifa_dev || !dev_iface->ifa_dev->dev) {"

I guess I have to use the outdated version of the driver at this address: https://github.com/raspberrypi/linux/tree/rpi-4.19.y/drivers/staging/wilc1000

AntoineBore, 2019-04-29:
FYI : So I have tried again. it still doesn't work but not at the same step.. I have traced the return of wilc_wlan_init() under wilc_wlan_initialize() and I can get 1.

Now I have this issue : wilc_sdio_cmd53..failed, err(-84) and Failed cmd53 [0], bytes read...

I will try to make it work, I'll keep you informed.

AdhamAbozaeid, 2019-04-29:
It's better to use the WILC driver on this repo since it has more fixes and improved performance.
From the error that you get when you define DISABLE_PWRSAVE_AND_SCAN_DURING_IP, it looks that you kernel version doesn't define struct in_ifaddr which should be defined here: https://elixir.bootlin.com/linux/v4.19/source/include/linux/inetdevice.h#L135

The error you get when you undefine DISABLE_PWRSAVE_AND_SCAN_DURING_IP is because member obtaining_ip is only defined when DISABLE_PWRSAVE_AND_SCAN_DURING_IP is defined, but it used unprotected in set_power_mgmt()

AdhamAbozaeid, 2019-04-29:
For the SDIO errors, try disabling CONFIG_CFG80211_DEFAULT_PS in your kernel .config file and see if this helps

AntoineBore, 2019-04-30:
So I have installed the last version of the driver. Now I have this error in dmesg :
I'm a little confuse because I thought there was no need for Reset and IRQ when you are not using the low power mode.

[   68.916962] mmc1: new high speed SDIO card at address 0001
[   98.334760] wilc_sdio: no symbol version for module_layout
[   98.334784] wilc_sdio: loading out-of-tree module taints kernel.
[   98.334798] wilc_sdio: module is from the staging directory, the quality is unknown, you have been warned.
[   98.347793] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
[   98.347808] (unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
[   98.354657] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
[   98.354676] (unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
[   98.356371] wifi_pm : 0
[   98.356396] wilc_sdio mmc1:0001:1: failed to get Reset GPIO, try default
[   98.356407] wilc_sdio mmc1:0001:1: failed to get default Reset GPIO
[   98.356413] wifi_pm : 1
[   98.356422] wilc_sdio mmc1:0001:1: failed to get Reset GPIO, try default
[   98.356431] wilc_sdio mmc1:0001:1: failed to get default Reset GPIO
[   98.362489] wilc_sdio mmc1:0001:1: Driver Initializing success
[   98.582868] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   98.582878] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   98.582913] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   98.582919] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   98.582953] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   98.582959] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   98.582992] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   98.582997] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   98.583030] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   98.583035] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   98.583070] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   98.583075] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   98.583109] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   98.583114] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   98.583147] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   98.583153] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   98.583190] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   98.583195] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   98.583230] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   98.583235] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   98.583268] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   98.583274] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   98.583306] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   98.583311] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   98.583345] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   98.583350] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   98.583383] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   98.583388] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   98.583422] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   98.583427] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   98.583460] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   98.583465] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   98.583498] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   98.583503] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   98.583538] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   98.583543] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   98.584102] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[f7470fd7] wlan0
[   98.584106] WILC POWER UP
[... cut ...]
[   98.780113] wilc_sdio mmc1:0001:1: failed to load default irq
[   98.780120] wilc_sdio mmc1:0001:1 p2p0: INFO [wlan_deinit_locks]De-Initializing Locks
[   98.780126] wilc_sdio mmc1:0001:1 p2p0: ERR [wilc_wlan_initialize:1204] WLAN initialization FAILED
[   98.780132] wilc_sdio mmc1:0001:1 p2p0: ERR [wilc_mac_open:1248] Failed to initialize wilc
[   98.780174] wilc_sdio mmc1:0001:1 p2p0: ERR [wilc_send_config_pkt:1903] Set Timed out
[   98.780182] wilc_sdio mmc1:0001:1 p2p0: ERR [handle_set_wfi_drv_handler:433] Failed to set driver handler

AdhamAbozaeid, 2019-04-30:
```
[ 98.356396] wilc_sdio mmc1:0001:1: failed to get Reset GPIO, try default
[ 98.356407] wilc_sdio mmc1:0001:1: failed to get default Reset GPIO
[ 98.356413] wifi_pm : 1
[ 98.356422] wilc_sdio mmc1:0001:1: failed to get Reset GPIO, try default
[ 98.356431] wilc_sdio mmc1:0001:1: failed to get default Reset GPIO
.......
[ 98.585250] wilc_sdio mmc1:0001:1: failed to get IRQ GPIO, load default
[ 98.585257] wilc_sdio mmc1:0001:1: failed to load default irq
```

from the logs above, it looks like the driver is failing to find the correct GPIOs (specially the IRQ GPIO) from the device tree file, and it's trying the default hardcoded GPIOs instead but still failing. Probably because these GPIOs are not available on your board.

AntoineBore, 2019-05-02:
I have hardcoded Reset, IRQ and chip_en to 3 available GPIOs and it worked.
Thank you again for your help. Unfortunately I think I won't succeed in doing this wifi module work...

Whatever I try, I always have a different problem and I fear that this solution won't be consistent for my Access Point need.

For your information, here is my last log :
[  137.958729] wilc_sdio mmc1:0001:1 wlan0: INFO [init_irq]IRQ request succeeded IRQ-NUM= 166
[  137.958743] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Initializing Threads ...
[  137.958753] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Creating kthread for transmission
[  137.958960] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Creating kthread for Debugging
[  137.959182] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]Detect chip WILC1000
[  137.959193] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]loading firmware mchp/wilc1000_wifi_firmware.bin
[  137.959205] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]WLAN firmware: mchp/wilc1000_wifi_firmware.bin
[  137.959733] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_firmware_download]Downloading Firmware ...
[  137.959762] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_firmware_download]Downloading firmware size = 135020
[  137.979936] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_firmware_download]Offset = 119704
[  137.985784] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_firmware_download]Offset = 135020
[  137.985799] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_firmware_download]Download Succeeded
[  137.985863] wilc_sdio mmc1:0001:1 wlan0: INFO [linux_wlan_start_firmware]Starting Firmware ...
[  137.990353] wilc_sdio mmc1:0001:1 wlan0: INFO [linux_wlan_start_firmware]Waiting for FW to get ready ...
[  138.492050] wilc_sdio mmc1:0001:1 wlan0: INFO [linux_wlan_start_firmware]Firmware start timed out
[  138.492066] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_initialize:1163] Failed to start firmware
[  138.492079] wilc_sdio mmc1:0001:1: wilc_sdio_disable_interrupt
[  138.492199] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinitializing Threads
[  138.492211] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinit debug Thread
[  138.972041] wilc_sdio mmc1:0001:1 wlan0: INFO [debug_thread]Exit debug thread
[  138.972106] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinitializing Threads
[  138.972185] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinit_locks]De-Initializing Locks
[  138.972199] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_initialize:1204] WLAN initialization FAILED
[  138.972211] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_mac_open:1248] Failed to initialize wilc
[  138.972267] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_send_config_pkt:1903] Set Timed out
[  138.972280] wilc_sdio mmc1:0001:1 wlan0: ERR [handle_set_wfi_drv_handler:433] Failed to set driver handler`

[  302.643689] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[b6e5313c] wlan0
[  302.643704] power up request for already powered up source Wifi
[  302.643710] Device already up. request source is Wifi
[  302.643726] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_init_host_int]Host[b6e5313c][761445c9]
[  302.643754] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
[  302.643768] wilc_sdio mmc1:0001:1 wlan0: WRN [wilc_wlan_initialize: 1206]wilc already initialized
[  302.643831] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_send_config_pkt:1903] Set Timed out
[  302.643845] wilc_sdio mmc1:0001:1 wlan0: ERR [handle_set_wfi_drv_handler:433] Failed to set driver handler
[  302.643867] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_send_config_pkt:1903] Set Timed out
[  302.643879] wilc_sdio mmc1:0001:1 wlan0: ERR [handle_set_operation_mode:459] Failed to set operation mode
[  302.643895] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_send_config_pkt:1886] Get Timed out
[  302.643908] wilc_sdio mmc1:0001:1 wlan0: ERR [handle_get_mac_address:480] Failed to get mac address
[  302.643944] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]Mac address: 00:00:00:00:00:00
[  302.643956] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_mac_open:1272] Wrong MAC address
[  302.643990] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_send_config_pkt:1903] Set Timed out
[  302.644002] wilc_sdio mmc1:0001:1 wlan0: ERR [handle_set_wfi_drv_handler:433] Failed to set driver handler
[  302.644036] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Deinitializing wilc  ...
[  302.644046] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Disabling IRQ
[  302.644057] wilc_sdio mmc1:0001:1: wilc_sdio_disable_interrupt
[  302.644169] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Deinitializing Threads
[  302.644181] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinitializing Threads
[  302.644191] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinit debug Thread
[  302.644201] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinitializing Threads
[  302.644211] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Deinitializing IRQ
[  302.645544] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_stop]Read RESET Reg fffffbff : Retry10
[  302.645556] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_stop]Bit 10 reset after : Retry 10
[  302.645738] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_stop]Read RESET Reg fffffbff : Retry10
[  302.649904] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Deinitializing WILC Wlan
[  302.649919] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Deinitializing Locks
[  302.649929] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinit_locks]De-Initializing Locks
[  302.649939] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_deinitialize]wilc deinitialization Done

AdhamAbozaeid, 2019-05-02:
from the logs, it looks like you are having problems to receive the first interrupt coming from WILC

AntoineBore, 2019-05-06:
Hi,

I have tried many times but it seems I won't succeed. The log I get is a bit different but I think I have the same problem with the interrupt. Maybe my wiring isn't good, especially the IRQ pin which is hard to solder with the WILC1000-SD.

[  130.282250] wilc_sdio mmc1:0001:1 wlan0: INFO [change_virtual_intf]Interface type = NL80211_IFTYPE_STATION
[  130.282322] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_send_config_pkt:1903] Set Timed out
[  130.282335] wilc_sdio mmc1:0001:1 wlan0: ERR [handle_set_wfi_drv_handler:433] Failed to set driver handler
[  130.282356] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_send_config_pkt:1903] Set Timed out
[  130.282367] wilc_sdio mmc1:0001:1 wlan0: ERR [handle_set_operation_mode:459] Failed to set operation mode
[  130.282382] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_send_config_pkt:1903] Set Timed out
[  130.282394] wilc_sdio mmc1:0001:1 wlan0: ERR [handle_power_management:2802] Failed to send power management
[  130.282410] wilc_sdio mmc1:0001:1 p2p0: ERR [wilc_send_config_pkt:1903] Set Timed out
[  130.282421] wilc_sdio mmc1:0001:1 p2p0: ERR [handle_power_management:2802] Failed to send power management
[  130.282773] wilc_sdio mmc1:0001:1 wlan0: INFO [del_virtual_intf]Deleting virtual interface
[  132.430939] wilc_sdio mmc1:0001:1 wlan0: INFO [change_virtual_intf]In Change virtual interface function
[  132.430959] wilc_sdio mmc1:0001:1 wlan0: INFO [change_virtual_intf]Wireless interface name =wlan0
[  132.430969] wilc_sdio mmc1:0001:1 wlan0: INFO [change_virtual_intf]Changing virtual interface, enable scan
[  132.430981] wilc_sdio mmc1:0001:1 wlan0: INFO [change_virtual_intf]Interface type = NL80211_IFTYPE_AP
[  132.431056] wilc_sdio mmc1:0001:1 wlan0: INFO [del_virtual_intf]Deleting virtual interface
[  132.431077] wilc_sdio mmc1:0001:1 wlan0: INFO [add_virtual_intf]Adding monitor interface[e52b5b5e]
[  132.431089] wilc_sdio mmc1:0001:1 wlan0: INFO [add_virtual_intf]Initializing mon ifc virtual device driver
[  132.431100] wilc_sdio mmc1:0001:1 wlan0: INFO [add_virtual_intf]Adding monitor interface[e52b5b5e]
[  132.431112] wilc_sdio mmc1:0001:1 wlan0: INFO [add_virtual_intf]Setting monitor flag in private structure
[  132.472970] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[e52b5b5e] wlan0
[  132.472984] power up request for already powered up source Wifi
[  132.472992] Device already up. request source is Wifi
[  132.473008] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_init_host_int]Host[e52b5b5e][fd2a8ead]
[  132.473026] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
[  132.473037] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_init_locks]Initializing Locks ...
[  132.473049] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
[  132.477418] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_initialize]WILC Initialization done
[  132.477431] wilc_sdio mmc1:0001:1: failed to get IRQ GPIO, load default
[  132.477483] wilc_sdio mmc1:0001:1 wlan0: INFO [init_irq]IRQ request succeeded IRQ-NUM= 166
[  132.477496] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Initializing Threads ...
[  132.477506] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Creating kthread for transmission
[  132.477729] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Creating kthread for Debugging
[  132.477945] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]Detect chip WILC1000
[  132.477957] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]loading firmware mchp/wilc1000_wifi_firmware.bin
[  132.477969] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]WLAN firmware: mchp/wilc1000_wifi_firmware.bin
[  132.478366] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_firmware_download]Downloading Firmware ...
[  132.478382] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_firmware_download]Downloading firmware size = 135020
[  132.499641] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_firmware_download]Offset = 119704
[  132.505877] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_firmware_download]Offset = 135020
[  132.505896] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_firmware_download]Download Succeeded
[  132.505961] wilc_sdio mmc1:0001:1 wlan0: INFO [linux_wlan_start_firmware]Starting Firmware ...
[  132.510268] wilc_sdio mmc1:0001:1 wlan0: INFO [linux_wlan_start_firmware]Waiting for FW to get ready ...
[  133.011827] wilc_sdio mmc1:0001:1 wlan0: INFO [linux_wlan_start_firmware]Firmware start timed out
[  133.011842] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_initialize:1163] Failed to start firmware
[  133.011855] wilc_sdio mmc1:0001:1: wilc_sdio_disable_interrupt
[  133.012010] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinitializing Threads
[  133.012022] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinit debug Thread
[  133.491798] wilc_sdio mmc1:0001:1 wlan0: INFO [debug_thread]Exit debug thread
[  133.491858] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinitializing Threads
[  133.491936] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinit_locks]De-Initializing Locks
[  133.491951] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_initialize:1204] WLAN initialization FAILED
[  133.491962] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_mac_open:1248] Failed to initialize wilc
[  133.492009] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_send_config_pkt:1903] Set Timed out
[  133.492021] wilc_sdio mmc1:0001:1 wlan0: ERR [handle_set_wfi_drv_handler:433] Failed to set driver handler
[... cut ...]
[  169.542371] wilc_sdio mmc1:0001:1 wlan0: ERR [handle_set_operation_mode:459] Failed to set operation mode
[  169.542399] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_send_config_pkt:1903] Set Timed out
[  169.542416] wilc_sdio mmc1:0001:1 wlan0: ERR [handle_power_management:2802] Failed to send power management
[  169.542432] wilc_sdio mmc1:0001:1 p2p0: ERR [wilc_send_config_pkt:1903] Set Timed out
[  169.542443] wilc_sdio mmc1:0001:1 p2p0: ERR [handle_power_management:2802] Failed to send power management
[  169.542607] wilc_sdio mmc1:0001:1 wlan0: INFO [del_virtual_intf]Deleting virtual interface

AdhamAbozaeid, 2019-05-06:
Do you have CONFIG_WILC_HW_OOB_INTR defined? if yes, then don't define it, and the muxed SDIO interrupt will be used, which uses SD DAT1 line for the interrupt, instead of an external OOB interrupt.

If you already have CONFIG_WILC_HW_OOB_INTR undefined, you can add logs to wilc_sdio_interrupt() which is the first entry point for the interrupt handling, and see if it's being triggered at all.

Another possibility is that you are using flywires to connect the SD board as far as I understand, which most probably with SDIO running at high speeds

AntoineBore, 2019-05-07:
I think that I have wiring problems but I have succeeded to go a little further ! I'm now able to see my softAP. Although, I cannot connect my laptop/phone to it. Here is the var/log/syslog:

raspberrypi kernel : [854.567594] mon.wlan: INFO [mgmt_tx_complete]Packet sent Size = 30 Add = 8fce6217
raspberrypi hostapd: wlan0: STA <MAC_ADDRESS> IEEE 802.11: did not acknowledge authentification response
...
...
raspberrypi kernel : [1001,615620] mon.wlan: INFO [mgmt_tx_complete]Packet sent Size = 30 Add = 8afc30a6
raspberrypi hostapd: wlan0: STA <MAC_ADDRESS> IEEE 802.11: did not acknowledge authentification response

Do you think it's also a wiring problem? SoftAP and hostapd work well with tp-link tl-wn722n so my kernel should be OK I guess..

AdhamAbozaeid, 2019-05-07:
I don't think it's a wiring problem since you aren't seeing SDIO errors.
Have you tested the STA interface as well? Are you testing in a busy environment?

This will need more logs, and maybe air sniffer logs if possible. Please open a ticket on salesforce so that the team can help you more with that.

AntoineBore, 2019-05-21:
Hi,

I am finally able to use the wilc1000 as an access point. 
The problem was due to the wiring. As I use a Raspi CM3+, I have to use wire to connect the dev board and even if those wires were clean, it couldn't support 50MHz.
To resolve the problem, I just lowered the SDIO clock speed to 25MHz.

closed 2019-05-21 by AntoineBore

AdhamAbozaeid, 2019-05-21:
Thanks Antoine for the update

mswingtra, 2019-05-22:
@AntoineBore Hi, I'm trying to make wilc1000 work with rpi CM3+ as well, and I'm pretty much facing right now the same issue as you (mentioned in https://github.com/linux4wilc/driver/issues/41#issuecomment-487449114). Could you tell me which version of the driver and which version of the linux kernel have you used for your successful attempt?

mswingtra, 2019-05-22:
@AdhamAbozaeid 
This is my dmesg output. I have compiled latest driver version from `dev` branch (commit a2ab7b3728afdc4a51cc8fe0423921a71b3f2500) on top of linux kernel version `rpi-4.14.y`.
 
```
[  355.636931] Initializing Locks ...
[  355.637178] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
[  355.637186] (unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
[  355.645952] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
[  355.645976] (unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
[  355.648427] wilc_sdio mmc1:0001:1: Driver Initializing success
[  355.777550] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  355.777560] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  355.777593] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  355.777598] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  355.777629] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  355.777634] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  355.777665] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  355.777670] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  355.777701] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  355.777706] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  355.777736] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  355.777741] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  355.777772] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  355.777777] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  355.777808] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  355.777813] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  355.777847] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  355.777852] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  355.777885] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  355.777890] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  355.777921] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  355.777927] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  355.777957] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  355.777962] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  355.777993] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  355.777998] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  355.778029] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  355.778034] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  355.778065] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  355.778070] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  355.778101] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  355.778106] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  355.778137] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[  355.778142] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  355.778218] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[b5c28000] wlan0
[  355.778222] WILC POWER UP
[  355.778228] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_init_host_int]Host[b5c28000][b96a8c00]
[  355.778238] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
[  355.778244] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
[  355.778250] wilc_sdio mmc1:0001:1: SDIO speed: 50000000
[  355.778678] wilc_sdio mmc1:0001:1: chipid 001003a0
[  355.779124] wilc_sdio mmc1:0001:1: wilc_sdio_cmd53..failed, err(-84)
[  355.779129] wilc_sdio mmc1:0001:1: Failed cmd53, write reg (00001118)...
[  355.779135] wilc_sdio mmc1:0001:1 wlan0: ERR [init_chip:1923] fail write reg 0x1118
[  355.779238] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_initialize:1065] Initializing WILC_Wlan FAILED
[  355.779243] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_initialize:1145] WLAN initialization FAILED
[... cut ...]
[  355.894934] wilc_sdio mmc1:0001:1: wilc_sdio_cmd53..failed, err(-84)
[  355.894940] wilc_sdio mmc1:0001:1: Failed cmd53, write reg (00001118)...
[  355.894946] wilc_sdio mmc1:0001:1 p2p0: ERR [init_chip:1923] fail write reg 0x1118
[  355.895053] wilc_sdio mmc1:0001:1 p2p0: ERR [wilc_wlan_initialize:1065] Initializing WILC_Wlan FAILED
[  355.895058] wilc_sdio mmc1:0001:1 p2p0: ERR [wilc_wlan_initialize:1145] WLAN initialization FAILED
[  355.895063] wilc_sdio mmc1:0001:1 p2p0: ERR [wilc_mac_open:1189] Failed to initialize wilc
```

AdhamAbozaeid, 2019-05-22:
@mswingtra , I see SDIO errors in the logs, so this might be the same problem with fly wires that Antoine was having.
It's very tricky to use SDIO with fly wires because of the noise. Try to use as short wires as possible, and twist them with ground wires, and lower the SDIO clock you are using.

AntoineBore, 2019-05-23:
@mswingtra, like @AdhamAbozaeid said, I think it's a wiring problem. Even if I succeed in making it work, I still often have this problem, even if I use really small wires and a low frequency (I'm using 1 MHz but I was able to make it work with 25 MHz).

mswingtra, 2019-05-23:
@AdhamAbozaeid @AntoineBore Thanks for advice guys! I'll try with shorter cables then.

commit a2ab7b37 (2019-04-29): staging: wilc1000: Compilation errors if compilation switches are disabled

Compilation and linking erros are observed if compilation switches
DISABLE_PWRSAVE_AND_SCAN_DURING_IP and WILC_DEBUGFS aren't defiend

Fixes #41