linux4wilc/driver #43: Concurrent AP-STA mode not working

BDP42, 2019-04-19:
I'm having problems getting concurrent AP-STA mode working. 

From the documentation on this it seems that the wlan0 and p2p0 interfaces are used, hostapd using one and the STA client using the other.

Can either interface be used for either STA or AP functionality?
Presumably the p2p0 interface can be used for regular STA client functionality ( i.e. it does not have to be a P2P client).

In my case I am running hostapd over wlan0 and wpa_supplicant over p2p0.
I'm using the master branch for WILC3000 driver and firmware.

I start hostapd first and then wpa_supplicant. 
I can see the Access Point from another device. However as soon wpa_supplicant connects to another AP the Access Point started by hostapd disappears. There are no obvious errors on the console.

If I start the STA first and then the AP, the AP can be seen for a short while (seconds) but then disappears like before.

The AP stays up if I start it on its own and I can connect to it as per normal.

Here's my hostapd config

```interface=wlan0
driver=nl80211
ssid=wilc1000_SoftAP
dtim_period=2
beacon_int=100
channel=7
hw_mode=g
max_num_sta=8
ap_max_inactivity=300
```

Here's my wpa_supplicant config

```at /data/misc/wifi/wilc_wpa_supplicant.conf                                   <
ctrl_interface=/data/misc/wifi/sockets
update_config=1
ap_scan=1

network={
        ssid="RCOM_700560"
        scan_ssid=1
        key_mgmt=WPA-PSK
        proto=RSN
        pairwise=CCMP
        auth_alg=OPEN
        psk="c36fd0298c02d0f1f1897a0a46b6841395f7967d"
}
```

Here's the console output
[console_output_concurrent_ap_sta.txt](https://github.com/linux4wilc/driver/files/3097740/console_output_concurrent_ap_sta.txt)

Here's the WILC3000 debug output
[wilc3000_debug_output_concurrent_ap_sta.txt](https://github.com/linux4wilc/driver/files/3097788/wilc3000_debug_output_concurrent_ap_sta.txt)

ShonkyCH, 2019-04-19:
Is your RCOM_700560 AP on channel 1? Looks like it. Your AP is on channel 7.

The AP needs to be on the same channel as the AP that the STA mode is connecting to.

We use p2p0 for the AP and wlan0 for STA although I don't know if you have to use them that way. Just made more sense to us.

AdhamAbozaeid, 2019-04-22:
you can use either interfaces for STA/AP functionalities. Just make sure to adjust your wilc_hostapd.conf file to point to the interface you want to use for AP.
Also as ShonkyCH pointed, both interfaces have to run on the same channel, so make sure you are starting the AP on the same channel your STA is connecting to

BDP42, 2019-04-23:
As suggested the problem was caused by the fact that the AP and STA were using different channels.
I changed them to use the same channel and concurrent AP/STA mode is now working.

closed 2019-04-23 by BDP42