linux4wilc/driver #45: wilc1000 soft AP configuration

ksathara, 2019-05-22:


ksathara, 2019-05-22:
Hi, From dmesg -w, I can see the point at which this kernel oops pops up.
[  616.574137] wilc_sdio mmc1:0001:1 wlan0: INFO [add_key]Adding key with cipher suite = fac04
[  616.574152] wilc_sdio mmc1:0001:1 wlan0: INFO [add_key]b83e41a0 b9eca000 2
[  616.574164] wilc_sdio mmc1:0001:1 wlan0: INFO [add_key]key d 68 c
[  616.574221] Unable to handle kernel NULL pointer dereference at virtual address 00000000
[  616.586832] pgd = 7f72c965
[  616.591729] [00000000] *pgd=00000000

There seems to be adding of some key happening just before the oops.

AdhamAbozaeid, 2019-05-22:
Can you add more logs to handle_key() to see where exactly it fails?
you can also try the driver and FW from the dev branches since they have some updates

AntoineBore, 2019-05-29:
I have a similar problem when I enable hostapd. If I start hostapd manually after the boot, so once the wilc1000 connection is complete, it works. But If hostapd is launched before the end of the wilc1000 initialization, the wifi module crash.

So I think you just need to start hostapd manually after the boot or find a solution so the hostapd service wait the end of the wilc1000 initialization. (I haven't found this solution yet but I will try to found it).

AdhamAbozaeid, 2019-08-08:
@ksathara , any updates on that?

closed 2019-11-04 by AdhamAbozaeid