linux4wilc/driver #46: p2p0 and scan

AntoineBore, 2019-05-29:
Hi,

I'm using the WILC1000 as an access point and I was wondering if it's possible to deactivate the p2p0. Indeed, direct Wi-Fi is useless in my application and this p2p0 just consume time to my CPU.

Another question: is it possible to disable the wilc1000 automatic network scan ? During the boot, I can see that the "cfg_scan_result" find several networks, but I don't want to connect to any (again, I'm using the module as an access point) so this is just a waste of time CPU.

AntoineBore, 2019-05-30:
Here are some logs. At the start, I have this for a long time:

[    7.222931] wilc_sdio mmc1:0001:1 p2p0: INFO [scan]Trigger Scan Request
[    7.227564] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 3.
[    7.227580] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry0: 33:33:0:0:0:1
[    7.227587] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry1: 1:0:5e:0:0:1
[    7.227594] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry2: 33:33:ff:67:75:bc
[    7.227739] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 4.
[    7.227748] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry0: 33:33:0:0:0:1
[    7.227755] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry1: 1:0:5e:0:0:1
[    7.227762] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry2: 33:33:ff:67:75:bc
[    7.227769] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry3: 33:33:0:0:0:fb
[    7.312614] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 3.
[    7.312631] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry0: 33:33:0:0:0:1
[    7.312638] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry1: 1:0:5e:0:0:1
[    7.312645] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry2: 33:33:0:0:0:fb
[    7.313167] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 2.
[    7.313181] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry0: 33:33:0:0:0:1
[    7.313188] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry1: 1:0:5e:0:0:1
[    7.410159] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_set_multicast_list]Setting mcast List with count = 3.
[    7.410178] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_set_multicast_list]Entry0: 33:33:0:0:0:1
[    7.410185] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_set_multicast_list]Entry1: 1:0:5e:0:0:1
[    7.410193] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_set_multicast_list]Entry2: 33:33:ff:67:75:bc
[    7.410252] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_set_multicast_list]Setting mcast List with count = 4.
[    7.410261] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_set_multicast_list]Entry0: 33:33:0:0:0:1
[    7.410267] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_set_multicast_list]Entry1: 1:0:5e:0:0:1
[    7.410274] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_set_multicast_list]Entry2: 33:33:ff:67:75:bc
[    7.410281] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_set_multicast_list]Entry3: 33:33:0:0:0:fb
[    7.442753] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_set_multicast_list]Setting mcast List with count = 3.
[    7.442771] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_set_multicast_list]Entry0: 33:33:0:0:0:1
[    7.442778] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_set_multicast_list]Entry1: 1:0:5e:0:0:1
[    7.442785] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_set_multicast_list]Entry2: 33:33:0:0:0:fb
[    7.443265] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_set_multicast_list]Setting mcast List with count = 2.
[    7.443277] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_set_multicast_list]Entry0: 33:33:0:0:0:1
[    7.443283] wilc_sdio mmc1:0001:1 p2p0: INFO [wilc_set_multicast_list]Entry1: 1:0:5e:0:0:1

After several minutes and after I turn p2p0 down manually, I keep having this:

[  813.371521] wilc_sdio mmc1:0001:1 wlan0: INFO [scan]Trigger Scan Request
[  813.372292] wilc_sdio mmc1:0001:1 wlan0: INFO [handle_scan]Don't scan. WLAN_IFC is in state [-2136139032]
[  813.372314] wilc_sdio mmc1:0001:1 wlan0: INFO [handle_scan_done]Abort running scan
[  813.387510] wilc_sdio mmc1:0001:1 wlan0: INFO [cfg_scan_result]Scan Aborted
[  818.391189] wilc_sdio mmc1:0001:1 wlan0: INFO [scan]Requested num of channel 13
[  818.391205] wilc_sdio mmc1:0001:1 wlan0: INFO [scan]Scan Request IE len =  126
[  818.391216] wilc_sdio mmc1:0001:1 wlan0: INFO [scan]Number of SSIDs 1
[  818.391227] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wfi_cfg_alloc_fill_ssid]Received one NULL SSID
[  818.391237] wilc_sdio mmc1:0001:1 wlan0: INFO [scan]Trigger Scan Request
[  818.391669] wilc_sdio mmc1:0001:1 wlan0: INFO [handle_scan]Don't scan. WLAN_IFC is in state [-2136139032]
[  818.391683] wilc_sdio mmc1:0001:1 wlan0: INFO [handle_scan_done]Abort running scan
[  818.406814] wilc_sdio mmc1:0001:1 wlan0: INFO [cfg_scan_result]Scan Aborted
[  819.671782] wilc_sdio mmc1:0001:1 wlan0: INFO [debug_thread]*** Debug Thread Running ***
[  823.413108] wilc_sdio mmc1:0001:1 wlan0: INFO [scan]Requested num of channel 13
[  823.413125] wilc_sdio mmc1:0001:1 wlan0: INFO [scan]Scan Request IE len =  126
[  823.413136] wilc_sdio mmc1:0001:1 wlan0: INFO [scan]Number of SSIDs 1
[  823.413148] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wfi_cfg_alloc_fill_ssid]Received one NULL SSID

Is this possible to turn down both of them?

ShonkyCH, 2019-05-30:
This will reduce the amount of noise a fair bit but not all of it:
```
echo 0x000 > /sys/kernel/debug/wilc/wilc_debug_region
```

AntoineBore, 2019-05-30:
If I understand, this will stop the debugging thread for running? But why does it run in the first place and why it's looking for hostpot since the module is used as an AP ?

ShonkyCH, 2019-05-30:
It will filter out the messages but not stop any particular threads from running. They will still be generated but the output functions will just drop them.

I don't know that it particularly understands the difference between modes. Just take the interface down and it will take minimal CPU. This would be the least of the issues.

AdhamAbozaeid, 2019-05-30:
Hi Antoine

By default, both wlan0 and p2p0 interfaces will be down, unless the user/wpa_supplicant brought them up. If the supplicant is bring the p2p0 interface up automatically, you will need either to disable that on your host, or disable it manually using ifconfig p2p0 down.
Which interface are you running the hostapd on?

WILC1000 doesn't do automatic scans; it only scans when the supplicant issues a scan request, which shouldn't happen in your case if you are only running the hostapd, and not the wpa_supplicant.

Another way to filter the debug level of the host is to use the dmesg -n _level_ command

AntoineBore, 2019-05-31:
I removed this wpa_supplicant and still have the p2p0, so for the moment, I added ifconfig p2p0 down in /etc/rc.local.
I'm running the hostapd on wlan0. I followed this tutorial: https://www.raspberrypi.org/documentation/configuration/wireless/access-point.md

AdhamAbozaeid, 2019-06-03:
If you disable the supplicant service, what happens when you modprobe wilc_sdio?

closed 2019-07-16 by AntoineBore