linux4wilc/driver #47: Driver instantiation problem due to RST/EN low on PI CM3

dir-ableton, 2019-06-04:
We've set up the WILC3000 according to the datasheet on our RPI 3 CM3 module. Part of the datasheet says explicitly "Connect to a host output that defaults low at power-up". We use GPIOs 37 and 39 for RST and EN respectively.

However this does not seem to agree with the kernel 4.14 MMC/SDHOST infrastructure. Without RST/EN set to HIGH the loaded driver is not probed. Only when forcing the lines high during MMC bus scanning (via unbind/bind on the driver) the driver is probed. It then complains about not being able to allocate the GPIO pins, which isn't too surprising as they are at that time allocated by the generic GPIO subsystem.

```
[ 2219.232232] wilc_sdio mmc1:0001:1: failed to get Reset GPIO, try default
[ 2219.239046] wilc_sdio mmc1:0001:1: failed to get default Reset GPIO
[ 2219.245603] wilc_sdio mmc1:0001:1: Driver Initializing success

```

Releasing the pins before loading the driver again fails to probe, the error in that case is -110. I haven't yet investigated deeply, but my strong suspicion is that the MMC subsystem tries to talk to the module before invoking `linux_sdio_probe`.

So while I can for the time ignore the warnings, I'm a bit confused as how to solve this problem. The PI doesn't allow for dts-based pulling high of the GPIOs in question. Should we re-route the PCB to take boot-time-high GPIOs? Should the driver be changed so that it can/will power up the module, force a re-scan and only then drives the pins low? 

Versions:

```
Linux version 4.14.74-rt44 (oe-user@oe-host) (gcc version 8.2.0 (GCC)) #1 SMP PREEMPT RT Mon Jun 3 12:48:48 UTC 2019

commit 720f7d4fe2cc8bb17ca9e06775d09768572dbbb6 (tag: wilc_linux_15_2, tag: devtool-patched, tag: devtool-base, ableton/gpio-setup-experiments, master, devtool)
```

AdhamAbozaeid, 2019-06-04:
The driver gets the gpio from the dts the file of the platform after the probe is called.
Accordingly, it requires that the chip_en be high before the driver is inserted.
For the "failed to get reset GPIO, try default" error you are getting, this means the driver couldn't retrieve the correct GPIO from the dts file, and is trying to use the hardcoded default GPIO in the driver.

dir-ableton, 2019-06-05:
I do not follow your logic here. Where is the connection between DTS and chip_en high? My DTS is correct, see the bottom of this post.

I double checked, the reason for falling back to the default is not the missing/misconfigured DTS, but the fact that to assert HIGH on CHIP_EN and RESET I need to grab these lines first. The error of gpiod_get is not `-ENOENT`, but `-EBUSY`.

I verified this with the following quick and dirty hack:

```
diff --git a/wilc/linux_wlan.c b/wilc/linux_wlan.c
index 38ad88a..bbac736 100644
--- a/wilc/linux_wlan.c
+++ b/wilc/linux_wlan.c
@@ -1796,6 +1796,8 @@ static void wilc_wlan_power(struct wilc *wilc, int power)

        gpio_reset = gpiod_get(wilc->dt_dev, "reset", GPIOD_ASIS);
        if (IS_ERR(gpio_reset)) {
+          if(PTR_ERR(gpio_reset) == -EBUSY )
+            dev_warn(wilc->dev, "reset gpio is busy GPIO, try default\r\n");
                dev_warn(wilc->dev, "failed to get Reset GPIO, try default\r\n");
                gpio_reset = gpio_to_desc(GPIO_NUM_RESET);
                if (!gpio_reset) {
@@ -1809,6 +1811,9 @@ static void wilc_wlan_power(struct wilc *wilc, int power)

        gpio_chip_en = gpiod_get(wilc->dt_dev, "chip_en", GPIOD_ASIS);
        if (IS_ERR(gpio_chip_en)) {
+          if(PTR_ERR(gpio_chip_en) == -EBUSY )
+            dev_warn(wilc->dev, "chip_en gpio is busy GPIO, try default\r\n");
                gpio_chip_en = gpio_to_desc(GPIO_NUM_CHIP_EN);
                if (!gpio_chip_en) {
                        dev_warn(wilc->dev,
```
This is to be expected, as I have to grab the lines for the reasons lined out in my first post.

Please be aware that the Raspberry PI GPIO host does NOT support assigning values through the DTS.

If the driver expects the lines to be driven high, IMHO this should be reflected somewhere in the documentation - neither the example DTS nor the README mention this and the datasheet explicitly asks for system startup low lines for those two signals.

But if you say the behaviour is to be expected, then I have to find a workaround. There is later kernel versions that allow for the gpiod_get call to share GPIO lines, I will give that a whirl.

```
# find /sys/firmware/devicetree/base/soc/sdio\@7e300000/
/sys/firmware/devicetree/base/soc/sdio@7e300000/
/sys/firmware/devicetree/base/soc/sdio@7e300000/compatible
/sys/firmware/devicetree/base/soc/sdio@7e300000/clocks
/sys/firmware/devicetree/base/soc/sdio@7e300000/status
/sys/firmware/devicetree/base/soc/sdio@7e300000/interrupts
/sys/firmware/devicetree/base/soc/sdio@7e300000/wilc_sdio@0
/sys/firmware/devicetree/base/soc/sdio@7e300000/wilc_sdio@0/compatible
/sys/firmware/devicetree/base/soc/sdio@7e300000/wilc_sdio@0/irq-gpios
/sys/firmware/devicetree/base/soc/sdio@7e300000/wilc_sdio@0/chip_en-gpios
/sys/firmware/devicetree/base/soc/sdio@7e300000/wilc_sdio@0/status
/sys/firmware/devicetree/base/soc/sdio@7e300000/wilc_sdio@0/bus-width
/sys/firmware/devicetree/base/soc/sdio@7e300000/wilc_sdio@0/reset-gpios
/sys/firmware/devicetree/base/soc/sdio@7e300000/wilc_sdio@0/phandle
/sys/firmware/devicetree/base/soc/sdio@7e300000/wilc_sdio@0/reg
/sys/firmware/devicetree/base/soc/sdio@7e300000/wilc_sdio@0/name
/sys/firmware/devicetree/base/soc/sdio@7e300000/brcm,overclock-50
/sys/firmware/devicetree/base/soc/sdio@7e300000/bus-width
/sys/firmware/devicetree/base/soc/sdio@7e300000/dma-names
/sys/firmware/devicetree/base/soc/sdio@7e300000/phandle
/sys/firmware/devicetree/base/soc/sdio@7e300000/reg
/sys/firmware/devicetree/base/soc/sdio@7e300000/pinctrl-0
/sys/firmware/devicetree/base/soc/sdio@7e300000/dmas
/sys/firmware/devicetree/base/soc/sdio@7e300000/non-removable
/sys/firmware/devicetree/base/soc/sdio@7e300000/name
/sys/firmware/devicetree/base/soc/sdio@7e300000/pinctrl-names
```

AdhamAbozaeid, 2019-06-05:
Thanks for the extra details on the issue you are having.
I agree that this should be highlighted in the documentation. The user guide mentions that the HW pull up resistor should be installed on the shield board for the same purpose.
This is also implemented in the example where the GPIO are declared as active high.

The proper way to fix that is to use mmc's power sequencer to control the chip_en once the SDIO driver is registered in the WILC driver, hence allowing the probe to succeed.
There are some trial that were done for that in branch linux-4.9-at91_wilc_dev_pwrseq, but it's not complete yet.

dir-ableton, 2019-06-11:
Thanks for the confirmation and clarification. Shall we close the  ticket for now?

AdhamAbozaeid, 2019-06-11:
Since this works for you now, I'll go ahead and close it.

closed 2019-06-11 by AdhamAbozaeid