linux4wilc/driver #48: Bluetooth is not responding

ruschigo, 2019-06-07:
Hi!

So i'm working around 3 weeks trying test the atwilc3000 module, wifi and bluetooth, right now i have i WIFI working well, but i get a huge problem with Bluetooth.

I have a uart at dev/ttyS2 working, i had made simple test like send some file by cp command or creating a user application to write through ttyS2 and i got the signals out from processor-uart-tx with an oscilloscope. I don't try anything more complex like a uart loopback, because it will give some problems.

I had the wilc_bt dev after modprobe the module wilc_sdio.

The problem is that after send the BT_POWER_UP and BT_DOWNLOAD_FW commands, the hci commands are send and can be probed by oscilloscope but i got no response from atwilc3000 device.

I had only success logs and i can use hciattach to link hci to my uart, and got 'Device setup complete', the signals send by hciconfig hci0 (reset or up) can be saw by oscilloscope, so can i do anything to get more information or try any other thing?

I'm new working with kernel source code, and driver development, so please try to be as simple as possible :) .

Any help will be very appreciate.

i will let some logs here, any more logs could help just ask.
 
#Logs from kernel  Uart
------------------------------------------------------------------------------------------
```
b0124000.serial: ttyS2 at MMIO 0xb0124000 (irq = 13, base_baud = 1500000) is a caninos-uart
device: 'ttyS2': device_add
PM: Adding info for No Bus:ttyS2
```

```
Bluetooth: Core ver 2.22
device class 'bluetooth': registering
NET: Registered protocol family 31
Bluetooth: HCI device and connection manager initialized
Bluetooth: HCI socket layer initialized
Bluetooth: L2CAP socket layer initialized
Bluetooth: SCO socket layer initialized

Bluetooth: HCI UART driver ver 2.3
Bluetooth: HCI UART protocol H4 registered
Bluetooth: HCI UART protocol BCSP registered
Bluetooth: HCI UART protocol ATH3K registered
Bluetooth: HCI UART protocol Three-wire (H5) registered
```

#Uart verification by setserial
-------------------------------------------------------------------------------------------
```
root@debian-armhf:/home/caninos# setserial /dev/ttyS2 -g -a
/dev/ttyS2, Line 2, UART: undefined, Port: 0x0000, IRQ: 13
	Baud_base: 1500000, close_delay: 50, divisor: 0
	closing_wait: 3000
	Flags: spd_normal low_latency
```

# attaching HCI to uart, and try identify device (Bluez)
--------------------------------------------------------------------------------------------
```
root@debian-armhf:/home/caninos# hciattach /dev/ttyS2 any 115200 noflow
Device setup complete
root@debian-armhf:/home/caninos# hciconfig -a
hci0:	Type: Primary  Bus: UART
	BD Address: 00:00:00:00:00:00  ACL MTU: 0:0  SCO MTU: 0:0
	DOWN 
	RX bytes:0 acl:0 sco:0 events:0 errors:0
	TX bytes:12 acl:0 sco:0 commands:3 errors:0
	Features: 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
	Packet type: DM1 DH1 HV1 
	Link policy: 
	Link mode: SLAVE ACCEPT 
```

#Try to reset device
---------------------------------------------------------------------------------------------
```
root@debian-armhf:/home/caninos# hciconfig hci0 reset
Can't init device hci0: Connection timed out (110)
root@debian-armhf:/home/caninos# hciconfig
hci0:	Type: Primary  Bus: UART
	BD Address: 00:00:00:00:00:00  ACL MTU: 0:0  SCO MTU: 0:0
	DOWN 
	RX bytes:0 acl:0 sco:0 events:0 errors:0
	TX bytes:36 acl:0 sco:0 commands:9 errors:0
```

#lsmod
---------------------------------------------------------------------------------------------
```
root@debian-armhf:/home/caninos# lsmod
Module                  Size  Used by
wilc_sdio             176128  0
cfg80211              499712  1 wilc_sdio
```

#echo BT_POWER_UP > /dev/wilc_bt
---------------------------------------------------------------------------------------------
```
at_pwr_dev: open()
AT PWR: bt_power_up
Device already up. request source is BT
at_pwr_dev: close()
```

#echo BT_DOWNLOAD_FW > /dev/wilc_bt 
---------------------------------------------------------------------------------------------
```
[11285.097407] at_pwr_dev: open()
[11285.100592] AT PWR: bt_download_fw
[11285.100601] Bluetooth firmware: mchp/wilc3000_ble_firmware.bin
[11285.100619] firmware_class: __allocate_fw_buf: fw-mchp/wilc3000_ble_firmware.bin buf=eb4e5600
[11285.100678] atmel wilc_bt: loading /lib/firmware/updates/4.14.13/mchp/wilc3000_ble_firmware.bin failed with error -2
[11285.100697] atmel wilc_bt: loading /lib/firmware/updates/mchp/wilc3000_ble_firmware.bin failed with error -2
[11285.100714] atmel wilc_bt: loading /lib/firmware/4.14.13/mchp/wilc3000_ble_firmware.bin failed with error -2
[11285.105163] atmel wilc_bt: firmware: direct-loading firmware mchp/wilc3000_ble_firmware.bin
[11285.105195] firmware_class: fw_set_page_data: fw-mchp/wilc3000_ble_firmware.bin buf=eb4e5600 data=f309a000 size=58184
[11285.106852] Downloading BT firmware size = 58184 ...
[11285.161973] firmware_class: __fw_free_buf: fw-mchp/wilc3000_ble_firmware.bin buf=eb4e5600 data=f309a000 size=58184
[11285.162082] Starting BT firmware
[11285.270459] BT Start Succeeded
[11285.270592] at_pwr_dev: close()
```

AdhamAbozaeid, 2019-06-10:
Hi @ruschigo 
Is this the sequence you are using to bring up BLE? Are you running wifi in parallel?
You should power up the device, and download the FW before trying to attach the HCI device, so the correct sequence should be:
```
modprobe wilc-sdio
echo BT_POWER_UP > /dev/wilc_bt
echo BT_DOWNLOAD_FW > /dev/wilc_bt
hciattach ttyS1 any 115200 noflow
hciconfig hci0 up
```

ruschigo, 2019-06-10:
@AdhamAbozaeid thanks or replying 

I following the sequence that you posted, but i had try many other ways too (just for test purpose), but there are no way that work.

`modprobe wilc-sdio
echo BT_POWER_UP > /dev/wilc_bt
echo BT_DOWNLOAD_FW > /dev/wilc_bt
hciattach ttyS2 any 115200 noflow
hciconfig hci0 up`

I had post my logs before out of order because i had some trash in my kern.log so i try to avoid that, sorry about, i will post a complete clean log since the modprobe begins.(in order this time).

I had found only one strange log, after the command BT_DOWNLOAD_FW i saw the log 'BT initialize timeout', i found this message at the power up device code, what it means?

`Feb 14 08:13:46 debian-armhf kernel: [  115.824974] wilc_driver begin __init
Feb 14 08:13:46 debian-armhf kernel: [  115.825005] bus: 'sdio': add driver wilc_sdio
Feb 14 08:13:46 debian-armhf kernel: [  115.828744] bus: 'sdio': driver_probe_device: matched device mmc1:0001:1 with driver wilc_sdio
Feb 14 08:13:46 debian-armhf kernel: [  115.828761] bus: 'sdio': really_probe: probing driver wilc_sdio with device mmc1:0001:1
Feb 14 08:13:46 debian-armhf kernel: [  115.828783] wilc_sdio mmc1:0001:1: no of_node; not parsing pinctrl DT
Feb 14 08:13:46 debian-armhf kernel: [  115.828803] wilc_sdio mmc1:0001:1: no default pinctrl state
Feb 14 08:13:46 debian-armhf kernel: [  115.828831] devices_kset: Moving mmc1:0001:1 to end of list
Feb 14 08:13:46 debian-armhf kernel: [  115.828971] printk: WILC_PROBE BEGIN
Feb 14 08:13:46 debian-armhf kernel: [  115.832688] wilc_sdio mmc1:0001:1: WILC_PROBE BEGIN
Feb 14 08:13:46 debian-armhf kernel: [  115.832700] wilc_sdio mmc1:0001:1: Initializing netdev
Feb 14 08:13:46 debian-armhf kernel: [  115.833355] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
Feb 14 08:13:46 debian-armhf kernel: [  115.833372] (unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
Feb 14 08:13:46 debian-armhf kernel: [  115.833451] device: 'phy0': device_add
Feb 14 08:13:46 debian-armhf kernel: [  115.833592] PM: Adding info for No Bus:phy0
Feb 14 08:13:46 debian-armhf kernel: [  115.833985] device: 'wlan0': device_add
Feb 14 08:13:46 debian-armhf kernel: [  115.834228] PM: Adding info for No Bus:wlan0
Feb 14 08:13:46 debian-armhf kernel: [  115.835291] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
Feb 14 08:13:46 debian-armhf kernel: [  115.835308] (unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
Feb 14 08:13:46 debian-armhf kernel: [  115.835375] device: 'phy1': device_add
Feb 14 08:13:46 debian-armhf kernel: [  115.835489] PM: Adding info for No Bus:phy1
Feb 14 08:13:46 debian-armhf kernel: [  115.835869] device: 'p2p0': device_add
Feb 14 08:13:46 debian-armhf kernel: [  115.836108] PM: Adding info for No Bus:p2p0
Feb 14 08:13:46 debian-armhf kernel: [  115.836949] wifi_pm : 0
Feb 14 08:13:46 debian-armhf kernel: [  115.836984] mmc mmc1:0001: GPIO lookup for consumer reset
Feb 14 08:13:46 debian-armhf kernel: [  115.836994] mmc mmc1:0001: using lookup tables for GPIO lookup
Feb 14 08:13:46 debian-armhf kernel: [  115.837005] mmc mmc1:0001: lookup for GPIO reset failed
Feb 14 08:13:46 debian-armhf kernel: [  115.837017] wilc_sdio mmc1:0001:1: failed to get Reset GPIO, try default
Feb 14 08:13:46 debian-armhf kernel: [  115.851197] mmc mmc1:0001: GPIO lookup for consumer chip_en
Feb 14 08:13:46 debian-armhf kernel: [  115.851211] mmc mmc1:0001: using lookup tables for GPIO lookup
Feb 14 08:13:46 debian-armhf kernel: [  115.851222] mmc mmc1:0001: lookup for GPIO chip_en failed
Feb 14 08:13:46 debian-armhf kernel: [  115.851244] ------------[ cut here ]------------
Feb 14 08:13:46 debian-armhf kernel: [  115.856045] WARNING: CPU: 2 PID: 724 at /home/caninos/Documentos/Driver_Wifi/linux-debian-master/linux/drivers/gpio/gpiolib.c:2161 gpiod_free+0x44/0x64
Feb 14 08:13:47 debian-armhf kernel: [  115.879337] Modules linked in: wilc_sdio(+) cfg80211
Feb 14 08:13:47 debian-armhf kernel: [  115.884474] CPU: 2 PID: 724 Comm: modprobe Not tainted 4.14.13 #1
Feb 14 08:13:47 debian-armhf kernel: [  115.890573] Hardware name: labrador
Feb 14 08:13:47 debian-armhf kernel: [  115.894071] Backtrace: 
Feb 14 08:13:47 debian-armhf kernel: [  115.896567] [<c010df78>] (dump_backtrace) from [<c010e2ac>] (show_stack+0x20/0x24)
Feb 14 08:13:47 debian-armhf kernel: [  115.904144]  r7:600f0013 r6:c15e91c4 r5:00000000 r4:c15e91c4
Feb 14 08:13:47 debian-armhf kernel: [  115.909820] [<c010e28c>] (show_stack) from [<c0e7a2b0>] (dump_stack+0x84/0x98)
Feb 14 08:13:47 debian-armhf kernel: [  115.917056] [<c0e7a22c>] (dump_stack) from [<c0160478>] (__warn+0xf8/0x110)
Feb 14 08:13:47 debian-armhf kernel: [  115.924000]  r7:00000009 r6:c12230ac r5:00000000 r4:00000000
Feb 14 08:13:47 debian-armhf kernel: [  115.929669] [<c0160380>] (__warn) from [<c0160560>] (warn_slowpath_null+0x30/0x38)
Feb 14 08:13:47 debian-armhf kernel: [  115.937242]  r9:00000000 r8:eb3c0800 r7:00000000 r6:eda065e0 r5:eda063c0 r4:eda065e0
Feb 14 08:13:47 debian-armhf kernel: [  115.945003] [<c0160530>] (warn_slowpath_null) from [<c06d1fc0>] (gpiod_free+0x44/0x64)
Feb 14 08:13:47 debian-armhf kernel: [  115.952909] [<c06d1f7c>] (gpiod_free) from [<c06d2bb4>] (gpiod_put+0x18/0x1c)
Feb 14 08:13:47 debian-armhf kernel: [  115.960025]  r5:eda063c0 r4:eb3d1800
[... cut ...]
Feb 14 08:16:01 debian-armhf kernel: [  250.074597] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry1: 33:33:0:0:0:1
Feb 14 08:16:01 debian-armhf kernel: [  250.074606] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry2: 33:33:ff:da:25:f2
Feb 14 08:16:01 debian-armhf kernel: [  250.074613] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry3: 33:33:0:0:0:fb
Feb 14 08:16:01 debian-armhf kernel: [  250.517701] wilc_sdio mmc1:0001:1 wlan0: INFO [get_tx_power]Got tx power 18`

AdhamAbozaeid, 2019-06-10:
From the logs, it looks like the wifi interfaces are up, then closed.
Are you starting the wpa_supplicant, or doing a ifconfig wlan0 up? can you try again without starting the wifi interfaces?
Also please note that the commands I posted are using ttys1, while from you first comment I think your setup should be using ttys2

ruschigo, 2019-06-10:
Sorry about the ttyS* mistake, I'm always use the ttyS2, i don't have any dev/ttyS1, only dev/ttyS2 and dev/ttyS3, i had enable just my UART2 and UART3, so it looks right.

i'm not doing any ifconfig commands.

the wpa_supplicant was active, i disable it with systemctl stop, remove and then mask it .(is it enough?)
I had wicd application that was calling wifi too, so i disabled it too.

now the last log that refers to wilc3000 driver is the download of BLE firware, but it steel not working

```
Feb 14 08:13:07 debian-armhf kernel: [   76.763973] printk: WILC_PROBE BEGIN
Feb 14 08:13:07 debian-armhf kernel: [   76.767630] wilc_sdio mmc1:0001:1: WILC_PROBE BEGIN
Feb 14 08:13:07 debian-armhf kernel: [   76.767639] wilc_sdio mmc1:0001:1: Initializing netdev
Feb 14 08:13:07 debian-armhf kernel: [   76.771416] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
Feb 14 08:13:07 debian-armhf kernel: [   76.771435] (unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
Feb 14 08:13:07 debian-armhf kernel: [   76.771491] device: 'phy0': device_add
Feb 14 08:13:07 debian-armhf kernel: [   76.771598] PM: Adding info for No Bus:phy0
Feb 14 08:13:07 debian-armhf kernel: [   76.771935] device: 'wlan0': device_add
Feb 14 08:13:07 debian-armhf kernel: [   76.772156] PM: Adding info for No Bus:wlan0
Feb 14 08:13:07 debian-armhf kernel: [   76.773076] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
Feb 14 08:13:07 debian-armhf kernel: [   76.773089] (unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
Feb 14 08:13:07 debian-armhf kernel: [   76.773147] device: 'phy1': device_add
Feb 14 08:13:07 debian-armhf kernel: [   76.773235] PM: Adding info for No Bus:phy1
Feb 14 08:13:07 debian-armhf kernel: [   76.773519] device: 'p2p0': device_add
Feb 14 08:13:07 debian-armhf kernel: [   76.773722] PM: Adding info for No Bus:p2p0
Feb 14 08:13:07 debian-armhf kernel: [   76.774534] wifi_pm : 0
Feb 14 08:13:07 debian-armhf kernel: [   76.774560] mmc mmc1:0001: GPIO lookup for consumer reset
Feb 14 08:13:07 debian-armhf kernel: [   76.774569] mmc mmc1:0001: using lookup tables for GPIO lookup
Feb 14 08:13:07 debian-armhf kernel: [   76.774578] mmc mmc1:0001: lookup for GPIO reset failed
Feb 14 08:13:07 debian-armhf kernel: [   76.774588] wilc_sdio mmc1:0001:1: failed to get Reset GPIO, try default
Feb 14 08:13:07 debian-armhf kernel: [   76.791053] mmc mmc1:0001: GPIO lookup for consumer chip_en
Feb 14 08:13:07 debian-armhf kernel: [   76.791064] mmc mmc1:0001: using lookup tables for GPIO lookup
Feb 14 08:13:07 debian-armhf kernel: [   76.791073] mmc mmc1:0001: lookup for GPIO chip_en failed
Feb 14 08:13:07 debian-armhf kernel: [   76.791095] ------------[ cut here ]------------
Feb 14 08:13:07 debian-armhf kernel: [   76.795845] WARNING: CPU: 2 PID: 698 at /home/caninos/Documentos/Driver_Wifi/linux-debian-master/linux/drivers/gpio/gpiolib.c:2161 gpiod_free+0x44/0x64
Feb 14 08:13:07 debian-armhf kernel: [   76.815302] Modules linked in: wilc_sdio(+) cfg80211
Feb 14 08:13:07 debian-armhf kernel: [   76.820488] CPU: 2 PID: 698 Comm: modprobe Not tainted 4.14.13 #1
Feb 14 08:13:08 debian-armhf kernel: [   76.826597] Hardware name: labrador
Feb 14 08:13:08 debian-armhf kernel: [   76.830105] Backtrace: 
Feb 14 08:13:08 debian-armhf kernel: [   76.832602] [<c010df78>] (dump_backtrace) from [<c010e2ac>] (show_stack+0x20/0x24)
Feb 14 08:13:08 debian-armhf kernel: [   76.840174]  r7:60010013 r6:c15e91c4 r5:00000000 r4:c15e91c4
Feb 14 08:13:08 debian-armhf kernel: [   76.845837] [<c010e28c>] (show_stack) from [<c0e7a2b0>] (dump_stack+0x84/0x98)
Feb 14 08:13:08 debian-armhf kernel: [   76.853050] [<c0e7a22c>] (dump_stack) from [<c0160478>] (__warn+0xf8/0x110)
Feb 14 08:13:08 debian-armhf kernel: [   76.860016]  r7:00000009 r6:c12230ac r5:00000000 r4:00000000
Feb 14 08:13:08 debian-armhf kernel: [   76.865716] [<c0160380>] (__warn) from [<c0160560>] (warn_slowpath_null+0x30/0x38)
Feb 14 08:13:08 debian-armhf kernel: [   76.873302]  r9:00000000 r8:ec6db9c0 r7:00000000 r6:eda065e0 r5:eda063c0 r4:eda065e0
Feb 14 08:13:08 debian-armhf kernel: [   76.881068] [<c0160530>] (warn_slowpath_null) from [<c06d1fc0>] (gpiod_free+0x44/0x64)
Feb 14 08:13:08 debian-armhf kernel: [   76.888994] [<c06d1f7c>] (gpiod_free) from [<c06d2bb4>] (gpiod_put+0x18/0x1c)
Feb 14 08:13:08 debian-armhf kernel: [   76.896114]  r5:eda063c0 r4:ec282800
Feb 14 08:13:08 debian-armhf kernel: [   76.899822] [<c06d2b9c>] (gpiod_put) from [<bf0b284c>] (wilc_wlan_power+0xf4/0x160 [wilc_sdio])
Feb 14 08:13:08 debian-armhf kernel: [   76.908680] [<bf0b2758>] (wilc_wlan_power [wilc_sdio]) from [<bf0b4bac>] (wilc_wlan_power_on_sequence+0x20/0x30 [wilc_sdio])
Feb 14 08:13:08 debian-armhf kernel: [   76.919873]  r9:00000000 r8:ec6db9c0 r7:00000000 r6:ec82da00 r5:bf0d08ec r4:ec282800
Feb 14 08:13:08 debian-armhf kernel: [   76.927746] [<bf0b4b8c>] (wilc_wlan_power_on_sequence [wilc_sdio]) from [<bf0d70d8>] (linux_sdio_probe+0xd8/0x140 [wilc_sdio])
Feb 14 08:13:08 debian-armhf kernel: [   76.939108]  r5:bf0d08ec r4:ec82da08
Feb 14 08:13:08 debian-armhf kernel: [   76.942744] [<bf0d7000>] (linux_sdio_probe [wilc_sdio]) from [<c0acc8c8>] (sdio_bus_probe+0x118/0x12c)
Feb 14 08:13:08 debian-armhf kernel: [   76.952026]  r8:bf0c6898 r7:bf0cf838 r6:ec82da00 r5:00000000 r4:ec82da08
[... cut ...]
Feb 14 08:14:42 debian-armhf kernel: [  171.125755] Bluetooth: hci0 command 0x1003 tx timeout
Feb 14 08:14:44 debian-armhf kernel: [  173.141687] Bluetooth: hci0 command 0x1001 tx timeout
Feb 14 08:14:46 debian-armhf kernel: [  175.157754] Bluetooth: hci0 command 0x1009 tx timeout
Feb 14 08:14:56 debian-armhf kernel: [  185.414283] Interface is neither WLAN0 nor P2P0
Feb 14 08:14:56 debian-armhf kernel: [  185.414740] PM: Removing info for No Bus:lo
Feb 14 08:17:07 debian-armhf kernel: [  316.467782] EXT4-fs (mmcblk2p2): error count since last fsck: 5101995
```

AdhamAbozaeid, 2019-06-10:
It's better to avoid starting wifi at all for now to make debugging easier.
I also see a strange log here
```
Feb 14 08:13:37 debian-armhf kernel: [  106.009211] firmware_class: __allocate_fw_buf: fw-mchp/wilc3000_ble_firmware.bin buf=ec7c19c0
Feb 14 08:13:37 debian-armhf kernel: [  106.010669] atmel wilc_bt: loading /lib/firmware/updates/4.14.13/mchp/wilc3000_ble_firmware.bin failed with error -2
Feb 14 08:13:37 debian-armhf kernel: [  106.010748] atmel wilc_bt: loading /lib/firmware/updates/mchp/wilc3000_ble_firmware.bin failed with error -2
Feb 14 08:13:37 debian-armhf kernel: [  106.010821] atmel wilc_bt: loading /lib/firmware/4.14.13/mchp/wilc3000_ble_firmware.bin failed with error -2
Feb 14 08:13:37 debian-armhf kernel: [  106.015933] atmel wilc_bt: firmware: direct-loading firmware mchp/wilc3000_ble_firmware.bin
```
where it looks like the kernel is looking for the FW binary in multiple places, but looks like it founds it successfully at last, so I don't think this is a real problem.
Do you have logs from the WILC? You can read these by attaching FTDI cable to the debug UART pins (UART_TXD) on the WILC3000 board. Note that this is different from the BT_TXD UART pin used for the BLE HCI commands

ruschigo, 2019-06-11:
Hello again, i really appreciate your help! thank you!

Yeah i have this uart connected to my pc, so i got it with Putty, has only a few informations;

```
@@@@@@@@@@@@@@@@@
BT FW Started
@@@@@@@@@@@@@@@@@

BT Linux MAC Firmware version: WILC_Linux_BT_FW_REL_15_2_RC3

BT RTOS MAC Firmware version: WILC_RTOS_BT_FW_REL_4_4_RC2

RF Version: 01.1
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
```

AdhamAbozaeid, 2019-06-11:
From the WILC logs, it shows that the FW is starting successfully.
Please open a salesforce ticket with all necessary logs and more information about the HW you are using so that the team can help you more.

ruschigo, 2019-06-11:
@AdhamAbozaeid thank you for all your support!

I will try open the ticket.

can you give one more information?, the ATCWILC3000 is setting the BT_RTS(P10) to HIGH all the time, is this correct?
The BT_CTS is in LOW...

I will keep the issue open, when i solve it then i close the issue.

AdhamAbozaeid, 2019-06-11:
The default setting for WILC at startup is that UART flow control will be disabled, and hence hciattach is called with noflow option

ruschigo, 2019-08-30:
Hello again!

I stay some time out, working with other things, but i start again testing the bluetooth...

Just to let some update here, i found the source of all my troubles with the bluetooth, my uart driver is not working properly, the baud rate does not change properly, only when i configure to console mode, it is strange because the console callbacks and usual uarts use the same routine to change the baudrate, but for some reason that i steel don't know the the callback do not change the uart clock divisor properly(i will work on this later).

For while and to keep going with tests of bluetooth i just set the baud rate directly in registers of uart, i know it is not the right way, but works...

Now if i did the properly way to start bluetooth, using 

- modprobe wilc-sdio
- echo BT_POWER_ON >  /dev/wilc_bt
- echo BT_FW_CHIP_WAKEUP >  /dev/wilc_bt
- echo BT_DOWNLOAD_FW >  /dev/wilc_bt
- echo BT_FW_CHIP_ALLOW_SLEEP >  /dev/wilc_bt
- hciattach ttyS2 any 115200 noflow
- hciconfig hci0 -a

i have this...

`caninos@debian-armhf:~$ hciconfig hci0 -a
hci0:	Type: Primary  Bus: UART
	BD Address: F8:F0:05:D9:2A:77  ACL MTU: 27:30  SCO MTU: 0:0
	UP RUNNING 
	RX bytes:11917 acl:0 sco:0 events:305 errors:0
	TX bytes:303 acl:0 sco:0 commands:39 errors:0
	Features: 0x00 0x00 0x00 0x00 0x60 0x00 0x00 0x00
	Packet type: DM1 DH1 HV1 
	Link policy: 
	Link mode: SLAVE ACCEPT 
Can't read local name on hci0: Input/output error (5)`

I can get the MAC address, but steel have some problem, the MTU is strange, using the bluetoothctl <scan on> it can find some devices, not all, some times connect with these devices, but the connection is closed so soon...

I do not find the source of the error 'Can't read local name on hci0: Input/output error (5)' the logs do not help me a lot:

`root@debian-armhf:/etc/bluetooth# sudo service bluetooth status
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2019-08-30 15:11:18 -03; 11s ago
     Docs: man:bluetoothd(8)
 Main PID: 1939 (bluetoothd)
   Status: "Running"
    Tasks: 1 (limit: 4641)
   Memory: 488.0K
   CGroup: /system.slice/bluetooth.service
           └─1939 /usr/lib/bluetooth/bluetoothd --noplugin=sap

Aug 30 15:11:18 debian-armhf systemd[1]: Starting Bluetooth service...
Aug 30 15:11:18 debian-armhf bluetoothd[1939]: Bluetooth daemon 5.50
Aug 30 15:11:18 debian-armhf systemd[1]: Started Bluetooth service.
Aug 30 15:11:18 debian-armhf bluetoothd[1939]: Excluding (cli) sap
Aug 30 15:11:18 debian-armhf bluetoothd[1939]: Bluetooth management interface 1.14 initialized
Aug 30 15:11:18 debian-armhf bluetoothd[1939]: Failed to set local name: Failed (0x03)
Aug 30 15:11:18 debian-armhf bluetoothd[1939]: Failed to set local name: Failed (0x03)
`

Other case is that the hcitool scan is not working too it give me this:

`root@debian-armhf:/etc/bluetooth# hcitool scan
Scanning ...
Inquiry failed: Operation not supported
`

it looks like now to software stack of bluetooth...

If you had seen something like this before or have any idea that could help me, let me know

Thanks

Best regards

ruschigo, 2019-09-16:
Hi!

Just to close this issue, i had found the 'problem' with the responses from my chip and what was expected by microchip documents.

Microchip documents uses a old version of firmware and driver, so the driver and firmware provide support or classic bluetooth, the new version of driver and firmware does not support classic bluetooth.

So the error 'Can't read local name on hci0: Input/output error (5)' should be fine, because ble not support this feature, about size o packets, i don't know why it is so small in ble.

The results using old firmware and driver is below;

root@debian-armhf:/home/caninos# hciconfig -a
hci0:	Type: Primary  Bus: UART
	BD Address: F8:F0:05:D9:2A:77  ACL MTU: 1021:9  SCO MTU: 255:4
	UP RUNNING PSCAN ISCAN 
	RX bytes:2201 acl:0 sco:0 events:144 errors:0
	TX bytes:4852 acl:0 sco:0 commands:141 errors:0
	Features: 0xff 0xff 0xcd 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH HOLD SNIFF PARK 
	Link mode: SLAVE ACCEPT 
	Name: 'BlueZ'
	Class: 0x0c0000
	Service Classes: Rendering, Capturing
	Device Class: Miscellaneous, 
	HCI Version: 4.0 (0x6)  Revision: 0x709
	LMP Version: 4.0 (0x6)  Subversion: 0x709
	Manufacturer: Atmel Corporation (19)

Bluez 5.50 is ok as well.

Thanks

Best Regards 

Igor Ruschi

closed 2019-09-16 by ruschigo

AdhamAbozaeid, 2019-09-16:
> Microchip documents uses a old version of firmware and driver, so the driver and firmware provide support or classic bluetooth, the new version of driver and firmware does not support classic bluetooth.

I believe you are pointing to section "8.14.11 BLE Peripheral Mode Example For BlueZ 5.28 and Earlier" of the linux user guide since other sections shows the logs from the newer versions.
We will make sure to update that on the next version of the document, so please let us know if you see other inconsistencies in the documentation.

ruschigo, 2019-09-16:
>I believe you are pointing to section "8.14.11 BLE Peripheral Mode Example For BlueZ 5.28 and Earlier" of the linux user guide since other sections shows the logs from the newer versions.

Yes, it is the section that you pointed above, sorry about generalize, but your are right, it is the only point where i saw some inconsistence...

Thanks!

Best regards

Igor Ruschi