linux4wilc/driver #49: failed to get GPIO

flgcpwd, 2019-06-17:
Hello,

I use a custom board using imx6 and kernel 4.14.  Wilc1000 is connected on SDIO.
My WILC1000 was not detected on mmc but it was due to reset and enable pin wich was low at boot. So i set this pins high in the boot.
Now the driver is loded and i can connect my board to a WIFI access point.

However sometime a get an error to load the firmware.
When i check the logs i can see that GPIOs (enable, irq and reset) are not set correctly.
I get some errors : "failed to get IRQ GPIO" , "failed to get Reset GPIO" and "failed to get Enable GPIO" in the logs.

My device tree is  : 
        wilc_sdio@1 {
                compatible = "microchip,wilc1000", "microchip,wilc3000", "atmel,wilc_sdio";
                irq-gpios = <&gpio4 2 IRQ_TYPE_EDGE_FALLING>;
                reset-gpios = <&gpio4 12 GPIO_ACTIVE_LOW>;
                chip_en-gpios = <&gpio4 14 GPIO_ACTIVE_HIGH>;
                status = "okay";
                reg = <1>;
        };

Is it right to set "reset-gpios" in device tree or should i write only "reset" ?
If someone has an idea to help me it would be great.

Thanks,

AdhamAbozaeid, 2019-06-17:
Hi @flgcpwd 

I'm using different notation for the gpios on my dts for sama5d4
https://github.com/linux4wilc/linux-at91/commit/2a9741c27c8e2ada7864f4fbfeb5784a6343dec2#diff-567adfad6cf693a5d6f76e3dbc929db1

I assume your dts might still be ok since it works sometimes, but it's still worth a try.
The other possibility is that maybe these GPIOs are registered with another device that sometimes claims them before the WILC driver.
Can you check what's the return errors from gpiod_get() in the driver?

flgcpwd, 2019-06-18:
But i use driver 15.2.1 which use device tree.
I tried to export gpio pins from user space and it works, so i don't think that pins are used byother devices.
The return code "wl->gpio_irq" is -2.

AdhamAbozaeid, 2019-06-18:
-2 is the ENOENT No such file or directory

flgcpwd, 2019-06-18:
I tried this : 

modprobe wilc_sdio -r
echo 108 > /sys/class/gpio/export
echo 110 > /sys/class/gpio/export
echo out > /sys/class/gpio/gpio108/direction
echo out > /sys/class/gpio/gpio110/direction
echo 0 > /sys/class/gpio/gpio110/value
echo 0 > /sys/class/gpio/gpio108/value
sleep 1s
echo 1 > /sys/class/gpio/gpio110/value
sleep 1s
echo 1 > /sys/class/gpio/gpio108/value
echo 108 > /sys/class/gpio/unexport
echo 110 > /sys/class/gpio/unexport
sleep 3s
cd /sys/devices/soc0/soc/2100000.aips-bus/2194000.usdhc/driver
echo -n "2194000.usdhc" > unbind
sleep 3s
modprobe wilc_sdio
sleep 3s
echo -n "2194000.usdhc" > bind

as decribed in this post https://github.com/linux4wilc/driver/issues/21 and it works, but i always get the error message

AdhamAbozaeid, 2019-06-18:
Which error message? You mean the "failed to get IRQ GPIO"

FloGerm, 2019-06-18:
Yes exact.

AdhamAbozaeid, 2019-06-18:
that means the driver is failing to get the GPIO from the dts, and it works because you are controlling it manually from the command line, so the chip_en and reset have the correct values

FloGerm, 2019-06-18:
So, do you explain why the driver don't find thé gpio? What is the right syntax? The post https://github.com/linux4wilc/driver/issues/21 don't use the dame pin name in device tree. I use driver 15.2.1 and linux 4.14.
Thank you for tout help

AdhamAbozaeid, 2019-06-18:
I'm not sure why it fails to retrieve it from the dts in your case.
I'm using the syntax here https://github.com/linux4wilc/linux-at91/commit/2a9741c27c8e2ada7864f4fbfeb5784a6343dec2#diff-567adfad6cf693a5d6f76e3dbc929db1 on sama5d4, and it works fine.
I asked flgcpwd to try with this syntax instead of the one he's using, but not sure if he did.

flgcpwd, 2019-06-19:
Yes i tried with your syntax :
wilc_sdio@1 {
compatible = "microchip,wilc1000", "microchip,wilc3000", "atmel,wilc_sdio";
gpio_irq = <&gpio4 2 IRQ_TYPE_EDGE_FALLING>;
gpio_reset = <&gpio4 12 GPIO_ACTIVE_LOW>;
gpio_chip_en = <&gpio4 14 GPIO_ACTIVE_HIGH>;
status = "okay";
reg = <1>;
};

But i get the same result.

deanoati, 2019-06-19:
Driver 15.2.1 specifies the DTS syntax here:
https://github.com/linux4wilc/driver/blob/a65d0745c311400e60639878025e816ca089352c/wilc/microchip%2Cwilc%2Csdio.txt

Here is how I do it in our custom imx6 based system, kernel 4.1, latest wilc release:
```
&usdhc1 {
	pinctrl-names = "default";
	pinctrl-0 = <&pinctrl_usdhc1>;
	no-1-8-v;
	broken-cd;
	keep-power-in-suspend;
	enable-sdio-wakeup;
	bus-width = <4>;
	// max-frequency = <50000000>;
	// fsl,delay-line = <2>;
	status = "okay";

    // WILC1000 WiFi module
    wilc_sdio@0 {
        compatible = "microchip,wilc1000-sdio";
        reset-gpios = <&gpio1 9 GPIO_ACTIVE_HIGH>;
        chip_en-gpios = <&gpio3 27 GPIO_ACTIVE_HIGH>;
        // the IRQ gpio actually isn't used, but we assign one to appease the driver
        // See https://github.com/linux4wilc/driver/issues/22
        irq-gpios = <&gpio5 0 GPIO_ACTIVE_HIGH>;
        status = "okay";
        reg = <0>;
        bus-width = <4>;
    };
};
```

AdhamAbozaeid, 2019-06-19:
You are right.. these are the latest syntax we are using that were changed here:
https://github.com/linux4wilc/linux-at91/commit/53d5527c845f7fa3cff517ca86e9f6014609e4e4#diff-ae70d3de9794fd2ce9190983ae795c0c
My bad!

AdhamAbozaeid, 2019-08-08:
@flgcpwd did you find a fix for retrieving the GPIOs from the dts file?

flgcpwd, 2019-08-26:
A use this syntax in the dts : 

&usdhc2 {
	pinctrl-names = "default";
	pinctrl-0 = <&pinctrl_usdhc2>;
	non-removable;
	no-1-8-v;
	keep-power-in-suspend;
	status = "okay";

    #address-cells = <1>;
    #size-cells = <0>;

	wilc_sdio@1 {
		compatible = "microchip,wilc1000", "microchip,wilc3000", "atmel,wilc_sdio";
		irq-gpios = <&gpio4 2 IRQ_TYPE_EDGE_FALLING>;
		reset-gpios = <&gpio4 12 GPIO_ACTIVE_LOW>;
		chip_en-gpios = <&gpio4 14 GPIO_ACTIVE_HIGH>;
		status = "okay";
		reg = <1>;
	};
};

Then in the application i reset the chip 

# Supprime le module
modprobe wilc_sdio -r

# Reset le chip
echo 108 > /sys/class/gpio/export
echo 110 > /sys/class/gpio/export
echo out > /sys/class/gpio/gpio108/direction
echo out > /sys/class/gpio/gpio110/direction
echo 0 > /sys/class/gpio/gpio110/value
echo 0 > /sys/class/gpio/gpio108/value
sleep 1s
echo 1 > /sys/class/gpio/gpio110/value
sleep 1s
echo 1 > /sys/class/gpio/gpio108/value
echo 108 > /sys/class/gpio/unexport
echo 110 > /sys/class/gpio/unexport
sleep 3s
cd /sys/devices/soc0/soc/2100000.aips-bus/2194000.usdhc/driver
echo -n "2194000.usdhc" > unbind
sleep 3s
modprobe wilc_sdio
sleep 3s
echo -n "2194000.usdhc" > bind
sleep 1

And use wpa_supplicant : 
wpa_supplicant -iwlan0 -Dnl80211 -c /etc/wpa_supplicant.conf &

flgcpwd, 2019-08-26:
> # My syntax in the dts :
> &usdhc2 {
> pinctrl-names = "default";
> pinctrl-0 = <&pinctrl_usdhc2>;
> non-removable;
> no-1-8-v;
> keep-power-in-suspend;
> status = "okay";
> 
> ```
> #address-cells = <1>;
> #size-cells = <0>;
> 
> wilc_sdio@1 {
> 	compatible = "microchip,wilc1000", "microchip,wilc3000", "atmel,wilc_sdio";
> 	irq-gpios = <&gpio4 2 IRQ_TYPE_EDGE_FALLING>;
> 	reset-gpios = <&gpio4 12 GPIO_ACTIVE_LOW>;
> 	chip_en-gpios = <&gpio4 14 GPIO_ACTIVE_HIGH>;
> 	status = "okay";
> 	reg = <1>;
> };
> ```
> 
> };
> 
> Then in the application i reset the chip
> 
> # Supprime le module
> modprobe wilc_sdio -r
> 
> # Reset le chip
> echo 108 > /sys/class/gpio/export
> echo 110 > /sys/class/gpio/export
> echo out > /sys/class/gpio/gpio108/direction
> echo out > /sys/class/gpio/gpio110/direction
> echo 0 > /sys/class/gpio/gpio110/value
> echo 0 > /sys/class/gpio/gpio108/value
> sleep 1s
> echo 1 > /sys/class/gpio/gpio110/value
> sleep 1s
> echo 1 > /sys/class/gpio/gpio108/value
> echo 108 > /sys/class/gpio/unexport
> echo 110 > /sys/class/gpio/unexport
> sleep 3s
> cd /sys/devices/soc0/soc/2100000.aips-bus/2194000.usdhc/driver
> echo -n "2194000.usdhc" > unbind
> sleep 3s
> modprobe wilc_sdio
> sleep 3s
> echo -n "2194000.usdhc" > bind
> sleep 1
> 
> #And use wpa_supplicant :
> wpa_supplicant -iwlan0 -Dnl80211 -c /etc/wpa_supplicant.conf &

closed 2019-08-26 by flgcpwd