linux4wilc/driver #51: Performance issues

dir-ableton, 2019-06-26:
This is a bit of a cry for help that I don't know where else to direct. Please feel free to point me to another forum/source of information.

We're using WILC3000 with a Raspberry Pi compute module 3 and an OS based on open embedded with a 4.14 preempt_rt kernel. The WIFI throughput is abysmal. We reach about 7MBit tops, and this with high fluctuations that sometimes even come to a stop.

In investigating this, I established that this is most likely NOT a hardware/wilc issue itself. I arrived at this conclusion due to using a stock PI3B with it's BRCMFMAC wifi chip. This achieves on the same AP bitrates around 30-35MBit. When using this with our own embedded Linux, the rates drop to the ones we observe with the WILC setup.

Using the 4.19 kernel + config from raspibian on our own hardware + wilc does NOT improve the situation.

So my current best guess is some userspace/configuration issue. Trying to investigate that I found one oddity that might pose a hint. Running `iw dev wlan0 station dump` to learn about the station properties, I get this on raspibian:
```
Station 08:62:66:3f:4b:10 (on wlan0)
	inactive time:	0 ms
	rx bytes:	20091006
	rx packets:	188060
	tx bytes:	1128647816
	tx packets:	741858
	tx failed:	165
	signal:  	-49 [-49] dBm
	tx bitrate:	24.0 MBit/s
	rx bitrate:	1.0 MBit/s
	authorized:	yes
	authenticated:	yes
	associated:	yes
	WMM/WME:	yes
	TDLS peer:	no
	DTIM period:	3
	beacon interval:100
	short slot time:yes
	connected time:	5455 seconds
```

The same command on our OE system yields a meager
```
     Station 6b:b9:c4:6a:83:80 (on wlan0)
        signal:         -71 dBm
```

Is this a signifier for a not optimal configured system?  Any pointers more than welcom, I've been banging my head on this one for a while now.

AdhamAbozaeid, 2019-06-26:
Hi @dir-ableton 
The signal strength is a little bit low, but you can improve that by getting closer to the AP, and maybe test in a quite environment.
However, I'm not sure if this will improve the throughput much since you say that when you use the stock PI3B you get higher throughput, so I doubt that this the problem is related to noisy environment.

dir-ableton, 2019-06-26:
The distance for that measurement was ~4m. Earlier with a different router I had 2m, tops. And the PI is in both cases on the same desk, achieving the much higher throughput (when driven by raspibiban, that is).

closed 2019-10-09 by dir-ableton