linux4wilc/driver #52: Modprobe WILC3000 failure to start up

jack-lane-fulfil, 2019-06-27:
Hi there, I'm getting the following issue at startup. I'm running a TS-7800-v2 with the June 2019 distribution installed. This is the error:

```
root@ts7800-v2:~# rmmod wilc_spi
root@ts7800-v2:~# modprobe wilc_spi                                             
[   93.490760] WILC_SPI spi0.3: Failed cmd response, cmd (ca), resp (ff)        
[   93.503244] WILC_SPI spi0.3: Failed cmd, read reg (003b0000)...              
[   93.512399] WILC_SPI spi0.3: Fail cmd read chip id...                        
[   93.740112] WILC_SPI: probe of spi0.3 failed with error -5
```

Any advice would be greatly appreciated. Thanks!

AdhamAbozaeid, 2019-06-27:
It's either that the chip isn't powered up successfully, or something is wrong with SPI communication.
You can start by making sure that the chip_enable and reset_n signals to the WILC module are correct.
If they are, then make sure the SPI lines are connected correctly to the host.
This can also happen if you are using fly wires to connect WILC's module to your host since fly wires are more prone to noise

jack-lane-fulfil, 2019-06-27:
Hi Adham,

Thank you for the advice! I am fairly sure the chip is powered on successfully -- I was having trouble with Wifi on the board (it was connecting extremely shakily), so I updated to a more recent Linux distribution, and now the wlan0 device isn't being found at all. Given that I could find and use it in the past, I think the hardware is (mostly?) ok.

How exactly do I test that the chip_enable and reset_n signals are correct? I have little experience with embedded systems issues.

And by SPI lines and fly wires, are you talking about the physical connections from the card to the host?

Thanks a ton!
-Jack

AdhamAbozaeid, 2019-06-27:
Hi jack
Yes, I think that means the HW is ok, unless the new kernel is using higher SPI clock frequency that is more affected by noise.

Check the WILC module and see how the chip_en and reset_n are connected. Are they tied high, or connected to the host's GPIOs? You can also probe them with a logic analyzer or a multi-meter and check their values.

Yes, by fly wires I'm referring to the physical connection between the WILC module and the host. Are you using your own designed daughter board that have WILC's module, and then this daughter board plugs in the TS-7800-v2? or are you using a WILC3000 shield board wired to TS-7800-v2 with fly wires?
I don't think that WILC3000 shield board can be plugged in directly to TS-7800-v2.

jack-lane-fulfil, 2019-06-27:
Hi Adham,

Thanks for the advice -- didn't realize that chip_en and reset_n are physical devices. It turns out that the new version of the linux distro I had didn't have the wifi enabled by default. This fixed it:

```
rmmod wilc_spi
source /sbin/ts7800.subr
enable_wifi
modprobe wilc_spi 
```

I'm getting a different error now, but I'll open up a new thread for that. Thanks for the help!

closed 2019-06-27 by jack-lane-fulfil

AdhamAbozaeid, 2019-06-27:
just to clarify, chip_en and reset_n are pins on the module that can be either derived from the host to power on/off the module, or hardwired with pull up resistors to be always on.