linux4wilc/driver #53: Infinite "failed cmd" sprayed after `dhclient` or `ip link set wlan0 up`

jack-lane-fulfil, 2019-06-27:
Hi!
After resolving that last bug, I have one more. Sometimes after I do `ip link set wlan0 up`, or once I set up a wpa_supplicant, `dhclient`, I get an infinite stream of this error:

```
[  310.538437] WILC_SPI spi0.3: Failed cmd response, cmd (ca), resp (00)        
[  310.545301] WILC_SPI spi0.3: Failed cmd, read reg (003b0000)...              
[  310.551658] WILC_SPI spi0.3: Failed cmd response, cmd (ca), resp (00)        
[  310.558116] WILC_SPI spi0.3: Failed cmd, read reg (003b0000)...              
[  310.566842] WILC_SPI spi0.3: Failed cmd response, cmd (ca), resp (00)        
[  310.573711] WILC_SPI spi0.3: Failed cmd, read reg (003b0000)...              
[  310.579678] WILC_SPI spi0.3: Failed cmd response, cmd (ca), resp (00)        
[  310.586881] WILC_SPI spi0.3: Failed cmd, read reg (003b0000)...              
[  310.595244] WILC_SPI spi0.3: Failed cmd response, cmd (ca), resp (00)  
```

For reference, this is the shell script I'm running. I've commented the two places where the error pops up. Thanks so much!!

```
ssid="MyNetwork"
ntwk_pass="MyPassword"

# bring up the new interface
ip link set wlan0 up;
### ERROR SPRAYED HERE SOMETIMES ####

# check for variable 1 and 2's existence. -n means "not empty string"

#wpa supplicant configuration
config_file="/wifi.config"
wpa_passphrase $ssid $ntwk_pass > $config_file

# set up wifi as a background process -- connects to the router
echo "starting wpa_supplicant..."
wpa_supplicant -D nl80211 -i wlan0 -c $config_file  &

# connect using DHCP
echo "connecting with DHCP...";
dhclient wlan0;

if [ $? == "0" ]; then
	echo "Successfully connected";
        ### ERROR SPRAYED HERE OTHER TIMES #####
else
	echo "DHCP connection failed"
	rm $config_file;
	exit 4;
fi;

# clean up the config file
rm $config_file
```

AdhamAbozaeid, 2019-06-28:
This might be related to powersave.
Try to disable default powersave state. Try recompiling the kernel with CONFIG_CFG80211_DEFAULT_PS disabled

AdhamAbozaeid, 2019-08-08:
@jack-lane-fulfil did disabling powersave have any effect?