linux4wilc/driver #54: modprobe wilc_sdio on RPI3 does nothing

doii-g, 2019-07-02:
Hi everyone!

I'm trying to get the **WILC3000** working on a **Raspberry Pi 3B**.

I successfully cross compiled the kernel with the driver files on my laptop (Debian Stretch) and transferred the new kernel to the SD card.

This is my procedure:

1. Download the toolchain to laptop:
`git clone https://github.com/raspberrypi/tools ~/tools`
1. Adjust PATH variable:
      1. `echo PATH=\$PATH:~/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin >> ~/.bashrc`
      1. `source ~/.bashrc`
1. Get the sources:
`git clone --depth=1 https://github.com/raspberrypi/linux`
1. Install Git and the build dependencies:
`sudo apt-get install git bc bison flex libssl-dev`
1. Raspberry Pi 2, Pi 3, Pi 3+, and Compute Module 3 default build configuration:
      1. `cd linux`
      1. `KERNEL=kernel7`
      1. `make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- bcm2709_defconfig`
1. Replace `drivers/staging/wilc` with the driver from the linux4wilc git repository:
`git clone https://github.com/linux4wilc/driver`
1. Modify the following line in linux/drivers/staging/Makefile:
From: `obj-$(CONFIG_WILC1000) += wilc1000/`
To: `obj-$(CONFIG_WILC) += wilc1000/`
1. Build kernel, modules, and device tree blobs:
`make –j4 ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- zImage modules dtbs`
1. Download and install default Raspbian to a SD card
1. Insert SD card into the Raspberry Pi and update the system
1. Insert SD card into laptop (rootfs partition: `/media/<user>/rootfs`; boot partition: `/media/<user>/boot`)
1. Install the modules
`sudo make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- INSTALL_MOD_PATH=/media/doii/rootfs/ modules_install`
1. Install kernel, modules and device tree blobs:
      1. `sudo cp /media/dominik/boot/$KERNEL.img /media/doii/boot/$KERNEL-backup.img`
      1. `sudo cp arch/arm/boot/zImage /media/doii/boot/$KERNEL.img`
      1. `sudo cp arch/arm/boot/dts/*.dtb /media/doii/boot/`
      1. `sudo cp arch/arm/boot/dts/overlays/*.dtb* /media/doii/boot/overlays/`
      1. `sudo cp arch/arm/boot/dts/overlays/README /media/doii/boot/overlays/`
1. Check if the driver is available:
`modinfo wilc_sdio`

This works pretty well and Raspbian boots up just fine.

Next I'd like to install the driver with `sudo modprobe wilc_sdio`.
With this command I get **`wilc_sdio: module is from the staging directory...`** on dmesg and nothing else happens.

Did I miss something or is something wrong with the driver or the kernel?

AdhamAbozaeid, 2019-07-02:
At this step, the host should probe the WILC's SDIO bus, and initialize the driver, so I the host wasn't able to communicate successfully with WILC.
Make sure that WILC's chip_en and reset_n lines are high, and that the SDIO connections are ok

doii-g, 2019-07-03:
**chip_en** is always high.
**reset_n** is low on default and doesn't change to high when loading the driver with modprobe.

We set the pin for **reset_n** (gpio12) to high via bash:
1. `echo "12" > /sys/class/gpio/export`
1. `echo "out" > /sys/class/gpio/gpio12/direction`
1. `echo "1" > /sys/class/gpio/gpio12/value`

With that we get some output on dmesg, but with some errors:

```
[   62.528627] wilc_sdio: module is from the staging directory, the quality is unknown, you have been warned.
[   84.194922] mmc1: new high speed SDIO card at address 0001
[   84.198490] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
[   84.198515] (unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
[   84.203997] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
[   84.204014] (unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
[   84.205767] wifi_pm : 0
[   84.205794] wilc_sdio mmc1:0001:1: failed to get Reset GPIO, try default
[   84.205804] wilc_sdio mmc1:0001:1: failed to get default Reset GPIO
[   84.205810] wifi_pm : 1
[   84.205819] wilc_sdio mmc1:0001:1: failed to get Reset GPIO, try default
[   84.205828] wilc_sdio mmc1:0001:1: failed to get default Reset GPIO
[   84.206117] wilc_sdio mmc1:0001:1: Driver Initializing success
[   84.423760] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   84.423771] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   84.423832] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   84.423838] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   84.423875] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   84.423880] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   84.423916] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   84.423922] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   84.423959] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   84.423965] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   84.424001] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   84.424006] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   84.424042] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   84.424047] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   84.424083] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   84.424089] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   84.424124] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   84.424130] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   84.424168] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   84.424174] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   84.424211] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   84.424217] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   84.424256] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   84.424262] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   84.424298] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   84.424303] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   84.424339] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   84.424344] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   84.424380] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   84.424385] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   84.424422] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   84.424427] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   84.424463] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   84.424468] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   84.424507] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   84.424513] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   84.424550] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[... cut ...]
[   84.659065] wilc_sdio mmc1:0001:1 p2p0: ERR [wilc_wlan_initialize:1204] WLAN initialization FAILED
[   84.659071] wilc_sdio mmc1:0001:1 p2p0: ERR [wilc_mac_open:1248] Failed to initialize wilc
[   84.659124] wilc_sdio mmc1:0001:1 p2p0: ERR [wilc_send_config_pkt:1903] Set Timed out
[   84.659132] wilc_sdio mmc1:0001:1 p2p0: ERR [handle_set_wfi_drv_handler:433] Failed to set driver handler
```

The error is always
```
wilc_sdio mmc1:0001:1: failed to get IRQ GPIO, load default
wilc_sdio mmc1:0001:1: failed to load default irq
```

AdhamAbozaeid, 2019-07-03:
This part tries to load the IRQ GPIO from the dts, and if it fails, it will try a hardcoded valued from the driver (GPIO_NUM).
SDIO uses muxed interrupts for data transfer, but another external IRQ is also used by WILC to be able to wakeup the host if it went into sleep mode

doii-g, 2019-07-04:
There a lot of dtb files in `linux/arch/arm/boot/dts` and `linux/arch/arm/boot/dts/overlays`. Which one should contain the information for the WILC3000?

Maybe there is something wrong with steps 13.iii and 13.iv in my procedure if a dts file is missing...

> 13. Install kernel, modules and device tree blobs:
>     1. sudo cp /media/dominik/boot/$KERNEL.img /media/doii/boot/$KERNEL-backup.img
>     1. sudo cp arch/arm/boot/zImage /media/doii/boot/$KERNEL.img
>     1. sudo cp arch/arm/boot/dts/*.dtb /media/doii/boot/
>     1. sudo cp arch/arm/boot/dts/overlays/*.dtb* /media/doii/boot/overlays/
>     1. sudo cp arch/arm/boot/dts/overlays/README /media/doii/boot/overlays/

AdhamAbozaeid, 2019-07-08:
This depends on the host and the build system you are using since the dts specifies all devices on the host.
For buildroot, this is in the .config file defined as BR2_LINUX_KERNEL_INTREE_DTS_NAME

doii-g, 2019-07-11:
We are cross compiling the Kernel on a Debian laptop. The .config file in the linux folder doesn't contain BR2_LINUX_KERNEL_INTREE_DTS_NAME.

I just did a `make clean` and made the kernel again, but I still run into the same problem. Do you have some more ideas for me?

AdhamAbozaeid, 2019-07-11:
This is part of the buildroot's .config file, not the linux .config file.
If you are using another build system, this might be different, since the .dts is compiled separately from the linux kernel, producing a .dtb file

AdhamAbozaeid, 2019-08-08:
closing since it's not a WILC issue

closed 2019-08-08 by AdhamAbozaeid