linux4wilc/driver #56: Sometimes frequent ERR level connect/disconnect messages

ChristianHackAMS, 2019-07-10:
On some units we are seeing repeated messages at ERROR level saying connected/disconnect followed by a message about an IP

```
2019-07-10T17:08:05.395+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [cfg_connect_result:205] Received WILC_MAC_STATUS_DISCONNECTED dev [d42dc000]
2019-07-10T17:08:06.415+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [cfg_connect_result:181] Connection Successful: BSSID: ac84c6751fe2
2019-07-10T17:08:26.846+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [clear_during_ip:123] Unable to Obtain IP
2019-07-10T17:13:10.005+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [cfg_connect_result:205] Received WILC_MAC_STATUS_DISCONNECTED dev [d42dc000]
2019-07-10T17:13:11.045+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [cfg_connect_result:181] Connection Successful: BSSID: ac84c6751fe2
2019-07-10T17:13:31.485+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [clear_during_ip:123] Unable to Obtain IP
2019-07-10T17:18:16.625+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [cfg_connect_result:205] Received WILC_MAC_STATUS_DISCONNECTED dev [d42dc000]
2019-07-10T17:18:17.635+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [cfg_connect_result:181] Connection Successful: BSSID: ac84c6751fe2
2019-07-10T17:18:38.045+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [clear_during_ip:123] Unable to Obtain IP
2019-07-10T17:23:21.245+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [cfg_connect_result:205] Received WILC_MAC_STATUS_DISCONNECTED dev [d42dc000]
2019-07-10T17:23:22.275+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [cfg_connect_result:181] Connection Successful: BSSID: ac84c6751fe2
2019-07-10T17:23:42.686+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [clear_during_ip:123] Unable to Obtain IP
2019-07-10T17:28:25.855+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [cfg_connect_result:205] Received WILC_MAC_STATUS_DISCONNECTED dev [d42dc000]
2019-07-10T17:28:26.915+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [cfg_connect_result:181] Connection Successful: BSSID: ac84c6751fe2
2019-07-10T17:28:47.325+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [clear_during_ip:123] Unable to Obtain IP
2019-07-10T17:33:30.465+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [cfg_connect_result:205] Received WILC_MAC_STATUS_DISCONNECTED dev [d42dc000]
2019-07-10T17:33:31.455+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [cfg_connect_result:181] Connection Successful: BSSID: ac84c6751fe2
2019-07-10T17:33:51.965+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [clear_during_ip:123] Unable to Obtain IP
2019-07-10T17:38:35.065+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [cfg_connect_result:205] Received WILC_MAC_STATUS_DISCONNECTED dev [d42dc000]
2019-07-10T17:38:36.095+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [cfg_connect_result:181] Connection Successful: BSSID: ac84c6751fe2
2019-07-10T17:38:56.605+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [clear_during_ip:123] Unable to Obtain IP
2019-07-10T17:48:26.135+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [cfg_connect_result:205] Received WILC_MAC_STATUS_DISCONNECTED dev [d42dc000]
2019-07-10T17:48:27.145+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [cfg_connect_result:181] Connection Successful: BSSID: ac84c6751fe2
2019-07-10T17:48:47.325+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [clear_during_ip:123] Unable to Obtain IP
2019-07-10T17:53:30.745+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [cfg_connect_result:205] Received WILC_MAC_STATUS_DISCONNECTED dev [d42dc000]
2019-07-10T17:53:31.785+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [cfg_connect_result:181] Connection Successful: BSSID: ac84c6751fe2
2019-07-10T17:53:51.965+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [clear_during_ip:123] Unable to Obtain IP
2019-07-10T18:03:01.645+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [cfg_connect_result:205] Received WILC_MAC_STATUS_DISCONNECTED dev [d42dc000]
2019-07-10T18:03:02.665+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [cfg_connect_result:181] Connection Successful: BSSID: ac84c6751fe2
2019-07-10T18:03:22.845+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [clear_during_ip:123] Unable to Obtain IP
2019-07-10T18:08:06.245+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [cfg_connect_result:205] Received WILC_MAC_STATUS_DISCONNECTED dev [d42dc000]
2019-07-10T18:08:07.305+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [cfg_connect_result:181] Connection Successful: BSSID: ac84c6751fe2
2019-07-10T18:08:27.485+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [clear_during_ip:123] Unable to Obtain IP
2019-07-10T18:13:10.855+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [cfg_connect_result:205] Received WILC_MAC_STATUS_DISCONNECTED dev [d42dc000]
2019-07-10T18:13:11.845+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [cfg_connect_result:181] Connection Successful: BSSID: ac84c6751fe2
2019-07-10T18:13:32.125+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [clear_during_ip:123] Unable to Obtain IP
2019-07-10T18:18:15.465+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [cfg_connect_result:205] Received WILC_MAC_STATUS_DISCONNECTED dev [d42dc000]
2019-07-10T18:18:16.485+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [cfg_connect_result:181] Connection Successful: BSSID: ac84c6751fe2
2019-07-10T18:18:36.766+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [clear_during_ip:123] Unable to Obtain IP
2019-07-10T18:23:20.075+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [cfg_connect_result:205] Received WILC_MAC_STATUS_DISCONNECTED dev [d42dc000]
2019-07-10T18:23:21.125+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [cfg_connect_result:181] Connection Successful: BSSID: ac84c6751fe2
2019-07-10T18:23:41.405+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [clear_during_ip:123] Unable to Obtain IP
2019-07-10T18:28:24.685+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [cfg_connect_result:205] Received WILC_MAC_STATUS_DISCONNECTED dev [d42dc000]
2019-07-10T18:28:25.665+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [cfg_connect_result:181] Connection Successful: BSSID: ac84c6751fe2
2019-07-10T18:28:46.045+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [clear_during_ip:123] Unable to Obtain IP
2019-07-10T18:33:29.295+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [cfg_connect_result:205] Received WILC_MAC_STATUS_DISCONNECTED dev [d42dc000]
2019-07-10T18:33:30.305+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [cfg_connect_result:181] Connection Successful: BSSID: ac84c6751fe2
2019-07-10T18:33:50.685+00:00 SM01-9999000003 kernel: wilc_sdio mmc1:0001:1 wlan0: ERR [clear_during_ip:123] Unable to Obtain IP
```

This is with firmware WILC_WIFI_FW_REL_15_2_1 Build: 11257 and driver commit a2ab7b3728afdc4a51cc8fe0423921a71b3f2500

During this time the module doesn't appear to disconnect and seems to keep working fine.

Also why is the driver saying "Unable to Obtain IP"? It shouldn't be aware of the higher network layers.

AdhamAbozaeid, 2019-07-11:
IP is used to delay starting powersave mode.
It looks like you are having several disconnects since directly after that a successful connection is established.
What kind of security is configured on the AP you are trying to connect to?

ChristianHackAMS, 2019-07-14:
IP as in Internet Protocol is used to delay power save? How does that work? A network interface driver really shouldn't have any knowledge of the higher networking layers should it? The unit does have an IP and is otherwise seemingly working.

WPA2-PSK is what we are using. 

Why would this be happening on almost exactly 5 minute boundaries?

Why does it "disconnect" and then immediately "reconnect" 1 second later? Seems like it's not really disconnecting?

Note: not all units do it. We have two at the moment that do it.

AdhamAbozaeid, 2019-07-15:
No. This is a legacy wrokaround to improve the connection time, and the plan is to remove it soon.

If this happens on a fixed period, it might be that the AP is disconnecting the STA for inactivity. It's hard to tell without having the air sniffer and the WILC logs. You can look at the air logs and see where the deauth/disassoc frames are initiated from.
The reconnection is issued from the supplicant layer; it automatically tries to reconnect to any known AP it has on its list that is also visible while scanning.
You can also monitor the supplicant connect/disconnect calls to the driver's connect() and disconnect() APIs in wilc_wfi_cfgoperations.c

AdhamAbozaeid, 2019-08-08:
@ChristianHackAMS , do you have updates on this?

ShonkyCH, 2019-08-12:
It still happens. At the moment, I don't have time to debug your code. We've already spent many many hours debugging this thing for you.

AdhamAbozaeid, 2019-08-12:
Since I can't reproduce this issue on my side, I can't provide more help here.
Please open a ticket on salesforce

closed 2019-08-12 by AdhamAbozaeid

ShonkyCH, 2019-08-12:
Great. Salesforce ticket opened #00449801