linux4wilc/driver #58: SPI buffer overrun reported initialization error

ChevalierDeBalibari, 2019-07-24:
Hello,

I are using WILC3000 on our internal product based on SAMA5D27-SOM1. I have both variants (SPI and SDIO) but both boards are making problems. 

Device tree binding:

```
spi0: spi@f8000000 {
pinctrl-names = "default";
pinctrl-0 = <&pinctrl_spi0_default>;
cs-gpios = <&pioA PIN_PA17 GPIO_ACTIVE_LOW>;
status = "okay";                          
    wilc_spi@0 {                                                                                                                                             
    reg = <0>;
    bus-width = <1>;  
    compatible = "microchip,wilc1000", "microchip,wilc3000";
    spi-max-frequency = <48000000>;
    irq-gpios = <&pioA PIN_PA24 GPIO_ACTIVE_HIGH>;
    chip_en-gpios = <&pioA PIN_PA25 GPIO_ACTIVE_HIGH>;
    reset-gpios = <&pioA PIN_PA26 GPIO_ACTIVE_HIGH>;
    pinctrl-names = "default";
    pinctrl-0 = <&pinctrl_wilc3000_default>;
    status = "okay"; /* Conflict with wilc-sdio */
}; 
``` 
pinctrl_wilc3000_default: wilc_default {
    irq {
        pinmux = <PIN_PA24__GPIO>; /* WILC3000 IRQ */
        bias-pull-up;
    };
    rst_en {
        pinmux = <PIN_PA26__GPIO>, /* WILC300 CHIP RST */
                        <PIN_PA25__GPIO>; /* WILC300j CHIP EN */
        bias-pull-down;
    };                                                                                                                                               
};

pinctrl_spi0_default: spi0_default {
    pinmux = <PIN_PA14__SPI0_SPCK>,
                   <PIN_PA15__SPI0_MOSI>,
                   <PIN_PA16__SPI0_MISO>,
                   <PIN_PA17__SPI0_NPCS0>;
    bias-disable;                                                                                                                                    
};

Relevant kernel log messages:

```
[    1.940000] WILC_SPI spi0.0: spiModalias: wilc1000, spiMax-Speed: 48000000
[    1.960000] WILC_SPI spi0.0: succesfully got gpio_reset
[    1.960000] WILC_SPI spi0.0: succesfully got gpio_chip_en
[    1.970000] WILC_SPI spi0.0: succesfully got gpio_reset
[    1.970000] WILC_SPI spi0.0: succesfully got gpio_chip_en
[    1.980000] WILC_SPI spi0.0: WILC SPI probe success
[   18.420000] WILC_SPI spi0.0: buffer overrun when reading data.
[   18.420000] WILC_SPI spi0.0: Failed cmd, read reg (003b0000)...
[   18.430000] WILC_SPI spi0.0: Reset and retry 10 3b0000
[   18.440000] WILC_SPI spi0.0: buffer overrun when reading data.
[   18.440000] WILC_SPI spi0.0: Failed cmd, read reg (003b0000)...
[   18.470000] WILC_SPI spi0.0: Reset and retry 9 3b0000
[   18.470000] WILC_SPI spi0.0: buffer overrun when reading data.
[   18.470000] WILC_SPI spi0.0: Failed cmd, read reg (003b0000)...
[   18.510000] WILC_SPI spi0.0: Reset and retry 8 3b0000
[   18.510000] WILC_SPI spi0.0: buffer overrun when reading data.
[   18.520000] WILC_SPI spi0.0: Failed cmd, read reg (003b0000)...
[   18.550000] WILC_SPI spi0.0: Reset and retry 7 3b0000
[   18.550000] WILC_SPI spi0.0: buffer overrun when reading data.
[   18.560000] WILC_SPI spi0.0: Failed cmd, read reg (003b0000)...
[   18.590000] WILC_SPI spi0.0: Reset and retry 6 3b0000
[   18.620000] WILC_SPI spi0.0: buffer overrun when reading data.
[   18.620000] WILC_SPI spi0.0: Failed cmd, read reg (003b0000)...
[   18.630000] WILC_SPI spi0.0: Reset and retry 5 3b0000
[   18.640000] WILC_SPI spi0.0: buffer overrun when reading data.
[   18.640000] WILC_SPI spi0.0: Failed cmd, read reg (003b0000)...
[   18.670000] WILC_SPI spi0.0: Reset and retry 4 3b0000
[   18.670000] WILC_SPI spi0.0: buffer overrun when reading data.
[   18.670000] WILC_SPI spi0.0: Failed cmd, read reg (003b0000)...
[   18.710000] WILC_SPI spi0.0: Reset and retry 3 3b0000
[   18.710000] WILC_SPI spi0.0: buffer overrun when reading data.
[   18.720000] WILC_SPI spi0.0: Failed cmd, read reg (003b0000)...
[   18.750000] WILC_SPI spi0.0: Reset and retry 2 3b0000
[   18.750000] WILC_SPI spi0.0: buffer overrun when reading data.
[   18.760000] WILC_SPI spi0.0: Failed cmd, read reg (003b0000)...
[   18.790000] WILC_SPI spi0.0: Reset and retry 1 3b0000
[   18.830000] [wilc start]: fail read reg 0x3b0000
[   18.830000] WILC1002 isn't suported 0
[   18.830000] WILC_SPI spi0.0: Unsupported chipid: 0
[   18.860000] Bluetooth firmware: mchp/wilc3000_ble_firmware.bin
[   20.620000] WILC_SPI spi0.0: buffer overrun when reading data.
[   20.620000] WILC_SPI spi0.0: Failed cmd, read reg (003b0090)...
[   20.660000] WILC_SPI spi0.0: Reset and retry 10 3b0090
[   20.660000] WILC_SPI spi0.0: buffer overrun when reading data.
[   20.660000] WILC_SPI spi0.0: Failed cmd, read reg (003b0090)...
[   20.710000] WILC_SPI spi0.0: Reset and retry 9 3b0090
[   20.710000] WILC_SPI spi0.0: buffer overrun when reading data.
[   20.720000] WILC_SPI spi0.0: Failed cmd, read reg (003b0090)...
[   20.760000] WILC_SPI spi0.0: Reset and retry 8 3b0090
[   20.760000] WILC_SPI spi0.0: buffer overrun when reading data.
[   20.770000] WILC_SPI spi0.0: Failed cmd, read reg (003b0090)...
[   20.810000] WILC_SPI spi0.0: Reset and retry 7 3b0090
[   20.840000] WILC_SPI spi0.0: buffer overrun when reading data.
[   20.840000] WILC_SPI spi0.0: Failed cmd, read reg (003b0090)...
[   20.890000] WILC_SPI spi0.0: Reset and retry 6 3b0090
[   20.940000] WILC_SPI spi0.0: buffer overrun when reading data.
[   20.940000] WILC_SPI spi0.0: Failed cmd, read reg (003b0090)...
[   20.990000] WILC_SPI spi0.0: Reset and retry 5 3b0090
[   20.990000] WILC_SPI spi0.0: buffer overrun when reading data.
[   21.000000] WILC_SPI spi0.0: Failed cmd, read reg (003b0090)...
[   21.040000] WILC_SPI spi0.0: Reset and retry 4 3b0090
[   21.070000] WILC_SPI spi0.0: buffer overrun when reading data.
[   21.070000] WILC_SPI spi0.0: Failed cmd, read reg (003b0090)...
[   21.080000] WILC_SPI spi0.0: Reset and retry 3 3b0090
[   21.150000] WILC_SPI spi0.0: buffer overrun when reading data.
[   21.150000] WILC_SPI spi0.0: Failed cmd, read reg (003b0090)...
[   21.160000] WILC_SPI spi0.0: Reset and retry 2 3b0090
[   21.160000] WILC_SPI spi0.0: buffer overrun when reading data.
[   21.170000] WILC_SPI spi0.0: Failed cmd, read reg (003b0090)...
[   21.210000] WILC_SPI spi0.0: Reset and retry 1 3b0090
[   21.210000] [wilc start]: fail read reg 0x3b0090 ...
[   22.120000] WILC_SPI spi0.0: buffer overrun when reading data.
[   22.120000] WILC_SPI spi0.0: Failed cmd, read reg (003b0400)...
[   22.130000] WILC_SPI spi0.0: Reset and retry 10 3b0400
[   22.140000] WILC_SPI spi0.0: buffer overrun when reading data.
[   22.140000] WILC_SPI spi0.0: Failed cmd, read reg (003b0400)...
[   22.190000] WILC_SPI spi0.0: Reset and retry 9 3b0400
[   22.190000] WILC_SPI spi0.0: buffer overrun when reading data.
[   22.190000] WILC_SPI spi0.0: Failed cmd, read reg (003b0400)...
[   22.250000] WILC_SPI spi0.0: Reset and retry 8 3b0400
[   22.250000] WILC_SPI spi0.0: buffer overrun when reading data.
[   22.250000] WILC_SPI spi0.0: Failed cmd, read reg (003b0400)...
[   22.300000] WILC_SPI spi0.0: Reset and retry 7 3b0400
[   22.300000] WILC_SPI spi0.0: buffer overrun when reading data.
[   22.310000] WILC_SPI spi0.0: Failed cmd, read reg (003b0400)...
[   22.350000] WILC_SPI spi0.0: Reset and retry 6 3b0400
[   22.360000] WILC_SPI spi0.0: buffer overrun when reading data.
[   22.360000] WILC_SPI spi0.0: Failed cmd, read reg (003b0400)...
[   22.380000] WILC_SPI spi0.0: Reset and retry 5 3b0400
[   22.400000] WILC_SPI spi0.0: buffer overrun when reading data.
[   22.410000] WILC_SPI spi0.0: Failed cmd, read reg (003b0400)...
[   22.430000] WILC_SPI spi0.0: Reset and retry 4 3b0400
[   22.440000] WILC_SPI spi0.0: buffer overrun when reading data.
[   22.440000] WILC_SPI spi0.0: Failed cmd, read reg (003b0400)...
[   22.460000] WILC_SPI spi0.0: Reset and retry 3 3b0400
[   22.470000] WILC_SPI spi0.0: buffer overrun when reading data.
[   22.470000] WILC_SPI spi0.0: Failed cmd, read reg (003b0400)...
[   22.500000] WILC_SPI spi0.0: Reset and retry 2 3b0400
[   22.510000] WILC_SPI spi0.0: buffer overrun when reading data.
[   22.510000] WILC_SPI spi0.0: Failed cmd, read reg (003b0400)...
[   22.530000] WILC_SPI spi0.0: Reset and retry 1 3b0400
```

AdhamAbozaeid, 2019-07-24:
Looks like all the SPI communication is not successful. I see a log that I suspect is thrown from the SPI driver on your host:
` buffer overrun when reading data`
Do you have more information about this error?

ChevalierDeBalibari, 2019-07-25:
buffer overrun when reading data message is can be found on line 585 of wilc_spi.c file.

AdhamAbozaeid, 2019-07-25:
yes you are right.. I overlooked this error.
It looks like the first part of the response coming from WILC is correct, since the 1st byte of the response should be the same as the command sent from the host, but not sure why it fails the length check on your side.
Please open a sales force ticket for this issue and the support team will take it from there

Mateusz-Gwara, 2019-10-24:
This looks similar to issue #53

ChevalierDeBalibari, 2019-11-07:
I fixed this with slightly different device tree configuration:

spi0: spi@f8000000 {
pinctrl-names = "default";
pinctrl-0 = <&pinctrl_spi0_default>;
cs-gpios = <&pioA PIN_PA17 GPIO_ACTIVE_LOW>;
status = "okay";                          
    wilc_spi@0 {                                                                                                                                             
    reg = <0>;
    bus-width = <1>;  
    compatible = "microchip,wilc1000", "microchip,wilc3000";
    spi-max-frequency = <24000000>;
    irq-gpios = <&pioA PIN_PA24 GPIO_ACTIVE_HIGH>;
    chip_en-gpios = <&pioA PIN_PA25 GPIO_ACTIVE_HIGH>;
    reset-gpios = <&pioA PIN_PA26 GPIO_ACTIVE_HIGH>;
    pinctrl-names = "default";
    pinctrl-0 = <&pinctrl_wilc3000_default>;
    status = "okay"; /* Conflict with wilc-sdio */
}; 

pinctrl_wilc3000_default: wilc_default {
    irq {
        pinmux = <PIN_PA24__GPIO>; /* WILC3000 IRQ */
        bias-pull-up;
    };
    rst_en {
        pinmux = <PIN_PA26__GPIO>, /* WILC3000 CHIP RST */
                        <PIN_PA25__GPIO>; /* WILC3000 CHIP EN */
        bias-pull-up;
    };                                                                                                                                               
};

pinctrl_spi0_default: spi0_default {
    pinmux = <PIN_PA14__SPI0_SPCK>,
                   <PIN_PA15__SPI0_MOSI>,
                   <PIN_PA16__SPI0_MISO>,
                   <PIN_PA17__SPI0_NPCS0>;
    bias-disable;                                                                                                                                    
};

AdhamAbozaeid, 2019-11-07:
Thanks for the update. I see the main change was the reduction of SPI clk to 24Mhz, which might point to a problem with noise on the SPI lines.
Can you give me more information on the setup you have? Are you using flywires to connect the host to WILC, or WILC MCHP board plugged into your SOM, a custom design?

closed 2019-12-04 by AdhamAbozaeid