linux4wilc/driver #59: WILC3000-SDIO initialization error

ChevalierDeBalibari, 2019-07-24:
Similar as in previous issue, wilc3000 connected to sdio0 interface of sama5d27-som1 based board... 

Device tree binding:

```
sdmmc0: sdio-host@a0000000 {
       //bus-width = <8>;
       max-frequency = <24000000>;
       pinctrl-names = "default";
       pinctrl-0 = <&pinctrl_sdmmc0_default>;
       non-removable;
       status = "okay";
                           
        wilc_sdio@0 {
             compatible = "microchip,wilc3000";
             irq-gpios = <&pioA PIN_PA24 GPIO_ACTIVE_HIGH>;
             chip_en-gpios = <&pioA PIN_PA25 GPIO_ACTIVE_HIGH>;
             reset-gpios = <&pioA PIN_PA26 GPIO_ACTIVE_LOW>;
             pinctrl-names = "default";
             pinctrl-0 = <&pinctrl_wilc3000_default>;
             status = "okay"; /* Conflict with wilc-spi */
             bus-width = <4>;
             reg = <0>;
        };                                                                                                                                                               
                          
};

pinctrl_wilc3000_default: wilc_default {
    irq {
        pinmux = <PIN_PA24__GPIO>; /* WILC3000 IRQ */
        bias-pull-up;
    };
    rst_en {
        pinmux = <PIN_PA26__GPIO>, /* WILC300 CHIP RST */
                        <PIN_PA25__GPIO>; /* WILC300j CHIP EN */
        bias-pull-up;
       //pull down is recommended in datasheet but causes mmc initialization error -110
       //bias-pull-down;
    };                                                                                                                                               
};

pinctrl_sdmmc0_default: sdmmc0_default {
       cmd_data {
           pinmux = <PIN_PA1__SDMMC0_CMD>,
                          <PIN_PA2__SDMMC0_DAT0>,
                          <PIN_PA3__SDMMC0_DAT1>,
                          <PIN_PA4__SDMMC0_DAT2>,
                          <PIN_PA5__SDMMC0_DAT3>;
          bias-disable;
      };
     conf_ck_cd {
         pinmux = <PIN_PA0__SDMMC0_CK>,
                        <PIN_PA13__SDMMC0_CD>;
        bias-disable;
    };
};

``` 

After system startup so far so good:

```
[    1.830000] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
[    1.840000] (unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
[    1.850000] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
[    1.860000] (unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
[    1.870000] wilc_sdio mmc0:0001:1: succesfully got gpio_reset
[    1.870000] wilc_sdio mmc0:0001:1: succesfully got gpio_chip_en
[    1.880000] wilc_sdio mmc0:0001:1: succesfully got gpio_reset
[    1.880000] wilc_sdio mmc0:0001:1: succesfully got gpio_chip_en
[... 87 lines cut ...]
[  654.030000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 0
[  654.040000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  654.050000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 0
[  654.060000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  654.070000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 0
[  654.080000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  654.090000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 0
[  654.090000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  654.100000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 0
[  654.110000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  654.130000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[c7a88000] wlan0
[  654.140000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_init_host_int]Host[c7a88000][c7a80800]
[  654.150000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
[  654.160000] wilc_sdio mmc0:0001:1 wlan0: INFO [wlan_init_locks]Initializing Locks ...
[  654.160000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
[  654.170000] wilc_sdio mmc0:0001:1: SDIO speed: 24000000
[  654.220000] wilc_sdio mmc0:0001:1: wilc_sdio_cmd52..failed, err(-110)
[  654.220000] wilc_sdio mmc0:0001:1: Fail cmd 52, enable csa...
[  654.220000] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_wlan_initialize:1123] Initializing WILC_Wlan FAILED
[  654.230000] wilc_sdio mmc0:0001:1 wlan0: INFO [wlan_deinit_locks]De-Initializing Locks
[  654.240000] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_wlan_initialize:1204] WLAN initialization FAILED
[  654.250000] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_mac_open:1248] Failed to initialize wilc
[  654.250000] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_send_config_pkt:1903] Set Timed out
[  654.260000] wilc_sdio mmc0:0001:1 wlan0: ERR [handle_set_wfi_drv_handler:433] Failed to set driver handler
```

ChevalierDeBalibari, 2019-07-24:
Update:

There is a workaround. Device seem to be working if I comment out 
```
reset-gpios = <&pioA PIN_PA26 GPIO_ACTIVE_LOW>;
```

AdhamAbozaeid, 2019-07-24:
Can you also try changing rest=gpios to ACTIVE_HIGH?

ChevalierDeBalibari, 2019-07-25:
Tried that, same output:

```
[   90.900000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   90.910000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   90.930000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   90.940000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   90.950000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   90.960000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   90.970000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   90.980000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   90.990000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   91.000000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   91.010000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   91.020000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   91.030000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   91.040000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   91.050000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   91.060000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   91.080000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   91.080000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   91.100000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   91.100000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   91.120000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   91.130000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   91.140000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   91.150000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   91.160000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   91.170000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   91.180000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   91.190000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   91.200000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   91.210000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   91.220000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   91.230000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   91.240000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Frame registering Frame Type: d0: Boolean: 1
[   91.250000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   91.260000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[c7a88000] wlan0
[   91.260000] WILC POWER UP
[   91.260000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_init_host_int]Host[c7a88000][c7a80800]
[   91.270000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
[   91.280000] wilc_sdio mmc0:0001:1 wlan0: INFO [wlan_init_locks]Initializing Locks ...
[   91.280000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
[   91.290000] wilc_sdio mmc0:0001:1: SDIO speed: 24000000
[   91.330000] wilc_sdio mmc0:0001:1: wilc_sdio_cmd52..failed, err(-110)
[   91.330000] wilc_sdio mmc0:0001:1: Fail cmd 52, enable csa...                                                                                                                             
[   91.340000] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_wlan_initialize:1123] Initializing WILC_Wlan FAILED                                                                                    
[   91.340000] wilc_sdio mmc0:0001:1 wlan0: INFO [wlan_deinit_locks]De-Initializing Locks
[   91.350000] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_wlan_initialize:1204] WLAN initialization FAILED
[   91.360000] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_mac_open:1248] Failed to initialize wilc                                                                                               
[   91.370000] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_send_config_pkt:1903] Set Timed out                                                                                                    
[... cut ...]
[   91.820000] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_wlan_initialize:1204] WLAN initialization FAILED
[   91.830000] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_mac_open:1248] Failed to initialize wilc                                                                                               
[   91.830000] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_send_config_pkt:1903] Set Timed out                                                                                                    
[   91.840000] wilc_sdio mmc0:0001:1 wlan0: ERR [handle_set_wfi_drv_handler:433] Failed to set driver handler
```

AdhamAbozaeid, 2019-10-17:
If you put the GPIOs low before inserting the kernel module, the SDIO probe function in the driver will not be called since the SDIO device (WILC) is powered down and won't respond to SDIO probes, and hence, the driver won't initialize since the probe function is the initialization point for the driver.

How's the WILC3000 SDIO board connected to the SOM? I'm suspecting it might be a problem with the SDIO Dat lines since in your original logs it seems like it's probing correctly, but fails afterwards.

ChevalierDeBalibari, 2019-10-20:
Weare using module, not  WILC3000 SDIO board. Module is connected to mmc0 interface directly and we also have EN, INT and RESETN lines connected to the PA24, PA25 and PA26. Moreover, EN and RESETN are pulled down through 1M resistors.

AdhamAbozaeid, 2019-10-21:
Please report on salesforce http://microchipsupport.force.com/
The team will help you debug the bus lines