linux4wilc/driver #60: WILC3000-SDIO Chip ID Detection Fail

prashanthkumarkk, 2019-07-29:
Hello,

We are working on iMX8 Quad Max custom board with Kernel BSP version **4.14.98**. But we are facing WILC3000 WIFI Chip ID detection issue.

Device tree binding:

```
&usdhc3 { /* USDHC3: SDIO WILC3000 WIFI */
        pinctrl-names = "default","state_100mhz", "state_200mhz";
        pinctrl-0 = <&pinctrl_usdhc3>, <&pinctrl_usdhc3_gpio>;
        pinctrl-1 = <&pinctrl_usdhc3_100mhz>, <&pinctrl_usdhc3_gpio>;
        pinctrl-2 = <&pinctrl_usdhc3_200mhz>, <&pinctrl_usdhc3_gpio>;
        vmmc-supply = <&reg_3p3v>;
        bus-width = <4>;
        no-1-8-v;
        non-removable;
        wifi-host;
        status = "okay";
        wilc_sdio@0 {
                compatible = "microchip,wilc3000";
                irq-gpios = <&gpio2 28 GPIO_ACTIVE_LOW>;
                reset-gpios = <&gpio2 29 GPIO_ACTIVE_LOW>;
                chip_en-gpios = <&gpio2 30 GPIO_ACTIVE_LOW>;
                status = "okay";
        };
};
```
After system startup driver initialization is good:

wilc1000_sdio mmc2:0001:1: Driver Initializing success

prashanthkumarkk, 2019-07-29:
Continuing from above...

After system startup driver initialization is good:

`wilc1000_sdio mmc2:0001:1: Driver Initializing success`

But if i try **ifconfig wlan0 up** i get:

```
~# ifconfig wlan0 up
 wilc1000_sdio mmc2:0001:1: chipid (00802221)
wilc1000_sdio mmc2:0001:1: has_thrpt_enh3 = 0...
wilc1000_sdio mmc2:0001:1: wilc_sdio_cmd52..failed, err(-110)
wilc1000_sdio mmc2:0001:1: Failed cmd52, set 0x10c data...
wilc1000_sdio mmc2:0001:1: wilc_sdio_cmd52..failed, err(-110)
wilc1000_sdio mmc2:0001:1: Failed cmd52, set 0x10c data...
wilc1000_sdio mmc2:0001:1: wilc_sdio_cmd52..failed, err(-110)
wilc1000_sdio mmc2:0001:1: Failed cmd52, set 0x10c data...
wilc1000_sdio mmc2:0001:1 wlan0: fail read reg 0x1118
wilc1000_sdio mmc2:0001:1 wlan0: WLAN Iinitialization FAILED
wilc1000_sdio mmc2:0001:1 wlan0: Failed to set driver handler
wilc1000_sdio mmc2:0001:1 wlan0: Host cmd 24 failed
wilc1000_sdio mmc2:0001:1 wlan0: [Host Interface] undefined
SIOCSIFFLAGS: Input/output error
```
The WILC3000 chip Id should detect as **003000d0**
But I'm getting a garbage chip ID value as **00802221** in which chip is not waking up.

The same WILC3000 driver with same DTS configuration is working fine in **4.9.51 Kernel BSP**.
The below prints are in 4.9.51 Kernel.

```
~# ifconfig wlan0 up
 wilc1000_sdio mmc2:0001:1 wlan0: Detect chip WILC3000
 wilc1000_sdio mmc2:0001:1 wlan0: loading firmware wilc3000_wifi_firmware.bin
 wilc1000_sdio mmc2:0001:1 wlan0: WILC Firmware Ver = WILC_WIFI_FW_REL_14_04 Build: 8556
```
We checked Reset & CHIP_EN  GPIOs, both are pulled up as spec.

Kindly help resolve above issue regarding chip ID detection issue in 4.14.98 Kernel BSP.

AdhamAbozaeid, 2019-07-29:
Are you using the same hardware in both cases?
If not, then the fact that the SDIO probes successfully but doesn't read the chip ID might point to a problem with the DAT lines.
If this is the exact same HW in both cases, then it might be a difference in the SDIO driver on 4.14. Maybe you can probe the SDIO lines with a scope and compare the plot in the successful/failed cases.
You can also open a salesforce ticket if you need further assistance with debugging

saravanakumar-sundaram, 2019-07-30:
Could you remove or comment the "non-removable" in dts file and try to bring up the interface.

prashanthkumarkk, 2019-07-30:
Hi,
I tried the same thing by removing "non-removable" property in dts file, but still getting same issue.

prashanthkumarkk, 2019-07-30:
We are using same hardware in both the case, only Kernel version is different.