linux4wilc/driver #61: wilc1000 fconnect to the access point

asaillard, 2019-08-05:
Hello,

I am working on Intel socfpga board with kernel 4.9 with wilc1000 using SDIO.
Sometimes, the driver fails to connect to the AP until I reboot.
I reproduced with 15.2 and the last dev branch.

log shows the following WARN:
> <4>[   20.461382] ------------[ cut here ]------------
> <4>[   20.516800] WARNING: CPU: 0 PID: 426 at /home/s03398/Workdir/OS_Nova-kernel-Rocko/src_yocto/build/tmp/work-shared/nova/kernel-source/net/wireless/sme.c:733 __cfg80211_connect_result+0x200/0x2a0
> <4>[   20.723340] CPU: 0 PID: 426 Comm: kworker/u2:2 Not tainted 4.9.76-rt61-ltsi-yocto-standard #1
> <4>[   20.825570] Hardware name: Altera SOCFPGA
> <4>[   20.873632] Workqueue: cfg80211 cfg80211_event_work
> <4>[   20.932557] Backtrace:
> <4>[   20.962103] [<8010a938>] (dump_backtrace) from [<8010ab84>] (show_stack+0x18/0x1c)
> <4>[   21.052862]  r7:00000009 r6:00000000 r5:805fa84e r4:00000000
> <4>[   21.120817] [<8010ab6c>] (show_stack) from [<80284ee0>] (dump_stack+0x20/0x28)
> <4>[   21.207641] [<80284ec0>] (dump_stack) from [<801184bc>] (__warn+0xc8/0xf8)
> <4>[   21.290348] [<801183f4>] (__warn) from [<801185a4>] (warn_slowpath_null+0x28/0x30)
> <4>[   21.381111]  r9:87389780 r8:87389734 r7:8659de0c r6:00000000 r5:00000000 r4:87389700
> <4>[   21.473960] [<8011857c>] (warn_slowpath_null) from [<804ed5d4>] (__cfg80211_connect_result+0x200/0x2a0)
> <4>[   21.586616] [<804ed3d4>] (__cfg80211_connect_result) from [<804cf5a8>] (cfg80211_process_wdev_events+0xdc/0x188)
> <4>[   21.709017]  r7:8659de0c r6:87389788 r5:87389700 r4:8659de00
> <4>[   21.776872] [<804cf4cc>] (cfg80211_process_wdev_events) from [<804cf6a8>] (cfg80211_process_rdev_events+0x54/0x6c)
> <4>[   21.901222]  r10:00000088 r9:00000000 r8:878f5900 r7:00000000 r6:87802000 r5:87330438
> <4>[   21.995082]  r4:87389700
> <4>[   22.025720] [<804cf654>] (cfg80211_process_rdev_events) from [<804cbbac>] (cfg80211_event_work+0x1c/0x24)
> <4>[   22.140517]  r5:87330120 r4:87330120
> <4>[   22.183354] [<804cbb90>] (cfg80211_event_work) from [<8012c080>] (process_one_work+0x1b4/0x2e4)
> <4>[   22.287663]  r5:87330120 r4:865e6d00
> <4>[   22.330520] [<8012becc>] (process_one_work) from [<8012c4ec>] (worker_thread+0x304/0x464)
> <4>[   22.428737]  r10:00000088 r9:865e6d18 r8:87802028 r7:8070ff00 r6:87802000 r5:87802000
> <4>[   22.522604]  r4:865e6d00
> <4>[   22.552945] [<8012c1e8>] (worker_thread) from [<801314d4>] (kthread+0xd4/0xec)
> <4>[   22.639555]  r10:00000000 r9:00000000 r8:00000000 r7:8012c1e8 r6:865e6d00 r5:00000000
> <4>[   22.745269]  r4:865ba9c0 r3:00000000
> <4>[   22.794556] [<80131400>] (kthread) from [<80107198>] (ret_from_fork+0x14/0x3c)
> <4>[   22.991489]  r7:00000000 r6:00000000 r5:80131400 r4:865ba9c0
> <4>[   23.161470] ---[ end trace 0000000000000002 ]---
 
full log attached:
[connect_result_WARN.txt](https://github.com/linux4wilc/driver/files/3467863/connect_result_WARN.txt)

asaillard, 2019-09-19:
I managed to get further in my investigation.
	
In a fast connect/disconnect sequence, cfg80211_connect_result() can fail to find the bss object which the driver is connecting to, the function exits without completing and logs the warning.
When this occurs, if the user space re-tries to connect with same bssid, it will always fail.

Detailed sequence of events on 4.9.x kernels
	
  * Connect request is initiated from user space. cfg80211_connect() stores
  ssid/ssid_len and calls rdev_connect()
  
  * wilc driver initiates the connection and sets vif->bssid
    
  * Connect succeeds. Driver calls cfg80211_connect_result() which adds new event to the
  cfg80211_wq worker thread
  
  * Disconnect request arrives from user space. cfg80211_disconnected() sets ssid_len to zero (does not call rdev_disconnect() while current_bss != NULL )
  
  * __cfg80211_connect_result() runs in worker thread which fails to find the bss because wdev->ssid_len is zero (logs the warning)
  
  * wilc driver refuses new connection requests to the same bss, because vif->bssid equals bss->bssid (returns -ALREADY).

on kernel > 4.9, since commit https://github.com/torvalds/linux/commit/0711d638786941ec02551dd9b4aa0d8341f7db5b, the warning is still logged, but at 4th step cfg80211_disconnected() calls rdev_disconnect.
the driver disconnect() clears vif->bssid, so the wilc driver will accept the connection retry on the same bss.

To overcome this warning, and the connection issue, the driver could store the bss object and upon connect completion pass it to kernel using cfg80211_connect_bss().
I take for exemple what has been done for another wireless driver in the commit https://github.com/torvalds/linux/commit/bcdd49b074d0ea48f54bc6d24a66efab8688b43a.

AdhamAbozaeid, 2019-09-19:
Thanks @asaillard for the update. I'll have a look and let you know if we can apply this to the WILC's driver as well

AdhamAbozaeid, 2019-10-12:
Hi @asaillard 
I'm getting back to this issue, and I'd like to thank you for the clear and detailed analysis.
To confirm the fix, I'm trying to reproduce the same issue on my setup before & after applying it.
Can you give me more details on how to reproduce?
I tried to reproduce it by calling enable_network/disable_network back to back, but couldn't see the warning.
`wpa_cli enable_network 0
wpa_cli disable_network 0`

asaillard, 2019-10-14:
Hi @AdhamAbozaeid,

I managed to reproduced it by adding a small delay between enable and disable_network:
wpa_cli enable_net 0 && sleep 1 && wpa_cli disable_net 0

AdhamAbozaeid, 2019-10-14:
Hi @asaillard 

I tested again with linux 4.9 on SAMA5d4 board, and still couldn't see it. How consistent is the error on your side? I tested around 10~15 times.
Can you try that again with the latest dev barnch?

`# wpa_cli enable_net 0 && sleep 1 && wpa_cli disable_net 0
Selected interface willc0_sdio mmc0:0001:1 wlan0: INFO [scan]Requested num of channel 14
wilc_sdio mmc0:0001:1 wlan0: INFO [scan]Scan Request IE len =  126
wilc_sdio mmc0:0001:1 wlan0: INFO [scan]Number of SSIDs 1
wilc_sdio mmc0:0001:1 wlan0: INFO [scan]Trigger Scan Request

OK
cfg_indicate_rx: Scan Notification Received
wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_scan_complete_received]Scan notification received
wilc_sdio mmc0:0001:1 wlan0: INFO [cfg_scan_result]Scan Done[d46a4000]
wlan0: Trying to associate with fa:f0:05:f1:3d:64 (SSID='WILC_SSID1' freq=2422 MHz)
Selected interface 'wlan0'
wilc_sdio mmc0:0001:1 wlan0: INFO [connect]Connecting to SSID [WILC_SSID1] on netdev [d46a4000] host if [d462be00]
wilc_sdio mmc0:0001:1 wlan0: INFO [connect]Adding key with cipher group fac02
wilc_sdio mmc0:0001:1 wlan0: INFO [connect]Authentication Type = 0
wilc_sdio mmc0:0001:1 wlan0: INFO [connect]In OPEN SYSTEM
wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_set_bssid]set bssid [fa:f0:05:f1:3d:64]
wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_send_connect_wid]send HOST_IF_WAITING_CONN_RESP
wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_send_connect_wid]set HOST_IF_WAITING_CONN_RESP
wlan0: CTRL-EVENT-DISCONNECTED bssid=fa:f0:05:f1:3d:64 reason=3 locally_generated=1
OK
# cfg_indicate_rx: Info message received
wilc_sdio mmc0:0001:1 wlan0: INFO [handle_rcvd_gnrl_async_info]Current State = 3,Received state = 1
wilc_sdio mmc0:0001:1 wlan0: INFO [cfg_connect_result]Connection response received=1 connect_stat[0]
wilc_sdio mmc0:0001:1 wlan0: INFO [cfg_connect_result]Connection Successful: BSSID: faf05f13d64
wilc_sdio mmc0:0001:1 wlan0: INFO [cfg_connect_result]Association request info elements length = 27
wilc_sdio mmc0:0001:1 wlan0: INFO [cfg_connect_result]Association response info elements length = 68
wlanwilco_sedioo  mmnc0g:0a00o1:f1 nwlfan 0:e IuNFeO  [wilc_wlan_set_bssid]set bssid [00:00:00:00:00:00]
wilc_sdio mmc0:0001:1 wlan0: INFO [disconnect]Disconnecting with reason code(3)

cfg_indicate_rx: Info message received
wilc_sdio mmc0:0001:1 wlan0: INFO [handle_rcvd_gnrl_async_info]Current State = 4,Received state = 0
wilc_sdio mmc0:0001:1 wlan0: INFO [cfg_connect_result]Received WILC_MAC_STATUS_DISCONNECTED dev [d46a4000]
wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_set_bssid]set bssid [00:00:00:00:00:00]
wlan0: CTRL-EVENT-DISCONNECTED bssid=fa:f0:05:f1:3d:64 reason=3 locally_generated=1
wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
nl80211: Was expecting local disconnect but got another disconnect event first
wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
`

asaillard, 2019-10-15:
Hi @AdhamAbozaeid 

On my board it is systematic. 
I can test your fix for you, if you want.
I reproduced the issue with the last dev branch:

> > [   42.866470] wilc_sdio mmc0:0001:1 wlan0: INFO [scan]Trigger Scan Request
<3>CTRL-EVENT-SCAN-STARTED
[   43.503973] cfg_indicate_rx: Scan Notification Received
[   43.509195] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_scan_complete_received]Scan notification received
<3>CTRL-EVENT-SCAN-RESULTS
<3>WPS-AP-AVAILABLE
<3>Trying to associate with 28:c6:8e:76:7e:9b (SSID='NETGEAR14' freq=2442 MHz)
[   43.934567] wilc_sdio mmc0:0001:1 wlan0: INFO [connect]Connecting to SSID [NETGEAR14] on netdev [87330000] host if [86d52e00]
[   43.945845] wilc_sdio mmc0:0001:1 wlan0: INFO [connect]Adding key with cipher group fac04
[   43.953993] wilc_sdio mmc0:0001:1 wlan0: INFO [connect]Authentication Type = 0
[   43.961208] wilc_sdio mmc0:0001:1 wlan0: INFO [connect]In OPEN SYSTEM
[   43.967652] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_set_bssid]set bssid [28:c6:8e:76:7e:9b]
[   43.976410] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_send_connect_wid]send HOST_IF_WAITING_CONN_RESP
[   44.064041] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_send_connect_wid]set HOST_IF_WAITING_CONN_RESP
<3>CTRL-EVENT-DISCONNECTED bssid=28:c6:8e:76:7e:9b reason=3 locally_generated=1
[   44.413994] cfg_indicate_rx: Info message received
[   44.418871] wilc_sdio mmc0:0001:1 wlan0: INFO [handle_rcvd_gnrl_async_info]Current State = 3,Received state = 1
[   44.454038] wilc_sdio mmc0:0001:1 wlan0: INFO [cfg_connect_result]Connection response received=1 connect_stat[0]
[   44.464239] wilc_sdio mmc0:0001:1 wlan0: INFO [cfg_connect_result]Connection Successful: BSSID: 28c68e767e9b
[   44.474029] wilc_sdio mmc0:0001:1 wlan0: INFO [cfg_connect_result]Association request info elements length = 22
[   44.484092] wilc_sdio mmc0:0001:1 wlan0: INFO [cfg_connect_result]Association response info elements length = 150
[   44.494380] ------------[ cut here ]------------
[   44.499034] WARNING: CPU: 0 PID: 355 at net/wireless/sme.c:733 __cfg80211_connect_result+0x200/0x2a0
[   44.508135] CPU: 0 PID: 355 Comm: kworker/u2:1 Not tainted 4.9.76-rt61-ltsi-01842-gb64c85cd1c9f #3
[   44.517055] Hardware name: Altera SOCFPGA
[   44.521058] Workqueue: cfg80211 cfg80211_event_work
[   44.525955] Backtrace:
[   44.528425] [<8010a954>] (dump_backtrace) from [<8010aba0>] (show_stack+0x18/0x1c)
[   44.535967]  r7:00000009 r6:00000000 r5:805f1638 r4:00000000
[   44.541619] [<8010ab88>] (show_stack) from [<80284fa4>] (dump_stack+0x20/0x28)
[   44.548827] [<80284f84>] (dump_stack) from [<80118544>] (__warn+0xc8/0xf8)
[   44.555716] [<8011847c>] (__warn) from [<8011862c>] (warn_slowpath_null+0x28/0x30)
[   44.563260]  r9:87331520 r8:873314d4 r7:86c0ec0c r6:00000000 r5:00000000 r4:873314a0
[   44.570986] [<80118604>] (warn_slowpath_null) from [<804e9880>] (__cfg80211_connect_result+0x200/0x2a0)
[   44.580354] [<804e9680>] (__cfg80211_connect_result) from [<804cb854>] (cfg80211_process_wdev_events+0xdc/0x188)
[   44.590490]  r7:86c0ec0c r6:87331528 r5:873314a0 r4:86c0ec00
[   44.596140] [<804cb778>] (cfg80211_process_wdev_events) from [<804cb954>] (cfg80211_process_rdev_events+0x54/0x6c)
[   44.606456]  r10:00000088 r9:00000000 r8:878f5900 r7:00000000 r6:87802000 r5:879ae438
[   44.614253]  r4:873314a0
[   44.616789] [<804cb900>] (cfg80211_process_rdev_events) from [<804c7e58>] (cfg80211_event_work+0x1c/0x24)
[   44.626324]  r5:879ae120 r4:879ae120
[   44.629898] [<804c7e3c>] (cfg80211_event_work) from [<8012c108>] (process_one_work+0x1b4/0x2e4)
[   44.638572]  r5:879ae120 r4:864f0b80
[   44.642144] [<8012bf54>] (process_one_work) from [<8012c574>] (worker_thread+0x304/0x464)
[   44.650294]  r10:00000088 r9:864f0b98 r8:87802028 r7:8070ff00 r6:87802000 r5:87802000
[   44.658109]  r4:864f0b80
[   44.660643] [<8012c270>] (worker_thread) from [<8013155c>] (kthread+0xd4/0xec)
[   44.667843]  r10:00000000 r9:00000000 r8:00000000 r7:8012c270 r6:864f0b80 r5:00000000
[   44.675641]  r4:86eb6080 r3:00000000
[   44.679213] [<80131488>] (kthread) from [<80107198>] (ret_from_fork+0x14/0x3c)
[   44.686410]  r7:00000000 r6:00000000 r5:80131488 r4:86eb6080
[   44.692047] ---[ end trace 0000000000000002 ]---
<3>No network configuration found for the current AP
<3>CTRL-EVENT-DISCONNECTED bssid=28:c6:8e:76:7e:9b reason=3 locally_generated=1
<3>CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
[   52.413970] cfg_indicate_rx: Info message received
[   52.418823] wilc_sdio mmc0:0001:1 wlan0: INFO [handle_rcvd_gnrl_async_info]Current State = 4,Received state = 0
[   52.428941] wilc_sdio mmc0:0001:1 wlan0: INFO [cfg_connect_result]Received WILC_MAC_STATUS_DISCONNECTED dev [87330000]
[   52.439611] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_set_bssid]set bssid [00:00:00:00:00:00]
Adding key with cipher group

AdhamAbozaeid, 2019-10-15:
Thanks for offering to verify the fix.
Attached is a patch that makes use of cfg_connect_bss()

[0001-staging-wilc1000-store-bss-object-and-use-cfg80211_c.zip](https://github.com/linux4wilc/driver/files/3731062/0001-staging-wilc1000-store-bss-object-and-use-cfg80211_c.zip)

asaillard, 2019-10-16:
I applied the patch and there is no more warning in the log, the fix corrects my issue.

> > [   20.018202] wilc_sdio mmc0:0001:1 wlan0: INFO [scan]Trigger Scan Request
<3>CTRL-EVENT-SCAN-STARTED
[   20.644128] cfg_indicate_rx: Scan Notification Received
[   20.649364] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_scan_complete_received]Scan notification received
<3>CTRL-EVENT-SCAN-RESULTS
<3>WPS-AP-AVAILABLE
<3>Trying to associate with 28:c6:8e:76:7e:9b (SSID='NETGEAR14' freq=2442 MHz)
[   21.074652] wilc_sdio mmc0:0001:1 wlan0: INFO [connect]Connecting to SSID [NETGEAR14] on netdev [87330000] host if [86442c00]
[   21.085958] wilc_sdio mmc0:0001:1 wlan0: INFO [connect]Adding key with cipher group fac04
[   21.094109] wilc_sdio mmc0:0001:1 wlan0: INFO [connect]Authentication Type = 0
[   21.101320] wilc_sdio mmc0:0001:1 wlan0: INFO [connect]In OPEN SYSTEM
[   21.107754] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_set_bssid]set bssid [28:c6:8e:76:7e:9b]
[   21.116511] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_send_connect_wid]send HOST_IF_WAITING_CONN_RESP
[   21.204161] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_send_connect_wid]set HOST_IF_WAITING_CONN_RESP
<3>CTRL-EVENT-DISCONNECTED bssid=28:c6:8e:76:7e:9b reason=3 locally_generated=1
[   22.014146] cfg_indicate_rx: Info message received
[   22.019003] wilc_sdio mmc0:0001:1 wlan0: INFO [handle_rcvd_gnrl_async_info]Current State = 3,Received state = 1
[   22.054257] wilc_sdio mmc0:0001:1 wlan0: INFO [cfg_connect_result]Connection response received=1 connect_stat[0]
[   22.064444] wilc_sdio mmc0:0001:1 wlan0: INFO [cfg_connect_result]Connection Successful: BSSID: 28c68e767e9b
[   22.074234] wilc_sdio mmc0:0001:1 wlan0: INFO [cfg_connect_result]Association request info elements length = 22
[   22.084333] wilc_sdio mmc0:0001:1 wlan0: INFO [cfg_connect_result]Association response info elements length = 150
<3>No network configuration found for the current AP[   22.096378] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_set_bssid]set bssid [00:00:00:00:00:00]
[   22.109695] wilc_sdio mmc0:0001:1 wlan0: INFO [disconnect]Disconnecting with reason code(3)
[   22.164146] cfg_indicate_rx: Info message received
[   22.168989] wilc_sdio mmc0:0001:1 wlan0: INFO [handle_rcvd_gnrl_async_info]Current State = 4,Received state = 0
<3>CTRL-EVENT-DISCONNECTED bssid=28:c6:8e:76:7e:9b reason=3 locally_generated=1[   22.172353] wilc_sdio mmc0:0001:1 wlan0: INFO [cfg_connect_result]Received WILC_MAC_STATUS_DISCONNECTED dev [87330000]
<3>CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
[   22.172362] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_set_bssid]set bssid [00:00:00:00:00:00]
[   22.212254] wilc_sdio mmc0:0001:1 wlan0: ERR [handle_rcvd_gnrl_async_info:965] conn_result is NULL

Thank you @AdhamAbozaeid for the correction

AdhamAbozaeid, 2019-10-16:
Thanks @asaillard  for confirming the fix. I'll submit it to the dev branch

commit 40d29c0a (2019-10-17): staging: wilc1000: store bss object and use cfg80211_connect_bss()

In a fast disconnect/connect sequence, cfg80211_connect_result() can
fail to find the bss object which the driver is connecting to. Detailed
sequence of events:
* Driver is connected in STA mode
* Disconnect request arrives from user space. Driver disconnects and
  calls cfg80211_disconnected() which adds new event to the
  cfg80211_wq worker thread
* Connect request arrives from user space. cfg80211_connect() stores
  ssid/ssid_len and calls rdev_connect()
* __cfg80211_disconnected() runs in worker thread and zero
  wdev->ssi [... cut]
closed 2019-10-17 by AdhamAbozaeid