linux4wilc/driver #62: More details on https://github.com/linux4wilc/driver/commit/bc0fc129b78e9f940bd5cd1455287760a755164a

tsifb, 2019-08-08:
@AdhamAbozaeid 
Can you provide some more details on the recent commit https://github.com/linux4wilc/driver/commit/bc0fc129b78e9f940bd5cd1455287760a755164a

It is concerning me that this fix (and related firmware commit) indicates that prior to this fix the WILC could be permanently damaged / disabled.

I need to determine if this fix needs to be urgently pushed out to products that have already been shipped and installed.

Can you detail:
1, the exactly sequence that can cause the problem to occur
2, what the impact is to the WILC module if it does occur
3, are there any workarounds or recoveries that can be used to restore WILC operation.
4, your assessment as to the urgency of performing an in-field upgrade to apply this fix.

We are about to release a new version of our product firmware very soon. Your assistance with understanding the impact of this issue would be helpful.

I also note that in May 2018 we reported salesforce ticket 00294134 - where two WILC modules became unusable (dead). Is this related?

AdhamAbozaeid, 2019-08-08:
@tsifb 
When the wlan interface isn't connected to an AP while powersave is enabled, WILC will be in sleep mode till it receive requests from the host.
If the user deinit the wlan interface, the driver would reset the WILC's FW without ensuring that it had enough time to execute its wakeup sequence, and might leave it in a mode that is stressful to the DC-DC converter, specially if the chip will be kept powered on after the wlan interface is deinit. Over long time, this might cause failure to the DC-DC.

deanoati, 2019-08-08:
As a datapoint, we also have had a couple of the Wilc modules mysteriously fail and stop working when we were working with the very buggy older drivers. Part of our recovery routine was to reset the module via GPIO and reload the driver. I never figured out exactly why this happened, only finding that they were 100% dead, with no uart debug output even. After reading about this bug it most certainly was the cause.

tsifb, 2019-08-09:
@AdhamAbozaeid thanks for the quick reply.
I'm upgrading to the dev branch HEAD to test. We don't use powersave... we has earlier problems related to that, and don't have the requirement to save power. But... I'm taking the opportunity to jump ahead in preparation for our next release.

AdhamAbozaeid, 2019-08-09:
Doing an early test would be great and will help us make a smooth transition between the releases. Thanks for taking the time to do that

closed 2019-08-09 by AdhamAbozaeid