linux4wilc/driver #63: wlan0 MAC address is incorrect in latest dev branch driver

tsifb, 2019-08-09:
Latest version:
ifconfig reports:-
wlan0     Link encap:Ethernet  HWaddr fa:f0:05:e5:4d:9a

Previously was f8:f0:05:.....

It should be f8:....

deanoati, 2019-08-09:
I noticed the same thing, but had enough other issues with the newer driver(see https://github.com/linux4wilc/driver/issues/50) that I wasn't going to report it yet.

Seems like the driver or firmware are reverting https://github.com/linux4wilc/firmware/commit/5264c2e04f3e6442c396e3f1ca9ddb41a818b8a7

AdhamAbozaeid, 2019-08-09:
The problem with the mac addresses is that now there's no notion of wlan0 and p2p0 interfaces in the driver. The user can remove all interfaces and add them manually with new names. I believe best we can now is to assign the first interface the original mac address to the first interface, and the altered address to the second interface.

deanoati, 2019-08-09:
i'd say assign the first STA interface instantiated the IEEE address (F8:...). and the first APmode/P2P interface perhaps the "locally administered" version (FA:...)?

The 2nd of an interface type would get trickier.

AdhamAbozaeid, 2019-08-09:
The problem is that the interface is assigned a mac address once it's opened, and the user can always start an interface in one mode, then change to the other.

AdhamAbozaeid, 2019-08-09:
Please verify with these binaries
[WILC3000.zip](https://github.com/linux4wilc/driver/files/3488108/WILC3000.zip)

tsifb, 2019-08-12:
I will test these binaries today.
I agree with @deanoati that the first interface, which is automatically created it seems, should get the base MAC.

Also,
When I added p2p0 as a managed interface, the default MAC is all zeros. I think at the least, the default MAC (if one is not specified) should be derived from the base MAC.

p2p0      Link encap:Ethernet  HWaddr 00:00:00:00:00:00
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

~# iw dev p2p0 info
Interface p2p0
        ifindex 7
        wdev 0x100000002
        addr 00:00:00:00:00:00
        type managed
        wiphy 1
        txpower 18.00 dBm

tsifb, 2019-08-12:
tested binaries added here.
wlan0 mac is now f8:...

Also, adding p2p0 has zeroed mac address, which becomes fa:... when hostapd is run. I guess this is intended behaviour - please confirm.
Also confirmed that when hostapd is killed, p2p0 interface is deleted.

p2p0      Link encap:Ethernet  HWaddr fa:f0:05:e5:4d:9a
          inet6 addr: fe80::f8f0:5ff:fee5:4d9a/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:10 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:836 (836.0 B)

I will test a little more, but given the change to dynamic interfaces, and the timing with our release cycle, it is unlikely we will move forward to this for our release. I will revisit in the near future to give it a thrashing.

AdhamAbozaeid, 2019-08-12:
Yes, the mac address is assigned after the interface is up, either using ifconfig up or hostapd or wpa_supplicant.

From testing I did here, I see that hostapd removes the interface it was using when it's killed. This applies to both p2p0 and wlan0. wpa_suplicant doesn't have the same behavior.

tsifb, 2019-08-12:
Closing this issue on the understanding that the MAC addressing modification to the firmware will be committed to firmware dev branch soon. (as per binaries attached here).

commit 724836df (2019-08-27): Assign wlan0 interface the original mac address

Adding the interfaces dynamically caused a regression causing wlan0
interface to be assigned the locally administrated mac address.
Since the interfaces are now added dynamically, and users can rename
them as they like, WILC will use the original mac address for the
interface that opens first, and the locally administrated address for
the other.

Fixes #63
closed 2019-08-12 by tsifb