linux4wilc/driver #64: Documentation for how to replicate p2p0 interface, now that virtual interfaces are dynamic.

tsifb, 2019-08-09:
Add some documentation with exact (example) commands / script to exactly replicate the old static p2p0 interface, now that this interface is no longer statically created.

This documentation should include how to set all interface parameters to match the old functionality. (including setting the MAC address as it was previously set. wlan0 MAC f8:... becomes f9:... for p2p0)

If there is already some doc and I missed it - please point me to it.

AdhamAbozaeid, 2019-08-09:
The user guide is updated, and will be published on the products pages https://www.microchip.com/wwwproducts/en/ATWILC1000 and https://www.microchip.com/wwwproducts/en/ATWILC3000 once released.

Before using the second interface, you need to add it using 
`iw phy0 interface add p2p0 type station`
Note that ph0 might change according to your system, so you need to know your wilc phy number using
`iw dev`
Also note that when the hostapd is deinitializing, it removes the interface it was using. Accordingly, you will have to manually add the interface if you killed hostapd, then restarted it.
wpa_supplicant doesn't do the same behavior, so you don't need to readd the interface if you killed the wpa_supplicant.

I will comment separately on the mac address problem in #63

closed 2019-08-09 by AdhamAbozaeid

tsifb, 2019-08-11:
Can you share the intended date for the document release?

My *hope* is that the team has carefully thought about how existing implementations will be able to be easily migrated forward to that existing functionality which is now relied upon can be maintained - or at the least, the changes are minimized going forward. Is this the case? Will the updated docs provide this guidance?

AdhamAbozaeid, 2019-08-12:
The final release is planned for September~October time frame.
I understand that the changes might affect the existing solutions, and require more changes on the user's side. Unfortunately the dynamic interface changes were inevitable to go through the mainlining process of the linux kernel.
The updated document will provide information on how to add the second virtual interface, and also a note on the difference in behavior between hostapd and wpa_supplicant when they are being killed