linux4wilc/driver #66: wilc can't start firmware

hxiaoqiang, 2019-08-09:
hi , I using WILC1000 on a ZYNQ Mpsoc custom board. driver and firmware version 15.2.
kernel version:  https://github.com/Xilinx/linux-xlnx/tree/xilinx-v2018.2(4.14.0)
here is my dts:
```
	amba_pl: amba_pl@0 {
		#address-cells = <2>;
		#size-cells = <2>;
		compatible = "simple-bus";
		ranges ;
		logisdhc_0: logisdhc@80000000 {
			compatible = "xlnx,logisdhc-2.1", "arasan,sdhci-8.9a";
			interrupt-names = "logisdhc_int";
			interrupt-parent = <&gic>;
			interrupts = <0 89 4>;
			reg = <0x0 0x80000000 0x0 0x10000>;
                        clock-names = "clk_xin", "clk_ahb";
                        clocks = <&clk 71>, <&clk 71>;
                        xlnx,device_id = <0>;
                         non-removable;
                         disable-wp;
            
                         #address-cells = <1>;
                         #size-cells = <0>;
            
                         wilc_sdio@0{
			       compatible      = "microchip,wilc1000", "microchip,wilc3000";
			       irq-gpios       = <&gpio 81 0>;
			       reset-gpios     = <&gpio 80 0>;
			       chip_en-gpios   = <&gpio 79 0>;
			        status          = "okay";
			        reg             = <0>;
			        bus-width       = <4>;
                         };
	 };
};
```
probe log:
```
[    2.022924] mmc0: new high speed SDIO card at address 0001
[    2.041120] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
[    2.041122] (unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
[    2.041669] (unnamed net_device) (uninitialized): INFO [wilc_create_wiphy]Registering wifi device
[    2.041671] (unnamed net_device) (uninitialized): INFO [wilc_wfi_cfg_alloc]Allocating wireless device
[    2.042083] wifi_pm : 0
[    2.042085] wifi_pm : 1
[    2.042177] wilc_sdio mmc0:0001:1: Driver Initializing success
```

ifconfig wlan0 up

```
root@root:~# ifconfig wlan0 up
[   20.809661] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[6cac2000] wlan0
[   20.817431] WILC POWER UP
[   20.820028] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_init_host_int]Host[6cac2000][6cec2000]
[   20.828274] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
[   20.835302] wilc_sdio mmc0:0001:1 wlan0: INFO [wlan_init_locks]Initializing Locks ...
[   20.843112] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
[   20.850846] wilc_sdio mmc0:0001:1: SDIO speed: 50000000
[   20.867598] wilc_sdio mmc0:0001:1: chipid 001003a0
[   20.872375] wilc_sdio mmc0:0001:1: chipid 001003a0
[   20.889289] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_initialize]WILC Initialization done
[   20.897654] wilc_sdio mmc0:0001:1 wlan0: INFO [wlan_initialize_threads]Initializing Threads ...
[   20.906332] wilc_sdio mmc0:0001:1 wlan0: INFO [wlan_initialize_threads]Creating kthread for transmission
[   20.915870] wilc_sdio mmc0:0001:1 wlan0: INFO [wlan_initialize_threads]Creating kthread for Debugging
[   20.925247] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_get_firmware]Detect chip WILC1000
[   20.933432] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_get_firmware]loading firmware mchp/wilc1000_wifi_firmware.bin
[   20.944106] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_get_firmware]WLAN firmware: mchp/wilc1000_wifi_firmware.bin
[   20.964246] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_firmware_download]Downloading Firmware ...
[   20.972780] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_firmware_download]Downloading firmware size = 135060
[   21.195042] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_firmware_download]Offset = 119744
[   21.204353] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_firmware_download]Offset = 135060
[   21.212536] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_firmware_download]Download Succeeded
[   21.220619] wilc_sdio mmc0:0001:1 wlan0: INFO [linux_wlan_start_firmware]Starting Firmware ...
[   21.229638] wilc_sdio mmc0:0001:1 wlan0: INFO [linux_wlan_start_firmware]Waiting for FW to get ready ...
[   22.251795] wilc_sdio mmc0:0001:1 wlan0: INFO [linux_wlan_start_firmware]Firmware start timed out
[   22.260595] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_wlan_initialize:1169] Failed to start firmware
[   22.269447] wilc_sdio mmc0:0001:1: wilc_sdio_disable_interrupt
[   22.275419] wilc_sdio mmc0:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinitializing Threads
[   22.284040] wilc_sdio mmc0:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinit debug Thread
[   22.292510] wilc_sdio mmc0:0001:1 wlan0: INFO [debug_thread]Exit debug thread
[   22.299625] wilc_sdio mmc0:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinitializing Threads
[   22.308320] wilc_sdio mmc0:0001:1 wlan0: INFO [wlan_deinit_locks]De-Initializing Locks
[   22.316167] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_wlan_initialize:1210] WLAN initialization FAILED
[   22.325188] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_mac_open:1254] Failed to initialize wilc
[   22.333550] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_send_config_pkt:1917] Set Timed out
[   22.341425] wilc_sdio mmc0:0001:1 wlan0: ERR [handle_set_wfi_drv_handler:433] Failed to set driver handler
ifconfig: SIOCSIFFLAGS: Input/output error
```
Any advice and suggestions will be greatly appreciated.
BestRegards
HSQ

AdhamAbozaeid, 2019-08-09:
You can trace the ISR in the driver and see if you are getting any interrupts from WILC at all.
Also you can double check if the driver was started or not by checking the WILC's UART lines and see if it's spitting any logs

AdhamAbozaeid, 2019-09-23:
Hi @hxiaoqiang 
Were yo able to get past the FW start step?

closed 2019-10-29 by AdhamAbozaeid

keshavaprashanth, 2021-09-13:
I am facing the same issue. Can i know how this issue was fixed ?