linux4wilc/driver #67: Wifi chip not starting correctly on reboot

MTaulin, 2019-09-04:
Hello,
im using the wilc1000 on a custom imx6 board connected via sdio interface. Im using the current dev driver with the Linux Kernel version 5.2.0 but the following issue also happened on earlier Kernel Versions.

I have the following problem: When i power on the device everything is working fine i can connect to wifi and also ssh on my device. The issue happens when i reboot the device(not power off). The device reboots but this time the wifi chip does not start.
I get following messages on bootlog:

`[   22.869041] WILC POWER UP` 
` [   22.872275] wilc_sdio mmc0:0001:1: SDIO speed: 50000000` 
` [   22.938215] wilc_sdio mmc0:0001:1: chipid 001003a0` 
` [   22.967199] wilc_sdio mmc0:0001:1: got gpio_irq successfully` 
` [   22.995381] wilc_handle_isr,>> UNKNOWN_INTERRUPT - 0x00000000` 
` [   23.002472] wilc_wlan_firmware_downloadDownloading firmware size = 135264` 
` [   23.057961] wilc_wlan_firmware_download Offset = 119884` 
` [   23.078521] wilc_wlan_firmware_download Offset = 135264` 
` [   23.613761] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_wlan_initialize:920] Failed to start firmware` 
` [   23.622600] wilc_sdio mmc0:0001:1: wilc_sdio_disable_interrupt` 
` [   23.988391] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_wlan_initialize:960] WLAN initialization FAILED` 
` [   23.997499] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_mac_open:1003] Failed to initialize wilc` 

When i power off and power on again the wifi is working. So i think something does not reset the wifi chip correctly on reboot.
I tested it with several of our devices and this is not happening on all devices. It's happening on like three out of ten devices

AdhamAbozaeid, 2019-09-04:
Hi @MTaulin 
Please open a sales force ticket for this issue.
http://microchipsupport.force.com

MTaulin, 2019-09-05:
opened a new Ticket:
 00459059 
Thanks

MTaulin, 2019-09-10:
Hello,
Solution was that the reset+en gpio weren't resetted correctly so the wifi chip could not start. Resetting the gpio's in bootloader solved the problem.

closed 2019-09-10 by MTaulin

AdhamAbozaeid, 2019-09-10:
Thanks for the update