linux4wilc/driver #68: p2p0 doesn't get linked

SarahWicker, 2019-09-13:
Hi i am not sure if here is the best place for the question but i used the repository from 
https://github.com/Avnet/u96v2-wilc-driver and they wrote that their source is this repository.

I followed your instruction on how to build the Linux image from "source" (using the Linux-xlnx but change the driver folder for the wilc module with yours) 

my problem is: the wpa_supplicant.conf is updated as you advised, and once i do : modprobe wilc-sdio i get (like in ATWILC1000-ATWILC3000-Wi-Fi-Link-Controller-Linux-User-Guide-DS70005328C.pdf) : drivers initilialized […] but after 
IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
and a tiny bit after
[ 2298.736288] IPv6: ADDRCONF(NETDEV_UP): p2p0: link is not ready
[ 2299.001221] IPv6: ADDRCONF(NETDEV_UP): p2p0: link is not ready
[ 2299.275837] IPv6: ADDRCONF(NETDEV_UP): p2p0: link is not ready
[ 2299.538213] IPv6: ADDRCONF(NETDEV_UP): p2p0: link is not ready
[ 2299.800886] IPv6: ADDRCONF(NETDEV_UP): p2p0: link is not ready
[ 2300.026968] IPv6: ADDRCONF(NETDEV_UP): p2p0: link is not ready
[ 2300.266363] IPv6: ADDRCONF(NETDEV_UP): p2p0: link is not ready
[ 2300.606788] IPv6: ADDRCONF(NETDEV_UP): p2p0: link is not ready
and this doesn't stop. it stop only if i put wlan0 down 

do you have an idea why? or of what it means?
thank you a lot
Sarah

AdhamAbozaeid, 2019-09-23:
Hi Sarah
I'm not sure why you are getting this after you do modprobe.
At this stage, none of the interfaces are up yet, until you call ifconfig p2p0 up,  ifconfig wlan0 up, or start the wpa_supplicant or the hostapd.
Also if you are using the dev branch, you won't be able to up the p2p0 interface unless you add it using 
iw phy0 interface add p2p0 type station
since the dev branch implemented the dynamic add/remove of virtual interface, and the driver doesn't add the p2p0 interface by default anymore.

what's the output of ifconfig at this stage?

AdhamAbozaeid, 2019-09-24:
It's weird that both interfaces are up before running the wpa_supplicant or the hostapd, but maybe something on your linux image is automatically upping the interfaces.
I'm also not sure about the history of this repor, but I'll close the ticket since it works for you now

closed 2019-09-24 by AdhamAbozaeid