linux4wilc/driver #70: Error on AP stop and loss of interface

jackmitch, 2019-09-18:
After setting up an AP with hostapd, when hostapd is killed the driver hits an error in deinit.

[   52.287616] WILC_SPI spi1.0 wlan0: INFO [wilc_wlan_initialize]WILC Firmware Ver = WILC_WIFI_FW_REL_15_3_RC2_DEV Build: 11457:11458M

commit `f4d2a68b279d86eb9da8e830e13dc1b4535a8c3e`

```
root@wilc:~/wifi-ap# dmesg | tail
[ 3077.076043] WILC_SPI spi1.0 wlan0: INFO [change_virtual_intf]In Change virtual interface function
[ 3077.084989] WILC_SPI spi1.0 wlan0: INFO [change_virtual_intf]Wireless interface name =wlan0
[ 3077.093387] WILC_SPI spi1.0 wlan0: INFO [change_virtual_intf]Interface type = NL80211_IFTYPE_STATION
[ 3077.102524] mon.wlan0: INFO [wilc_wfi_deinit_mon_interface]In Deinit monitor interface
[ 3077.110471] mon.wlan0: INFO [wilc_wfi_deinit_mon_interface]Unregister monitor netdev
[ 3077.119551] mon.wlan0 (unregistering): INFO [wilc_wfi_deinit_mon_interface]Deinit monitor interface done
[ 3077.129142] WILC_SPI spi1.0 wlan0: ERR [wilc_send_config_pkt:1814] Set Timed out
[ 3077.136626] WILC_SPI spi1.0 wlan0: ERR [wilc_set_operation_mode:1816] Failed to set driver handler
[ 3077.234628] WILC_SPI spi1.0 wlan0 (unregistering): ERR [wilc_send_config_pkt:1814] Set Timed out
[ 3077.243539] WILC_SPI spi1.0 wlan0 (unregistering): ERR [wilc_set_operation_mode:1816] Failed to set driver handler
```

AdhamAbozaeid, 2019-09-18:
@jackmitch , the interface becomes unavailable because the hostapd removes the virtual interface while deinitializing, so you have to re-add it using
iw phy0 interface add wlan0 type station
before restarting the hostapd

The error in the log above is an error to send a config packet that changes the operation mode of the interface while deinitializing, so shouldn't cause a problem

jackmitch, 2019-09-18:
Hi, is it correct that I should then be left with no remaining wireless interfaces after killing hostapd? I've never seen this behaviour from any other driver, I've always been left with the initial interface that the AP was started on.

jackmitch, 2019-09-18:
As a quick follow up, the 15.2.1 driver behaves as I would expect leaving the `wlan0` interface behind after hostapd has been killed.

AdhamAbozaeid, 2019-09-18:
That's correct. Because release 15.3 introduced the dynamic addition/removal of virtual interface that wasn't supported int 15.2.
 del_virtual_intf() in 15.2 was empty

AdhamAbozaeid, 2019-09-18:
It doesn't look correct to me, but I see hostapd is calling del_virtual_intf() while deinitializing.
Same behavior isn't observed when wpa_supplicant is deinitialized

closed 2019-09-23 by AdhamAbozaeid